syssi/esphome-pipsolar #3: Test PIP 2424MSE1

lucasimons, 2021-09-01:
Hi,
I tried with a 24v inverter from easun which however is identical to the 2424MSE1 of MPP solar looking at the esphome log I see that it sends something but receives nothing I also checked the wiring and I had reversed the red and white cable on the rj45 side, now I believe they are right
[image]

syssi, 2021-09-01:
Could you make a better photo of the rj45 connector? I want to make sure you have connected the correct pins. Ignore the color of the wires.

syssi, 2021-09-01:
@avion23 Is the rs232 converter wired correctly?

avion23, 2021-09-01:
I'll check again this weekend and upload photos

lucasimons, 2021-09-01:
[image] I use the original cable that came with the inverter and  it have only 4 cable

lucasimons, 2021-09-01:
I also have this converter but I search on internet how to connect them but I find different connections [image]

syssi, 2021-09-01:
You have to connect RX, TX and GND to the Sub-D connector / pins. This circuit could/should help:

[image]

At the pin header you've to connect RX, TX, VCC3.3 and GND.

syssi, 2021-09-01:
Don't connect the blue wire. In my opinion this is 12V VCC of the inverter. We don't need this voltage. ;-)

rem72, 2021-09-01:
The blue wire is 12v, don't connect it. 12v can drive a buck converter to power the ESP. Thus no need of external power.
Try this pinout
orange-white -> pin 2
solid orange -> pin 3
solid brown -> pin 5

[image]

lucasimons, 2021-09-01:
Which converter do you use?

rem72, 2021-09-01:
> Which converter do you use?

I recently buyed a Voltronic Axpert VMIII 3kw/24v but I'm still having some issues with pipsolar.  
I think that  is similar to yours.
It communicates but seems that protocol is not good.

lucasimons, 2021-09-01:
I also have a 2.4kw easun rebranded voltronic, but looking at the logs I don't get any response from the inverter ... Do you have any signs of life from the inverter?

syssi, 2021-09-02:
Did you try the increase the log level of the UART component to `VERY_VERBOSE`? This would dump every transmitted/received byte to the log. This log would help to debug protocol issues.

lucasimons, 2021-09-02:
Yes but i see that the esp send the command but the inverter not responding

[08:50:45][D][pipsolar:757]: timeout command to poll: QMOD
[08:50:45][VV][uart_esp8266:111]:     Wrote 0b01010001 (0x51)
[08:50:45][VV][uart_esp8266:111]:     Wrote 0b01000110 (0x46)
[08:50:45][VV][uart_esp8266:111]:     Wrote 0b01001100 (0x4C)
[08:50:45][VV][uart_esp8266:111]:     Wrote 0b01000001 (0x41)
[08:50:45][VV][uart_esp8266:111]:     Wrote 0b01000111 (0x47)
[08:50:45][VV][uart_esp8266:101]:     Wrote 0b10011000 (0x98)
[08:50:45][VV][uart_esp8266:101]:     Wrote 0b01110100 (0x74)
[08:50:45][VV][uart_esp8266:101]:     Wrote 0b00001101 (0x0D)
[08:50:45][D][pipsolar:840]: Sending polling command : QFLAG with length 5
[08:50:45][C][captive_portal:148]: Captive Portal:
[08:50:45][C][api:095]: API Server:
[08:50:45][C][api:096]:   Address: sensor2.local:6053
[08:50:45][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=15936 (now=16938)
[08:50:46][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=16936 (now=17936)
[08:50:47][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=17936 (now=18936)
[08:50:48][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=18936 (now=19942)
[08:50:49][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=19936 (now=20938)
[08:50:50][D][pipsolar:757]: timeout command to poll: QFLAG
[08:50:50][VV][uart_esp8266:111]:     Wrote 0b01010001 (0x51)
[08:50:50][VV][uart_esp8266:111]:     Wrote 0b01010000 (0x50)
[08:50:50][VV][uart_esp8266:111]:     Wrote 0b01001001 (0x49)
[08:50:50][VV][uart_esp8266:111]:     Wrote 0b01010010 (0x52)
[08:50:50][VV][uart_esp8266:111]:     Wrote 0b01001001 (0x49)
[08:50:50][VV][uart_esp8266:101]:     Wrote 0b11111000 (0xF8)
[08:50:50][VV][uart_esp8266:101]:     Wrote 0b01010100 (0x54)
[08:50:50][VV][uart_esp8266:101]:     Wrote 0b00001101 (0x0D)
[08:50:50][D][pipsolar:840]: Sending polling command : QPIRI with length 5
[08:50:50][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=20936 (now=21942)
[08:50:51][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=21936 (now=22936)
[08:50:52][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=22936 (now=23936)
[08:50:53][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=23936 (now=24936)
[08:50:54][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=24936 (now=25939)
[08:50:55][D][pipsolar:757]: timeout command to poll: QPIRI
[08:50:55][VV][uart_esp8266:111]:     Wrote 0b01010001 (0x51)
[08:50:55][VV][uart_esp8266:111]:     Wrote 0b01010000 (0x50)
[08:50:55][VV][uart_esp8266:111]:     Wrote 0b01001001 (0x49)
[08:50:55][VV][uart_esp8266:111]:     Wrote 0b01000111 (0x47)
[08:50:55][VV][uart_esp8266:111]:     Wrote 0b01010011 (0x53)
[08:50:55][VV][uart_esp8266:101]:     Wrote 0b10110111 (0xB7)
[08:50:55][VV][uart_esp8266:101]:     Wrote 0b10101001 (0xA9)
[08:50:55][VV][uart_esp8266:101]:     Wrote 0b00001101 (0x0D)
[08:50:55][D][pipsolar:840]: Sending polling command : QPIGS with length 5
[08:50:55][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=25936 (now=26936)
[08:50:56][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=26936 (now=27943)
[08:50:57][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=27936 (now=28936)
[08:50:58][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=28936 (now=29939)
[08:50:59][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=29936 (now=30937)
[08:51:00][D][pipsolar:757]: timeout command to poll: QPIGS
[08:51:00][VV][uart_esp8266:111]:     Wrote 0b01010001 (0x51)
[08:51:00][VV][uart_esp8266:111]:     Wrote 0b01001101 (0x4D)
[08:51:00][VV][uart_esp8266:111]:     Wrote 0b01001111 (0x4F)
[08:51:00][VV][uart_esp8266:111]:     Wrote 0b01000100 (0x44)
[08:51:00][VV][uart_esp8266:101]:     Wrote 0b01001001 (0x49)
[08:51:00][VV][uart_esp8266:101]:     Wrote 0b11000001 (0xC1)
[08:51:00][VV][uart_esp8266:101]:     Wrote 0b00001101 (0x0D)
[08:51:00][D][pipsolar:840]: Sending polling command : QMOD with length 4
[08:51:00][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=30936 (now=31936)
[08:51:01][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=31936 (now=32938)
[08:51:02][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=32936 (now=33937)
[08:51:03][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=33936 (now=34939)
[08:51:04][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=34936 (now=35937)
[08:51:05][D][pipsolar:757]: timeout command to poll: QMOD
[08:51:05][VV][uart_esp8266:111]:     Wrote 0b01010001 (0x51)
[08:51:05][VV][uart_esp8266:111]:     Wrote 0b01000110 (0x46)
[08:51:05][VV][uart_esp8266:111]:     Wrote 0b01001100 (0x4C)
[08:51:05][VV][uart_esp8266:111]:     Wrote 0b01000001 (0x41)
[08:51:05][VV][uart_esp8266:111]:     Wrote 0b01000111 (0x47)
[... 11 lines cut ...]
[08:51:10][VV][uart_esp8266:111]:     Wrote 0b01010000 (0x50)
[08:51:10][VV][uart_esp8266:111]:     Wrote 0b01001001 (0x49)
[08:51:10][VV][uart_esp8266:111]:     Wrote 0b01010010 (0x52)
[08:51:10][VV][uart_esp8266:111]:     Wrote 0b01001001 (0x49)
[... cut ...]
[08:51:15][VV][uart_esp8266:111]:     Wrote 0b01010011 (0x53)
[08:51:15][VV][uart_esp8266:101]:     Wrote 0b10110111 (0xB7)
[08:51:15][VV][uart_esp8266:101]:     Wrote 0b10101001 (0xA9)
[08:51:15][VV][uart_esp8266:101]:     Wrote 0b00001101 (0x0D)
[08:51:15][D][pipsolar:840]: Sending polling command : QPIGS with length 5
[08:51:15][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=45936 (now=46940)
[08:51:16][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=46936 (now=47936)

rem72, 2021-09-02:
I got it working, but not all values are good. Panel voltage and power are 0v and 0w, as you can see in my screenshot.
Althought it's charging batteries and powering home from PV's.
[image]

lucasimons, 2021-09-02:
In my converter , the smallest haven't led and the biggest ha a led but seems that not blinking [image]

rem72, 2021-09-02:
Now I got it working with a small one.

lucasimons, 2021-09-02:
Now I connect the tx of the converter with the tx of wemos d1 mini and I see blinking the led but I don't see any data in the log.... [image] the converter is the same of your converter female

rem72, 2021-09-02:
Ah! And I used the hardware serial of the wemos d1 instead of a software one, this can be done if you change the logger uart. 
Which pins do you use for comms?
This is my code:

esphome:
  name: solar
  platform: ESP8266
  board: d1_mini

# Enable logging
logger:
  hardware_uart: UART1
  level: verbose

# Enable Home Assistant API
api:

ota:
  password: !secret OTAPass

wifi:
  ssid: !secret Wifi
  password: !secret WifiPass

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Solar Fallback Hotspot"
    password: !secret OTAPass

captive_portal:

uart:
  - id: uart_bus
    tx_pin: TX
    rx_pin: RX
    # most devices use 2400 as baud_rate
    baud_rate: 2400

pipsolar:
  - uart_id: uart_bus
    id: inverter0

sensor:
  - platform: pipsolar
    pipsolar_id: inverter0

    grid_rating_current:
      id: inverter0_grid_rating_current
      name: inverter0_grid_rating_current
    ac_output_active_power:
      id: inverter0_ac_output_active_power
      name: inverter0_ac_output_active_power
    ac_output_apparent_power:
      id: inverter0_ac_output_apparent_power
      name: inverter0_ac_output_apparent_power
    ac_output_voltage:
      id: inverter0_ac_output_voltage
      name: inverter0_ac_output_voltage
    battery_capacity_percent:
      id: inverter0_battery_capacity_percent
      name: inverter0_battery_capacity_percent
    battery_charging_current:
      id: inverter0_battery_charging_current
      name: inverter0_battery_charging_current
    battery_discharge_current:
      id: inverter0_battery_discharge_current
      name: inverter0_battery_discharge_current
    battery_voltage:
      id: inverter0_battery_voltage
      name: inverter0_battery_voltage
    grid_voltage:
      id: inverter0_grid_voltage
      name: inverter0_grid_voltage
    inverter_heat_sink_temperature:
      id: inverter0_inverter_heat_sink_temperature
      name: inverter0_inverter_heat_sink_temperature
    output_load_percent:
      id: inverter0_output_load_percent
      name: inverter0_output_load_percent
    pv_charging_power:
      id: inverter0_pv_charging_power
      name: inverter0_pv_charging_power
    pv_input_voltage:
      id: inverter0_pv_input_voltage
      name: inverter0_pv_input_voltage
    

    
text_sensor:
  - platform: pipsolar
    pipsolar_id: inverter0
    device_mode:
      id: inverter0_device_mode
      name: inverter0_device_mode
    last_qpigs:
      id: inverter0_last_qpigs
      name: inverter0_last_qpigs
    last_qpiri:
      id: inverter0_last_qpiri
      name: inverter0_last_qpiri
      
switch:
  - platform: pipsolar
    pipsolar_id: inverter0
    output_source_priority_utility:
      name: inverter0_output_source_priority_utility
    output_source_priority_solar:
      name: inverter0_output_source_priority_solar
    output_source_priority_battery:
      name: inverter0_output_source_priority_battery
    input_voltage_range:
      name: inverter0_input_voltage_range
    pv_power_balance:
      name: inverter0_pv_power_balance

rem72, 2021-09-02:
On the other hand, I think decoding of QPIGS isn't fine. 
Sissy, here is the verbose log:

[10:51:33][D][text_sensor:015]: 'inverter0_last_qpigs': Sending state '(226.7 49.9 226.7 49.9 0498 0479 016 427 27.00 005 100 0035 01.9 255.1 00.00 00000 10010110 00 00 00510 110'
[10:51:33][V][sensor:037]: 'inverter0_grid_voltage': Received new state 226.699997
[10:51:33][D][sensor:131]: 'inverter0_grid_voltage': Sending state 226.70000 V with 1 decimals of accuracy
[10:51:33][V][sensor:037]: 'inverter0_ac_output_voltage': Received new state 226.699997
[10:51:33][D][sensor:131]: 'inverter0_ac_output_voltage': Sending state 226.70000 V with 1 decimals of accuracy
[10:51:33][V][sensor:037]: 'inverter0_ac_output_apparent_power': Received new state 498.000000
[10:51:33][D][sensor:131]: 'inverter0_ac_output_apparent_power': Sending state 498.00000 VA with 1 decimals of accuracy
[10:51:33][V][sensor:037]: 'inverter0_ac_output_active_power': Received new state 479.000000
[10:51:33][D][sensor:131]: 'inverter0_ac_output_active_power': Sending state 479.00000 W with 1 decimals of accuracy
[10:51:33][V][sensor:037]: 'inverter0_output_load_percent': Received new state 16.000000
[10:51:33][D][sensor:131]: 'inverter0_output_load_percent': Sending state 16.00000 % with 1 decimals of accuracy
[10:51:33][V][sensor:037]: 'inverter0_battery_voltage': Received new state 27.000000
[10:51:33][D][sensor:131]: 'inverter0_battery_voltage': Sending state 27.00000 V with 1 decimals of accuracy
[10:51:33][V][sensor:037]: 'inverter0_battery_charging_current': Received new state 5.000000
[10:51:33][D][sensor:131]: 'inverter0_battery_charging_current': Sending state 5.00000 A with 1 decimals of accuracy
[10:51:33][V][sensor:037]: 'inverter0_battery_capacity_percent': Received new state 100.000000
[10:51:33][D][sensor:131]: 'inverter0_battery_capacity_percent': Sending state 100.00000 % with 1 decimals of accuracy
[10:51:33][V][sensor:037]: 'inverter0_inverter_heat_sink_temperature': Received new state 35.000000
[10:51:33][D][sensor:131]: 'inverter0_inverter_heat_sink_temperature': Sending state 35.00000 °C with 1 decimals of accuracy
[10:51:33][V][sensor:037]: 'inverter0_pv_input_voltage': Received new state 0.900000
[10:51:33][D][sensor:131]: 'inverter0_pv_input_voltage': Sending state 0.90000 V with 1 decimals of accuracy
[10:51:33][V][sensor:037]: 'inverter0_battery_discharge_current': Received new state 0.000000
[10:51:33][D][sensor:131]: 'inverter0_battery_discharge_current': Sending state 0.00000 A with 1 decimals of accuracy
[10:51:33][V][sensor:037]: 'inverter0_pv_charging_power': Received new state 0.000000
[10:51:33][D][sensor:131]: 'inverter0_pv_charging_power': Sending state 0.00000 W with 1 decimals of accuracy
[10:51:33][V][app:082]: A component took a long time in a loop() cycle (0.21 s).
[10:51:33][V][app:083]: Components should block for at most 20-30ms in loop().

rem72, 2021-09-02:
But if you use the hardware uart = uart0 you have to send the logger output to the secondary uart = uart1.
Check these lines:

logger:
hardware_uart: UART1

lucasimons, 2021-09-02:
Yes i see your code and i update my code, 
now i see that write  ant read somethings

[11:23:14][D][pipsolar:840]: Sending polling command : QPIRI with length 5
[11:23:14][VV][uart_esp8266:132]:     Read 0b10101010 (0xAA)
[11:23:14][VV][uart_esp8266:132]:     Read 0b01001010 (0x4A)
[11:23:14][VV][uart_esp8266:132]:     Read 0b10010101 (0x95)
[11:23:14][VV][uart_esp8266:132]:     Read 0b11100001 (0xE1)
[11:23:14][VV][uart_esp8266:132]:     Read 0b11111111 (0xFF)
[11:23:15][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=242368 (now=243373)
[11:23:16][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=243368 (now=244369)
[11:23:17][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=244368 (now=245372)
[11:23:18][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=245368 (now=246368)
[11:23:19][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=246368 (now=247376)
[11:23:19][D][pipsolar:757]: timeout command to poll: QPIRI
[11:23:19][VV][uart_esp8266:111]:     Wrote 0b01010001 (0x51)
[11:23:19][VV][uart_esp8266:111]:     Wrote 0b01010000 (0x50)
[11:23:19][VV][uart_esp8266:111]:     Wrote 0b01001001 (0x49)
[11:23:19][VV][uart_esp8266:111]:     Wrote 0b01000111 (0x47)
[11:23:19][VV][uart_esp8266:111]:     Wrote 0b01010011 (0x53)
[11:23:19][VV][uart_esp8266:101]:     Wrote 0b10110111 (0xB7)
[11:23:19][VV][uart_esp8266:101]:     Wrote 0b10101001 (0xA9)
[11:23:19][VV][uart_esp8266:101]:     Wrote 0b00001101 (0x0D)
[11:23:19][D][pipsolar:840]: Sending polling command : QPIGS with length 5
[11:23:19][VV][uart_esp8266:132]:     Read 0b10101010 (0xAA)
[11:23:19][VV][uart_esp8266:132]:     Read 0b11101010 (0xEA)
[11:23:19][VV][uart_esp8266:132]:     Read 0b01101010 (0x6A)
[11:23:19][VV][uart_esp8266:132]:     Read 0b11110101 (0xF5)
[11:23:19][VV][uart_esp8266:132]:     Read 0b11111111 (0xFF)
[11:23:20][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=247368 (now=248368)
[11:23:21][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=248368 (now=249375)
[11:23:22][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=249368 (now=250368)
[11:23:23][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=250368 (now=251371)
[11:23:24][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=251368 (now=252370)
[11:23:24][D][pipsolar:757]: timeout command to poll: QPIGS
[11:23:24][VV][uart_esp8266:111]:     Wrote 0b01010001 (0x51)
[11:23:24][VV][uart_esp8266:111]:     Wrote 0b01001101 (0x4D)
[11:23:24][VV][uart_esp8266:111]:     Wrote 0b01001111 (0x4F)
[11:23:24][VV][uart_esp8266:111]:     Wrote 0b01000100 (0x44)
[11:23:24][VV][uart_esp8266:101]:     Wrote 0b01001001 (0x49)
[11:23:24][VV][uart_esp8266:101]:     Wrote 0b11000001 (0xC1)
[11:23:24][VV][uart_esp8266:101]:     Wrote 0b00001101 (0x0D)
[11:23:24][D][pipsolar:840]: Sending polling command : QMOD with length 4
[11:23:24][VV][uart_esp8266:132]:     Read 0b11111111 (0xFF)
[11:23:24][VV][uart_esp8266:132]:     Read 0b11111110 (0xFE)
[11:23:24][VV][uart_esp8266:132]:     Read 0b11111111 (0xFF)
[11:23:25][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=252368 (now=253370)
[11:23:26][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=253368 (now=254368)
[11:23:27][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=254368 (now=255368)
[11:23:28][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=255368 (now=256373)
[11:23:29][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=256368 (now=257370)
[11:23:29][D][pipsolar:757]: timeout command to poll: QMOD
[11:23:29][VV][uart_esp8266:111]:     Wrote 0b01010001 (0x51)
[11:23:29][VV][uart_esp8266:111]:     Wrote 0b01000110 (0x46)
[11:23:29][VV][uart_esp8266:111]:     Wrote 0b01001100 (0x4C)
[11:23:29][VV][uart_esp8266:111]:     Wrote 0b01000001 (0x41)
[11:23:29][VV][uart_esp8266:111]:     Wrote 0b01000111 (0x47)
[11:23:29][VV][uart_esp8266:101]:     Wrote 0b10011000 (0x98)
[11:23:29][VV][uart_esp8266:101]:     Wrote 0b01110100 (0x74)
[11:23:29][VV][uart_esp8266:101]:     Wrote 0b00001101 (0x0D)
[11:23:29][D][pipsolar:840]: Sending polling command : QFLAG with length 5
[11:23:29][VV][uart_esp8266:132]:     Read 0b11101010 (0xEA)
[11:23:29][VV][uart_esp8266:132]:     Read 0b00001010 (0x0A)
[11:23:29][VV][uart_esp8266:132]:     Read 0b01000110 (0x46)
[11:23:29][VV][uart_esp8266:132]:     Read 0b11111111 (0xFF)

syssi, 2021-09-02:
Good job! Please try to understand what you have changed in detail. So we can improve the configuration example at this repository.

@rem72 We can give it a try to understand and correct the QPIGS readings later the day!

lucasimons, 2021-09-02:
I tried to put back the connections tx to tx and rx to rx but now in the log I only see something writing but it doesn't read anything, the right connection should be the inverted one so rx to tx @rem72

lucasimons, 2021-09-02:
Ah but you have the inverter that can mix the network with the panels because I noticed the voltage of the panels at 259v but I have the version with 40a mppt regulator but maximum voltage 100v anyway it shouldn't change anything because at the time I tried through software watchpower and PC and I had qpigs with the same organization

[image] ...

 Instead how did you connect the converter to the wemos? @rem72

syssi, 2021-09-02:
This is the code which needs to be improved to support your device:

https://github.com/syssi/esphome-pipsolar/blob/2424mse1/components/pipsolar/pipsolar.cpp#L449-L467

I've pushed a feature branch to improve the device support of the 2424mse1. This feature branch can be easily used by adding the following lines to the yaml:

```
external_components:
  - source: github://syssi/esphome-pipsolar@2424mse1
    refresh: 0s
```

This overrides the official component. Let's try introduce the required changes for your device here.

rem72, 2021-09-02:
@lucasimons
This is my actual configuration:
[image]
[image]
[image]

Thanks for the improvement, 
Maybe the right config can be done at yaml level instead of .cpp, 
I'll give a try and let you know the result.

lucasimons, 2021-09-02:
Thanks now i have two wemos that the connection are correct, one is with the small converter in the first pics of this thread, one with the big converter i can use they to test the new code...

syssi, 2021-09-02:
@rem72 At the moment it's not possible to change the QPIGS parsing via the yaml configuration. Is soon we've created a proper parser for your device we can introduce a device toggle at the configuration layer.

syssi, 2021-09-02:
@lucasimons I've created a feature branch only (without any changes). It will take some time to update the parser. You could give it a try!

lucasimons, 2021-09-02:
unfortunately I don't know how to help in modifying the code, I currently use [solpiplog](https://github.com/njfaria/SolPipLog) and it works but I don't know if you can get something by looking at that code

avion23, 2021-09-03:
Here are the promised pictures.

I know it's difficult to see. I fanned out the network cable to 1234 from the top and then 5678 to the right.

lucasimons, 2021-09-03:
Which inverter do you have?

andreashergert1984, 2021-09-03:
> Here are the promised pictures.
> 
> I know it's difficult to see. I fanned out the network cable to 1234 from the top and then 5678 to the right.
> If you build something like this then include a bulk capacitor (bottom). The ESP-Module sometimes sips 500mA
> 
> [image]
> 
> blue / white is rx / tx (I don't remember which one).
> [image]

remeber the ESP is NOT TTL (5V) level.
You need you uart signal at 3.3v Level. I had issues when directly connection the TTL-Level(5V). therefor perhaps the max3232 chip is better. Is made a board where i can directly connect 2 Invertes (but one works also ).. if there is the need i can make the data available.. or even let a pcb-fab do some for shipping?

[image]

lucasimons, 2021-09-03:
> Hi,
> I tried with a 24v inverter from easun which however is identical to the 2424MSE1 of MPP solar looking at the esphome log I see that it sends something but receives nothing I also checked the [...]

@andreashergert1984 this is my converter but seems that write something but not receive anything from the inverter

syssi, 2021-09-03:
I've pushed a fix for the QPIGS response:

https://github.com/syssi/esphome-pipsolar/commit/3fe7d85796518b445954d0dde61e5e8a2b549822

To illustrate the QPIGS parser I've introduced some comments:

https://github.com/syssi/esphome-pipsolar/compare/40f50a89ae1699e00398244966413b9569bb439c...3fe7d85796518b445954d0dde61e5e8a2b549822

Please give it a try by adding this external/fixed component to your YAML:

```
external_components:
  - source: github://syssi/esphome-pipsolar@2424mse1
    refresh: 0s
```

avion23, 2021-09-03:
@lucasimons you also need 3.3V on the inverter side. The way it is right now, the high voltage side is not powered. The converter board has a little charge pump which generates the ~15V from your 3.3V.

Do not connect the 12V ;) Then you need a new converter.

I have some PIP-5048MS (?) Clone

andreashergert1984, 2021-09-03:
> @lucasimons you also need 3.3V on the inverter side. The way it is right now, the high voltage side is not powered. The converter board has a little charge pump which generates the ~15V from your 3.3V.
> 
> Do not connect the 12V ;) Then you need a new converter.
> 
> I have some PIP-5048MS (?) Clone

i actually us the ~15V to power the ESP32 Board. but it needs both inverters to deliver enough power.. so i will observe this...

lucasimons, 2021-09-03:
> This is my code
> 
> ```
> esphome:
>   name: sensor2
>   platform: ESP8266
>   board: d1_mini
> 
> 
> captive_portal:
> 
> # Enable logging
> logger:
>   baud_rate: 0
>   level: VERY_VERBOSE
>   logs:
>     uart: VERY_VERBOSE 
>     uart_esp8266: VERY_VERBOSE 
>     api.service: WARN
>     ota: WARN
>     sensor: DEBUG
> # Enable Home Assistant API
> api:
> 
> ota:
> 
> 
> 
> 
> substitutions:
>   name: pipsolar
> 
> 
> 
> uart:
>   id: uart0
>   baud_rate: 2400
>   tx_pin: GPIO1
>   rx_pin: GPIO3
> 
> pipsolar:
>   uart_id: uart0
>   id: inverter0
> 
> 
> sensor:
>   - platform: pipsolar
>     pipsolar_id: inverter0
>     # QPIRI
>     grid_rating_voltage:
>       name: "Inverter grid_rating_voltage"
>     grid_rating_current:
>       name: "Inveter grid_rating_current"
>     ac_output_rating_voltage:
>       name: "Inveter ac_output_rating_voltage"
>     ac_output_rating_frequency:
>       name: "Inveter ac_output_rating_frequency"
>     ac_output_rating_current:
>       name: "Inveter ac_output_rating_current"
>     ac_output_rating_apparent_power:
>       name: "Inveter ac_output_rating_apparent_power"
>     ac_output_rating_active_power:
>       name: "Inveter ac_output_rating_active_power"
>     battery_rating_voltage:
>       name: "Inveter battery_rating_voltage"
>     battery_recharge_voltage:
>       name: "Inveter battery_recharge_voltage"
>     battery_under_voltage:
>       name: "Inveter battery_under_voltage"
>     battery_bulk_voltage:
[... 126 lines cut ...]
>       name: "Inveter output_source_priority_utility"
>     output_source_priority_solar:
>       name: "Inveter output_source_priority_solar"
>     output_source_priority_battery:
>       name: "Inveter output_source_priority_battery"
>     input_voltage_range:
>       name: "Inveter input_voltage_range"
>     pv_ok_condition_for_parallel:
>       name: "Inveter pv_ok_condition_for_parallel"
>     pv_power_balance:
>       name: "Inveter pv_power_balance"
> 
> output:
>   - platform: pipsolar
>     pipsolar_id: inverter0
>     battery_recharge_voltage:
>       id: inverter0_battery_recharge_voltage_out
> ```
> 
> 
>  This are the connections 
> 
> [image]

I try the new code with the two converter but I don't see any response

andreashergert1984, 2021-09-03:
> I try the new code with the two converter but I don't see any response

could you provide a schematic of your connections?

syssi, 2021-09-03:
@lucasimons If your inverter always responds with a very short answer you could try to increase the baud rate (4800, 9600, 19200).

andreashergert1984, 2021-09-03:
> @lucasimons If your inverter always responds with a very short answer you could try to increase the baud rate (4800, 9600, 19200).

good point. i would first try to directly connect a PC to the inverter.. and check if there an answer gets out at least an error (NACK) or else.

lucasimons, 2021-09-03:
I am also using an rp4 with solpiplog and that works but it is connected via usb, however now I try to see if the esp sends something by connecting it to the PC via a ttl to usb converter

andreashergert1984, 2021-09-03:
> I am also using an rp4 with solpiplog and that works but it is connected via usb, however now I try to see if the esp sends something by connecting it to the PC via a ttl to usb converter

i would do a step by step check.. does you esp send the commands correctly out ? just loop your rx and tx pin on esp..  to another/same/arduino and listen to uart data. 
Then check the level shifter.. i also had some dead on arrival.

syssi, 2021-09-03:
@lucasimons Which baudrate do you use at solpiplog?

lucasimons, 2021-09-03:
i connect pc to the wemos and i think that works, 
[image]

i don t know the baudrate of solpiplog baudrate i think that is 2400

syssi, 2021-09-03:
@lucasimons If you type something into the putty termin. Do you receive these bytes at the ESP?

lucasimons, 2021-09-03:
yes i try to write something and the esp recive the data
[image]

syssi, 2021-09-03:
The received exception on the left doesn't look good! ;-)

lucasimons, 2021-09-03:
it actually doesn't look good but I clicked a few letters and / or numbers then hit enter,,,,could i write something like test?

syssi, 2021-09-03:
If you press "space" your ESP should receive `20` each time. Receiving an exception means your ESP crashed.

rem72, 2021-09-04:
@syssi , I've changed whatever I need to get all values from qpigs.
Need to redefine this variable to float:
 `PIPSOLAR_SENSOR(battery_voltage_scc, QPIGS, float)`

At the parser in pipsolar.cpp

`sscanf(                                                                                              // NOLINT
            tmp,                                                                                             // NOLINT
            "(%f %f %f %f %d %d %d %d %f %d %d %d %f %f %f %d %1d%1d%1d%1d%1d%1d%1d%1d %d %d %d %1d%1d%1d",  // NOLINT
            &value_grid_voltage_, &value_grid_frequency_, &value_ac_output_voltage_,                         // NOLINT
            &value_ac_output_frequency_,                                                                     // NOLINT
            &value_ac_output_apparent_power_, &value_ac_output_active_power_, &value_output_load_percent_,   // NOLINT
            &value_bus_voltage_, &value_battery_voltage_, &value_battery_charging_current_,                  // NOLINT
            &value_battery_capacity_percent_, &value_inverter_heat_sink_temperature_,                        // NOLINT
            &value_pv_input_current_for_battery_, &value_pv_input_voltage_, &value_battery_voltage_scc_,     // NOLINT
            &value_battery_discharge_current_, &value_add_sbu_priority_version_,                             // NOLINT
            &value_configuration_status_, &value_scc_firmware_version_, &value_load_status_,                 // NOLINT
            &value_battery_voltage_to_steady_while_charging_, &value_charging_status_,                       // NOLINT
            &value_scc_charging_status_, &value_ac_charging_status_,                                         // NOLINT
            &value_battery_voltage_offset_for_fans_on_, &value_eeprom_version_, &value_pv_charging_power_,   // NOLINT
            &value_charging_to_floating_mode_, &value_switch_on_,                                            // NOLINT
            &value_dustproof_installed_);    `

This makes the work for me.
[image]

Thank you so much for your help.

syssi, 2021-09-04:
> I've pushed a fix for the QPIGS response:
> 
> [3fe7d85](https://github.com/syssi/esphome-pipsolar/commit/3fe7d85796518b445954d0dde61e5e8a2b549822)
> 
> To illustrate the QPIGS parser I've introduced some comments:
> 
> [40f50a8...3fe7d85](https://github.com/syssi/esphome-pipsolar/compare/40f50a89ae1699e00398244966413b9569bb439c...3fe7d85796518b445954d0dde61e5e8a2b549822)
> 
> Please give it a try by adding this external/fixed component to your YAML:
> 
> ```
> external_components:
>   - source: github://syssi/esphome-pipsolar@2424mse1
>     refresh: 0s
> ```

@rem72 Did you see this message? You changed the type of `pv_input_current_for_battery`, right?

rem72, 2021-09-06:
> @rem72 At the moment it's not possible to change the QPIGS parsing via the yaml configuration. Is soon we've created a proper parser for your device we can introduce a device toggle at the configuration layer.

Thank so much for your effort, Contact me if u need testing the configuration selector. 
I finally get the component installed as local, and updated the code to obtain my data. Had to modify the header pipsolar.h too:

pipsolar.h:
line 70 `PIPSOLAR_SENSOR(battery_voltage_scc, QPIGS, float)` -> int to float

pipsolar.cpp:
starting at line 451
`sscanf(                                                                                              // NOLINT
            tmp,                                                                                             // NOLINT
            "(%f %f %f %f %d %d %d %d %f %d %d %d %f %f %f %d %1d%1d%1d%1d%1d%1d%1d%1d %d %d %d %1d%1d%1d",  // NOLINT
            &value_grid_voltage_, &value_grid_frequency_, &value_ac_output_voltage_,                         // NOLINT
            &value_ac_output_frequency_,                                                                     // NOLINT
            &value_ac_output_apparent_power_, &value_ac_output_active_power_, &value_output_load_percent_,   // NOLINT
            &value_bus_voltage_, &value_battery_voltage_, &value_battery_charging_current_,                  // NOLINT
            &value_battery_capacity_percent_, &value_inverter_heat_sink_temperature_,                        // NOLINT
            &value_pv_input_current_for_battery_, &value_pv_input_voltage_, &value_battery_voltage_scc_,     // NOLINT
            &value_battery_discharge_current_, &value_add_sbu_priority_version_,                             // NOLINT
            &value_configuration_status_, &value_scc_firmware_version_, &value_load_status_,                 // NOLINT
            &value_battery_voltage_to_steady_while_charging_, &value_charging_status_,                       // NOLINT
            &value_scc_charging_status_, &value_ac_charging_status_,                                         // NOLINT
            &value_battery_voltage_offset_for_fans_on_, &value_eeprom_version_, &value_pv_charging_power_,   // NOLINT
            &value_charging_to_floating_mode_, &value_switch_on_,                                            // NOLINT
            &value_dustproof_installed_);    `

This is my code, but sure it can be improved.
The final result:

[image]

lucasimons, 2021-09-06:
Is there a possibility to use the rj45 port to test with watchpower if there is a problem with that port?  Do you need an rj45 to rs232 and rs232 to usb cable or is there a direct rj45 to usb cable?

lucasimons, 2021-09-15:
I have bought this [cable](https://m.media-amazon.com/images/I/41G7S6UB7fL.jpg) from amazon and i have tried to connect the pc to the inverter with the rj45 port but it don t send any data, how i can do a test ? now i m using the solpiplog connected with usb and it works i have tried to connect the rp4 at th rj45 port but not work, so i think that the rj45 port of my inverter doesn t work ... Can i use the usb port to connect with the esp?

avion23, 2021-09-15:
RS232 is pretty simple. You have +15V and -15V. The voltages alternate rapidly while sending or receiving data.

The idle voltage is -15V. So you can just troubleshoot the connections with a multimeter.

I've added a picture to https://github.com/syssi/esphome-pipsolar/issues/3#issuecomment-912366449 where the top of the connector is Pin 1

lucasimons, 2021-09-15:
I believe that my inverter is not sending anything, but I can try, do you know if the two rj45 and usb outputs work in parallel or if one disables the other?

avion23, 2021-09-15:
I don't know. But just unplug the USB. Or try it out.
It's 2400 baud. You could also connect LEDs and confirm it visually.

lucasimons, 2021-09-16:
Today I tried via pc but nothing the inverter seems to send nothing, but if I connect the usb to the raspberry with solpiplog it works, I wanted to see if the serial port works but via pc with watch power I could not do it there are other software?

dominicknoetze, 2021-10-03:
> remeber the ESP is NOT TTL (5V) level. You need you uart signal at 3.3v Level. I had issues when directly connection the TTL-Level(5V). therefor perhaps the max3232 chip is better. Is made a board [...]

@andreashergert1984
Are you willing to share the schematics for your custom made dual inverter board?
I have my inverter running perfect with a pref board and a max3232 but I must admit it don't look  good.
Thanx

[239 more replies]