syssi/esphome-pipsolar #8: PIP 8048 support (Aka Axpert max)

SeByDocKy, 2021-12-11:
Hello,

Amazing job you did .... Is-il possible to extend your work to support as well the PIP8048 (axpert max I & II). Protocol changed just a bit...... In general in your amazing work, will be  fantastic to have an option to choose a particular PIP model even if I understood it's a huge work to be compatible with all PIP models

Here is the link for the PIP8048 protocol : https://www.transfernow.net/dl/202112112SpO2rcW
I did a custom sensor for it... but you work is even better....
Kind regards,

SeByDocKy, 2022-01-11:
Hello,

I looked into your official pipsolar integration in ESPhome. I think in order to made compatible with the PIP9048, it requires "just" to add the QPIGS2 command il the list of sended commands definied in pipsolar.h & pipsolar.cpp. 

The QPIGS2 command is quiet small to decode : 
```Computer: QPIGS2 <CRC><cr>
Device: (BB.B CCC.C DDDDD <CRC><cr>
Data Description Notes
a ( Start byte
b BB.B PV2 Input current E is an Integer ranging from 0 to 9. The units
is A.
MAX Communication Protocol
7
c CCC.C PV2 Input voltage U is an Integer ranging from 0 to 9. The units
is V.
d DDDDD PV2 Charging power M is an Integer ranging from 0 to 9. The uni```

Another small should be nice in your orgilnal QPIGS. Just change the name of the two variables/entities 
_value_pv_input_current_for_battery__ into _value_pv1_input_current__ and _value_pv_input_voltage__ into _value_pv1_input_voltage__

as _value_pv2_input_current__ and _value_pv2_input_voltage__ would be added by the QPIGS2 command....

Hope you will read this msg.

syssi, 2022-01-11:
Could you provide some response examples of your device?

syssi, 2022-01-11:
I'm a bit confused. The issue title mentions a PIP8084. You mentioned a PIP9048. Could you provide an explanation?

SeByDocKy, 2022-01-11:
> I'm a bit confused. The issue title mentions a PIP8084. You mentioned a PIP9048. Could you provide an explanation?

Sorry was a typo ... it's the 8048 (aka, axpert max I & II)

SeByDocKy, 2022-01-11:
I also maybe saw a tiny mistake in your code tor the QPIGS -> value_pv1_input_current_ ... it's scanned with a %d and should be a %f according to the manual (13th items in the scanf instructions)

syssi, 2022-01-11:
The scanf is a bit different per model/firmware/response. It's important to have some response example for improvements here.

SeByDocKy, 2022-01-11:
Ok I modified all your code to add the QPIGS2 command and everything related.... It's compiling... Need to wait sun to test results on my PIP8048. I am really impressed by your general programming level skills.... A masterpiece.

syssi, 2022-01-11:
I'm not the author. Just a contributor. Like you? ;-)

SeByDocKy, 2022-01-11:
> I'm not the author. Just a contributor. Like you? ;-)

Common :) ....  you did the JKBMS & the PIPsolar :)....  there are masterpieces...

SeByDocKy, 2022-01-13:
Hi before to test my modification, I am trying to collect at least some informations from the official pipisolar integration

Very strange..... I collected this log : 

```INFO Reading configuration /config/esphome/test_clapet.yaml...
WARNING 'test_clapet': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Detected timezone 'Europe/Paris'
INFO Starting log output from test_clapet.local using esphome API
INFO Successfully connected to test_clapet.local
[14:48:17][I][app:102]: ESPHome version 2022.2.0-dev compiled on Jan 13 2022, 14:45:39
[14:48:17][C][wifi:488]: WiFi:
[14:48:17][C][wifi:350]:   Local MAC: 3C:71:BF:A1:F0:80
[14:48:17][C][wifi:351]:   SSID: [redacted]
[14:48:17][C][wifi:352]:   IP Address: 192.168.1.87
[14:48:17][C][wifi:354]:   BSSID: [redacted]
[14:48:17][C][wifi:355]:   Hostname: 'test_clapet'
[14:48:17][C][wifi:357]:   Signal strength: -60 dB ▂▄▆█
[14:48:17][C][wifi:361]:   Channel: 12
[14:48:17][C][wifi:362]:   Subnet: 255.255.255.0
[14:48:17][C][wifi:363]:   Gateway: 192.168.1.1
[14:48:17][C][wifi:364]:   DNS1: 192.168.1.1
[14:48:17][C][wifi:365]:   DNS2: 192.168.1.1
[14:48:17][C][logger:233]: Logger:
[14:48:17][C][logger:234]:   Level: DEBUG
[14:48:17][C][logger:235]:   Log Baud Rate: 115200
[14:48:17][C][logger:236]:   Hardware UART: UART0
[14:48:17][C][uart.arduino_esp32:105]: UART Bus:
[14:48:17][C][uart.arduino_esp32:106]:   TX Pin: GPIO16
[14:48:17][C][uart.arduino_esp32:107]:   RX Pin: GPIO17
[14:48:17][C][uart.arduino_esp32:109]:   RX Buffer Size: 256
[14:48:17][C][uart.arduino_esp32:111]:   Baud Rate: 2400 baud
[14:48:17][C][uart.arduino_esp32:112]:   Data Bits: 8
[14:48:17][C][uart.arduino_esp32:113]:   Parity: NONE
[14:48:17][C][uart.arduino_esp32:114]:   Stop bits: 1
[14:48:17][C][homeassistant.time:010]: Home Assistant Time:
[14:48:17][C][homeassistant.time:011]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[14:48:17][C][pipsolar:860]: Pipsolar:
[14:48:17][C][pipsolar:861]: used commands:
[14:48:17][C][pipsolar:864]: QPIRI
[14:48:17][C][pipsolar:864]: QPIGS
[14:48:17][C][pipsolar:864]: QMOD
[14:48:17][C][pipsolar.text_sensor:010]: Pipsolar TextSensor 'pvbrain_axpertdevice_mode'
[14:48:17][C][pipsolar.text_sensor:010]: Pipsolar TextSensor 'pvbrain_axpert_last_qpigs'
[14:48:17][C][pipsolar.text_sensor:010]: Pipsolar TextSensor 'pvbrain_axpert_last_qpiri'
[14:48:17][C][captive_portal:144]: Captive Portal:
[14:48:17][C][ota:082]: Over-The-Air Updates:
[14:48:17][C][ota:083]:   Address: test_clapet.local:3232
[14:48:17][C][api:134]: API Server:
[14:48:17][C][api:135]:   Address: test_clapet.local:6053
[14:48:17][C][api:139]:   Using noise encryption: NO
[14:48:18][C][mdns:084]: mDNS:
[14:48:18][C][mdns:085]:   Hostname: test_clapet
[14:48:18][D][pipsolar:838]: Sending polling command : QPIGS with length 5
[14:48:18][D][pipsolar:772]: checking crc on incoming message
[14:48:18][D][pipsolar:775]: CRC OK
[14:48:18][D][pipsolar:448]: Decode QPIGS
[14:48:18][D][text_sensor:067]: 'pvbrain_axpert_last_qpigs': Sending state 'QPIGS'
[14:48:18][D][sensor:125]: 'pvbrain_axpert_ac_output_apparent_power': Sending state 0.00000 VA with 1 decimals of accuracy
[14:48:18][D][sensor:125]: 'pvbrain_axpert_ac_output_active_power': Sending state 0.00000 W with 1 decimals of accuracy
[14:48:18][D][sensor:125]: 'pvbrain_axpert_pv1_input_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[14:48:19][D][pipsolar:838]: Sending polling command : QMOD with length 4
[14:48:19][D][pipsolar:772]: checking crc on incoming message
[14:48:19][D][pipsolar:775]: CRC OK
[14:48:19][D][pipsolar:471]: Decode QMOD
[14:48:19][D][text_sensor:067]: 'pvbrain_axpertdevice_mode': Sending state 'M'
[14:48:20][D][pipsolar:838]: Sending polling command : QPIRI with length 5
[14:48:20][D][pipsolar:772]: checking crc on incoming message
[14:48:20][D][pipsolar:775]: CRC OK
[14:48:20][D][pipsolar:429]: Decode QPIRI
[14:48:20][D][text_sensor:067]: 'pvbrain_axpert_last_qpiri': Sending state 'QPIRI'
[... 154 lines cut ...]
#     pv2_input_voltage:
#       id: pvbrain_axpert_pv2_input_voltage
#       name: pvbrain_axpert_pv2_input_voltage  
  
text_sensor:
   - platform: pipsolar
     pipsolar_id: pip8048
     device_mode:
       id: pvbrain_axpert_device_mode
       name: pvbrain_axpertdevice_mode
     last_qpigs:
       id: pvbrain_axpert_last_qpigs
       name: pvbrain_axpert_last_qpigs
     last_qpiri:
       id: pvbrain_axpert_last_qpiri
       name: pvbrain_axpert_last_qpiri
```

Seems I am receiving anything ....
I tried to chage GPIO, etc .... Here is a small videoshot or the led's activity....: https://www.youtube.com/watch?v=6rha8HuPOes
I am not sure first of all if my physical connexion is OK. I correctly wired
RJ45(1,2,9) into the DB9(3,2,8). I check continuitues no problem. Of course I tried by permuting TXpin & RXpin in the yaml....

If I unplugged the RJ45 from the inverter .... I continue to receive a CRC OK .... so there is a big problem somewhere....
I will attempt with a direct connexion to watchpower to eventually discard a problem with the RS232 part of my inverter.

SeByDocKy, 2022-01-16:
Hi @syssi,

I solved my problem of bad wiring and also tested my fork to integrate the GPIGS2 command. It works perfectly....

SeByDocKy, 2022-02-22:
Hi @syssi 

Got a new question concerning the PIPsolar integration. With my PIP8048, I need to define the BATTERY_BULK_VOLTAGE output command. In a first attempt, I added the new command in the output/__init__py file pointing the PCCV command as decribed in the manual. It compiles fine... and I can see in the log that the command is correctly sent .... but no changes in the associaced QPRI command after a while. For the battery_float_voltage it's working...

Here is my output/__init__py
```import esphome.codegen as cg
import esphome.config_validation as cv
from esphome import automation
from esphome.components import output
from esphome.const import CONF_ID, CONF_VALUE
from .. import PIPSOLAR_COMPONENT_SCHEMA, CONF_PIPSOLAR_ID, pipsolar_ns

DEPENDENCIES = ["pipsolar"]

PipsolarOutput = pipsolar_ns.class_("PipsolarOutput", output.FloatOutput)
SetOutputAction = pipsolar_ns.class_("SetOutputAction", automation.Action)

CONF_POSSIBLE_VALUES = "possible_values"

# 3.11 PCVV<nn.n><cr>: Setting battery C.V. (constant voltage) charging voltage 48.0V ~ 58.4V for 48V unit
# battery_bulk_voltage;
# battery_recharge_voltage;     12V unit: 11V/11.3V/11.5V/11.8V/12V/12.3V/12.5V/12.8V
#                               24V unit: 22V/22.5V/23V/23.5V/24V/24.5V/25V/25.5V
#                               48V unit: 44V/45V/46V/47V/48V/49V/50V/51V
# battery_under_voltage;        40.0V ~ 48.0V for 48V unit
# battery_float_voltage;        48.0V ~ 58.4V for 48V unit
# battery_type;  00 for AGM, 01 for Flooded battery
# current_max_ac_charging_current;
# output_source_priority; 00 / 01 / 02
# charger_source_priority;  For HS: 00 for utility first, 01 for solar first, 02 for solar and utility, 03 for only solar charging
#                           For MS/MSX: 00 for utility first, 01 for solar first, 03 for only solar charging
# battery_redischarge_voltage;  12V unit: 00.0V12V/12.3V/12.5V/12.8V/13V/13.3V/13.5V/13.8V/14V/14.3V/14.5
#                               24V unit: 00.0V/24V/24.5V/25V/25.5V/26V/26.5V/27V/27.5V/28V/28.5V/29V
#                               48V unit: 00.0V48V/49V/50V/51V/52V/53V/54V/55V/56V/57V/58V

CONF_BATTERY_BULK_VOLTAGE = "battery_bulk_voltage"
CONF_BATTERY_RECHARGE_VOLTAGE = "battery_recharge_voltage"
CONF_BATTERY_UNDER_VOLTAGE = "battery_under_voltage"
CONF_BATTERY_FLOAT_VOLTAGE = "battery_float_voltage"
CONF_BATTERY_TYPE = "battery_type"
CONF_CURRENT_MAX_AC_CHARGING_CURRENT = "current_max_ac_charging_current"
CONF_CURRENT_MAX_CHARGING_CURRENT = "current_max_charging_current"
CONF_OUTPUT_SOURCE_PRIORITY = "output_source_priority"
CONF_CHARGER_SOURCE_PRIORITY = "charger_source_priority"
CONF_BATTERY_REDISCHARGE_VOLTAGE = "battery_redischarge_voltage"

TYPES = {
    CONF_BATTERY_BULK_VOLTAGE: (
	    [44.0, 45.0, 46.0, 47.0, 48.0, 49.0, 50.0, 51.0],
        "PCCV%02.1f",
	),
    CONF_BATTERY_RECHARGE_VOLTAGE: (
        [44.0, 45.0, 46.0, 47.0, 48.0, 49.0, 50.0, 51.0],
        "PBCV%02.1f",
    ),
    CONF_BATTERY_UNDER_VOLTAGE: (
        [40.0, 40.1, 42, 43, 44, 45, 46, 47, 48.0],
        "PSDV%02.1f",
    ),
    CONF_BATTERY_FLOAT_VOLTAGE: ([48.0, 49.0, 50.0, 51.0], "PBFT%02.1f"),
    CONF_BATTERY_TYPE: ([0, 1, 2], "PBT%02.0f"),
    CONF_CURRENT_MAX_AC_CHARGING_CURRENT: ([2, 10, 20], "MUCHGC0%02.0f"),
    CONF_CURRENT_MAX_CHARGING_CURRENT: ([10, 20, 30, 40], "MCHGC0%02.0f"),
    CONF_OUTPUT_SOURCE_PRIORITY: ([0, 1, 2], "POP%02.0f"),
    CONF_CHARGER_SOURCE_PRIORITY: ([0, 1, 2, 3], "PCP%02.0f"),
    CONF_BATTERY_REDISCHARGE_VOLTAGE: (
        [0, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58],
        "PBDV%02.1f",
    ),
}

CONFIG_SCHEMA = PIPSOLAR_COMPONENT_SCHEMA.extend(
    {
        cv.Optional(type): output.FLOAT_OUTPUT_SCHEMA.extend(
            {
                cv.Required(CONF_ID): cv.declare_id(PipsolarOutput),
                cv.Optional(CONF_POSSIBLE_VALUES, default=values): cv.All(
                    cv.ensure_list(cv.positive_float), cv.Length(min=1)
                ),
            }
        )
        for type, (values, _) in TYPES.items()
    }
)

async def to_code(config):
    paren = await cg.get_variable(config[CONF_PIPSOLAR_ID])

    for type, (_, command) in TYPES.items():
        if type in config:
            conf = config[type]
            var = cg.new_Pvariable(conf[CONF_ID])
            await output.register_output(var, conf)
            cg.add(var.set_parent(paren))
            cg.add(var.set_set_command(command))
            if (CONF_POSSIBLE_VALUES) in conf:
                cg.add(var.set_possible_values(conf[CONF_POSSIBLE_VALUES]))

@automation.register_action(
    "output.pipsolar.set_level",
    SetOutputAction,
    cv.Schema(
        {
            cv.Required(CONF_ID): cv.use_id(CONF_ID),
            cv.Required(CONF_VALUE): cv.templatable(cv.positive_float),
        }
    ),
)
def output_pipsolar_set_level_to_code(config, action_id, template_arg, args):
    paren = yield cg.get_variable(config[CONF_ID])
    var = cg.new_Pvariable(action_id, template_arg, paren)
    template_ = yield cg.templatable(config[CONF_VALUE], args, float)
    cg.add(var.set_level(template_))
    yield var

```

syssi, 2022-02-22:
Could you provide the manual another time? The wetransfer link isn't available anymore. Are you sure the command is called `PCCV`? Do you mean `PCVV` (Constant Voltage Charging Voltage)?

SeByDocKy, 2022-02-22:
> Could you provide the manual another time? The wetransfer link isn't available anymore. Are you sure the command is called `PCCV`? Do you mean `PCVV` (Constant Voltage Charging Voltage)?

Here is : https://drive.google.com/file/d/112Adm0MECE3c6t64uYWZcAL4N0FucdSJ/view?usp=sharing

p21 ....

SeByDocKy, 2022-02-22:
Mea culpa it's PCVV and not PCCV :)

syssi, 2022-03-12:
I've pushed your customized version to the feature branch `pip8048`:

```
external_components:
  - source: github://syssi/esphome-pipsolar@pip8048
    refresh: 0s
```

SeByDocKy, 2022-04-13:
Hi @syssi, just reading more deeply the axpert max protocol manual, and I discovered a very interesting output command "PBATCD" which allow to enable/disable charging/discharging for the inverter.  I think it can be extremely usefull for example with my PVbrain coupling a BMS and inverter. For example when the float voltage is reached, thanks to this command we can disable charging to let the BMS do is balancing job. In fact to be more general we can do "smarter" the  couple inverter,BMS like can allow CAN exchange with a seplos BMS for example

SeByDocKy, 2022-04-13:
To be more précise

```3.30 PBATCD<abc><cr>: Battery charge/discharge controlling command
Computer: PBATCD<abc><CRC><cr>
Device: (ACK<CRC><cr> if device accepts this command, otherwise, responds (NAK<CRC><cr>
a = Discharge completely on/off
b = Discharge on/off, but standby allowed (so small discharge allowed)
c = Charge completely on/off
Detail:
abc:
Charger Discharger
1 1 1 Enabled charger Enabled discharger
0 1 1
Enabled charger, depends on Prog16 setting if
AC source valid, charge 2A from AC, even if
prog. 16 is “only solar”. If prog. 16 is any other
setting, ignore and let charging from AC source
continue normally.
Disabled discharger and shut down unit
completely when insufficient PV or Grid is
present.
1 0 1
Enabled charger, depends on Prog16 setting if
AC source valid, charge 2A from AC, even if
prog. 16 is “only solar”. If prog. 16 is any other
setting, ignore and let charging from AC source
continue normally.
Disabled discharger but keep unit stay at standby
mode.
1 1 0 Disabled charger Enabled discharger
0 1 0 Disabled charger
Disabled discharger and shut down unit
completely when no PV or Grid is present.
1 0 0 Disabled charger
Disabled discharger but keep unit stay at standby
mode.
0 0 1 N/A N/A
0 0 0
Cleaned the enable/disable charger flags and
return to previous charger status.
Cleaned the enable/disable discharger flags and
return to previous discharger status.
```

I would like to  choose between PBATCD 111 <cr> and PBATCD 110 <cr>

SeByDocKy, 2022-05-03:
I think the custom select written by @andreashergert1984 could be modified to add the PBATCD with something like
```select:
  - platform: pipsolar
    pipsolar_id: pip8048
    charging_discharging_controls:
      id: charging_discharging_controls_select 
      name: charging_discharging_controls_select
      optionsmap:
        "Charging enable/discharging enable": "PBATCD111"
        "Charging disable/discharging enable": "PBATCD110"
        "Charging enable/discharging disable": "PBATCD011"
        "Charging disable/discharging disable": "PBATCD010"
      statusmap:
        "0": "Charging enable/discharging enable"
        "1": "Charging disable/discharging enable"
        "2": "Charging enable/discharging disable"
        "3": "Charging disable/discharging disable"
```

syssi, 2022-06-22:
Please try a clean build:

```
esphome clean fv-menic.yaml
esphome run fv-menic.yaml
```

PetrDubi, 2022-06-23:
Thank you for your repply. I was use esphome version 2021.10.x which is last in Home Assistant. Today I install last stable version from github (2022.6.2), use Clean build files and Install. Now it has compile without problems. Thank you

PetrDubi, 2022-06-28:
Good afternoon,
I have one question. Do you plan in future add querry for PV generated energy and output load energy?
Best regards

syssi, 2022-06-28:
Could you provide example requests and responses of the values you are looking for?

PetrDubi, 2022-06-28:
Query total PV generated energy
Computer: QET<CRC><cr>
Device: (NNNNNNNN<CRC><cr>
Generated energy, N: 0~9, unit: Wh Generated energy,

Query PV generated energy of year
Computer: QEYyyyy<cr>
Device: (NNNNNNNN<CRC><cr>
yyyy: Year, y: 0~9
NNNNNNNN: Generated energy, N: 0~9, unit: Wh

Query PV generated energy of month
Computer: QEMyyyymm <CRC><cr>
Device: (NNNNNNNN<CRC><cr>
yyyy: Year, y: 0~9
mm: Month, m: 0~9 
NNNNNNNN: Generated energy, N: 0~9, unit: Wh

Query PV generated energy of day
Computer: QEDyyyymmdd<CRC><cr>
Device: (NNNNNNNN<CRC><cr>
yyyy: Year, y: 0~9
mm: Month, m: 0~9
dd: Day,  d: 0~9
NNNNNNNN: Generated energy, N: 0~9, unit: Wh

Query total output load energy
Computer: QLT<CRC><cr>
Device: (NNNNNNNN<CRC><cr>
NNNNNNNN: Output load energy, N: 0~9, unit: Wh

Query output load energy of year
Computer: QLYyyyy<cr>
Device: (NNNNNNNN<CRC><cr>
yyyy: Year, y: 0~9
NNNNNNNN: Output load energy, N: 0~9, unit: Wh

Query output load energy of month
Computer: QLMyyyymm <CRC><cr>
Device: (NNNNNNNN<CRC><cr>
yyyy: Year, y: 0~9
mm: Month, m: 0~9 
NNNNNNNN: Output load energy, N: 0~9, unit: Wh

Query output load energy of day
Computer: QLDyyyymmdd<CRC><cr>
Device: (NNNNNNNN<CRC><cr>
yyyy: Year, y: 0~9
mm: Month, m: 0~9
dd: Day,  d: 0~9
NNNNNNNN: Output load energy, N: 0~9, unit: Wh

For year, month and day is required date and I suggest querry for this year, this month and today.

AtarianComputing, 2022-07-13:
> Hi @syssi,
> 
> I solved my problem of bad wiring and also tested my fork to integrate the GPIGS2 command. It works perfectly.... Your code is just incredible :)

Could you elaborate your bad wiring? I have this exact same problem with my inverter but I've tried three different TTL converters and all possible wiring and config options. I can't believe I've missed something but you never know...

rem72, 2022-11-22:
> I think the custom select written by @andreashergert1984 could be modified to add the PBATCD with something like
> 
> ```
>   - platform: pipsolar
>     pipsolar_id: pip8048
>     charging_discharging_controls:
>       id: charging_discharging_controls_select 
>       name: charging_discharging_controls_select
>       optionsmap:
>         "Charging enable/discharging enable": "PBATCD111"
>         "Charging disable/discharging enable": "PBATCD110"
>         "Charging enable/discharging disable": "PBATCD011"
>         "Charging disable/discharging disable": "PBATCD010"
>       statusmap:
>         "0": "Charging enable/discharging enable"
>         "1": "Charging disable/discharging enable"
>         "2": "Charging enable/discharging disable"
>         "3": "Charging disable/discharging disable"
> ```

Hi all, recently I moved fron my old axpert VMII 24V to a MAXII 8KW, than you so much for make my life good doing this.
Now I guess if you finally managed to control the charger of the inverter, would be great if it coud be controlled to charge the battery in case low voltage or just cheap grid price.
Thanks for your great job guys.

SeByDocKy, 2022-11-23:
I will try to modify the custom select in order to add & test the chargin/discharging  options...
Not sure it will work.....

SeByDocKy, 2022-11-23:
@rem72  I modified the custom select to accept the 
```- platform: pipsolar
    pipsolar_id: pip8048
    charging_discharging_controls:
      id: charging_discharging_controls_select 
      name: charging_discharging_controls_select
      optionsmap:
        "Charging enable/discharging enable": "PBATCD111"
        "Charging disable/discharging enable": "PBATCD110"
        "Charging enable/discharging disable": "PBATCD011"
        "Charging disable/discharging disable": "PBATCD010"
      statusmap:
        "0": "Charging enable/discharging enable"
        "1": "Charging disable/discharging enable"
        "2": "Charging enable/discharging disable"
        "3": "Charging disable/discharging disable"```

Need to be tested ...

rem72, 2022-11-23:
Hi, Itried to compile but havin errors:
Compiling /data/solar2/.pioenvs/solar2/src/main.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:866:14: error: 'class esphome::pipsolar::Pipsolar' has no member named 'set_charging_discharging_controls_select'
  866 |   inverter0->set_charging_discharging_controls_select(charging_discharging_controls_select);
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [/data/solar2/.pioenvs/solar2/src/main.cpp.o] Error 1
========================= [FAILED] Took 58.34 seconds =========================

rem72, 2022-11-23:
This is the only code I've changed:

`select:
  - platform: pipsolar
    pipsolar_id: inverter0
    output_source_priority:
      id: inverter0_output_source_priority_select
      name: inverter0_output_source_priority_select
      optionsmap:
        "Utility first": "POP00"
        "Solar only": "POP01"
        "Solar Battery Utility": "POP02"
      statusmap:
        "0": "Utility first"
        "1": "Solar only"
        "2": "Solar Battery Utility"

  - platform: pipsolar
    pipsolar_id: inverter0
    charging_discharging_controls:
      id: charging_discharging_controls_select 
      name: charging_discharging_controls_select
      optionsmap:
        "Charging enable/discharging enable": "PBATCD111"
        "Charging disable/discharging enable": "PBATCD110"
        "Charging enable/discharging disable": "PBATCD011"
        "Charging disable/discharging disable": "PBATCD010"
      statusmap:
        "0": "Charging enable/discharging enable"
        "1": "Charging disable/discharging enable"
        "2": "Charging enable/discharging disable"
        "3": "Charging disable/discharging disable"`

SeByDocKy, 2022-11-24:
Hummmm strange .... coz it's compiling for me. Are u sure to have the refresh: 0s option in the external component part ?

rem72, 2022-11-25:
Hi, I'm still havin' the same issue. I Tried clenaning build files.
What it's weird is that when I verify the code, all seems fine, no warning, no errors. It's only whe I compile it.
This is the external component code, seems well.

external_components:
  - source: github://syssi/esphome-pipsolar@pip8048
    refresh: 0s

rem72, 2022-11-26:
I tried also creating a new device from 0, but the same compilation issue, it's weird.

rem72, 2022-11-26:
I think this line is for the pipsolar.h:
`  PIPSOLAR_SELECT(charging_discharging_controls_select, QPIRI)
`
And these are for pipsolar.cpp:
```
       // special for charging_discharging_controls select
        if (this->charging_discharging_controls_select_) {
          std::string value = esphome::to_string(value_charging_discharging_controls_);
          this->charging_discharging_controls_select_->map_and_publish(value);
        }

``` Can you check them and update the fork?

SeByDocKy, 2022-11-26:
Well ... was compiling with my current... version. I updated the ESPhome dev now I got the error... effectively I forgot some element in the main pipsolar files.

SeByDocKy, 2022-11-26:
Hummm seems a bit more complex as expected :(

SeByDocKy, 2022-11-26:
Need also to implement the QBATCD sensor command as well

SeByDocKy, 2022-11-26:
Well .... I added the QBATCD command in the loop .... but I think the select is not fully functionnal. @syssi  can you help me ? espcially around line 444 in pipsolar.cpp  I don't know how to handle this part with the select...

syssi, 2022-11-27:
I will have a look but it will take some time.

dr3amr, 2023-01-12:
> Hi @syssi,
> 
> I solved my problem of bad wiring and also tested my fork to integrate the GPIGS2 command. It works perfectly.... Your code is just incredible :)

@SeByDocKy Can you share your wiring solution ? I also do not seem to be able to connect to my 8048 MAX. I have latest provided example / firmware flashed to ESP32, using GPIO pins 1 and 3 (as are default in example). 

ESP32   ->  RX goes to TX, TX goes to RX    ->   TTL  Adapter female DB9 -> straight male to male DB9 -> female DB9 RS232 to RJ45 official cable.

[image]
[image]

Is the oficial cable not good ? Saw somewhere that between the male and female DB9 connectors the RX and TX are inversed. The official cable pinout are RJ45 (1-TX,2-RX,4-VCC12v,8-GND) -> DB9 (2,3,VCC12v not connected, 5) tested for continuity. Does RX and TX need to be inversed between the official cable and the TTL DB9 connector ? meaning RJ45 (1,2,8) -> DB9 (3,2,5) ? The single led on the TTL adaptor does not blink at all.
I'm using an ESP32-CAM (esp32s) could this be a problem ? Can try with other esp32, also with esp8266.

SeByDocKy, 2023-01-12:
@dr3amr  you will be able to find my wiring in my github : https://github.com/SeByDocKy/ESPhome
look for pvbrain.yaml in the subsitutions parts

syssi, 2023-01-12:
My 2 cents: If you have the choice please use a RS232 converter board with 3.3V TTL f.e. MAX3232.

dr3amr, 2023-01-12:
> My 2 cents: If you have the choice please use a RS232 converter board with 3.3V TTL f.e. MAX3232.

I am, I am using an adapter similar to one bellow, it is powered from ESP32 side via 3.3v as shown in previous update's picture.

https://s13emagst.akamaized.net/products/17056/17055154/images/res_5738b5b253feac14eaaf98dd87922eae.jpg

Problem is, there is no writings on the adapter to let user know which pin is expecting to be TX and which RX ... thus the problem. I need to find a way to reverse TX and RX coming from the Inverter via the RJ45 cable.

Also another question, is it normal that one of the RX or TX (pin 2 or 3 do not recall) has 12v when measured with the GND (pin 5) on the DB9 connector (RJ45 - DB9 official cable) ?

syssi, 2023-01-12:
You can check the continuity between MAX3232 and the RS232 jack. The specifications of the MAX3232 will be clear.

Yes, the RS232 interface has a logic level of +12V/-12V:

> Valid RS-232 signals are either in the range of +3 V to +15 V or the range −3 V to −15 V with respect to the common ground (Figure 1-1). To be more specific, the driver output is Logic 0 when the voltage is between +5 V and +15 V and is Logic 1 when the voltage is between -5 V to -15 V.

https://www.ti.com/lit/an/slla544/slla544.pdf

You should be able to measure/see (TX PIN2 <-> GND PIN5) the outgoing traffic (`TX`) of your inverter as voltage around 12V.

dr3amr, 2023-01-12:
> You can check the continuity between MAX3232 and the RS232 jack. The specifications of the MAX3232 will be clear.
>
> Yes, the RS232 interface has a logic level of +12V/-12V: [...]

Thanks, will check for continuity. So TX on RS232 should go to T1 OUT (pin 14) of MAX3232 and RX on RS232 should go to R1 IN (pin 13) of MAX3232 ? 

11 | T1 IN | Transmission pin of first converter module for Microcontroller(Tx)
12 | R1 OUT | Reception pin of first converter module for Microcontroller(Rx)
13 | R1 IN | Reception pin of first converter module for RS232 cable
14 | T1 OUT | Transmission pin of first converter module for RS232 cable

[image]

dr3amr, 2023-01-12:
@syssi  @SeByDocKy Checked and seems TX pin 1 (RJ45) is reaching Pin 2 (RS232 female) on cable end, then straight connection to pin 2 of the TTL Adapter (RS232 female), then goes to pin 14 (T1 OUT of MAX3232 (should be to R1 IN?). Also RX pin 2 (RJ45) is reaching Pin 3 (RS232 female) on cable end, then straight connection to pin 3 of the TTL Adapter (RS232 female), then goes to pin 13 (R1 IN of MAX3232 (should be to T1 OUT?) ?

dr3amr, 2023-01-12:
> @syssi @SeByDocKy Checked and seems TX pin 1 (RJ45) is reaching Pin 2 (RS232 female) on cable end, then straight connection to pin 2 of the TTL Adapter (RS232 female), then goes to pin 14 (T1 OUT of MAX3232 (should be to R1 IN?). Also RX pin 2 (RJ45) is reaching Pin 3 (RS232 female) on cable end, then straight connection to pin 3 of the TTL Adapter (RS232 female), then goes to pin 13 (R1 IN of MAX3232 (should be to T1 OUT?) ? Forgive me as I am new to this :(

dr3amr, 2023-01-16:
@syssi @SeByDocKy I was able in the end to fix communication. Understanding which pins go from the RS232 female connector to the MAX3232 Adapter was key. Also another key point was understanding that the CROSS communication (RX-TX, TX-RX) is needed between inverter and the MAX3232 TTL Adapter, and from the Adapter to the ESP32 card it needs to be STRAIGHT (RX-RX, TX-TX).

Question: which sensor tells when inverter is either consuming from GRID or is in Bypass GRID mode ?

SeByDocKy, 2023-01-16:
There is not direct information from any command to know if it's in grid-tied or not but you can do a basic binary_sensor testing for exemple the frequency information

```
- platform: template
    name: ${name}_${inverter}_grid_input_on
    id: ${name}_${inverter}_grid_input_on
    lambda: |-
      if (id(${name}_${inverter}_grid_voltage).state > 0) {
        // Grid is connected
        return true;
      } else {
        // Grid is disconnected.
        return false;
      }
```

dr3amr, 2023-01-16:
> There is not direct information from any command to know if it's in grid-tied or not but you can do a basic binary_sensor testing for exemple with the frequency information
> 
> ```
> - platform: template
>     name: ${name}_${inverter}_grid_input_on
>     id: ${name}_${inverter}_grid_input_on
>     lambda: |-
>       if (id(${name}_${inverter}_grid_voltage).state > 0) {
>         // Grid is connected
>         return true;
>       } else {
>         // Grid is disconnected.
>         return false;
>       }
> ```

Hi, I do not mean when GRID is connected or not. I mean when Inverter consumes / takes energy from GRID (connected is almost all the time). There are sensors for AC Output, could there also be an AC Input Power for example ? Should be 0w when power is taken from Battery or Solar, but should have >0w when for example "Charger_source_priority" is set to Solar and utility, and not Solar only.

SeByDocKy, 2023-01-16:
Neither sensor as well :( .to know how much from Grid the inverter takes...

dr3amr, 2023-01-16:
> Neither sensor as well :( .to know how much from Grid the inverter takes...

At least we have Device_mode which matches with the LED strip. Values: "B" (Battery - turquoise), "L" (Grid-line - magenta), and I think also "S" (Solar - white). Will check the solar part tomorrow when there is some sun (light). Depending on the mode and both battery and ov outut w, you can figure from which source the ac_output_active_power W are from.

madurapa, 2024-08-28:
You can check the device mode if it is "L" meaning the inverter takes power from the grid.

```  - platform: template
    name: "${name} Conditioned Power"
    id: conditioned_power
    unit_of_measurement: "W"
    accuracy_decimals: 3
    internal: true
    lambda: |-
      if (id(${name}_${inverter}_device_mode).state == "L") {
        return id(${name}_${inverter}_ac_output_active_power).state;
      } else {
        return 0;
      }
```

dr3amr, 2025-08-10:
> @dr3amr <https://github.com/dr3amr> you can check the device mode if it
>     pipsolar_id: ${pipsolar_id}

Doing this instead:

{% set offgridpower = states('binary_sensor.off_grid_status') %}
{% set devicemode = states('sensor.esp_mppsolar_device_mode') %}
{% if devicemode == 'L' %}
{% set result = 'Grid Bypass' %}
{% elif devicemode == 'B' %}
    {% if offgridpower == 'on' %}
      {% set result = 'Solar + Battery' %}
    {% else %}
        {% set result = 'Battery' %}
    {% endif %}
{% endif %}
{{ result }}

The off_grid_status is another template cheching if any of the PV is > 0.