syssi/esphome-pipsolar #9: Compatibility for HMS 3K-24V

diybateriaslifepo4, 2022-01-03:
[image]
[image]
[image]

as we have talked about in the other issue to try to solve the problem with this model, problem is temperature and SBU not working

syssi, 2022-01-03:
```
external_components:
  - source: github://syssi/esphome-pipsolar@hms-3k-24v
    refresh: 0s
```

I've added two QPIGS response to the parser: https://github.com/syssi/esphome-pipsolar/commit/9ea662c191a1ff3a50e81d4f3924073d559141a6

IMO the different fields & values doesn't align with your device. Could you try to ask the seller for some specs / protocol description? A description for every value/field would be awesome:

```
Pos.:    1     2     3    4     5   ...  
Values: 001.0 00.0 229.0 50.0 0000 0000 000 413 27.23 000 100 0672 0000 000.0 00.00 00001 10010000 00 04 00000 000
```

diybateriaslifepo4, 2022-01-03:
I bought it from an aliexpress seller, I have asked him for various things like firmware update and he does not offer any support.
But I will try again, thank you.

diybateriaslifepo4, 2022-01-03:
> ```
> external_components:
>   - source: github://syssi/esphome-pipsolar@hms-3k-24v
>     refresh: 0s
> ```
> 
> I've added an example response (extracted from your logs) to the parser: [9ea662c](https://github.com/syssi/esphome-pipsolar/commit/9ea662c191a1ff3a50e81d4f3924073d559141a6)
> 
> IMO the different fields & values doesn't align with your device. Could you try to ask the seller for some specs / protocol description? A description for every value/field would be awesome:
> 
> ```
> Pos.:    1     2     3    4     5   ...  
> Values: 001.0 00.0 229.0 50.0 0000 0000 000 413 27.23 000 100 0672 0000 000.0 00.00 00001 10010000 00 04 00000 000
> ```

I have tried this, but nothing has changed, the temperature is still wrong and the SBU mode can not be activated
I leave you the current log in case you see something else.

[00:03:34][V][text_sensor:016]: 'pipsolar last_qflag': Received new state (EbDajkuvxyz
[00:03:34][D][text_sensor:067]: 'pipsolar last_qflag': Sending state '(EbDajkuvxyz'
[00:03:34][VV][scheduler:185]: Running interval 'update' with interval=1000 last_execution=223195 (now=224196)
[00:03:34][D][pipsolar:890]: Sending polling command : QPIRI with length 5
[00:03:35][D][pipsolar:824]: checking crc on incoming message
[00:03:35][D][pipsolar:827]: CRC OK
[00:03:35][D][pipsolar:429]: Decode QPIRI
[00:03:35][V][text_sensor:016]: 'pipsolar last_qpiri': Received new state (230.0 13.0 230.0 50.0 13.0 3000 2400 24.0 23.5 23.0 27.8 27.2 2 30 50 0 1 2 - 01 1 0 24.0 0 0
[00:03:35][D][text_sensor:067]: 'pipsolar last_qpiri': Sending state '(230.0 13.0 230.0 50.0 13.0 3000 2400 24.0 23.5 23.0 27.8 27.2 2 30 50 0 1 2 - 01 1 0 24.0 0 0'
[00:03:35][D][sensor:113]: 'pipsolar grid_rating_voltage': Sending state 230.00000 V with 1 decimals of accuracy
[00:03:35][D][sensor:113]: 'pipsolar grid_rating_current': Sending state 13.00000 A with 1 decimals of accuracy
[00:03:35][D][sensor:113]: 'pipsolar ac_output_rating_voltage': Sending state 230.00000 V with 1 decimals of accuracy
[00:03:35][D][sensor:113]: 'pipsolar ac_output_rating_frequency': Sending state 50.00000 Hz with 1 decimals of accuracy
[00:03:35][D][sensor:113]: 'pipsolar ac_output_rating_current': Sending state 13.00000 A with 1 decimals of accuracy
[00:03:35][D][sensor:113]: 'pipsolar ac_output_rating_apparent_power': Sending state 3000.00000 VA with 1 decimals of accuracy
[00:03:35][D][sensor:113]: 'pipsolar ac_output_rating_active_power': Sending state 2400.00000 W with 1 decimals of accuracy
[00:03:35][D][sensor:113]: 'pipsolar battery_rating_voltage': Sending state 24.00000 V with 1 decimals of accuracy
[00:03:35][D][sensor:113]: 'pipsolar battery_recharge_voltage': Sending state 23.50000 V with 1 decimals of accuracy
[00:03:35][D][sensor:113]: 'pipsolar battery_under_voltage': Sending state 23.00000 V with 1 decimals of accuracy
[00:03:35][D][sensor:113]: 'pipsolar battery_bulk_voltage': Sending state 27.80000 V with 1 decimals of accuracy
[00:03:35][D][sensor:113]: 'pipsolar battery_float_voltage': Sending state 27.20000 V with 1 decimals of accuracy
[00:03:35][D][sensor:113]: 'pipsolar battery_type': Sending state 2.00000  with 1 decimals of accuracy
[00:03:35][D][sensor:113]: 'pipsolar current_max_ac_charging_current': Sending state 30.00000 A with 1 decimals of accuracy
[00:03:35][D][sensor:113]: 'pipsolar current_max_charging_current': Sending state 50.00000 A with 1 decimals of accuracy
[00:03:35][D][sensor:113]: 'pipsolar input_voltage_range': Sending state 0.00000  with 1 decimals of accuracy
[00:03:35][D][sensor:113]: 'pipsolar output_source_priority': Sending state 1.00000  with 1 decimals of accuracy
[00:03:35][D][sensor:113]: 'pipsolar charger_source_priority': Sending state 2.00000  with 1 decimals of accuracy
[00:03:35][D][sensor:113]: 'pipsolar output_mode': Sending state 0.00000  with 1 decimals of accuracy
[00:03:35][D][sensor:113]: 'pipsolar battery_redischarge_voltage': Sending state 0.00000  with 1 decimals of accuracy
[00:03:35][D][sensor:113]: 'pipsolar pv_power_balance': Sending state 0.00000  with 1 decimals of accuracy
[00:03:35][VV][scheduler:185]: Running interval 'update' with interval=1000 last_execution=224195 (now=225197)
[00:03:35][D][pipsolar:890]: Sending polling command : QPIGS with length 5
[00:03:36][D][pipsolar:824]: checking crc on incoming message
[00:03:36][D][pipsolar:827]: CRC OK
[00:03:36][D][pipsolar:448]: Decode QPIGS
[00:03:36][V][text_sensor:016]: 'pipsolar last_qpigs': Received new state (001.0 00.0 228.0 50.0 0000 0000 000 413 27.23 000 100 0629 0000 000.0 00.00 00001 10010000 00 04 00000 000
[00:03:36][D][text_sensor:067]: 'pipsolar last_qpigs': Sending state '(001.0 00.0 228.0 50.0 0000 0000 000 413 27.23 000 100 0629 0000 000.0 00.00 00001 10010000 00 04 00000 000'
[00:03:36][D][sensor:113]: 'pipsolar grid_voltage': Sending state 1.00000 V with 1 decimals of accuracy
[00:03:36][D][sensor:113]: 'pipsolar grid_frequency': Sending state 0.00000 Hz with 1 decimals of accuracy
[00:03:36][D][sensor:113]: 'pipsolar ac_output_voltage': Sending state 228.00000 V with 1 decimals of accuracy
[00:03:36][D][sensor:113]: 'pipsolar ac_output_frequency': Sending state 50.00000 Hz with 1 decimals of accuracy
[00:03:36][D][sensor:113]: 'pipsolar ac_output_apparent_power': Sending state 0.00000 VA with 1 decimals of accuracy
[00:03:36][D][sensor:113]: 'pipsolar ac_output_active_power': Sending state 0.00000 W with 1 decimals of accuracy
[... cut ...]
[00:03:37][D][pipsolar:523]: Decode QMOD
[00:03:37][V][text_sensor:016]: 'pipsolar last_qmod': Received new state (B
[00:03:37][D][text_sensor:067]: 'pipsolar last_qmod': Sending state '(B'
[00:03:37][V][text_sensor:016]: 'pipsolar device_mode': Received new state B
[00:03:37][D][text_sensor:067]: 'pipsolar device_mode': Sending state 'B'
[00:03:37][VV][scheduler:185]: Running interval 'update' with interval=1000 last_execution=226195 (now=227198)
[00:03:37][D][pipsolar:890]: Sending polling command : QFLAG with length 5

diybateriaslifepo4, 2022-01-03:
One more thing, the equipment is without solar panels or 220v input connected, I have it on the test table connected to a power supply.

diybateriaslifepo4, 2022-01-03:
I have just done a test, I have connected the inverter to the watchpower software, and I have activated the SBU mode, then I have connected it again to the esp8266 and it appears that it is in SBU mode correctly.
I'll leave you the log for what it's worth:

[00:12:01][VV][scheduler:185]: Running interval 'update' with interval=1000 last_execution=730195 (now=731195)
[00:12:01][D][pipsolar:890]: Sending polling command : QPIRI with length 5
[00:12:02][D][pipsolar:824]: checking crc on incoming message
[00:12:02][D][pipsolar:827]: CRC OK
[00:12:02][D][pipsolar:429]: Decode QPIRI
[00:12:02][V][text_sensor:016]: 'pipsolar last_qpiri': Received new state (230.0 13.0 230.0 50.0 13.0 3000 2400 24.0 23.5 23.0 27.8 27.2 2 30 50 0 2 2 - 01 1 0 24.0 0 0
[00:12:02][D][text_sensor:067]: 'pipsolar last_qpiri': Sending state '(230.0 13.0 230.0 50.0 13.0 3000 2400 24.0 23.5 23.0 27.8 27.2 2 30 50 0 2 2 - 01 1 0 24.0 0 0'
[00:12:02][D][sensor:113]: 'pipsolar grid_rating_voltage': Sending state 230.00000 V with 1 decimals of accuracy
[00:12:02][D][sensor:113]: 'pipsolar grid_rating_current': Sending state 13.00000 A with 1 decimals of accuracy
[00:12:02][D][sensor:113]: 'pipsolar ac_output_rating_voltage': Sending state 230.00000 V with 1 decimals of accuracy
[00:12:02][D][sensor:113]: 'pipsolar ac_output_rating_frequency': Sending state 50.00000 Hz with 1 decimals of accuracy
[00:12:02][D][sensor:113]: 'pipsolar ac_output_rating_current': Sending state 13.00000 A with 1 decimals of accuracy
[00:12:02][D][sensor:113]: 'pipsolar ac_output_rating_apparent_power': Sending state 3000.00000 VA with 1 decimals of accuracy
[00:12:02][D][sensor:113]: 'pipsolar ac_output_rating_active_power': Sending state 2400.00000 W with 1 decimals of accuracy
[00:12:02][D][sensor:113]: 'pipsolar battery_rating_voltage': Sending state 24.00000 V with 1 decimals of accuracy
[00:12:02][D][sensor:113]: 'pipsolar battery_recharge_voltage': Sending state 23.50000 V with 1 decimals of accuracy
[00:12:02][D][sensor:113]: 'pipsolar battery_under_voltage': Sending state 23.00000 V with 1 decimals of accuracy
[00:12:02][D][sensor:113]: 'pipsolar battery_bulk_voltage': Sending state 27.80000 V with 1 decimals of accuracy
[00:12:02][D][sensor:113]: 'pipsolar battery_float_voltage': Sending state 27.20000 V with 1 decimals of accuracy
[00:12:02][D][sensor:113]: 'pipsolar battery_type': Sending state 2.00000  with 1 decimals of accuracy
[00:12:02][D][sensor:113]: 'pipsolar current_max_ac_charging_current': Sending state 30.00000 A with 1 decimals of accuracy
[00:12:02][D][sensor:113]: 'pipsolar current_max_charging_current': Sending state 50.00000 A with 1 decimals of accuracy
[00:12:02][D][sensor:113]: 'pipsolar input_voltage_range': Sending state 0.00000  with 1 decimals of accuracy
[00:12:02][D][sensor:113]: 'pipsolar output_source_priority': Sending state 2.00000  with 1 decimals of accuracy
[00:12:02][D][sensor:113]: 'pipsolar charger_source_priority': Sending state 2.00000  with 1 decimals of accuracy
[00:12:02][D][sensor:113]: 'pipsolar output_mode': Sending state 0.00000  with 1 decimals of accuracy
[00:12:02][D][sensor:113]: 'pipsolar battery_redischarge_voltage': Sending state 0.00000  with 1 decimals of accuracy
[00:12:02][D][sensor:113]: 'pipsolar pv_power_balance': Sending state 0.00000  with 1 decimals of accuracy
[00:12:02][VV][scheduler:185]: Running interval 'update' with interval=1000 last_execution=731195 (now=732197)
[00:12:02][D][pipsolar:890]: Sending polling command : QPIGS with length 5
[00:12:03][D][pipsolar:824]: checking crc on incoming message
[00:12:03][D][pipsolar:827]: CRC OK
[00:12:03][D][pipsolar:448]: Decode QPIGS
[00:12:03][V][text_sensor:016]: 'pipsolar last_qpigs': Received new state (001.0 00.0 229.0 50.0 0000 0000 000 414 27.29 000 100 0575 0000 000.0 00.00 00001 10010000 00 04 00000 000
[00:12:03][D][text_sensor:067]: 'pipsolar last_qpigs': Sending state '(001.0 00.0 229.0 50.0 0000 0000 000 414 27.29 000 100 0575 0000 000.0 00.00 00001 10010000 00 04 00000 000'
[00:12:03][D][sensor:113]: 'pipsolar grid_voltage': Sending state 1.00000 V with 1 decimals of accuracy
[00:12:03][D][sensor:113]: 'pipsolar grid_frequency': Sending state 0.00000 Hz with 1 decimals of accuracy
[00:12:03][D][sensor:113]: 'pipsolar ac_output_voltage': Sending state 229.00000 V with 1 decimals of accuracy
[00:12:03][D][sensor:113]: 'pipsolar ac_output_frequency': Sending state 50.00000 Hz with 1 decimals of accuracy
[00:12:03][D][sensor:113]: 'pipsolar ac_output_apparent_power': Sending state 0.00000 VA with 1 decimals of accuracy
[00:12:03][D][sensor:113]: 'pipsolar ac_output_active_power': Sending state 0.00000 W with 1 decimals of accuracy
[00:12:03][D][sensor:113]: 'pipsolar output_load_percent': Sending state 0.00000 % with 1 decimals of accuracy
[00:12:03][D][sensor:113]: 'pipsolar bus_voltage': Sending state 414.00000 V with 1 decimals of accuracy
[00:12:03][D][sensor:113]: 'pipsolar battery_voltage': Sending state 27.29000 V with 1 decimals of accuracy
[00:12:03][D][sensor:113]: 'pipsolar battery_charging_current': Sending state 0.00000 A with 1 decimals of accuracy
[00:12:03][D][sensor:113]: 'pipsolar battery_capacity_percent': Sending state 100.00000 % with 1 decimals of accuracy
[00:12:03][D][sensor:113]: 'pipsolar inverter_heat_sink_temperature': Sending state 575.00000 °C with 1 decimals of accuracy
[00:12:03][D][sensor:113]: 'pipsolar pv_input_current_for_battery': Sending state 0.00000 A with 1 decimals of accuracy
[... cut ...]
[00:12:03][D][text_sensor:067]: 'pipsolar last_qmod': Sending state '(B'
[00:12:03][V][text_sensor:016]: 'pipsolar device_mode': Received new state B
[00:12:03][D][text_sensor:067]: 'pipsolar device_mode': Sending state 'B'
[00:12:04][VV][scheduler:185]: Running interval 'update' with interval=1000 last_execution=733195 (now=734195)
[00:12:04][D][pipsolar:890]: Sending polling command : QFLAG with length 5
[00:12:04][D][pipsolar:824]: checking crc on incoming message
[00:12:04][D][pipsolar:827]: CRC OK
[00:12:04][D][pipsolar:531]: Decode QFLAG

diybateriaslifepo4, 2022-01-03:
[image]
so it appears to me as correctly activated, if I select another of the two modes it changes correctly, but I can no longer select SBU again.

syssi, 2022-01-03:
Good job! Using watchpower is a good idea. It's possible to capture the command to switch the modes from watchtower? It looks like sensing the mode works fine but the command to change the mode is bogus.

diybateriaslifepo4, 2022-01-04:
> Good job! Using watchpower is a good idea. It's possible to capture the command to switch the modes from watchtower? It looks like sensing the mode works fine but the command to change the mode is bogus.

I don't know how to do it, if you teach me I will do it 😅

syssi, 2022-01-04:
@diybateriaslifepo4 Could you provide your yaml configuration?

syssi, 2022-01-04:
The important parts of your logs:

```
# Try to enable POP02 (Solar->Battery->Utility)

[09:12:08][D][switch:013]: 'pipsolar output_source_priority_battery' Turning ON.
[09:12:08][D][pipsolar:908]: got command: POP02
[09:12:08][D][pipsolar:900]: Command queued successfully: POP02 with length 5 at position 3
[09:12:08][D][pipsolar:824]: checking crc on incoming message
[09:12:08][D][pipsolar:827]: CRC OK
[09:12:08][D][pipsolar:523]: Decode QMOD
[09:12:08][V][text_sensor:016]: 'pipsolar last_qmod': Received new state (B
[09:12:08][D][text_sensor:067]: 'pipsolar last_qmod': Sending state '(B'
[09:12:08][V][text_sensor:016]: 'pipsolar device_mode': Received new state B
[09:12:08][D][text_sensor:067]: 'pipsolar device_mode': Sending state 'B'
[09:12:08][D][pipsolar:859]: Sending command from queue: POP02 with length 5
[09:12:08][D][pipsolar:042]: response length for command OK
[09:12:08][D][pipsolar:824]: checking crc on incoming message
[09:12:08][D][pipsolar:827]: CRC OK
[09:12:08][D][pipsolar:048]: command not successful
#                                                           ^^^ failed
```

```
# Try to enable POP01: Solar->Utility->Battery
[10:37:00][D][switch:013]: 'pipsolar output_source_priority_solar' Turning ON.
[10:37:00][D][pipsolar:856]: got command: POP01
[10:37:00][D][pipsolar:848]: Command queued successfully: POP01 with length 5 at position 0
[10:37:00][D][pipsolar:807]: Sending command from queue: POP01 with length 5
[10:37:00][D][pipsolar:042]: response length for command OK
[10:37:00][D][pipsolar:772]: checking crc on incoming message
[10:37:00][D][pipsolar:775]: CRC OK
[10:37:00][D][pipsolar:046]: command successful
                                                             ^^^^^^^^^ success
```

The device responds with "ACK" on success and "NACK" if the command failed.

syssi, 2022-01-04:
I've found this issue: https://github.com/jblance/mpp-solar/issues/46

And this important detail:

> Does your CRC generator work with the POP02 (battery output source priority)command? Mine gives E2 0A as the CRC but the box expects E2 0B.
https://forums.aeva.asn.au/viewtopic.php?p=53712#p53712

We should increase the log level of the uart component. ESPHome will print every transmitted & received byte to the log. The important part is the POP02<CRC> request. 

```
$ echo -n POP02 | hexdump -C
00000000  50 4f 50 30 32                                    |POP02|
```

We are looking for something like this:

```
50.4F.50.30.32.E2.0A
# or
50.4F.50.30.32.E2.0B
```

syssi, 2022-01-04:
How to increase the log level:

```
logger:
  level: VERY_VERBOSE
  logs:
    uart: VERY_VERBOSE
    uart.arduino_esp8266: VERY_VERBOSE
    # Make some components less verbose
    api.service: WARN
    ota: WARN
```

diybateriaslifepo4, 2022-01-04:
> @diybateriaslifepo4 Could you provide your yaml configuration?

substitutions:
  name: pipsolar

esphome:
  name: ${name}
  platform: ESP8266
  board: d1_mini
  
external_components:
  - source: github://syssi/esphome-pipsolar@hms-3k-24v
    refresh: 0s

# Enable logging
logger:
  baud_rate: 0
  level: VERY_VERBOSE
  logs:
    uart: WARN
    uart_esp8266: WARN
    api.service: WARN
    ota: WARN
    sensor: DEBUG
    pipsolar: VERY_VERBOSE
  
# Enable Home Assistant API
api:

ota:
  password: "xxxxxxxxx"

wifi:
  ssid: "xxxxx"
  password: "xxxxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp8266Ex-Pip-Solar"
    password: "OZ4dkBrdr9E2"

captive_portal:

uart:
  id: uart0
  baud_rate: 2400
  tx_pin: GPIO1
  rx_pin: GPIO3

pipsolar:
  uart_id: uart0
  id: inverter0
#  update_interval: 1s

sensor:
  - platform: pipsolar
    pipsolar_id: inverter0
    # QPIRI
    grid_rating_voltage:
      name: "${name} grid_rating_voltage"
    grid_rating_current:
      name: "${name} grid_rating_current"
    ac_output_rating_voltage:
      name: "${name} ac_output_rating_voltage"
    ac_output_rating_frequency:
      name: "${name} ac_output_rating_frequency"
    ac_output_rating_current:
      name: "${name} ac_output_rating_current"
    ac_output_rating_apparent_power:
      name: "${name} ac_output_rating_apparent_power"
[... 127 lines cut ...]
      name: "${name} over_temperature_restart_function"
#    backlight_on:
#      name: "${name} backlight_on"

switch:
  - platform: pipsolar
    pipsolar_id: inverter0
    output_source_priority_utility:
      name: "${name} output_source_priority_utility" #UsB
    output_source_priority_solar:
      name: "${name} output_source_priority_solar" #SUb
    output_source_priority_battery:
      name: "${name} output_source_priority_battery" #SbU
    input_voltage_range:
      name: "${name} input_voltage_range"
#    pv_ok_condition_for_parallel:
#      name: "${name} pv_ok_condition_for_parallel"
    pv_power_balance:
      name: "${name} pv_power_balance"

output:
  - platform: pipsolar
    pipsolar_id: inverter0
    battery_recharge_voltage:
      id: inverter0_battery_recharge_voltage_out

diybateriaslifepo4, 2022-01-04:
> 

`substitutions:
  name: pipsolar

esphome:
  name: ${name}
  platform: ESP8266
  board: d1_mini
  
external_components:
  - source: github://syssi/esphome-pipsolar@hms-3k-24v
    refresh: 0s

# Enable logging
logger:
  baud_rate: 0
  level: VERY_VERBOSE
  logs:
    uart: WARN
    uart_esp8266: WARN
    api.service: WARN
    ota: WARN
    sensor: DEBUG
    pipsolar: VERY_VERBOSE
  
# Enable Home Assistant API
api:

ota:

wifi:
  ssid: "xxxxxxxxx"
  password: "xxxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp8266Ex-Pip-Solar"
    password: "OZ4dkBrdr9E2"

captive_portal:

uart:
  id: uart0
  baud_rate: 2400
  tx_pin: GPIO1
  rx_pin: GPIO3

pipsolar:
  uart_id: uart0
  id: inverter0
#  update_interval: 1s

sensor:
  - platform: pipsolar
    pipsolar_id: inverter0
    # QPIRI
    grid_rating_voltage:
      name: "${name} grid_rating_voltage"
    grid_rating_current:
      name: "${name} grid_rating_current"
    ac_output_rating_voltage:
      name: "${name} ac_output_rating_voltage"
    ac_output_rating_frequency:
      name: "${name} ac_output_rating_frequency"
    ac_output_rating_current:
      name: "${name} ac_output_rating_current"
    ac_output_rating_apparent_power:
      name: "${name} ac_output_rating_apparent_power"
    ac_output_rating_active_power:
[... 126 lines cut ...]
      name: "${name} over_temperature_restart_function"
#    backlight_on:
#      name: "${name} backlight_on"

switch:
  - platform: pipsolar
    pipsolar_id: inverter0
    output_source_priority_utility:
      name: "${name} output_source_priority_utility" #UsB
    output_source_priority_solar:
      name: "${name} output_source_priority_solar" #SUb
    output_source_priority_battery:
      name: "${name} output_source_priority_battery" #SbU
    input_voltage_range:
      name: "${name} input_voltage_range"
#    pv_ok_condition_for_parallel:
#      name: "${name} pv_ok_condition_for_parallel"
    pv_power_balance:
      name: "${name} pv_power_balance"

output:
  - platform: pipsolar
    pipsolar_id: inverter0
    battery_recharge_voltage:
      id: inverter0_battery_recharge_voltage_out`

diybateriaslifepo4, 2022-01-04:
> @diybateriaslifepo4 Could you provide your yaml configuration?

[https://linkode.org/#Q3vnEtNCyBZ7rHxziKy032]

lucasimons, 2022-02-21:
> > @diybateriaslifepo4 Could you provide your yaml configuration?
> 
> [https://linkode.org/#Q3vnEtNCyBZ7rHxziKy032](url)

Hi, have you managed to get the various activations to work sbu uti, ect? do you have a picture of the connection of the rj45 to the converter?