syssi/esphome-pipsolar #12: EAsun iSolar SM IV 5.6kW support

bogdanr, 2022-03-31:
Hello,

I have this EAsun iSolar SM IV inverter but most commands seem to report values equal to zero.

```
[16:39:23][D][pipsolar:775]: CRC OK
[16:39:23][D][pipsolar:448]: Decode QPIGS
[16:39:23][D][text_sensor:067]: 'inverter0_last_qpigs': Sending state 'QPIGS'
[16:39:23][D][sensor:125]: 'inverter0_grid_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[16:39:23][D][sensor:125]: 'inverter0_grid_frequency': Sending state 0.00000 Hz with 1 decimals of accuracy
[16:39:23][D][sensor:125]: 'inverter0_ac_output_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[16:39:23][D][sensor:125]: 'inverter0_ac_output_frequency': Sending state 0.00000 Hz with 1 decimals of accuracy
[16:39:23][D][sensor:125]: 'inverter0_ac_output_apparent_power': Sending state 0.00000 VA with 1 decimals of accuracy
[16:39:23][D][sensor:125]: 'inverter0_ac_output_active_power': Sending state 0.00000 W with 1 decimals of accuracy
[16:39:23][D][sensor:125]: 'inverter0_output_load_percent': Sending state 0.00000 % with 1 decimals of accuracy
[16:39:23][D][sensor:125]: 'inverter0_bus_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[16:39:23][D][sensor:125]: 'inverter0_battery_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[16:39:23][D][sensor:125]: 'inverter0_battery_capacity_percent': Sending state 0.00000 % with 1 decimals of accuracy
[16:39:23][D][sensor:125]: 'inverter0_inverter_heat_sink_temperature': Sending state 0.00000 °C with 1 decimals of accuracy
[16:39:23][D][sensor:125]: 'inverter0_pv_input_current_for_battery': Sending state 0.00000 A with 1 decimals of accuracy
[16:39:23][D][sensor:125]: 'inverter0_pv_input_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[16:39:23][D][sensor:125]: 'inverter0_battery_voltage_scc': Sending state 0.00000 V with 1 decimals of accuracy
[16:39:24][D][sensor:125]: 'inverter0_battery_discharge_current': Sending state 0.00000 A with 1 decimals of accuracy
[16:39:24][D][sensor:125]: 'inverter0_pv_charging_power': Sending state 0.00000 W with 1 decimals of accuracy
[16:39:24][D][pipsolar:838]: Sending polling command : QPIWS with length 5
[16:39:24][D][pipsolar:772]: checking crc on incoming message
[16:39:24][D][pipsolar:775]: CRC OK
[16:39:24][D][pipsolar:525]: Decode QPIWS
[16:39:25][D][pipsolar:838]: Sending polling command : QMOD with length 4
[16:39:25][D][pipsolar:772]: checking crc on incoming message
[16:39:25][D][pipsolar:775]: CRC OK
```

How could I make this work?

syssi, 2022-03-31:
Could you enable the debug output of the `uart` component and provide some more logs? It looks like the response to `QPIGS` is just `QPIGS`.

```
uart:
  id: uart0
  baud_rate: 9600
  tx_pin: XXX
  rx_pin: YYY
  debug:
    direction: BOTH
```

bogdanr, 2022-03-31:
```
[18:07:58][D][pipsolar:429]: Decode QPIRI
[18:07:58][D][sensor:125]: 'inverter0_grid_rating_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:58][D][sensor:125]: 'inverter0_grid_rating_current': Sending state 0.00000 A with 1 decimals of accuracy
[18:07:58][D][sensor:125]: 'inverter0_battery_rating_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:58][D][sensor:125]: 'inverter0_input_voltage_range': Sending state 0.00000  with 1 decimals of accuracy
[18:07:58][D][sensor:125]: 'inverter0_output_source_priority': Sending state 0.00000  with 1 decimals of accuracy
[18:07:58][D][sensor:125]: 'inverter0_charger_source_priority': Sending state 0.00000  with 1 decimals of accuracy
[18:07:58][D][sensor:125]: 'inverter0_machine_type': Sending state 0.00000  with 1 decimals of accuracy
[18:07:58][D][sensor:125]: 'inverter0_topology': Sending state 0.00000  with 1 decimals of accuracy
[18:07:58][D][sensor:125]: 'inverter0_battery_redischarge_voltage': Sending state 0.00000  with 1 decimals of accuracy
[18:07:58][D][uart_debug:114]: <<< 51:50:49:52:49:F8:54:0D
[18:07:59][D][pipsolar:838]: Sending polling command : QPIGS with length 5
[18:07:59][D][uart_debug:114]: >>> 51:50:49:47:53:B7:A9:0D
[18:07:59][D][pipsolar:772]: checking crc on incoming message
[18:07:59][D][pipsolar:775]: CRC OK
[18:07:59][D][pipsolar:448]: Decode QPIGS
[18:07:59][D][text_sensor:067]: 'inverter0_last_qpigs': Sending state 'QPIGS'
[18:07:59][D][sensor:125]: 'inverter0_grid_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:59][D][sensor:125]: 'inverter0_grid_frequency': Sending state 0.00000 Hz with 1 decimals of accuracy
[18:07:59][D][sensor:125]: 'inverter0_ac_output_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:59][D][sensor:125]: 'inverter0_ac_output_frequency': Sending state 0.00000 Hz with 1 decimals of accuracy
[18:07:59][D][sensor:125]: 'inverter0_ac_output_apparent_power': Sending state 0.00000 VA with 1 decimals of accuracy
[18:07:59][D][sensor:125]: 'inverter0_ac_output_active_power': Sending state 0.00000 W with 1 decimals of accuracy
[18:07:59][D][sensor:125]: 'inverter0_output_load_percent': Sending state 0.00000 % with 1 decimals of accuracy
[18:07:59][D][sensor:125]: 'inverter0_bus_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:59][D][sensor:125]: 'inverter0_battery_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:59][D][sensor:125]: 'inverter0_battery_capacity_percent': Sending state 0.00000 % with 1 decimals of accuracy
[18:07:59][D][sensor:125]: 'inverter0_inverter_heat_sink_temperature': Sending state 0.00000 °C with 1 decimals of accuracy
[18:07:59][D][sensor:125]: 'inverter0_pv_input_current_for_battery': Sending state 0.00000 A with 1 decimals of accuracy
[18:07:59][D][sensor:125]: 'inverter0_pv_input_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:59][D][sensor:125]: 'inverter0_battery_voltage_scc': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:59][D][sensor:125]: 'inverter0_battery_discharge_current': Sending state 0.00000 A with 1 decimals of accuracy
[18:07:59][D][sensor:125]: 'inverter0_pv_charging_power': Sending state 0.00000 W with 1 decimals of accuracy
[18:07:59][D][uart_debug:114]: <<< 51:50:49:47:53:B7:A9:0D
[18:08:00][D][pipsolar:838]: Sending polling command : QPIWS with length 5
[18:08:00][D][uart_debug:114]: >>> 51:50:49:57:53:B4:DA:0D
[18:08:00][D][pipsolar:772]: checking crc on incoming message
[18:08:00][D][pipsolar:775]: CRC OK
[18:08:00][D][pipsolar:525]: Decode QPIWS
[18:08:00][D][uart_debug:114]: <<< 51:50:49:57:53:B4:DA:0D
[18:08:01][D][pipsolar:838]: Sending polling command : QMOD with length 4
[18:08:01][D][uart_debug:114]: >>> 51:4D:4F:44:49:C1:0D
[18:08:01][D][pipsolar:772]: checking crc on incoming message
[18:08:01][D][pipsolar:775]: CRC OK
[18:08:01][D][pipsolar:471]: Decode QMOD
[18:08:01][D][text_sensor:067]: 'inverter0_device_mode': Sending state 'M'
[18:08:01][D][uart_debug:114]: <<< 51:4D:4F:44:49:C1:0D
[18:08:02][D][pipsolar:838]: Sending polling command : QPIRI with length 5
[18:08:02][D][uart_debug:114]: >>> 51:50:49:52:49:F8:54:0D
[18:08:02][D][pipsolar:772]: checking crc on incoming message
[18:08:02][D][pipsolar:775]: CRC OK
```
But please not my baud_rate is 2400.

syssi, 2022-03-31:
Sorry for the confusion. I didn't want to suggest another baudrate. The important part was:

```
  debug:
    direction: BOTH
```

syssi, 2022-04-02:
Could you provide a photos of your wiring?

bogdanr, 2022-04-07:
I just received a new adapter and I tried with this one instead. The messages changed but they still don't look good to me.
```
[19:31:40][D][pipsolar:755]: timeout command to poll: QPIWS
[19:31:40][D][uart_debug:114]: <<< FF:7F:EF:FF:FF:DF:FF:FD:DF:FF:FD:FD:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:DF:FF:FF:FF:FF:FF:7F:FF:FF:FF:FF:FF:FF:FF:FF:7F:FF
[19:31:40][D][pipsolar:838]: Sending polling command : QMOD with length 4
[19:31:40][D][uart_debug:114]: >>> 51:4D:4F:44:49:C1:0D
[19:31:41][D][uart_debug:114]: <<< FF:FF:DF:FF:FF:FF:FF:FF:FD:FF:FF:FF:FF:EF:DF:FF:FF:FF:FF:FF:FF:F7:FF:FD:FF:FB:FF:FF:FF:DF:DF:F7:FF:FF:FF:FF:FF:FF:FF:FF:FF:DF:FF:FF:FF:FF:FF:FF:FF:ED:FF:FF:DF:DF:FF:FF:FF:DF:FF:FF:FD:FF:FF:FF:FF:DF:DF:FF:FF:FF:BF:FF:F7:FF:FF:FF:FF:FF:FF:F7:F7:FF:FF:FD:FD:FF:FF:FD:DF:FF:EF:DF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:BF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FB:FF:BD:FF:DF:FD:DF:FF:FF:FF:FF:DF:FF:FF:FF:FF:FF:FF:DF:FF:FF:FF:FF:FF:FF:D7:EF:FF:BF:FF:FD:FF:FF:FF:FF:FF
[19:31:42][D][uart_debug:114]: <<< FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:D7:FF:9D:FF:FF:FF:FF:FF:FB:AF:FF:FF:FF:FF:7F:FF:FF:FF:FF:FF:FF:FF:FF:FD:FF:FD:FF:FF:FF:FF:FF:FF:FF:FF:FF:9F:FF:FF:FF:FF:FF:FF:FF:EF:FF:EF:FF:FF:FF:FF:FF:FF:7F:FF:FF:FF:FF:FF:FF:DF:FF:FF:FF:FF:DF:FF:FF:DF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:F7:FF:FF:FF:FF:FF:BD:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:EF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:DF:F7:FF:FD:DF:DF:FF:FF:FF:FF:FF:DB:FF:FF:FF:FF:DF:FF:FF:FF:FF:FF:DD
[19:31:43][D][uart_debug:114]: <<< FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FB:FF:FF:BF:FF:EF:DF:FF:FF:FF:FF:FD:FF:FF:FF:FF:FD:FD:FF:FF:FF:FF:BF:EF:FF:FF:FD:FF:FF:FF:FF:FF:FF:FD:DF:FD:FF:FF:FF:FF:DF:FF:FF:FF:FF:DF:FF:FF:FF:FF:FF:FF:FF:FF:FF:7F:FF:FF:FF:FF:FF:FF:FF:DF:FF:FF:FF:BF:FF:FF:FF:F5:FF:FF:FD:FF:FF:FD:FF:DF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:DF:DF:FF:FF:FF:FF:FF:DF:FF:FF:FF:FF:FF:FF:FD:7F:DF:FF:FB:EF:FF:FF:FF:FF:FF:FF:FF:FF:DF:DF:DF:FF:FF:D7:FF:F7:DF:FF:FF:FF:FF:FF:FF
[19:31:44][D][uart_debug:114]: <<< FF:FF:FF:7F:F7:DF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:DF:7F:EF:FF:FF:FF:FF:FF:FF:FF:FF:77:FF:DF:7F:FF:FF:FF:FF:FF:EF:FF:FF:FF:FB:FF:FF:BF:BF:FF:FF:FF:EF:DF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FD:FD:FF:7F:FD:FF:FF:FF:FD:DF:F7:FF:DF:FF:FF:FD:DF:DD:FF:FF:FF:FF:FF:FD:FF:FF:FF:FF:FF:FD:FF:DF:FF:FF:FF:FF:FD:FF:ED:FF:FF:FF:DF:FF:FF:DF:FF:FF:DF:FD:BD:FF:FD:DF:FF:FF:FF:FF:FF:FF:FF:FB:FF:FF:FF:FD:FF:FF:FF:FF:FF:FF:FF:FD:DF:FF:FF:FF:FD:FF:FF:FF
[19:31:45][D][uart_debug:114]: <<< FB:FF:FF:FF:FF:FF:BF:FF:FF:FF:FF:FF:FF:BF:FF:FD:FF:FD:FF:FF:FF:FF:FF:FF:FF:DF:FF:DF:F7:FF:FF:FF:FF:EF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:7F:FF:FF:FF:FF:FF:FD:BD:FF:FF:FF:FF:FF:FF:FF:DF:FF:FF:FF:DF:FF:FF:FF:FF:DF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:7F:FF:FF:FF:FF:FF:FF:DF:FF:DF:FF:DF:FF:FF:FF:FF:7F:FF:FF:FF:FF:FF:FF:7F:FF:FF:FF:FF:DF:7F:FF:FF:FF:FF:FF:FF:FF:FF:FB:FF:FF:DF:FF:FF:EF:FF:FF:EF:FF:FF:FF:FF:DF:FF:FF:FF:FF:FF:FF:FF
[19:31:45][D][pipsolar:755]: timeout command to poll: QMOD
[19:31:45][D][uart_debug:114]: <<< EF:FF:FF:FF:FF:DF:FB:FF:FF:FF:FF:FF:F7:DF:FF:FF:FF:FF:FF
[19:31:45][D][pipsolar:838]: Sending polling command : QPIRI with length 5
[19:31:45][D][uart_debug:114]: >>> 51:50:49:52:49:F8:54:0D
[19:31:47][D][uart_debug:114]: <<< FF:FF:FF:FF:FF:EF:FF:FF:FF:FF:FF:FF:EF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FD:FD:FF:FF:FF:FF:FF:FF:DF:FF:DF:FF:FF:FF:FB:FF:FF:FF:FF:DF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:BF:FF:FF:FF:FD:DF:FF:7F:FF:FF:FF:DF:FF:FF:FF:FF:FF:FF:FD:DF:DF:FF:DF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:F7:FF:F7:FF:DF:FF:FF:FF:FF:FF:FF:FF:F7:DF:FF:FF:FF:DB:FF:F7:FF:EF:FF:F7:FF:FF:FF:EF:FF:FF:FF:FF:FF:FF:FF:BF:FF:FF:6F:FF:7D:FF:FF:FF:FF:FF:FD:FF:DF:BF
[19:31:47][D][uart_debug:114]: <<< DF:FF:DF:FF:FF:FF:FF:DF:FF:FF:FF:FF:FF:FF:D3:FF:F7:FF:FF:FF:F7:FF:FF:FF:FF:FF:FF:FF:FF:F7:FF:FF:FF:DF:FF:FF:DF:EF:FF:FF:FD:FE:FF:FF:FF:FF:EF:DF:FF:FF:FD:FF:FD:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FD:FF:FF:FF:FF:FF:FF:FF:F7:FF:F7:FF:FF:FF:FD:7F:FF:FF:EF:DF:FF:BF:DF:FF:FF:FF:FF:FF:FF:FF:FF:CF:FF:3F:FF:FF:FF:FF:EF:FF:FF:FF:FF:7F:FF:FF:FF:FF:DF:FF:FF:FF:FF:FF:FF:F7:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF
[19:31:49][D][uart_debug:114]: <<< FF:EF:FF:F7:FB:FF:7F:FF:FF:FF:EF:FF:FF:FF:FF:D7:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FB:FF:FF:FF:FF:FF:FF:FF:FF:FF:DF:FF:DF:FF:FF:F7:FF:DF:FF:DF:FF:FF:FF:FF:FF:FD:FF:FF:FF:FF:FF:FF:FF:FD:FF:FF:FF:FF:FF:FF:FF:EF:BF:FF:FF:DB:FF:FF:FF:FF:FF:FF:FE:FD:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:7F:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:EF:FB:FF:FF:DD:FF:FF:DF:FF:FF:FD:FD:FF:FF:FF:FF:D7:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:BF:FF:FF:FF:FF:FF:FF:FF:EF:DF:FF:FF:EF:FF:FF:FF:FF
[19:31:50][D][uart_debug:114]: <<< FF:FF:FF:FF:FD:FF:FF:FF:F7:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:F7:FB:FF:DF:FF:BD:FF:FF:FF:DF:FF:FF:FF:DF:FB:FF:EF:DF:FF:FF:FF:FD:FF:EF:FF:FF:FF:FF:F7:F7:FF:FF:FD:FF:FF:FF:FF:FF:F7:FF:DF:FB:FF:DF:DF:FF:FF:FF:FF:FF:EB:FF:FF:FF:FF:FF:DF:FD:FF:EF:7F:FF:7F:FF:FF:FB:FF:FF:FF:FF:FF:FF:FF:FF:DF:DF:DF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:F7:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:DF:FF:FD:FF:FF:DF:FF:FF:FF:DF:DF:DF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:F7
[19:31:50][D][pipsolar:755]: timeout command to poll: QPIRI
[19:31:50][D][uart_debug:114]: <<< FF:FF:FF:FF:FF:FF:EF:FF:FF:DF:FF:FF:FF:FF:DF:FF:FF:FF:DF:FF:FB:FF:EF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:7F:FF:EF:FF:EF:FF:FF:FF:FF:F7:FF:FB:FF:FF:FF:FD:FF:FF:FF:F7:FF:FF:FF:FF:FF:FF:FF:EF:FF:FF:FF:FF:FF:FF:DF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:DF:FF:FF:FF:EF:FF:FF:FF:FF: [... cut ...]
[19:31:56][D][uart_debug:114]: <<< FF:DF:FF:FF:FF:7F:FF:FF:FF:FF:DF:FF:FF:FF:FF:FF:FF:FF:EF:F7:FF:FF:FF:F7:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:DF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:DF:FF:FB:FF:FF:FF:FF:FF:FF:FF:FF:DF:FF:FF:FF:FF:FF:FB:BB:FF:FF:FF:F7:FF:FF:F7:FF:FF:F7:FF:F7:7F:FF:FF:EF:D7:FF:FF:DF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:7F:FF:FF:FF:DF:FF:FF:FF:FF:FF:FF:FF:FF:FF:F7:FF:FF:7F:FF:DF:FF:FF:FF:DF:FF:FF:FF:FF:FF:7F:FF:FF:FF:FF:FF:FD:FF:FF
```
Does this tell you anything? I will also try to make some photos and send them.

syssi, 2022-04-07:
Did you connect GND properly? Please make sure the baudrate is 2400 baud.

bogdanr, 2022-04-07:
Dude, you are the best!
This new adapter had a different pinout even though the board is quite similar. I incorrectly assumed the pinout would be the same but GND was switched with TX on this board as opposed with the other board. It's impressive that the chip was powered though.

**Thanks a lot!**
In conclusion for others reading this thread in the future:
* syssi is the best
* not all adapters actually work. I have one with MAX232 which works.
* try to rotate RX/TX until you get it right.

closed 2022-04-07 by bogdanr

syssi, 2022-04-07:
You are welcome! If you want to do me a favor please provide a photo of your rs232 converter + wiring. It looks like a proper connection/wiring is the hardest part of the journey. ;-)

bogdanr, 2022-04-07:
[image]
[image]

syssi, 2022-04-07:
To be very precise: You are using a **MAX3232**. This is the 3.3V version of the MAX232. This is important & good because driving a MAX232 with 3.3V is out of spec and driving a MAX232 with 5V leads to a logic level (TTL) of 5V too. The GPIOs of a ESP aren't 5V tolerant.