syssi/esphome-pipsolar #13: How much solar/grid/battery power?

bogdanr, 2022-04-11:
`ac_output_active_power` and  `ac_output_rating_active_power` are very useful for seeing how much power I am using.
Still, these don't tell me anything if this energy comes from the grid, battery or solar? Is there any way to get this data?

SeByDocKy, 2022-04-12:
As far I read the protocol document, I never seen such information avalable. Once can substract the . ac_output_active_power with the (PV_power + battering_charging) but it won't be super accurate.

SeByDocKy, 2022-04-12:
Yes on the PIP8048 wtih two strings pv_charging_power is replace with pv1_charging_power etc....
maybe you can use the pip8048 branch to test with your easun.

jowser7, 2022-04-12:
It may be helpful to note, the voltronic PV power cuts off its input power if the batteries are full or reached your float / bulk voltage setting.  I have perfect sunny day, but you can see from the chart when I reached full charge state the blue line PV input current drops off. The purple is my battery discharge current out.  
[image]

Looking at PV power curve
[image]

If you read the binary sensor     ac_charging_status:   and its status is on, then maybe that can deduce how often the AC is being used.   I use an external energy monitoring plug for the AC in, and AC out of the inverter that I can then do that subtraction independent of the inverter. 

Does the Easun app show power with that string variable PV1. . ...? or is it also 0W?

syssi, 2022-04-13:
@bogdanr Could you try to use the feature branch:

```
external_components:
  - source: github://syssi/esphome-pipsolar@pip8048
    refresh: 0s
```

And update/change your config yaml a bit: https://github.com/syssi/esphome-pipsolar/pull/10/files#diff-dbc81de7d9fb76af6e67b9c8b02f69c8ea7f35b243a2898f26520c759e0d7b6dL122-R145

Scorcerer, 2022-04-20:
Wouldn't the answer be in implementing QED QEY and rest of them? This would allow you to see how much PV actually generated and used, and then rest of it was used from the grid....

bogdanr, 2022-06-18:
I shorted my inverter and it took a while to get parts to fix it.
I confirm the changes from branch PIP8048 improve things. I now have a sensor for looking at the solar production (pv1_charging_power).
I can also see a battery_discharge_current sensor but I am missing a sensor which shows charging current.
I suggest you merge this pull request.

closed 2022-06-18 by bogdanr