syssi/esphome-pipsolar #14: Adding QED, QEM, QEY and similar commands.

Scorcerer, 2022-04-20:
Hi!
For some time I'm using esphome-pipsolar, and it's awesome! The only thing I'm missing is the ability to count how many Wh did the inverter generate (obviously for some HA-graph greatness and tracking PV production).
Would it be possible to add here? I can test it, if i ever learn how to use a branch on docker...

syssi, 2022-04-22:
Could you provide some more details about the commands? Do you have an idea which models does support these commands?

SeByDocKy, 2022-04-22:
I don't advise to use the builtin statistics capacities of these inverter. There are not really accurate and you can only log infos up to the 01/01 of the current year.. Use a dedicaced PZEM or similar device.... and use a total_daily_energy sensor to integrate data in kWh ....

Scorcerer, 2022-04-25:
Hiho, thanks for your responses :)
My inverter is like this:
[image]
It shows daily/monthly/yearly/total generation, and I thought that since this is an additive value, it should be simple to push it into HA for graphing.
And I was thinking about getting just a rudimentary reading for now - since it's all in software, it could give me an approximate idea when/how much is the inverter producing, without any additional costs.
I also saw those commands listed in PDF document linked here:
https://github.com/syssi/esphome-pipsolar/issues/3#issuecomment-1036986944