syssi/esphome-pipsolar #15: pip8048 branch with last dev compile error

SeByDocKy, 2022-04-30:
Hi @syssi,
After updating my esphome dev to the last version, I can't compile anymore my model :(.
Got a problem with the select part :( ...

```File "C:\Users\SeByDocKy\Desktop\SeBy\Maison\Home assistant\47\.esphome\external_components\8e1d245d\components\pipsolar\select\__init__.py", line 5, in <module>
    from esphome.jsonschema import jschema_composite
ImportError: cannot import name 'jschema_composite' from 'esphome.jsonschema' (C:\Users\SeByDocKy\anaconda3\lib\site-packages\esphome\jsonschema.py)←[0m
←[32mINFO Detected timezone 'Europe/Paris'←[0m
Failed config

select.pipsolar: [source pvbrain.yaml:1266]

  Platform not found: 'select.pipsolar'.```

syssi, 2022-04-30:
Could you provide the `select:` section of your configuration yaml?

SeByDocKy, 2022-04-30:
```select:
  - platform: pipsolar
    pipsolar_id: pip8048
    output_source_priority:
      id: pvbrain_axpert_output_source_priority_select 
      name: pvbrain_axpert_output_source_priority_select
      optionsmap:
        "Utility first": "POP00"
        "Solar only": "POP01"
        "Solar Battery Utility": "POP02"
      statusmap:
        "0": "Utility first"
        "1": "Solar only"
        "2": "Solar Battery Utility"```

According to : https://github.com/esphome/esphome/commit/05dc97099a4efe6810ca682939f322106deb0f2d

seems the import line need to be removed.

syssi, 2022-04-30:
I'm on tour today and will have a looking this evening. Did you try `esphome clean pvbrain.yaml`?

SeByDocKy, 2022-04-30:
Yes I tried to clean, install a former ESPhome dev, etc... :( no success :(

syssi, 2022-04-30:
Fixed by https://github.com/syssi/esphome-pipsolar/commit/5f40093ab8aa16c213aa34ff25f7f0c2acb963e8

Please try again!

closed 2022-04-30 by syssi

SeByDocKy, 2022-04-30:
Yep it's working  again :) thanks :)

BradleyGeldenhuys, 2023-03-07:
I am having the same issue as above when trying to add the "select:" into configuration.

[image]

syssi, 2023-03-07:
@BradleyGeldenhuys Could you provide a screenshot/copy of your `external_components` section? Which branch / component do you use exactly?

BradleyGeldenhuys, 2023-03-07:
Thanks for the quick response.

I am using ESPHome Current version: 2023.2.4 in Home Assistant 

I was not using external_components and after reading some more added the following.

external_components:
  - source: github://syssi/esphome-pipsolar@2424mse1
    refresh: 0s

I then did a clean build files.

[image]

INFO Reading configuration /config/esphome/inverter-1.yaml...
INFO Updating https://github.com/syssi/esphome-pipsolar.git@2424mse1
INFO Deleting /data/inverter-1/.pioenvs
INFO Deleting /data/inverter-1/.piolibdeps
INFO Done!

However when I try install I get the followig error.

In file included from src/esphome/components/pipsolar/output/pipsolar_output.h:3,
                 from src/esphome/components/pipsolar/output/pipsolar_output.cpp:1:
src/esphome/components/pipsolar/output/../pipsolar.h:7:10: fatal error: esphome/components/pipsolar/select/pipsolar_select.h: No such file or directory
    7 | #include "esphome/components/pipsolar/select/pipsolar_select.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/inverter-1/.pioenvs/inverter-1/src/esphome/components/pipsolar/output/pipsolar_output.cpp.o] Error 1

BradleyGeldenhuys, 2023-03-07:
I just read this comment again https://github.com/syssi/esphome-pipsolar/issues/3#issuecomment-1434562703

I added the select component after adding the external_component and compiled correctly.

Thank you