syssi/esphome-pipsolar #16: Pip 5048MGX not reporting all data and mismatching data entry.

ans-gw, 2022-05-06:
i use component pipsolar original Esphome.
I have some missing and mismatching data :Pv Input Power---Pv input Voltage, Pv Input ampere,Battery discharge Ampere.

This 2 lines are from Voltronic/homeassistant Log.
QPIGS reply size (110 bytes)
QPIGS: 110 bytes read: (231.0 50.0 231.0 50.0 1848 1827 036 397 53.10 011 054 0048 01.1 335.3 00.00 00000 00010111 00 00 **00722** 010
    00722=Pw input power

 QPIRI reply size (103 bytes)
 QPIRI: 103 bytes read: (230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 50.0 47.5 54.7 53.6 2 010 040 0 1 3 9 00 0 0 52.0 0 1 035

This 2 lines are from Pipsolar original Esphome component.
Inveter last_qpigs': Sending state '(230.9 49.9 230.9 49.9 0461 0412 009 393 52.70 003 050 0048 00.3 322.5 00.00 00000 00010111 00 00 **00194** 010'          00194=Pv Input Power
dware UART: UART0
[11:36:47][D][sensor:125]: 'Inveter grid_voltage': Sending state 230.89999 V with 1 decimals of accuracy
[11:36:47][D][sensor:125]: 'Inveter grid_frequency': Sending state 49.90000 Hz with 1 decimals of accuracy
[11:36:47][D][sensor:125]: 'Inveter ac_output_voltage': Sending state 230.89999 V with 1 decimals of accuracy
[11:36:47][D][sensor:125]: 'Inveter ac_output_frequency': Sending state 49.90000 Hz with 1 decimals of accuracy
[11:36:47][D][sensor:125]: 'Inveter ac_output_apparent_power': Sending state 461.00000 VA with 1 decimals of accuracy
[11:36:47][D][sensor:125]: 'inverter0_ac_output_active_power': Sending state 412.00000 W with 1 decimals of accuracy
[11:36:47][D][sensor:125]: 'Inveter output_load_percent': Sending state 9.00000 % with 1 decimals of accuracy
[11:36:47][D][sensor:125]: 'Inveter bus_voltage': Sending state 393.00000 V with 1 decimals of accuracy
[11:36:47][D][sensor:125]: 'Inveter battery_voltage': Sending state 52.70000 V with 1 decimals of accuracy
[11:36:47][D][sensor:125]: 'Inveter battery_charging_current': Sending state 3.00000 A with 1 decimals of accuracy
[11:36:47][D][sensor:125]: 'Inveter battery_capacity_percent': Sending state 50.00000 % with 1 decimals of accuracy
[11:36:48][D][sensor:125]: 'Inveter inverter_heat_sink_temperature': Sending state 48.00000 °C with 1 decimals of accuracy
[11:36:48][D][sensor:125]: 'Inveter pv_input_current_for_battery': Sending state 0.00000 A with 1 decimals of accuracy
[11:36:48][D][sensor:125]: 'Inveter pv_input_voltage': Sending state 0.30000 V with 1 decimals of accuracy
[11:36:48][D][sensor:125]: 'Inveter battery_voltage_scc': Sending state 322.50000 V with 1 decimals of accuracy
[11:36:48][D][sensor:125]: 'Inveter battery_discharge_current': Sending state 0.00000 A with 1 decimals of accuracy
[11:36:48][D][sensor:125]: 'Inveter eeprom_version': Sending state 0.00000  with 1 decimals of accuracy
[11:36:48][D][sensor:125]: 'Inveter pv_charging_power': Sending state 0.00000 W with 1 decimals of accuracy

Inveter last_qpiri': Sending state '(230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 50.0 47.5 54.7 53.6 2 010 040 0 0 3 9 00 0 0 52.0 0 1 035'
[11:36:50][D][sensor:125]: 'inverter0_grid_rating_voltage': Sending state 230.00000 V with 1 decimals of accuracy
[11:36:50][D][sensor:125]: 'inverter0_grid_rating_current': Sending state 21.70000 A with 1 decimals of accuracy
[11:36:50][D][sensor:125]: 'inverter0_battery_rating_voltage': Sending state 48.00000 V with 1 decimals of accuracy
[11:36:50][D][sensor:125]: 'inverter0_battery_recharge_voltage': Sending state 50.00000 V with 1 decimals of accuracy
[11:36:50][D][sensor:125]: 'inverter0_battery_under_voltage': Sending state 47.50000 V with 1 decimals of accuracy
[11:36:50][D][sensor:125]: 'inverter0_battery_bulk_voltage': Sending state 54.70000 V with 1 decimals of accuracy
[11:36:50][D][sensor:125]: 'inverter0_battery_float_voltage': Sending state 53.60000 V with 1 decimals of accuracy
[11:36:50][D][sensor:125]: 'inverter0_output_source_priority': Sending state 0.00000  with 1 decimals of accuracy
[11:36:50][D][sensor:125]: 'inverter0_charger_source_priority': Sending state 3.00000  with 1 decimals of accuracy
[11:36:50][D][sensor:125]: 'inverter0_battery_redischarge_voltage': Sending state 52.00000  with 1 decimals of accuracy
[11:36:51][D][pipsolar:838]: Sending polling command : QPIGS with length 5

This pip fro solpiplog.
[image]

thanks for helps

syssi, 2022-05-11:
Please use this feature branch:

```
external_components:
  - source: github://syssi/esphome-pipsolar@2424mse1
    refresh: 0s
```

The format of the QPIGS response should match here:

https://github.com/syssi/esphome-pipsolar/commit/3c5b8bb40052a171ffbd8a9fe76791ac1765df6b

syssi, 2022-07-19:
Closing this because of inactivity.

closed 2022-07-19 by syssi