syssi/esphome-pipsolar #18: Voltronic Axpert MKS IV 5.6kW compatible

amagr0, 2022-06-09:
Hello Sebastian,

It's amazing your contribution for the community, I found you in esphome-jk-bms and now here.
I'm struggling to put the [docker-voltronic-homeassistant](https://github.com/ned-kelly/docker-voltronic-homeassistant) to work, and then I found in HA forum your post about this ESPhome version, that I'd like to try since I'm more familiar with ESPhome than Docker stuff. Actually I was getting data in HA from my inverter, but sometimes I lose connection. In Docker I was getting "unhealthy" status, I decided to install everything again but now I'm not getting any data at all in HA, so I'm here.

I was looking for your ESPhome yaml file and did not find any way to define the QPIRI, QPIWS, QMOD and QPIGS value, like in the Docker version?
How is this made?

Do you think my inverter will by compatible?

Thanks
André Magro

syssi, 2022-06-09:
Nice to see you here, too! :-) How many PV strings does your inverter support? One or two?

amagr0, 2022-06-09:
It only have ha one [PV input](https://voltronicpower.com/en-US/product/detail/axpert-mks-iv-3.6kva-5.6kva)

I don't know if this information is relevant, but I leave it here:

Main CPU Firmware Version: 00074.01
Secondary CPU Firmware Version: 00030.00

syssi, 2022-06-09:
In this case you should give it a try with the official component included in espbome. If it doesn't work out of the box please increase the log level to debug and post some logs here.

amagr0, 2022-06-09:
Hello @syssi,

I only see the reference to configure mqtt in secrets, but is not also necessary to declare in the yaml file?

[image]

[image]

syssi, 2022-06-10:
You have to extend the `secrets.yaml` by additional key-values. If you use Home Assistant please remove the `mqtt` section and add `api:` instead.

amagr0, 2022-06-16:
Hello @syssi,

After being blessed with the miracle of life, I became a dad! Now I'm back to this fight.

So, I'm getting timeouts:
[image]

Just a note, based on this comment I connected the inverter directly to the ESP32. So i don't know if the timeouts are due to this direct connection. I have a RS485 module, do you think I can connect it to the inverter and ESP32, or should it be a RS232 module?
[image]

This is how it is connected.
[image]

And this is the yaml code.

esphome:
  name: inv-voltronic

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:

api:
  encryption:
    key: "y9H5HQINyIv3kfx07vNKzHlfrP1Pqi77U5ZaPI+Fc4U="

ota:
  password: "5a15ebc8158de6088366b81c1d3142d0"

wifi:
  ssid: !secret wifi_ssid2
  password: !secret wifi_password2

  ap:
    ssid: "Inv-Voltronic Fallback Hotspot"
    password: "Aym9t8ksVJHj"

captive_portal:

uart:
  - id: uart_bus
    tx_pin: GPIO18
    rx_pin: GPIO19
    baud_rate: 2400

pipsolar:
  - uart_id: uart_bus
    id: inverter0
   
sensor:
  - platform: pipsolar
    pipsolar_id: inverter0
    grid_rating_voltage:
      id: inverter0_grid_rating_voltage
      name: inverter0_grid_rating_voltage
    grid_rating_current:
      id: inverter0_grid_rating_current
      name: inverter0_grid_rating_current
    ac_output_apparent_power:
      id: inverter0_ac_output_apparent_power
      name: inverter0_ac_output_apparent_power
    ac_output_active_power:
      id: inverter0_ac_output_active_power
      name: inverter0_ac_output_active_power
    
Tks

syssi, 2022-06-16:
Congratulations! :-) You should disconnect the ESP as soon as possible. The logic level of a RS232 connection is -12V / +12V. The logic level of an ESP is 0V to 3.3V. The level shifter / RS232 to TTL converter is super important. At the worst case the GPIOs of your ESP are already toasted.

syssi, 2022-06-16:
Hmmm.. if you are sure the logic level of the RJ45 jack is 0V/3.3V and not -12V/+12V you should try to swap RX/TX because the TX pin of the inverter must be connected to the RX pin of the ESP and vice versa.

amagr0, 2022-06-16:
> Congratulations! :-) You should disconnect the ESP as soon as possible. The logic level of a RS232 connection is -12V / +12V. The logic level of an ESP is 0V to 3.3V. The level shifter / RS232 to TTL converter is super important. At the worst case the GPIOs of your ESP are already toasted.

I probably toasted the GPIOs of the ESP.
Between the TX and GND from inverter I'm getting -13.4V.

Do you think there is any problem using the RS485 module?

nuvolare12, 2022-12-06:
Hello, have you resolve the problem timeout?

VirtualFab, 2022-12-08:
Hi Syssi I had the timeout issue and solved it swapping Rx and Tx (I am using the Max3232 level shifter as per your schematics).
But now I read all zeroes, here is my log:

> INFO Reading configuration /config/esphome/voltronic1.yaml...
> INFO Starting log output from pipsolar/debug
> INFO Connected to MQTT broker!
> [09:57:35][D][sensor:127]: 'pipsolar output_mode': Sending state 0.00000  with 1 decimals of accuracy
> [09:57:35][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
> [09:57:35][D][pipsolar:838]: Sending polling command : QPIGS with length 5
> [09:57:35][D][uart_debug:158]: <<< "QPIGS\xB7\xA9\r"
> [09:57:35][D][pipsolar:772]: checking crc on incoming message
> [09:57:35][D][pipsolar:775]: CRC OK
> [09:57:35][D][pipsolar:448]: Decode QPIGS
> [09:57:35][D][sensor:127]: 'pipsolar grid_voltage': Sending state 0.00000 V with 1 decimals of accuracy
> [09:57:35][D][sensor:127]: 'pipsolar ac_output_voltage': Sending state 0.00000 V with 1 decimals of accuracy
> [09:57:35][D][sensor:127]: 'pipsolar ac_output_apparent_power': Sending state 0.00000 VA with 1 decimals of accuracy
> [09:57:35][D][sensor:127]: 'pipsolar ac_output_active_power': Sending state 0.00000 W with 1 decimals of accuracy
> [09:57:35][D][sensor:127]: 'pipsolar output_load_percent': Sending state 0.00000 % with 1 decimals of accuracy
> [09:57:35][D][sensor:127]: 'pipsolar battery_voltage': Sending state 0.00000 V with 1 decimals of accuracy
> [09:57:35][D][sensor:127]: 'pipsolar battery_charging_current': Sending state 0.00000 A with 1 decimals of accuracy
> [09:57:35][D][sensor:127]: 'pipsolar battery_capacity_percent': Sending state 0.00000 % with 1 decimals of accuracy
> [09:57:35][D][sensor:127]: 'pipsolar inverter_heat_sink_temperature': Sending state 0.00000 °C with 1 decimals of accuracy
> [09:57:35][D][sensor:127]: 'pipsolar pv_input_current_for_battery': Sending state 0.00000 A with 1 decimals of accuracy
> [09:57:35][D][sensor:127]: 'pipsolar pv_input_voltage': Sending state 0.00000 V with 1 decimals of accuracy
> [09:57:35][D][sensor:127]: 'pipsolar battery_voltage_scc': Sending state 0.00000 V with 1 decimals of accuracy
> [09:57:35][D][sensor:127]: 'pipsolar battery_discharge_current': Sending state 0.00000 A with 1 decimals of accuracy
> [09:57:35][D][sensor:127]: 'pipsolar pv_charging_power': Sending state 0.00000 W with 1 decimals of accuracy
> [09:57:36][D][uart_debug:158]: >>> "QMODI\xC1\r"
> [09:57:36][D][pipsolar:838]: Sending polling command : QMOD with length 4
> [09:57:36][D][uart_debug:158]: <<< "QMODI\xC1\r"
> [09:57:36][D][pipsolar:772]: checking crc on incoming message
> [09:57:36][D][pipsolar:775]: CRC OK
> [09:57:36][D][pipsolar:471]: Decode QMOD
> [09:57:36][D][text_sensor:067]: 'pipsolar device_mode': Sending state 'M'
> [09:57:37][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
> [09:57:37][D][pipsolar:838]: Sending polling command : QPIRI with length 5
> [09:57:37][D][uart_debug:158]: <<< "QPIRI\xF8T\r"
> [09:57:37][D][pipsolar:772]: checking crc on incoming message
> [09:57:37][D][pipsolar:775]: CRC OK
> [09:57:37][D][pipsolar:429]: Decode QPIRI
> [09:57:37][D][sensor:127]: 'pipsolar grid_rating_voltage': Sending state 0.00000 V with 1 decimals of accuracy
> [09:57:37][D][sensor:127]: 'pipsolar output_source_priority': Sending state 0.00000  with 1 decimals of accuracy
> [09:57:37][D][sensor:127]: 'pipsolar charger_source_priority': Sending state 0.00000  with 1 decimals of accuracy
> [09:57:37][D][sensor:127]: 'pipsolar output_mode': Sending state 0.00000  with 1 decimals of accuracy

I am using MQTT, here it is my config:

> substitutions:
>   name: pipsolar
> 
> esphome:
>   name: ${name}
>   platform: ESP32
>   board: esp-wrover-kit
> 
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
> 
> ota:
> 
> logger:
>   baud_rate: 0
> 
> mqtt:
>   broker: !secret mqtt_host
>   username: !secret mqtt_username
>   password: !secret mqtt_password
>   id: mqtt_client
> 
[... 78 lines cut ...]
>     ac_charging_status:
>       name: "${name} ac_charging_status"
>     charging_to_floating_mode:
>       name: "${name} charging_to_floating_mode"
> 
> switch:
>   - platform: pipsolar
>     pipsolar_id: inverter0
>     output_source_priority_utility:
>       name: "${name} output_source_priority_utility"
>     output_source_priority_solar:
>       name: "${name} output_source_priority_solar"
>     output_source_priority_battery:
>       name: "${name} output_source_priority_battery"
> 
> output:
>   - platform: pipsolar
>     pipsolar_id: inverter0
>     battery_recharge_voltage:
>       id: inverter0_battery_recharge_voltage_out
> 

As Amagr0 does, I too was wondering if there is the need to set QPIRI, QPIWS, QMOD and QPIGS values sowehow, like in the Docker version (which I didn't tried since the EspHome integration would be perfect).

Thank you for your work, please help us (me and nuvolare12) to get your integration working!

closed 2023-04-14 by syssi