syssi/esphome-pipsolar #19: Easun SMW 8k (aka Axpert MAX 8) support

syssi, 2022-06-24:
Good evening,
I have one more problem. Sensor battery_discharge_current have value 0.0A for all time and not respond with reality.
Can you help me? Inverter is Easun SMW 8k (aka Axpert MAX 8).
In manual is:

```
QPIGS<cr>: Device general status parameters inquiry
Computer: QPIGS <CRC><cr>
Device: (BBB.B CC.C DDD.D EE.E FFFF GGGG HHH III
JJ.JJ KK K OOO TTTT EE. E
UUU.U WW.WW PPPPP b7b6b5b4b3b2b1b0 QQ VV MMMMM b10b9b8 Y ZZ AAAA
<CRC><
```

PPPPP Battery discharge current P is an Integer ranging from 0 to 9. The units is A.

Thank you Petr

_Originally posted by @PetrDubi in https://github.com/syssi/esphome-pipsolar/issues/8#issuecomment-1165808123_

syssi, 2022-06-24:
As first step let's compare the parser of the different known QPIGS responses. For example the 2424MSE1 and PIP-5048Mg FW71.85 reponses:

```
226.7 49.9 226.7 49.9 0498 0479 016 427 27.00 005 100 0035 01.9 255.1 00.00 00000 10010110 00 00 00510 110 # 2424MSE1
225.8 49.9 225.8 49.9 0609 0565 020 427 27.00 005 100 0035 02.2 259.9 00.00 00000 10010110 00 00 00590 110 # 2424MSE1
232.6 50.0 229.9 49.9 0391 0312 007 402 54.40 042 072 0066 0042 284.6 00.00 00000 00010010 00 00 02901 010 # PIP-5048Mg FW71.85

001.0 00.0 229.0 50.0 0000 0000 000 415 27.29 000 100 0479 0000 000.0 00.00 00001 10010000 00 04 00000 000 # HMS 3K-24V
001.0 00.0 229.0 50.0 0000 0000 000 413 27.23 000 100 0672 0000 000.0 00.00 00001 10010000 00 04 00000 000 # HMS 3K-24V
```

syssi, 2022-06-24:
The next step is to check the QPIGS message parser. I've added a lot of comments here to improve the readability:

https://github.com/syssi/esphome-pipsolar/blob/2424mse1/components/pipsolar/pipsolar.cpp#L449-L517

It looks like you are asking about value 16:

https://github.com/syssi/esphome-pipsolar/blob/2424mse1/components/pipsolar/pipsolar.cpp#L471
https://github.com/syssi/esphome-pipsolar/blob/2424mse1/components/pipsolar/pipsolar.cpp#L502
https://github.com/syssi/esphome-pipsolar/blob/2424mse1/components/pipsolar/pipsolar.cpp#L219

The value is published as `battery_discharge_current`. Could you explain what's not working for you and could you provide an example response of your inverter incl. a discharge current greater 0. I would like to feed the omponent with your payload to reproduce the issue.

PetrDubi, 2022-06-26:
To friday I used pipsolar from esphome in Home Assistant, My configuration was function good but without values from PV panels and battery discharge current (value was zero). When I was change my cofiguration for use pipsolar from your github then I was can't compilled this configuration. After install last version esphome from github I can compile your example confgiuration, but my configuration was fail compilation with error for pipsolar_select. After many tests i found problem at friday - in my configuration I not use section from your example called select: When I was copy section select from example to my comfiguration then I was compile program and upload it to esp. All values from PV panels is ok, but battery discharge current was zero (friday evening).
At saturday I was outside of my home and today i check values from inverter and in history I was found battery discharge  current with non zero values (from friday evening to today). I was test this and when battery is discharging, then esphome viewed actual value. At friday was battery discharge current zero. beacuse it was small consumption and PV panels power was higher then consumption and I was don't check this value at inverter at friday evening - my big mistake..
I'm very sorry, I'm trotl and all values are OK :-)

syssi, 2022-06-26:
Alright. No worries! Let's close this issue. :-)

closed 2022-06-26 by syssi