syssi/esphome-pipsolar #21: Easun ISolar SML 3K - Not Working

AtarianComputing, 2022-07-13:
[image]

It has a RJ45 COM-port that I've used to this day to connect it to my server via the RJ45/DB9 cable they provided using the WatchPower software. I can confirm that everything works there.

The said cable contains three wires that I have verified to be the same pinouts as described here. RJ45 pins 1,2,8 go to DB9 pins 2,3,5 respectively.

I have used three different TTL converters with MAX3238, MAX3232, MAX202, with the same results. The one converter that has LEDs either pulses the TX light without any activity from RX. The times when I get the null results regardless of whether the RJ45 cable is plugged or not, the RX and TX lights pulse simultaneously. Someone here with the same problem said that it turned out to be bad wiring.

I've tried every conceivable combination of wiring, different wires, different GPIOs with no improvements. Clearly I'm not getting any data from the inverter. Could my model be so incompatible that I'm not getting anything despite it working with what I understand to be a universal software, WatchPower?

Here is my current configuration:
```
substitutions:
  name: atarian-solar
  
esphome:
  name: ${name}

esp32:
  board: nodemcu-32s
  
#external_components:
#  - source: github://syssi/esphome-pipsolar@2424mse1
#    refresh: 0s
#    components: [pipsolar]
#external_components:
#  - source: github://andreashergert1984/esphome@test_newcrc
#    refresh: 0s
#    components: [pipsolar]
    
# Enable logging
logger:
  level: VERY_VERBOSE
  logs:
    uart: VERY_VERBOSE
    uart.arduino_esp8266: VERY_VERBOSE
    # Make some components less verbose
    api.service: WARN
    ota: WARN
  
# Enable Home Assistant API
api:
  encryption:
    key: "gX2lzJ1mNneZVElxRBhPTM3W8SOgACnL8FzsJ08GLQw="

ota:
  password: "d0cfe82ff281d1617e7530fe9a8ddd0a"

wifi:
  ssid: !secret wifi_linksys_ssid
  password: !secret wifi_linksys_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Atarian-Solar Fallback Hotspot"
    password: "VIgd5loU3AeJ"
    
uart:
  - id: uart_bus
    tx_pin: GPIO19
    rx_pin: GPIO18
    # most devices use 2400 as baud_rate
    baud_rate: 2400
    debug:
      direction: BOTH
    
pipsolar:
  uart_id: uart_bus
  id: inverter0
  update_interval: 10s
[... 111 lines cut ...]
[20:02:24][V][sensor:074]: 'atarian-solar battery_capacity_percent': Received new state 0.000000
[20:02:24][D][sensor:125]: 'atarian-solar battery_capacity_percent': Sending state 0.00000 % with 1 decimals of accuracy
[20:02:24][V][sensor:074]: 'atarian-solar inverter_heat_sink_temperature': Received new state 0.000000
[20:02:24][D][sensor:125]: 'atarian-solar inverter_heat_sink_temperature': Sending state 0.00000 °C with 1 decimals of accuracy
[20:02:24][V][sensor:074]: 'atarian-solar pv_input_current_for_battery': Received new state 0.000000
[20:02:24][D][sensor:125]: 'atarian-solar pv_input_current_for_battery': Sending state 0.00000 A with 1 decimals of accuracy
[20:02:24][V][sensor:074]: 'atarian-solar pv_input_voltage': Received new state 0.000000
[20:02:24][D][sensor:125]: 'atarian-solar pv_input_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[20:02:24][V][sensor:074]: 'atarian-solar battery_voltage_scc': Received new state 0.000000
[20:02:24][D][sensor:125]: 'atarian-solar battery_voltage_scc': Sending state 0.00000 V with 1 decimals of accuracy
[20:02:24][V][sensor:074]: 'atarian-solar battery_discharge_current': Received new state 0.000000
[20:02:24][D][sensor:125]: 'atarian-solar battery_discharge_current': Sending state 0.00000 A with 1 decimals of accuracy
[20:02:24][V][sensor:074]: 'atarian-solar battery_voltage_offset_for_fans_on': Received new state 0.000000
[20:02:24][D][sensor:125]: 'atarian-solar battery_voltage_offset_for_fans_on': Sending state 0.00000 V with 1 decimals of accuracy
[20:02:24][V][sensor:074]: 'atarian-solar eeprom_version': Received new state 0.000000
[20:02:24][D][sensor:125]: 'atarian-solar eeprom_version': Sending state 0.00000  with 1 decimals of accuracy
[20:02:24][V][sensor:074]: 'atarian-solar  pv_charging_power': Received new state 0.000000
[20:02:24][D][sensor:125]: 'atarian-solar  pv_charging_power': Sending state 0.00000 W with 1 decimals of accuracy
[20:02:24][V][component:199]: Component esphome.coroutine took a long time for an operation (0.35 s).
[20:02:24][V][component:200]: Components should block for at most 20-30ms.
```

But again, it will display this even with the RJ45 unplugged.

I'm at a total loss here...

AtarianComputing, 2022-07-13:
> Show a photo of your wiring

Which iteration/combination? :) The way I've set it up now is that the COM-cable I built has a DB9 housing so a photo would only reveal the wiring between the converter and ESP. 

Actually, hmb, I'll draw it. Tonight or tomorrow.

AtarianComputing, 2022-07-13:
Alright. Here is the actual converter that's currently plugged in and how it's wired as of now:

[image]

I've verified the connections to the IC by continuity checks. The schematic also confirms it:

[image]

I've also switched the TX/RX on both ends to rule out any confusion on my part.

AtarianComputing, 2022-07-14:
I just realized that WatchPower has a debug feature. Here it is.
[WatchPower-Debug.zip](https://github.com/syssi/esphome-pipsolar/files/9109411/WatchPower-Debug.zip)

syssi, 2022-07-18:
```
[20:02:14][D][uart_debug:114]: <<< 51:50:49:47:53:B7:A9:0D
[20:02:23][VV][scheduler:196]: Running interval 'update' with interval=10000 last_execution=19325 (now=29325)
[20:02:24][D][pipsolar:838]: Sending polling command : QPIGS with length 5
[20:02:24][D][uart_debug:114]: >>> 51:50:49:47:53:B7:A9:0D
[20:02:24][D][pipsolar:772]: checking crc on incoming message
```

The incoming message is an echo of the request / transmitted message. This looks like bad wiring. Please keep in mind the MAX202 provides a logic level of 5V. There is a chance you have already toasted your RX GPIO.

AtarianComputing, 2022-07-18:
> I've checked your debug logs from WatchPower especially the QPIGS response:
> 
> ```
> 233.0 50.0 233.0 50.0 0318 0301 012 427 27.07 003 095 0498 0003 034.8 27.21 00000 10010110 00 04 00105 000
> ```
> 
> It's the standard format and supported by the official component: https://esphome.io/components/pipsolar.html

Thanks for confirming this.

> In other words: As soon as you manage to make a proper connection between the inverter and the ESP it should work.
> 
> About the logic level: If you power the MAX202 with 5V the logic level at the output pins are 5V too. The ESP isn't 5V tolerant if you take it strictly. Please don't try to power the MAX202 with 3.3V this is out of spec. Please use 3.3V components f.e. MAX3232.

Ah, of course. Didn't even consider that. Ok, once I get back home I'll start looking into it again. Away for 2 week vacation atm. Cheers.

KM192, 2023-02-04:
Did you succeed to connect SML III via pipsolar ?
I have the same troubles.

KHackmann, 2023-04-10:
In welchem Port am Inverter steckt der RJ45 Stecker?

syssi, 2023-04-14:
Summary: 

- WatchPower is able to talk to the inverter
- The inverter responds to QPIGS and QPIRI
- The exact protocol version isn't know. This YAML should be used to identify the protocol version: https://github.com/syssi/esphome-pipsolar/blob/main/tests/esp8266-test-protocols.yaml

closed 2023-04-14 by syssi