syssi/esphome-pipsolar #22: Getting CRC NOK with SunGoldPower 6048 (Axpert King II rebrand)

derekyle, 2022-07-17:
So I am seeing a lot of CRC NOK as well as a few CRC OK. Which makes me think that I must have things wired up correctly, maybe? But I am just getting everything reported as 0.0 or off.

Could this be a compatibility issue? The SunGoldPower has identical specs and looks physically identical on the outside (except the paint job) as the Voltronic Axpert King II which is also the same as the MPP Solar lvx6048, so I am assuming they are all the same unit.

Here are some of the logs:

`[18:07:00][D][pipsolar:836]: Sending polling command : QPIRI with length 5
[18:07:00][D][pipsolar:772]: checking crc on incoming message
[18:07:00][D][pipsolar:775]: CRC OK
[18:07:00][D][pipsolar:429]: Decode QPIRI
[18:07:00][D][text_sensor:067]: 'inverter0_last_qpiri': Sending state 'QPIRI'
[18:07:00][D][sensor:124]: 'Inverter Battery Recharge Voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:00][D][sensor:124]: 'Inverter Battery Bulk Voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:00][D][sensor:124]: 'Inverter Battery Float Voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:00][D][sensor:124]: 'Inverter Output Source Priority': Sending state 0.00000  with 1 decimals of accuracy
[18:07:00][D][sensor:124]: 'Inverter Charger Source Priority': Sending state 0.00000  with 1 decimals of accuracy
[18:07:00][D][sensor:124]: 'Inverter Output Mode': Sending state 0.00000  with 1 decimals of accuracy
[18:07:00][D][sensor:124]: 'Inverter Battery Re-discharge Voltage': Sending state 0.00000  with 1 decimals of accuracy
[18:07:01][D][pipsolar:836]: Sending polling command : QPIWS with length 5
[18:07:01][D][pipsolar:772]: checking crc on incoming message
[18:07:01][D][pipsolar:775]: CRC OK
[18:07:01][D][pipsolar:525]: Decode QPIWS
[18:07:02][D][pipsolar:836]: Sending polling command : QMOD with length 4
[18:07:02][D][pipsolar:772]: checking crc on incoming message
[18:07:02][D][pipsolar:781]: CRC NOK expected: 4A 4E but got: 49 C1
[18:07:03][D][pipsolar:836]: Sending polling command : QPIGS with length 5
[18:07:03][D][pipsolar:772]: checking crc on incoming message
[18:07:03][D][pipsolar:775]: CRC OK
[18:07:03][D][pipsolar:448]: Decode QPIGS
[18:07:03][D][text_sensor:067]: 'inverter0_last_qpigs': Sending state 'QPIGS'
[18:07:03][D][sensor:124]: 'Inverter Grid Voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:03][D][sensor:124]: 'Inverter Grid Frequency': Sending state 0.00000 Hz with 1 decimals of accuracy
[18:07:03][D][sensor:124]: 'Inverter AC Output Voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:03][D][sensor:124]: 'Inverter AC Output Frequency': Sending state 0.00000 Hz with 1 decimals of accuracy
[18:07:03][D][sensor:124]: 'Inverter AC Output Apparent Power': Sending state 0.00000 VA with 1 decimals of accuracy
[18:07:03][D][sensor:124]: 'Inverter AC Output Active Power': Sending state 0.00000 W with 1 decimals of accuracy
[18:07:03][D][sensor:124]: 'Inverter Output Load Percent': Sending state 0.00000 % with 1 decimals of accuracy
[18:07:03][D][sensor:124]: 'Inverter Bus Voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:03][D][sensor:124]: 'Inverter Battery Voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:03][D][sensor:124]: 'Inverter Battery Charging Current': Sending state 0.00000 A with 1 decimals of accuracy
[18:07:03][D][sensor:124]: 'Inverter Battery Capacity Percent': Sending state 0.00000 % with 1 decimals of accuracy
[18:07:03][D][sensor:124]: 'Inverter Inverter Heat Sink Temperature': Sending state 0.00000 °C with 1 decimals of accuracy
[18:07:03][D][sensor:124]: 'Inverter PV Input Current for Battery': Sending state 0.00000 A with 1 decimals of accuracy
[18:07:03][D][sensor:124]: 'Inverter PV Input Voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:03][D][sensor:124]: 'Inverter Battery Voltage SCC': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:03][D][sensor:124]: 'Inverter Battery Discharge Current': Sending state 0.00000 A with 1 decimals of accuracy
[18:07:03][D][sensor:124]: 'Inverter PV Charging Power': Sending state 0.00000 W with 1 decimals of accuracy
[18:07:04][D][pipsolar:836]: Sending polling command : QPIRI with length 5
[18:07:04][D][pipsolar:772]: checking crc on incoming message
[18:07:04][D][pipsolar:781]: CRC NOK expected: F8 54 but got: F8 0
[18:07:05][D][pipsolar:836]: Sending polling command : QPIWS with length 5
[18:07:05][D][pipsolar:772]: checking crc on incoming message
[18:07:05][D][pipsolar:775]: CRC OK
[18:07:05][D][pipsolar:525]: Decode QPIWS
[18:07:06][D][pipsolar:836]: Sending polling command : QMOD with length 4
[18:07:06][D][pipsolar:772]: checking crc on incoming message
[18:07:06][D][pipsolar:775]: CRC OK
[18:07:06][D][pipsolar:471]: Decode QMOD
[18:07:06][D][text_sensor:067]: 'inverter0_device_mode': Sending state 'M'
[18:07:07][D][pipsolar:836]: Sending polling command : QPIGS with length 5
[18:07:07][D][pipsolar:772]: checking crc on incoming message
[18:07:07][D][pipsolar:775]: CRC OK
[18:07:07][D][pipsolar:448]: Decode QPIGS
[18:07:07][D][text_sensor:067]: 'inverter0_last_qpigs': Sending state 'QPIGS'
[18:07:07][D][sensor:124]: 'Inverter Grid Voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:07][D][sensor:124]: 'Inverter Grid Frequency': Sending state 0.00000 Hz with 1 decimals of accuracy
[18:07:07][D][sensor:124]: 'Inverter AC Output Voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:07][D][sensor:124]: 'Inverter AC Output Frequency': Sending state 0.00000 Hz with 1 decimals of accuracy
[18:07:07][D][sensor:124]: 'Inverter AC Output Apparent Power': Sending state 0.00000 VA with 1 decimals of accuracy
[18:07:07][D][sensor:124]: 'Inverter AC Output Active Power': Sending state 0.00000 W with 1 decimals of accuracy
[18:07:07][D][sensor:124]: 'Inverter Output Load Percent': Sending state 0.00000 % with 1 decimals of accuracy
[18:07:07][D][sensor:124]: 'Inverter Bus Voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:07][D][sensor:124]: 'Inverter Battery Voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:07][D][sensor:124]: 'Inverter Battery Charging Current': Sending state 0.00000 A with 1 decimals of accuracy
[18:07:07][D][sensor:124]: 'Inverter Battery Capacity Percent': Sending state 0.00000 % with 1 decimals of accuracy
[18:07:07][D][sensor:124]: 'Inverter Inverter Heat Sink Temperature': Sending state 0.00000 °C with 1 decimals of accuracy
[18:07:07][D][sensor:124]: 'Inverter PV Input Current for Battery': Sending state 0.00000 A with 1 decimals of accuracy
[18:07:07][D][sensor:124]: 'Inverter PV Input Voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:07][D][sensor:124]: 'Inverter Battery Voltage SCC': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:07][D][sensor:124]: 'Inverter Battery Discharge Current': Sending state 0.00000 A with 1 decimals of accuracy
[18:07:07][D][sensor:124]: 'Inverter PV Charging Power': Sending state 0.00000 W with 1 decimals of accuracy
[18:07:08][D][pipsolar:836]: Sending polling command : QPIRI with length 5
[18:07:08][D][pipsolar:772]: checking crc on incoming message
[18:07:08][D][pipsolar:775]: CRC OK
[18:07:08][D][pipsolar:429]: Decode QPIRI
[18:07:08][D][text_sensor:067]: 'inverter0_last_qpiri': Sending state 'QPIRI'
[18:07:08][D][sensor:124]: 'Inverter Battery Recharge Voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:08][D][sensor:124]: 'Inverter Battery Bulk Voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:08][D][sensor:124]: 'Inverter Battery Float Voltage': Sending state 0.00000 V with 1 decimals of accuracy
[18:07:08][D][sensor:124]: 'Inverter Output Source Priority': Sending state 0.00000  with 1 decimals of accuracy
[18:07:08][D][sensor:124]: 'Inverter Charger Source Priority': Sending state 0.00000  with 1 decimals of accuracy
[18:07:08][D][sensor:124]: 'Inverter Output Mode': Sending state 0.00000  with 1 decimals of accuracy
[18:07:08][D][sensor:124]: 'Inverter Battery Re-discharge Voltage': Sending state 0.00000  with 1 decimals of accuracy`

Any help is greatly appreciated! And thanks again for all your volunteer work on this project. I'm already using your code to integrate my jkbms and smart shunt into Home Assistant.

derekyle, 2022-07-17:
As an update, I discovered that I was supposed to have the 3.3 volt pin connected from the esp8266 to the TTL side of the RS232 converter. So now, I am getting all CRC OK messages every second or so. However, now, instead of seeing 0 on everything, all of my sensors are showing as "unknown" in home assistant. And I don't see any of the debug messages like above that say "sending state...". It's just all CRC OK messages.

[20:34:54][D][pipsolar:836]: Sending polling command : QPIGS with length 5
[20:34:54][D][pipsolar:772]: checking crc on incoming message
[20:34:54][D][pipsolar:775]: CRC OK
[20:34:55][D][pipsolar:836]: Sending polling command : QPIRI with length 5
[20:34:55][D][pipsolar:772]: checking crc on incoming message
[20:34:55][D][pipsolar:775]: CRC OK
[20:34:56][D][pipsolar:836]: Sending polling command : QPIWS with length 5
[20:34:56][D][pipsolar:772]: checking crc on incoming message
[20:34:56][D][pipsolar:775]: CRC OK
[20:34:57][D][pipsolar:836]: Sending polling command : QMOD with length 4
[20:34:57][D][pipsolar:772]: checking crc on incoming message
[20:34:57][D][pipsolar:775]: CRC OK
[20:34:58][D][pipsolar:836]: Sending polling command : QPIGS with length 5
[20:34:58][D][pipsolar:772]: checking crc on incoming message
[20:34:58][D][pipsolar:775]: CRC OK
[20:34:59][D][pipsolar:836]: Sending polling command : QPIRI with length 5
[20:34:59][D][pipsolar:772]: checking crc on incoming message
[20:34:59][D][pipsolar:775]: CRC OK
[20:35:00][D][pipsolar:836]: Sending polling command : QPIWS with length 5
[20:35:00][D][pipsolar:772]: checking crc on incoming message
[20:35:00][D][pipsolar:775]: CRC OK
[20:35:01][D][pipsolar:836]: Sending polling command : QMOD with length 4
[20:35:01][D][pipsolar:772]: checking crc on incoming message
[20:35:01][D][pipsolar:775]: CRC OK
[20:35:02][D][pipsolar:836]: Sending polling command : QPIGS with length 5
[20:35:02][D][pipsolar:772]: checking crc on incoming message
[20:35:02][D][pipsolar:775]: CRC OK
[20:35:03][D][pipsolar:836]: Sending polling command : QPIRI with length 5
[20:35:03][D][pipsolar:772]: checking crc on incoming message
[20:35:03][D][pipsolar:775]: CRC OK
[20:35:04][D][pipsolar:836]: Sending polling command : QPIWS with length 5
[20:35:04][D][pipsolar:772]: checking crc on incoming message
[20:35:04][D][pipsolar:775]: CRC OK
[20:35:05][D][pipsolar:836]: Sending polling command : QMOD with length 4
[20:35:05][D][pipsolar:772]: checking crc on incoming message
[20:35:05][D][pipsolar:775]: CRC OK
[20:35:06][D][pipsolar:836]: Sending polling command : QPIGS with length 5
[20:35:06][D][pipsolar:772]: checking crc on incoming message
[20:35:06][D][pipsolar:775]: CRC OK
[20:35:07][D][pipsolar:836]: Sending polling command : QPIRI with length 5
[20:35:07][D][pipsolar:772]: checking crc on incoming message
[20:35:07][D][pipsolar:775]: CRC OK
[20:35:08][D][pipsolar:836]: Sending polling command : QPIWS with length 5
[20:35:08][D][pipsolar:772]: checking crc on incoming message
[20:35:08][D][pipsolar:775]: CRC OK
[20:35:09][D][pipsolar:836]: Sending polling command : QMOD with length 4
[20:35:09][D][pipsolar:772]: checking crc on incoming message
[20:35:09][D][pipsolar:775]: CRC OK
[20:35:10][D][pipsolar:836]: Sending polling command : QPIGS with length 5
[20:35:10][D][pipsolar:772]: checking crc on incoming message
[20:35:10][D][pipsolar:775]: CRC OK
[20:35:11][D][pipsolar:836]: Sending polling command : QPIRI with length 5
[20:35:11][D][pipsolar:772]: checking crc on incoming message
[20:35:11][D][pipsolar:775]: CRC OK
[20:35:12][D][pipsolar:836]: Sending polling command : QPIWS with length 5
[20:35:12][D][pipsolar:772]: checking crc on incoming message
[20:35:12][D][pipsolar:775]: CRC OK
[20:35:13][D][pipsolar:836]: Sending polling command : QMOD with length 4
[20:35:13][D][pipsolar:772]: checking crc on incoming message
[20:35:13][D][pipsolar:775]: CRC OK
[20:35:14][D][pipsolar:836]: Sending polling command : QPIGS with length 5
[20:35:14][D][pipsolar:772]: checking crc on incoming message
[20:35:14][D][pipsolar:775]: CRC OK
[20:35:15][D][pipsolar:836]: Sending polling command : QPIRI with length 5
[20:35:15][D][pipsolar:772]: checking crc on incoming message
[20:35:15][D][pipsolar:775]: CRC OK
[20:35:16][D][pipsolar:836]: Sending polling command : QPIWS with length 5
[20:35:16][D][pipsolar:772]: checking crc on incoming message
[20:35:16][D][pipsolar:775]: CRC OK

syssi, 2022-07-17:
Could you provide your config yaml and enable the debug output of the `uart` component?

syssi, 2022-07-17:
I've an idea what's happening: I are receiving an echo of your transmitted message. If you enable the debug output of the `uart` message you will see the same sequence of bytes going out and coming in. I assume your wiring isn't correct.

derekyle, 2022-07-17:
Here are the logs with UART debug turned on.

[09:49:28][D][pipsolar:836]: Sending polling command : QPIGS with length 5
[09:49:28][D][uart_debug:114]: >>> 51:50:49:47:53:B7:A9:0D
[09:49:28][D][pipsolar:772]: checking crc on incoming message
[09:49:28][D][pipsolar:775]: CRC OK
[09:49:28][D][uart_debug:114]: <<< 28:4E:41:4B:73:73:0D
[09:49:29][VV][scheduler:195]: Running interval 'update' with interval=1000 last_execution=149623 (now=150623)
[09:49:29][D][pipsolar:836]: Sending polling command : QPIRI with length 5
[09:49:29][D][uart_debug:114]: >>> 51:50:49:52:49:F8:54:0D
[09:49:29][D][pipsolar:772]: checking crc on incoming message
[09:49:29][D][pipsolar:775]: CRC OK
[09:49:29][D][uart_debug:114]: <<< 28:4E:41:4B:73:73:0D
[09:49:30][VV][scheduler:195]: Running interval 'update' with interval=1000 last_execution=150623 (now=151630)
[09:49:30][D][pipsolar:836]: Sending polling command : QPIWS with length 5
[09:49:30][D][uart_debug:114]: >>> 51:50:49:57:53:B4:DA:0D
[09:49:30][D][pipsolar:772]: checking crc on incoming message
[09:49:30][D][pipsolar:775]: CRC OK
[09:49:30][D][uart_debug:114]: <<< 28:4E:41:4B:73:73:0D
[09:49:31][VV][scheduler:195]: Running interval 'update' with interval=1000 last_execution=151623 (now=152626)
[09:49:31][D][pipsolar:836]: Sending polling command : QMOD with length 4
[09:49:31][D][uart_debug:114]: >>> 51:4D:4F:44:49:C1:0D
[09:49:31][D][pipsolar:772]: checking crc on incoming message
[09:49:31][D][pipsolar:775]: CRC OK
[09:49:31][D][uart_debug:114]: <<< 28:4E:41:4B:73:73:0D
[09:49:32][VV][scheduler:195]: Running interval 'update' with interval=1000 last_execution=152623 (now=153625)
[09:49:32][D][pipsolar:836]: Sending polling command : QPIGS with length 5
[09:49:32][D][uart_debug:114]: >>> 51:50:49:47:53:B7:A9:0D
[09:49:32][D][pipsolar:772]: checking crc on incoming message
[09:49:32][D][pipsolar:775]: CRC OK
[09:49:32][D][uart_debug:114]: <<< 28:4E:41:4B:73:73:0D
[09:49:33][VV][scheduler:195]: Running interval 'update' with interval=1000 last_execution=153623 (now=154623)
[09:49:33][D][pipsolar:836]: Sending polling command : QPIRI with length 5
[09:49:33][D][uart_debug:114]: >>> 51:50:49:52:49:F8:54:0D

derekyle, 2022-07-17:
And here is my config yaml:

esphome:
  name: hybrid-inverter

esp8266:
  board: d1_mini

substitutions:
  inverter_name: Inverter

# Enable logging
logger:
  baud_rate: 0
  level: VERY_VERBOSE
  logs:
    uart: VERY_VERBOSE
    uart.arduino_esp8266: VERY_VERBOSE
    # Make some components less verbose
    api.service: WARN
    ota: WARN
  hardware_uart: uart1

# Enable Home Assistant API
api:

ota:
  password: "pass"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Hybrid-Inverter Fallback Hotspot"
    password: "4tDGDhiVtlKs"

captive_portal:
    
uart:
  - id: uart_bus
    tx_pin: TX
    rx_pin: RX
    # most devices use 2400 as baud_rate
    baud_rate: 2400
    debug:
      direction: BOTH

pipsolar:
  - uart_id: uart_bus
    id: inverter0

text_sensor:
  - platform: pipsolar
    pipsolar_id: inverter0
    device_mode:
      id: inverter0_device_mode
      name: inverter0_device_mode
    last_qpigs:
      id: inverter0_last_qpigs
      name: inverter0_last_qpigs
    last_qpiri:
      id: inverter0_last_qpiri
   
sensor:
  - platform: pipsolar
    pipsolar_id: inverter0
    grid_voltage:
      name: "${inverter_name} Grid Voltage"
    grid_frequency:
      name: "${inverter_name} Grid Frequency"
    ac_output_voltage:
      name: "${inverter_name} AC Output Voltage"
    ac_output_frequency:
      name: "${inverter_name} AC Output Frequency"
    ac_output_apparent_power:
      name: "${inverter_name} AC Output Apparent Power"
    ac_output_active_power:
      name: "${inverter_name} AC Output Active Power"
    output_load_percent:
      name: "${inverter_name} Output Load Percent"
    bus_voltage:
      name: "${inverter_name} Bus Voltage"
    battery_voltage:
      name: "${inverter_name} Battery Voltage"
    battery_charging_current:
      name: "${inverter_name} Battery Charging Current"
    battery_capacity_percent:
      name: "${inverter_name} Battery Capacity Percent"
    battery_bulk_voltage:
      name: "${inverter_name} Battery Bulk Voltage"
    battery_float_voltage:
      name: "${inverter_name} Battery Float Voltage"
    battery_redischarge_voltage:
      name: "${inverter_name} Battery Re-discharge Voltage"
    battery_recharge_voltage:
      name: "${inverter_name} Battery Recharge Voltage"
    inverter_heat_sink_temperature:
      name: "${inverter_name} Inverter Heat Sink Temperature"
    pv_input_current_for_battery:
      name: "${inverter_name} PV Input Current for Battery"
    pv_input_voltage:
      name: "${inverter_name} PV Input Voltage"
    pv_charging_power:
      name: "${inverter_name} PV Charging Power"
    battery_voltage_scc:
      name: "${inverter_name} Battery Voltage SCC"
    battery_discharge_current:
      name: "${inverter_name} Battery Discharge Current"
    output_source_priority:
      name: "${inverter_name} Output Source Priority"
    charger_source_priority:
      name: "${inverter_name} Charger Source Priority"
    output_mode:
      name: "${inverter_name} Output Mode"
      
binary_sensor:
  - platform: pipsolar
    pipsolar_id: inverter0
    add_sbu_priority_version:
      name: "${inverter_name} Add SBU Priority Version"
    configuration_status:
      name: "${inverter_name} configuration_status"
    load_status:
      name: "${inverter_name} Load Status"
    battery_voltage_to_steady_while_charging:
      name: "${inverter_name} Battery Voltage to Steady While Charging"
    charging_status:
      name: "${inverter_name} Charging Status"
    scc_charging_status:
      name: "${inverter_name} SCC charging status"
    ac_charging_status:
      name: "${inverter_name} AC charging status"
    charging_to_floating_mode:
      name: "${inverter_name} Charging to floating mode"
    switch_on:
      name: "${inverter_name} Switch On"
    fault_inverter_fault:
      name: "${inverter_name} Fault"
    fault_inverter_voltage_too_low:
      name: "${inverter_name} Fault Voltage Too Low"
    warning_fan_lock:
      name: "${inverter_name} Fault Fan Lock"
    warning_battery_low_alarm:
      name: "${inverter_name} Battery Low Alarm"
    fault_inverter_over_current:
      name: "${inverter_name} Fault Over Current"
    warning_over_load:
      name: "${inverter_name} Warning Over Load"
    fault_code:
      name: "${inverter_name} Fault Code"
      
switch:
  - platform: pipsolar
    pipsolar_id: inverter0
    output_source_priority_utility:
      name: "${inverter_name} output_source_priority_utility"
    output_source_priority_solar:
      name: "${inverter_name} output_source_priority_solar"
    output_source_priority_battery:
      name: "${inverter_name} output_source_priority_battery"
    input_voltage_range:
      name: "${inverter_name} input_voltage_range"

derekyle, 2022-07-17:
It looks like the same sequence of bytes going out everytime and a different sequence coming back each time. And I suspect this is coming from the inverter because swapping tx/Rx on either side of the converter results in all timeouts. And unplugging the rj45 on the inverter or switching the rj45 over to the BMS plug also gives timeouts.

syssi, 2022-07-17:
Thanks for the logs of the raw messages. You are right. This is a response of the inverter (no echos here). I will try to decode the response manually. It looks like the inverter responds with the same answer every time (independent of the request/command).

syssi, 2022-07-17:
The meaning of `28:4E:41:4B:73:73:0D` is `NAK`. The inverter rejects all requests.

derekyle, 2022-07-17:
Does this mean that the rs232-ttl converter wiring is likely correct?

syssi, 2022-07-18:
Yes. The wiring is correct!

syssi, 2022-07-18:
May be the protocol of your inverter is a bit different like this: https://github.com/jblance/mpp-solar/issues/33

Could you try to use mppsolar (it already supports the different protocol). Please ping me if you have success.

SeByDocKy, 2022-07-18:
As I already written, I used without any problem for a frien the oroginal PIPsolar code with an King II .... so I am pretty sure it's a problem of wiring and the choice of a right adapter. I remember I spent some time to make it working. If you want I can ask him some photos.... and the GPIO I used.

SeByDocKy, 2022-07-18:
Here is the photo : 
[image]

The DB9 cable was the one shipped with the inverter and the RS232 TTL converter is this model : https://fr.aliexpress.com/item/1005003095210072.html?spm=a2g0o.order_list.0.0.5aea5e5bLKmgTm&gatewayAdapt=glo2fra

Be aware to take the right gender according to your DB9 cable.

syssi, 2022-07-18:
This traffic

```
[09:49:31][D][uart_debug:114]: >>> 51:4D:4F:44:49:C1:0D
[09:49:31][D][pipsolar:772]: checking crc on incoming message
[09:49:31][D][pipsolar:775]: CRC OK
[09:49:31][D][uart_debug:114]: <<< 28:4E:41:4B:73:73:0D
```

shows the inverter is able to receive data and responds with `NAK`. IMO the wiring is fine and the protocol is probably a bit different. I found a bug report above at the mppsolar project which describes a similar behavior which could be solved by sending commands to the inverter without appending a checksum.

derekyle, 2022-07-24:
I haven't had a chance to try the jblance code directly yet. But I will soon. However, I did get an answer back from my dealer and they provided this excel with the protocol details.

[https://docs.google.com/spreadsheets/d/1_Gpx5ZK-nOyW5oE5jj7XfJ9twNGWV18e/edit?usp=sharing&ouid=101779422375876088393&rtpof=true&sd=true](url)

derekyle, 2022-07-24:
@SeByDocKy sorry for the confusion, I updated the title of this issue. It seems that my unit is a bit different than the King II and is actually identical to the Infinisolar V II. The biggest difference being that mine and the infinisolar are both split phase and have a few higher ratings. Otherwise, they look identical on the outside. But it seems like maybe there was a change to the protocol on these newer split phase models.

derekyle, 2022-09-24:
@syssi Ok, I was finally able to get around to trying out mpp-solar on my device. After playing around with it a little bit, it appears that I am able to successfully query my device using the PI18 protocol that mpp-solar implements. Is this project utilizing the PI18 protocol?

syssi, 2022-09-24:
Could you provide an request and response example of the `QPIGS` command or could you provide some details about the PI18 protocol?

syssi, 2022-09-24:
I've checked the [PI18 protocol implementation](https://github.com/jblance/mpp-solar/blob/master/mppsolar/protocols/pi18.py) of mppsolar and the [protocol definition](https://github.com/jblance/mpp-solar/blob/master/docs/protocols/PI18_InfiniSolar-V-protocol-20170926.pdf). The pipsolar component uses a different command set and doesn't support `^P0...` commands yet.

derekyle, 2022-09-24:
There is no QPIGS command for the PI18 protocol. So it gives me:

'full_command not found for QPIGS in protocol b'PI18''

But the GS command gives the following:

```
Parameter                       Value           Unit
grid_voltage                    118.2           V   
grid_frequency                  59.9            Hz  
ac_output_voltage               0.0             V   
ac_output_frequency             0.0             Hz  
ac_output_apparent_power        0               VA  
ac_output_active_power          0               W   
output_load_percent             0               %   
battery_voltage                 54.3            V   
battery_voltage_from_scc        0.0             V   
battery_voltage_from_scc2       0.0             V   
battery_discharge_current       0               A   
battery_charging_current        0               A   
battery_capacity                100             %   
inverter_heat_sink_temperature  47              °C  
mppt1_charger_temperature       0               °C  
mppt2_charger_temperature       0               °C  
pv1_input_power                 0               W   
pv2_input_power                 0               W   
pv1_input_voltage               0.0             V   
pv2_input_voltage               0.0             V   
setting_value_configuration_state       Nothing changed     
mppt1_charger_status            abnormal            
mppt2_charger_status            abnormal            
load_connection                 disconnect          
battery_power_direction         charge              
dc/ac_power_direction           AC-DC               
line_power_direction            input               
```

derekyle, 2022-10-11:
Interesting. I wonder if there is a setting on the Inverter that might need
to be changed. Others have reported this working on an identical Voltronic
model. But maybe SunGoldPower has made some modifications to their units
beyond the blue paint and logo on the case.

syssi, 2026-05-19:
Pi18 support is available now. Please use: esp8266-pi18-example.yaml / esp32-pi18-example.yaml