syssi/esphome-pipsolar #24: Voltronic Axpert MKS IV 5.6kW no PV data

ardupic, 2022-09-08:
Hi, i've flashed your code on a esp8266 nodemcu amica. Most of data look ok but i can't retreive power voltage and current of panels.
how can i fix this issue?
thanks

syssi, 2022-09-25:
Could you enable the debug output of the `uart` component by:

```
uart:
  id: uart0
  baud_rate: 2400
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH
    dummy_receiver: true
    after:
      delimiter: "\r"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
```

And provide a log capture? I would like to check the responses of your inverter.

ardupic, 2022-10-24:
thank syssi, 
i'v solved using your 2424mse1 branch. super!!! could you explain me how modify code for different protocols?
thank for all

closed 2022-11-27 by syssi