syssi/esphome-pipsolar #25: I have a few qestions about this PIP 2424.:

ninlee, 2022-09-19:
I have a few qestions about this PIP 2424.:
-1 how can set Solar Charge priority:PCP00,PCP01,PCP02.PCP03?
-2 how to send other command(change bulk and float voltage,ecc,ecc)?? before using MQTT was very easy.
I no clearly understand if this brach allows(as lv5048).

charger_source_priority:
      id: inverter0_charger_source_priority_select
      name: inverter0_charger_source_priority_select
      optionsmap:
        "Utility first": "PCP00"
        "Solar first": "PCP01"
        "Solar and Utility": "PCP02"
        "Solar only": "PCP03"
      statusmap:
        "0": "Utility first"
        "1": "Solar first"
        "2": "Solar and Utility"
        "3": "Solar only"

I have no clear how to use output and automation?
  - platform: pipsolar
    pipsolar_id: inverter0
    battery_recharge_voltage:
      id: inverter0_battery_recharge_voltage_out

also thus not working :select..
select:
  - platform: pipsolar
    pipsolar_id: inverter0
    output_source_priority:
      id: inverter0_output_source_priority_select
      name: inverter0_output_source_priority_select
      optionsmap:
        "Utility first": "POP00"
        "Solar only": "POP01"
        "Solar Battery Utility": "POP02"
      statusmap:
        "0": "Utility first"
        "1": "Solar only"
        "2": "Solar Battery Utility"          
      

Can you kindly make some example?...im afraid to make some damage to inverter.
Thank you for your time.

_Originally posted by @ans-gw in https://github.com/syssi/esphome-pipsolar/issues/3#issuecomment-1156627223_

ans-gw, 2022-09-19:
> I have a few qestions about this PIP 2424.: -1 how can set Solar Charge priority:PCP00,PCP01,PCP02.PCP03? -2 how to send other command(change bulk and float voltage,ecc,ecc)?? before using MQTT was very easy. I no clearly understand if this brach allows(as lv5048).
> 
> charger_source_priority: id: inverter0_charger_source_priority_select name: inverter0_charger_source_priority_select optionsmap: "Utility first": "PCP00" "Solar first": "PCP01" "Solar and Utility": "PCP02" "Solar only": "PCP03" statusmap: "0": "Utility first" "1": "Solar first" "2": "Solar and Utility" "3": "Solar only"
> 
> I have no clear how to use output and automation?
> 
> * platform: pipsolar
>   pipsolar_id: inverter0
>   battery_recharge_voltage:
>   id: inverter0_battery_recharge_voltage_out
> 
> also thus not working :select.. select:
> 
> * platform: pipsolar
>   pipsolar_id: inverter0
>   output_source_priority:
>   id: inverter0_output_source_priority_select
>   name: inverter0_output_source_priority_select
>   optionsmap:
>   "Utility first": "POP00"
>   "Solar only": "POP01"
>   "Solar Battery Utility": "POP02"
>   statusmap:
>   "0": "Utility first"
>   "1": "Solar only"
>   "2": "Solar Battery Utility"
> 
> Can you kindly make some example?...im afraid to make some damage to inverter. Thank you for your time.
> 
> _Originally posted by @ans-gw in [#3 (comment)](https://github.com/syssi/esphome-pipsolar/issues/3#issuecomment-1156627223)_

I agree,need a clear explanation...I m still waiting.

20after4, 2024-03-17:
I tried to implement this using templates and lambdas, however, it turns out that the set_level action only accepts values between 0 and 1, so setting the output_source_priority doesn't work as expected.

Here's my code, which mostly works but never sets a value other than 0 or 1:

```yaml
select:
  - platform: template
    id: ${inverter_id}_output_source_priority_select
    name: "${inverter_name} Output Source Priority"
    options:
      - 'Utility'
      - 'Solar'
      - 'Battery'
    update_interval: 30s
    lambda: !lambda |-
      auto sensor = id(${inverter_id}_output_source_priority);
      if (sensor->has_state()) {
        int index = static_cast<int>(sensor->state);
        auto option = id(${inverter_id}_output_source_priority_select).at(index);
        if (option.has_value()) {
          return option.value();
        }
      }
      return std::string("Utility");
    set_action:
      then:
        - lambda: |-
            float value = 1;
            auto select = id(${inverter_id}_output_source_priority_select);
            auto index = select->index_of(x.c_str());
            ESP_LOGI("pip", "set_action: %s", x.c_str());
            ESP_LOGI("pip", "state: %s", select->state.c_str());
            if (index.has_value()) {
              value = static_cast<float>(index.value());
              ESP_LOGI("pip", "value: %f", value);
              id(${inverter_id}_output_source_priority_out).set_level(value);
            }

  - platform: template
    id: ${inverter_id}_charger_source_priority_select
    name: "${inverter_name} Charge Source Priority"
    options:
      - 'Utility First'
      - 'Solar First'
      - 'Solar + Utility'
      - 'Solar Only'
    lambda: !lambda |-
      auto sensor = id(${inverter_id}_charger_source_priority);
      if (sensor->has_state()) {
        int index = static_cast<int>(sensor->state);
        auto option = id(${inverter_id}_charger_source_priority_select).at(index);
        if (option.has_value()) {
          return option.value();
        }
      }
      return std::string("Solar Only");
    set_action:
      then:
        - output.set_level:
            id: ${inverter_id}_charger_source_priority_out
            level: !lambda |-
              float value = 1;
              auto select = id(${inverter_id}_charger_source_priority_select);
              auto index = select->index_of(x.c_str());
              ESP_LOGI("pip", "set_action: %s", x.c_str());
              ESP_LOGI("pip", "previous state: %s", select->state.c_str());
              if (index.has_value()) {
                value = static_cast<float>(index.value());
                return value;
              }
              return 1.0;
```

20after4, 2024-03-17:
ok I got it to work by using the pip8048 branch and the following additions to my config:

```yaml
external_components:
  - source: https://github.com/syssi/esphome-pipsolar@pip8048
    components: [ pipsolar ]

select:
  - platform: pipsolar
    pipsolar_id: ${inverter_id}
    charger_source_priority:
      id: ${inverter_id}_charger_source_priority_select
      name: ${inverter_name} Charger Source Priority Select
      optionsmap:
        "Utility first": "PCP00"
        "Solar first": "PCP01"
        "Solar and utility": "PCP02"
        "Solar only": "PCP03"
      statusmap:
        "0": "Utility first"
        "1": "Solar first"
        "2": "Solar and utility"
        "3": "Solar only"
    output_source_priority:
      id: ${inverter_id}_output_source_priority_select
      name: ${inverter_name} Output Source Priority
      optionsmap:
        "Utility first": "POP00"
        "Solar only": "POP01"
        "Solar Battery Utility": "POP02"
      statusmap:
        "0": "Utility first"
        "1": "Solar only"
        "2": "Solar Battery Utility"
```

I also had to change various sensor names to match the updated names from the new branch. Essentially just replace `pv_`  with `pv1_`  e.g. `pv_input_current` becomes `pv1_input_current`