syssi/esphome-pipsolar #26: pip8048 branch have no more switch working

SeByDocKy, 2022-09-23:
I just updated last ESPhome dev and now gor error while compile the PVbrain using the pip8048 branch.

Here is the log of compilation
```    name: pvbrain_axpert_input_voltage_range
(base) PS C:\Users\SeByDocKy\Desktop\SeBy\Maison\Home assistant\47> esphome compile pvbrain.yaml
←[32mINFO Reading configuration pvbrain.yaml...←[0m
←[32mINFO Updating https://github.com/syssi/esphome-jk-bms.git@main←[0m
←[32mINFO Updating https://github.com/syssi/esphome-pipsolar.git@pip8048←[0m
←[31mERROR Unable to load component pipsolar.switch:
Traceback (most recent call last):
  File "C:\Users\SeByDocKy\anaconda3\lib\site-packages\esphome\loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "C:\Users\SeByDocKy\anaconda3\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "C:\Users\SeByDocKy\Desktop\SeBy\Maison\Home assistant\47\.esphome\external_components\8e1d245d\components\pipsolar\switch\__init__.py", line 38, in <module>
    cv.Optional(CONF_ICON, default=ICON_POWER): switch.icon,
AttributeError: module 'esphome.components.switch' has no attribute 'icon'←[0m
←[32mINFO Detected timezone 'Europe/Paris'←[0m
Failed config

switch.pipsolar: [source pvbrain.yaml:503]

  Platform not found: 'switch.pipsolar'.
  platform: pipsolar
  pipsolar_id: pip8048
  output_source_priority_utility:
    name: pvbrain_axpert_output_source_priority_utility
  output_source_priority_solar:
    name: pvbrain_axpert_output_source_priority_solar
  output_source_priority_battery:
    name: pvbrain_axpert_output_source_priority_battery
  input_voltage_range:
    name: pvbrain_axpert_input_voltage_range
(base) PS C:\Users\SeByDocKy\Desktop\SeBy\Maison\Home assistant\47>
```

syssi, 2022-09-23:
I know what needs to be changed. I will care about these changes tomorrow.

SeByDocKy, 2022-09-23:
Great :)

syssi, 2022-09-24:
Fixed.

closed 2022-09-24 by syssi

SeByDocKy, 2022-09-24:
Compile fine now :) Thanks :) 👍