syssi/esphome-pipsolar #27: Steca Solarix PLI 5000-48 // Axpert MKS 5K-48

ma-schmitz, 2022-09-25:
Hello,

I use the above inverter and would like to read it with ESPHome.
Unfortunately, I do not get proper answers from this inverter and just do not know where to look further.
I have attached the log files and my ESPHome configuration. Maybe someone wants to help me.

I use a nodemcu-32s, MAX3232 RS232 to TTL and a DC-DC converter.

Thanks in advance

**Log-File**

`INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from steca-solar-inverter.local using esphome API
INFO Successfully connected to steca-solar-inverter.local
[11:29:46][I][app:102]: ESPHome version 2022.3.1 compiled on Sep 24 2022, 11:29:08
[11:29:46][C][wifi:491]: WiFi:
[11:29:46][C][wifi:353]:   Local MAC: 78:21:84:9F:6C:20
[11:29:46][C][wifi:354]:   SSID: 'Schmitz'[redacted]
[11:29:46][C][wifi:355]:   IP Address: 192.168.11.164
[11:29:46][C][wifi:357]:   BSSID: 80:2A:A8:2D:CE:DC[redacted]
[11:29:46][C][wifi:358]:   Hostname: 'steca-solar-inverter'
[11:29:46][C][wifi:360]:   Signal strength: -76 dB ▂▄▆█
[11:29:46][C][wifi:364]:   Channel: 1
[11:29:46][C][wifi:365]:   Subnet: 255.255.255.0
[11:29:46][C][wifi:366]:   Gateway: 192.168.11.1
[11:29:46][C][wifi:367]:   DNS1: 192.168.11.84
[11:29:46][C][wifi:368]:   DNS2: 0.0.0.0
[11:29:46][C][logger:233]: Logger:
[11:29:46][C][logger:234]:   Level: DEBUG
[11:29:46][C][logger:235]:   Log Baud Rate: 115200
[11:29:46][C][logger:236]:   Hardware UART: UART0
[11:29:46][C][uart.arduino_esp32:107]: UART Bus:
[11:29:46][C][uart.arduino_esp32:108]:   TX Pin: GPIO22
[11:29:46][C][uart.arduino_esp32:109]:   RX Pin: GPIO23
[11:29:46][C][uart.arduino_esp32:111]:   RX Buffer Size: 256
[11:29:46][C][uart.arduino_esp32:113]:   Baud Rate: 2400 baud
[11:29:46][C][uart.arduino_esp32:114]:   Data Bits: 8
[11:29:46][C][uart.arduino_esp32:115]:   Parity: NONE
[11:29:46][C][uart.arduino_esp32:116]:   Stop bits: 1
[11:29:46][D][pipsolar:838]: Sending polling command : QPIRI with length 5
[11:29:46][C][pipsolar:860]: Pipsolar:
[11:29:46][C][pipsolar:861]: used commands:
[11:29:46][C][pipsolar:864]: QPIRI
[11:29:46][C][pipsolar:864]: QMOD
[11:29:46][C][pipsolar:864]: QPIGS
[11:29:46][C][pipsolar:864]: QFLAG
[11:29:46][C][pipsolar.switch:010]: Pipsolar Switch 'steca-solar-inverter output_source_priority_utility'
[11:29:46][C][pipsolar.switch:010]:   Icon: 'mdi:power'
[11:29:46][D][pipsolar:772]: checking crc on incoming message
[11:29:46][D][pipsolar:775]: CRC OK
[11:29:46][C][pipsolar.switch:010]: Pipsolar Switch 'steca-solar-inverter output_source_priority_solar'
[11:29:46][C][pipsolar.switch:010]:   Icon: 'mdi:power'
[11:29:46][D][pipsolar:429]: Decode QPIRI
[11:29:46][D][text_sensor:067]: 'steca-solar-inverter last_qpiri': Sending state 'QPIRI'
[11:29:46][C][pipsolar.switch:010]: Pipsolar Switch 'steca-solar-inverter output_source_priority_battery'
[11:29:46][C][pipsolar.switch:010]:   Icon: 'mdi:power'
[11:29:46][D][sensor:125]: 'steca-solar-inverter grid_rating_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[11:29:46][D][sensor:125]: 'steca-solar-inverter grid_rating_current': Sending state 0.00000 A with 1 decimals of accuracy
[11:29:46][D][sensor:125]: 'steca-solar-inverter ac_output_rating_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[11:29:46][D][sensor:125]: 'steca-solar-inverter ac_output_rating_frequency': Sending state 0.00000 Hz with 1 decimals of accuracy
[11:29:46][D][sensor:125]: 'steca-solar-inverter ac_output_rating_current': Sending state 0.00000 A with 1 decimals of accuracy
[11:29:46][D][sensor:125]: 'steca-solar-inverter ac_output_rating_apparent_power': Sending state 0.00000 VA with 1 decimals of accuracy
[11:29:46][D][sensor:125]: 'steca-solar-inverter ac_output_rating_active_power': Sending state 0.00000 W with 1 decimals of accuracy
[11:29:46][D][sensor:125]: 'steca-solar-inverter current_max_ac_charging_current': Sending state 0.00000 A with 1 decimals of accuracy
[11:29:46][D][sensor:125]: 'steca-solar-inverter current_max_charging_current': Sending state 0.00000 A with 1 decimals of accuracy
[11:29:46][D][sensor:125]: 'steca-solar-inverter output_source_priority': Sending state 0.00000  with 1 decimals of accuracy
[11:29:46][D][switch:037]: 'steca-solar-inverter output_source_priority_utility': Sending state ON
[... 162 lines cut ...]
    pipsolar_id: inverter0
    charging_status:
      name: "${name} charging_status"
    scc_charging_status:
      name: "${name} scc_charging_status"
    ac_charging_status:
      name: "${name} ac_charging_status"
    backlight_on:
      name: "${name} backlight_on"

switch:
  - platform: pipsolar
    pipsolar_id: inverter0
    output_source_priority_utility:
      name: "${name} output_source_priority_utility"
    output_source_priority_solar:
      name: "${name} output_source_priority_solar"
    output_source_priority_battery:
      name: "${name} output_source_priority_battery"

output:
  - platform: pipsolar
    pipsolar_id: inverter0
    battery_recharge_voltage:
      id: inverter0_battery_recharge_voltage_out`

syssi, 2022-09-25:
Please enable the debug output of the `uart` component by

```
uart:
  - id: uart_bus
    tx_pin: GPIO22
    rx_pin: GPIO23
    # most devices use 2400 as baud_rate
    baud_rate: 2400
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
```

and flash the ESP again. Please provide a new log capture. This time the raw traffic should be part of the log. I assume the component doesn't receive a valid response of the inverter it's probably an echo of your request.

ma-schmitz, 2022-09-25:
Hello,

hier are the logs:

`[11:06:22][C][uart.arduino_esp32:107]: UART Bus:
[11:06:22][C][uart.arduino_esp32:108]:   TX Pin: GPIO23
[11:06:22][C][uart.arduino_esp32:109]:   RX Pin: GPIO22
[11:06:22][C][uart.arduino_esp32:111]:   RX Buffer Size: 256
[11:06:22][C][uart.arduino_esp32:113]:   Baud Rate: 2400 baud
[11:06:22][C][uart.arduino_esp32:114]:   Data Bits: 8
[11:06:22][C][uart.arduino_esp32:115]:   Parity: NONE
[11:06:22][C][uart.arduino_esp32:116]:   Stop bits: 1
[11:06:22][D][pipsolar:838]: Sending polling command : QPIRI with length 5
[11:06:22][C][pipsolar:860]: Pipsolar:
[11:06:22][C][pipsolar:861]: used commands:
[11:06:22][C][pipsolar:864]: QPIRI
[11:06:22][C][pipsolar:864]: QMOD
[11:06:22][C][pipsolar:864]: QPIGS
[11:06:22][C][pipsolar:864]: QFLAG
[11:06:22][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[11:06:22][C][pipsolar.switch:010]: Pipsolar Switch 'steca-solar-inverter output_source_priority_utility'
[11:06:22][C][pipsolar.switch:010]:   Icon: 'mdi:power'
[11:06:22][D][pipsolar:772]: checking crc on incoming message
[11:06:22][D][pipsolar:775]: CRC OK
[11:06:22][C][pipsolar.switch:010]: Pipsolar Switch 'steca-solar-inverter output_source_priority_solar'
[11:06:22][C][pipsolar.switch:010]:   Icon: 'mdi:power'
[11:06:22][D][pipsolar:429]: Decode QPIRI
[11:06:22][D][text_sensor:067]: 'steca-solar-inverter last_qpiri': Sending state 'QPIRI'
[11:06:22][C][pipsolar.switch:010]: Pipsolar Switch 'steca-solar-inverter output_source_priority_battery'
[11:06:22][C][pipsolar.switch:010]:   Icon: 'mdi:power'
[11:06:22][D][sensor:125]: 'steca-solar-inverter grid_rating_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[11:06:22][D][sensor:125]: 'steca-solar-inverter grid_rating_current': Sending state 0.00000 A with 1 decimals of accuracy
[11:06:22][D][sensor:125]: 'steca-solar-inverter ac_output_rating_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[11:06:22][D][sensor:125]: 'steca-solar-inverter ac_output_rating_frequency': Sending state 0.00000 Hz with 1 decimals of accuracy
[11:06:22][D][sensor:125]: 'steca-solar-inverter ac_output_rating_current': Sending state 0.00000 A with 1 decimals of accuracy
[11:06:22][D][sensor:125]: 'steca-solar-inverter ac_output_rating_apparent_power': Sending state 0.00000 VA with 1 decimals of accuracy
[11:06:22][D][sensor:125]: 'steca-solar-inverter ac_output_rating_active_power': Sending state 0.00000 W with 1 decimals of accuracy
[11:06:22][D][sensor:125]: 'steca-solar-inverter current_max_ac_charging_current': Sending state 0.00000 A with 1 decimals of accuracy
[11:06:22][D][sensor:125]: 'steca-solar-inverter current_max_charging_current': Sending state 0.00000 A with 1 decimals of accuracy
[11:06:22][D][sensor:125]: 'steca-solar-inverter output_source_priority': Sending state 0.00000  with 1 decimals of accuracy
[11:06:23][D][switch:037]: 'steca-solar-inverter output_source_priority_utility': Sending state ON
[11:06:23][D][switch:037]: 'steca-solar-inverter output_source_priority_solar': Sending state OFF
[11:06:23][D][switch:037]: 'steca-solar-inverter output_source_priority_battery': Sending state OFF
[11:06:23][D][sensor:125]: 'steca-solar-inverter charger_source_priority': Sending state 0.00000  with 1 decimals of accuracy
[11:06:23][D][sensor:125]: 'steca-solar-inverter output_mode': Sending state 0.00000  with 1 decimals of accuracy
[11:06:23][D][sensor:125]: 'steca-solar-inverter battery_redischarge_voltage': Sending state 0.00000  with 1 decimals of accuracy
[11:06:23][C][mdns:084]: mDNS:
[11:06:23][C][mdns:085]:   Hostname: steca-solar-inverter
[11:06:23][D][uart_debug:158]: <<< "QPIRI\xF8T\r"
[11:06:23][C][ota:085]: Over-The-Air Updates:
[11:06:23][C][ota:086]:   Address: steca-solar-inverter.local:3232
[11:06:23][C][ota:089]:   Using Password.
[11:06:23][C][api:138]: API Server:
[11:06:23][C][api:139]:   Address: steca-solar-inverter.local:6053
[11:06:23][C][api:143]:   Using noise encryption: NO
[11:06:23][D][pipsolar:838]: Sending polling command : QMOD with length 4
[11:06:23][D][uart_debug:158]: >>> "QMODI\xC1\r"
[11:06:23][D][pipsolar:772]: checking crc on incoming message
[11:06:23][D][pipsolar:775]: CRC OK
[11:06:23][D][pipsolar:471]: Decode QMOD
[11:06:23][D][text_sensor:067]: 'steca-solar-inverter last_qmod': Sending state 'QMOD'
[11:06:23][D][text_sensor:067]: 'steca-solar-inverter device_mode': Sending state 'M'
[11:06:23][D][uart_debug:158]: <<< "QMODI\xC1\r"
[11:06:24][D][pipsolar:838]: Sending polling command : QPIGS with length 5
[11:06:24][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[11:06:24][D][pipsolar:772]: checking crc on incoming message
[11:06:24][D][pipsolar:775]: CRC OK
[11:06:24][D][pipsolar:448]: Decode QPIGS
[11:06:24][D][text_sensor:067]: 'steca-solar-inverter last_qpigs': Sending state 'QPIGS'
[11:06:24][D][uart_debug:158]: <<< "QPIGS\xB7\xA9\r"
[... 92 lines cut ...]
[11:06:34][D][uart_debug:158]: <<< "QFLAG\x98t\r"
[11:06:34][D][pipsolar:838]: Sending polling command : QPIRI with length 5
[11:06:34][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[11:06:34][D][pipsolar:772]: checking crc on incoming message
[11:06:34][D][pipsolar:775]: CRC OK
[11:06:34][D][pipsolar:429]: Decode QPIRI
[11:06:34][D][text_sensor:067]: 'steca-solar-inverter last_qpiri': Sending state 'QPIRI'
[11:06:35][D][sensor:125]: 'steca-solar-inverter grid_rating_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[... cut ...]
[11:06:35][D][sensor:125]: 'steca-solar-inverter output_mode': Sending state 0.00000  with 1 decimals of accuracy
[11:06:35][D][sensor:125]: 'steca-solar-inverter battery_redischarge_voltage': Sending state 0.00000  with 1 decimals of accuracy
[11:06:35][D][uart_debug:158]: <<< "QPIRI\xF8T\r"
[11:06:35][D][pipsolar:838]: Sending polling command : QMOD with length 4
[11:06:35][D][uart_debug:158]: >>> "QMODI\xC1\r"
[11:06:35][D][pipsolar:772]: checking crc on incoming message
[11:06:35][D][pipsolar:775]: CRC OK`

syssi, 2022-09-25:
```
[11:06:23][D][uart_debug:158]: >>> "QMODI\xC1\r"
[11:06:23][D][uart_debug:158]: <<< "QMODI\xC1\r"

[11:06:24][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[11:06:24][D][uart_debug:158]: <<< "QPIGS\xB7\xA9\r"

[11:06:25][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[11:06:25][D][uart_debug:158]: <<< "QFLAG\x98t\r"

[11:06:26][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[11:06:27][D][uart_debug:158]: <<< "QPIRI\xF8T\r"

[11:06:27][D][uart_debug:158]: >>> "QMODI\xC1\r"
[11:06:27][D][uart_debug:158]: <<< "QMODI\xC1\r"

[11:06:28][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[11:06:28][D][uart_debug:158]: <<< "QPIGS\xB7\xA9\r"

[11:06:29][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[11:06:29][D][uart_debug:158]: <<< "QFLAG\x98t\r"

[11:06:30][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[11:06:31][D][uart_debug:158]: <<< "QPIRI\xF8T\r"

[11:06:31][D][uart_debug:158]: >>> "QMODI\xC1\r"
[11:06:32][D][uart_debug:158]: <<< "QMODI\xC1\r"

[11:06:32][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[11:06:33][D][uart_debug:158]: <<< "QPIGS\xB7\xA9\r"

[11:06:33][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[11:06:34][D][uart_debug:158]: <<< "QFLAG\x98t\r"

[11:06:34][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[11:06:35][D][uart_debug:158]: <<< "QPIRI\xF8T\r"
```

It looks like your wiring isn't correct or your inverter responds with the same message your sending. Please double check your setup. Sometimes the RX/TX lines between the inverter and the RS232 converter must be swapped. Sometimes the RX/TX lines between the RS232 converter and the ESP are the root cause and must be swapped.

ma-schmitz, 2022-09-25:
so, I never swapped the RS232 side, now at least I get data.
But if I am not mistaken there is still a lot missing.
Or am I thinking wrong and now everything is as expected?

[18:17:39][C][logger:233]: Logger:
[18:17:39][C][logger:234]:   Level: DEBUG
[18:17:39][C][logger:235]:   Log Baud Rate: 115200
[18:17:39][C][logger:236]:   Hardware UART: UART0
[18:17:39][C][uart.arduino_esp32:107]: UART Bus:
[18:17:39][C][uart.arduino_esp32:108]:   TX Pin: GPIO22
[18:17:39][C][uart.arduino_esp32:109]:   RX Pin: GPIO23
[18:17:39][C][uart.arduino_esp32:111]:   RX Buffer Size: 256
[18:17:39][C][uart.arduino_esp32:113]:   Baud Rate: 2400 baud
[18:17:39][C][uart.arduino_esp32:114]:   Data Bits: 8
[18:17:39][C][uart.arduino_esp32:115]:   Parity: NONE
[18:17:39][C][uart.arduino_esp32:116]:   Stop bits: 1
[18:17:39][D][pipsolar:838]: Sending polling command : QPIGS with length 5
[18:17:39][C][pipsolar:860]: Pipsolar:
[18:17:39][C][pipsolar:861]: used commands:
[18:17:39][C][pipsolar:864]: QPIRI
[18:17:39][C][pipsolar:864]: QMOD
[18:17:39][C][pipsolar:864]: QPIGS
[18:17:39][C][pipsolar:864]: QFLAG
[18:17:39][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[18:17:39][C][pipsolar.switch:010]: Pipsolar Switch 'steca-solar-inverter output_source_priority_utility'
[18:17:39][C][pipsolar.switch:010]:   Icon: 'mdi:power'
[18:17:39][C][pipsolar.switch:010]: Pipsolar Switch 'steca-solar-inverter output_source_priority_solar'
[18:17:39][C][pipsolar.switch:010]:   Icon: 'mdi:power'
[18:17:39][C][pipsolar.switch:010]: Pipsolar Switch 'steca-solar-inverter output_source_priority_battery'
[18:17:39][C][pipsolar.switch:010]:   Icon: 'mdi:power'
[18:17:39][C][mdns:084]: mDNS:
[18:17:39][C][mdns:085]:   Hostname: steca-solar-inverter
[18:17:39][C][ota:085]: Over-The-Air Updates:
[18:17:39][C][ota:086]:   Address: steca-solar-inverter.local:3232
[18:17:39][C][ota:089]:   Using Password.
[18:17:39][C][api:138]: API Server:
[18:17:39][C][api:139]:   Address: steca-solar-inverter.local:6053
[18:17:39][C][api:143]:   Using noise encryption: NO
[18:17:40][D][pipsolar:772]: checking crc on incoming message
[18:17:40][D][pipsolar:775]: CRC OK
[18:17:40][D][pipsolar:448]: Decode QPIGS
[18:17:40][D][text_sensor:067]: 'steca-solar-inverter last_qpigs': Sending state '(234.9 50.0 230.4 49.9 0530 0448 010 345 49.90 000 042 0026 0000 085.1 49.86 00008 00110110 00 00 00034 110'
[18:17:40][D][uart_debug:158]: <<< "(234.9 50.0 230.4 49.9 0530 0448 010 345 49.90 000 042 0026 0000 085.1 49.86 00008 00110110 00 00 00034 1102\xD0\r"
[18:17:40][D][pipsolar:838]: Sending polling command : QFLAG with length 5
[18:17:40][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[18:17:40][D][pipsolar:772]: checking crc on incoming message
[18:17:40][D][pipsolar:775]: CRC OK
[18:17:40][D][pipsolar:479]: Decode QFLAG
[18:17:40][D][text_sensor:067]: 'steca-solar-inverter last_qflag': Sending state '(EbuvxyDajkz'
[18:17:40][D][uart_debug:158]: <<< "(EbuvxyDajkz8\x83\r"
[18:17:41][D][pipsolar:838]: Sending polling command : QPIRI with length 5
[18:17:41][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[18:17:42][D][pipsolar:772]: checking crc on incoming message
[18:17:42][D][pipsolar:775]: CRC OK
[18:17:42][D][pipsolar:429]: Decode QPIRI
[18:17:42][D][text_sensor:067]: 'steca-solar-inverter last_qpiri': Sending state '(230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 46.0 45.5 53.2 53.1 2 02 010 1 1 3 9 01 0 0 51.0 0 1 060'
[18:17:42][D][sensor:125]: 'steca-solar-inverter grid_rating_voltage': Sending state 230.00000 V with 1 decimals of accuracy
[18:17:42][D][sensor:125]: 'steca-solar-inverter grid_rating_current': Sending state 21.70000 A with 1 decimals of accuracy
[18:17:42][D][sensor:125]: 'steca-solar-inverter ac_output_rating_voltage': Sending state 230.00000 V with 1 decimals of accuracy
[18:17:42][D][sensor:125]: 'steca-solar-inverter ac_output_rating_frequency': Sending state 50.00000 Hz with 1 decimals of accuracy
[18:17:42][D][sensor:125]: 'steca-solar-inverter ac_output_rating_current': Sending state 21.70000 A with 1 decimals of accuracy
[18:17:42][D][sensor:125]: 'steca-solar-inverter ac_output_rating_apparent_power': Sending state 5000.00000 VA with 1 decimals of accuracy
[18:17:42][D][sensor:125]: 'steca-solar-inverter ac_output_rating_active_power': Sending state 5000.00000 W with 1 decimals of accuracy
[18:17:42][D][sensor:125]: 'steca-solar-inverter current_max_ac_charging_current': Sending state 2.00000 A with 1 decimals of accuracy
[18:17:42][D][sensor:125]: 'steca-solar-inverter current_max_charging_current': Sending state 10.00000 A with 1 decimals of accuracy
[18:17:42][D][sensor:125]: 'steca-solar-inverter output_source_priority': Sending state 1.00000  with 1 decimals of accuracy
[18:17:42][D][sensor:125]: 'steca-solar-inverter charger_source_priority': Sending state 3.00000  with 1 decimals of accuracy
[18:17:42][D][sensor:125]: 'steca-solar-inverter output_mode': Sending state 0.00000  with 1 decimals of accuracy
[18:17:42][D][sensor:125]: 'steca-solar-inverter battery_redischarge_voltage': Sending state 51.00000  with 1 decimals of accuracy
[18:17:42][D][uart_debug:158]: <<< "(230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 46.0 45.5 53.2 53.1 2 02 010 1 1 3 9 01 0 0 51.0 0 1 060\x91~\r"
[18:17:42][D][pipsolar:838]: Sending polling command : QMOD with length 4
[18:17:42][D][uart_debug:158]: >>> "QMODI\xC1\r"
[18:17:42][D][pipsolar:772]: checking crc on incoming message
[18:17:42][D][pipsolar:775]: CRC OK
[18:17:42][D][pipsolar:471]: Decode QMOD
[... cut ...]
[18:17:44][D][pipsolar:772]: checking crc on incoming message
[18:17:44][D][pipsolar:775]: CRC OK
[18:17:44][D][pipsolar:479]: Decode QFLAG
[18:17:44][D][text_sensor:067]: 'steca-solar-inverter last_qflag': Sending state '(EbuvxyDajkz'
[18:17:45][D][uart_debug:158]: <<< "(EbuvxyDajkz8\x83\r"
[18:17:45][D][pipsolar:838]: Sending polling command : QPIRI with length 5
[18:17:45][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[18:17:46][D][pipsolar:772]: checking crc on incoming message
[18:17:46][D][pipsolar:775]: CRC OK
[18:17:46][D][pipsolar:429]: Decode QPIRI

syssi, 2022-09-25:
Could you provide some more log lines? It looks like the `QPIGS` respone isn't processed.

ma-schmitz, 2022-09-26:
so, I have tried many things again.
It looks good so far - I am getting data from the inverter.

What doesn't work so well is changing the modes (outpu source priority).
If you change from POP01 to POP00, it happens very quickly. If you change back, however, it takes between 2 and 10 minutes until the command is executed. What is the reason for this?

POP02 does not change to any mode, so this error message appears:

[11:21:45][D][switch:013]: 'steca-solar-inverter output_source_priority_battery' Turning ON.
[11:21:45][D][pipsolar:856]: got command: POP02
[11:21:45][D][pipsolar:848]: Command queued successfully: POP02 with length 5 at position 0
[11:21:45][D][uart_debug:158]: >>> "POP02\xE2\n"
[11:21:45][D][pipsolar:807]: Sending command from queue: POP02 with length 5
[11:21:45][D][uart_debug:158]: >>> "\r"
[11:21:45][D][pipsolar:062]: response length for command POP02 not OK: with length 5
[11:21:45][D][uart_debug:158]: <<< "(NAK\rs\r"

[espHOme_config.txt](https://github.com/syssi/esphome-pipsolar/files/9645056/espHOme_config.txt)
[steca-solar-inverter_logs.txt](https://github.com/syssi/esphome-pipsolar/files/9645061/steca-solar-inverter_logs.txt)

syssi, 2022-09-26:
Please update your ESPHome version. The inverter doesn't accept the POP02 command because of a wrong CRC. This was fixed recently:  https://github.com/esphome/esphome/commit/e0555e140f8e8b7debe26c498dbf115cfd52bff5

ma-schmitz, 2022-09-26:
ok, yes now the switch to POP02 works.
Is there a solution for the slow change from POP00 to another mode (POP01 / POP02)? The change to POP00 always works immediately, the change to another mode between 2 and 10 minutes.

syssi, 2022-09-26:
I don't have an answer here. Could you check the `uart` traffic? I assume the "delayed commands" are pushed immediately to the inverter and the inverter does process the command/instruction that late?!

ma-schmitz, 2022-09-26:
yes, the commands are sent immediately and the inverter delays them.

Is it possible to get "solar charge priority" as a switch?

And before I forget: Thank you very much for the effort and support.

syssi, 2022-09-26:
The modes are available as select box in future but it's work in progress.

ma-schmitz, 2022-09-26:
I just noticed that not all commands are supported.
For example, I use MCHGC060 - unfortunately this is not released.
Can this be changed?

[14:47:09][D][switch:013]: 'current_max_charging_current_60' Turning ON.
[14:47:09][D][pipsolar.output:018]: Will not write: MCHGC060 as it is not in list of allowed values
[14:47:09][D][pipsolar:772]: checking crc on incoming message
[14:47:09][D][pipsolar:775]: CRC OK
[14:47:09][D][pipsolar:429]: Decode QPIRI

syssi, 2022-09-26:
We are talking about this code fragment: https://github.com/esphome/esphome/blob/dev/esphome/components/pipsolar/output/__init__.py#L50-L74

There is a config setting called `possible_values`. Here you can define deviating values to the sensitive default.

You could try something like this:

```
sensor:
  - platform: pipsolar
    pipsolar_id: inverter0
    current_max_charging_current:
      name: "${name} current_max_charging_current"
      possible_values:
        - "MCHGC010"
        - "MCHGC020"
        - "MCHGC030"
        - "MCHGC040"
        - "MCHGC050"
        - "MCHGC060"
```

ma-schmitz, 2022-09-26:
Thanks for the quick reply, but now I get errors when compiling.

current_max_ac_charging_current: 
    name: steca-solar-inverter current_max_ac_charging_current
  current_max_charging_current: 
    name: steca-solar-inverter current_max_charging_current
    
    [possible_values] is an invalid option for [current_max_charging_current]. Did you mean [on_value]?
    possible_values: 
      - MCHGC010
      - MCHGC020
      - MCHGC030
      - MCHGC040
      - MCHGC050
      - MCHGC060
  input_voltage_range:

syssi, 2022-09-26:
My fault. I made two mistakes:

1. We are talking about outputs not sensors
2. The `possible_values` must be floats (10.0, 20.0, 30.0, ...)

SeByDocKy, 2022-09-26:
You can check the PVbrain.yaml (https://github.com/Bandit-17/PVBRAIN/blob/main/README.md) that give exemple how to proceed

syssi, 2022-09-26:
@SeByDocKy Thanks for your hint: https://github.com/Bandit-17/PVBRAIN/blob/main/pvbrain.yaml#L349-L369

closed 2022-09-27 by ma-schmitz