syssi/esphome-pipsolar #31: add split phase sensors to mppsolar lv5048 inverter

magnetus26, 2022-09-29:
hello Sebastian,
do you think its possible to add a sensor to my slit phase inverter to monitor both sides ?

I will like to be able to monitor
grid voltage phase A
grid voltage phase B
grid current phase A
grid current phase B
ac output voltage phase A
ac output voltage phase B

the inverter uses QPIGS protocol
with your guideness I can log whatever info you need

thank you

73 de KP4MI
Miguel

magnetus26, 2022-09-29:
please let me know if this log is of some use:

[logs_mppsolar-inverter-0_logs.txt](https://github.com/syssi/esphome-pipsolar/files/9679693/logs_mppsolar-inverter-0_logs.txt)

thanks,
Miguel
73 de KP4MI

magnetus26, 2022-10-28:
Hello Sebastian,
have you had some time to look at this?
thanks
best regards,
Miguel

magnetus26, 2023-04-14:
no news on this one?
thanks
Miguel

syssi, 2023-04-14:
Hi Miguel! I hope you are fine. I've extended the protocol test by some commands of your split phase converter:

https://github.com/syssi/esphome-pipsolar/commit/261dc2c58c9b5c5aad122dffa667ae9524d7fb13

Could you flash the YAML and provide some logs? Thanks in advance!

magnetus26, 2023-04-14:
> 261dc2c

should I just clean the build and flash the same YAML file or is there
needed to ne added to the yaml?

thanks
Miguel

syssi, 2023-04-14:
You can just flash the YAML but you have to change the tx_pin/rx_pin probably. If it doesn't work out of the box please provide your current/working YAML. I will provide a drop-in replacement just for testing.

magnetus26, 2023-04-14:
logs:
[logs_mppsolar-inverter-0_logs.txt](https://github.com/syssi/esphome-pipsolar/files/11235371/logs_mppsolar-inverter-0_logs.txt)

current (test) yaml:
[esp8266-test-protocols.txt](https://github.com/syssi/esphome-pipsolar/files/11235414/esp8266-test-protocols.txt)

my (original) yaml:
[mppsolar-inverter-0.txt](https://github.com/syssi/esphome-pipsolar/files/11235425/mppsolar-inverter-0.txt)

syssi, 2023-04-14:
It looks like I messed up the timing of the procotol test YAML. I've pushed an update. Could you try to most recent version of the protocol test? The one with a 1 second delay between the commands instead of 2 seconds:

https://github.com/syssi/esphome-pipsolar/blob/main/tests/esp8266-test-protocols.yaml

magnetus26, 2023-04-14:
most recent version logs:
[logs_mppsolar-inverter-0_logs (1).txt](https://github.com/syssi/esphome-pipsolar/files/11235855/logs_mppsolar-inverter-0_logs.1.txt)
same failure. :(

syssi, 2023-04-14:
```
logger:
  baud_rate: 0
  level: DEBUG
```

magnetus26, 2023-04-14:
[logs_mppsolar-inverter-0_logs.txt](https://github.com/syssi/esphome-pipsolar/files/11236180/logs_mppsolar-inverter-0_logs.txt)
 
we got something now

syssi, 2023-04-14:
Perfect! Please flash your production setup again. You have retrieved all required informations the extend the pipsolar component.

```
[14:44:03][I][main:491]: Testing PI41 split phase / multiple strings commands...
[14:44:03][D][uart_debug:158]: >>> "QPIGS2h-\r"
[14:44:03][D][uart_debug:158]: <<< "(121.7 60.0 120.2 59.9 0384 0282 012 0000 000.0 54.70 0000010\xE0A\r"
[14:44:04][D][uart_debug:158]: >>> "QP2GS0\x14\x05\r"
[14:44:04][D][uart_debug:158]: <<< "(11 55355535553555 B 00 121.8 60.00 120.1 59.96 0360 0262 012 54.7 000 065 000.0 00 00000010\v\"\r"
[14:44:05][D][uart_debug:158]: >>> "QP2GS1\x04$\r"
[14:44:05][D][uart_debug:158]: <<< "(00 00000000000000 P 00 000.0 00.00 000.0 00.00 0000 0000 000 00.0 000 000 000.0 00 00000000\xAF-\r"
```

magnetus26, 2023-04-14:
great!! do we need to create new sensors to read both phases?
logs from my production setup with your changes:
[logs_mppsolar-inverter-0_logs (1).txt](https://github.com/syssi/esphome-pipsolar/files/11236330/logs_mppsolar-inverter-0_logs.1.txt)

syssi, 2023-04-14:
Your production YAML has the same GPIO conflict. Please disable the serial logging here too and provide another log:

```
logger:
  baud_rate: 0
  level: DEBUG
```

magnetus26, 2023-04-14:
ok see attached:
[logs_mppsolar-inverter-0_logs.txt](https://github.com/syssi/esphome-pipsolar/files/11236563/logs_mppsolar-inverter-0_logs.txt)

syssi, 2023-04-14:
This looks much better.

magnetus26, 2023-04-14:
what will be the next step to add the sensors for both phases?
I was looking at the cpp and py files and looks very complicated.( to me ) 
or theres another way where not that much code editing will be needed?
73 de KP4MI

syssi, 2023-04-14:
Some (de)coding is required. ;-)

magnetus26, 2023-04-14:
coffee time !!

magnetus26, 2023-05-10:
hello Syssi
is there a way to send individual commands and see their response like POP00, POP01  ect, for debuggin or to experient? idk if possible, how can be done using esphome as a serial port and a console software.

magnetus26, 2023-06-14:
> Some (de)coding is required. ;-)

hello syssi, while waiting (if this ever happends lol). 
can you explain me how the output component works with an example?
I will like to create a switch or a selector for charger_source_priority

```
output:
  - platform: pipsolar
    pipsolar_id: inverter0
    battery_recharge_voltage:
      id: inverter0_charger_source_priority

on_...:
  then:
  - output
```

but honestly dont know how to do it, 
can you help me with the yaml code
thanks
Miguel

magnetus26, 2023-07-21:
> Perfect! Please flash your production setup again. You have retrieved all required informations the extend the pipsolar component.
>
> ```
> [14:44:03][I][main:491]: Testing PI41 split phase / multiple [...]

i dont u desrstand much but I know there are some QPIGS2 commands in your code for pv2. 
can you tell me what files deal with the sensors so I can try to make a version witb both QPIGS and QPIGS2 sensors for my split phase unit. 
i know pipsolar.cpp is one of them. if i know what other files are involve i can try to modify them to add the sensors from the other phase of.my inverter (QPIGS2 sensors). 
ill be using pip8048 as my base  branch. all i need is a list of the files that needed to be modify and i start my trial and error from there
73 de KP4MI
thanks
Miguel