syssi/esphome-pipsolar #32: problem with calling current_max_charging_current

marcinbc, 2022-10-18:
Hi, I have a problem with calling the current_max_charging_current service, the current_max_ac_charging_current service on the same settings
works great
[logs_rekuperator_upload.txt](https://github.com/syssi/esphome-pipsolar/files/9814867/logs_rekuperator_upload.txt)

marcinbc, 2022-11-14:
[21:58:10][D][number:054]: 'pvbrain_axpert_current_max_charging_current_number' - Setting number value
[21:58:10][D][number:113]:   New number value: 20.000000
[21:58:10][D][number:012]: 'pvbrain_axpert_current_max_charging_current_number': Sending state 20.000000
[21:58:10][D][pipsolar.output:015]: Will write: MCHGC020 out of value 20.000000 / 20
[21:58:10][D][pipsolar:856]: got command: MCHGC020
[21:58:10][D][pipsolar:847]: Command queued successfully: MCHGC020 with length 8 at position 2
[21:58:10][D][uart_debug:158]: >>> "MCHGC020\xD1u\r"
[21:58:10][D][pipsolar:807]: Sending command from queue: MCHGC020 with length 8
[21:58:11][D][uart_debug:158]: <<< "(NAKss\r"
[21:58:11][D][pipsolar:042]: response length for command OK
[21:58:11][D][pipsolar:772]: checking crc on incoming message
[21:58:11][D][pipsolar:775]: CRC OK
[21:58:11][D][pipsolar:048]: command not successful

Gio-dot, 2022-12-15:
> [21:58:10][D][number:054]: 'pvbrain_axpert_current_max_charging_current_number' - Setting number value [21:58:10][D][number:113]: New number value: 20.000000 [21:58:10][D][number:012]: [...]

Hi, i have the same issue on my MPP Solar PIP 8048, but with command `current_max_ac_charging_current`. Did you manage to solve it? Can you share the whole esphome configuration?

Gio-dot, 2022-12-15:
Ok, i solved my issue.. look here: https://github.com/syssi/esphome-pipsolar/issues/41