syssi/esphome-pipsolar #35: Voltronic Axpert VM III TWIN 6 kW no data

eruc, 2022-10-24:
Hi, I am using esp8266 for communication with the Voltronic Axpert VMIII Twin 6 Kw. I can't get the complete data in ESPhome .. The wiring is done correctly, i tried various yaml codes including: main, 2424mse1, hms-3k-24v, pip8048, and they all have the same result ie no data. I tried with another code but it sends me back to then displayed data. I am wondering if there is a solution for this new model? Thank you.

these are the data obtained in Log

[22:20:13][D][uart_debug:158]: >>> "^P005PIq\xC2\x8B\r"
[22:20:15][D][uart_debug:158]: >>> "QPI\xC2\xBE\xC2\xAC\r"
[22:20:15][D][uart_debug:158]: <<< "(NAKss\r"
[22:20:17][D][uart_debug:158]: >>> "QDIq\e\r"
[22:20:17][D][uart_debug:158]: <<< "(230.0 50.0 0030 42.0 54.0 56.4 46.0 60 0 0 2 0 0 0 0 0 1 1 1 0 1 0 54.0 0 1\xFD\xD8\r"

[[image]<br /><a target='_blank' href='https://it.imgbb.com/'></a><br />](url)

[[image]<br /><a target='_blank' href='https://it.imgbb.com/'></a><br />](url)

lucasimons, 2023-01-02:
@eruc Do you have any updates, did you manage to get it working?

CastiRudolf, 2023-05-05:
@syssi I am experiencing the same issue on the same model.

Here is my log output when running the above test
[17:31:33][I][main:424]: Testing PI30/PI30MAX/PI30REVO/PI41 commands...
[17:31:33][I][main:427]: This is the set of commands supported by the pipsolar component!
[17:31:33][D][uart_debug:158]: >>> "QPI\xBE\xAC\r"
[17:31:33][D][uart_debug:158]: <<< "(PI30\x9A\v\r"
[17:31:34][D][uart_debug:158]: >>> "QDIq\e\r"
[17:31:35][D][uart_debug:158]: <<< "(230.0 50.0 0030 42.0 54.0 56.4 46.0 60 0 0 2 0 0 0 0 0 1 1 0 0 1 0 54.0 0 1 000\x9E`\r"
[17:31:35][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[17:31:35][D][uart_debug:158]: <<< "(EkxDabdjlnuvyz\x8D}\r"
[17:31:36][D][uart_debug:158]: >>> "QMN\xBBd\r"
[17:31:36][D][uart_debug:158]: <<< "(MKS2-6000\x10\xD7\r"
[17:31:37][D][uart_debug:158]: >>> "QMODI\xC1\r"
[17:31:37][D][uart_debug:158]: <<< "(L\x06\a\r"
[17:31:38][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[17:31:39][D][uart_debug:158]: <<< "(222.5 49.9 222.5 49.9 0289 0213 004 401 54.50 000   0038 00.0 000.0 00.00 00000 00010101 00 00 00000 010 0 00 0000O\xE1\r"
[17:31:39][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[17:31:40][D][uart_debug:158]: <<< "(230.0 26.0 230.0 50.0 26.0 6000 6000 48.0 48.0 48.0 56.5 56.5 9 030 010 0 0 2 9 00 0 0 54.0 0 1 000\x99\xB5\r"
[17:31:40][D][uart_debug:158]: >>> "QPIWS\xB4\xDA\r"
[17:31:40][D][uart_debug:158]: <<< "(100000000000000000000000000000000000\x15\xCF\r"
[17:31:41][D][uart_debug:158]: >>> "QT\'\xFF\r"
[17:31:41][D][uart_debug:158]: <<< "(20230102075052\xC4\x16\r"

CastiRudolf, 2023-05-08:
I just want to add that currently I am monitoring my inverter with a Raspberry Pi. 
I am using the following settings - 
qpiri=103
qpiws=40
qmod=5
qpigs=120

Hope this helps :-)

WallK, 2023-12-02:
This (and other TWIN models) have longer QPIGS
I've made a change that exists, for example, in pip8048 branch that supports it

Use
```
external_components:
  - source: github://syssi/esphome-pipsolar@pip8048
```

It works for my PIP6048MT (which is the same as VM3 TWIN)

baturinivan, 2024-03-30:
> It works for my PIP6048MT (which is the same as VM3 TWIN)

It works for me with VM III Twin (24V, 4.2kW). Failed to compile initially, but after change to idf versus arduino and tweaking with version, this solution is finally working.

Edmondo-diRonza, 2024-05-06:
> > It works for my PIP6048MT (which is the same as VM3 TWIN)
> 
> It works for me with VM III Twin (24V, 4.2kW). Failed to compile initially, but after change to idf versus arduino and tweaking with version, this solution is finally working.
Hi, I have an Axpert VMIV Twin (48V, 6KW) with what I believe is the same issue of a long QPIGS response. I tried to use the pip8084 branch, but I encountered an error during compilation. I'm using the recommended version of esp-idf. What kind of tweaks did you make to get it working? Which version? Thank you in advance for helping me.

baturinivan, 2024-05-06:
`
esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: recommended

external_components:
  - source: github://syssi/esphome-pipsolar@pip8048
    refresh: 0s

uart:
  - id: uart_inv
    tx_pin: GPIO1
    rx_pin: GPIO3
    baud_rate: 2400

pipsolar:
  - uart_id: uart_inv
    id: inverter0
    update_interval: 5s

binary_sensor:
  - platform: pipsolar
    pipsolar_id: inverter0
    add_sbu_priority_version:
      name: "inv add_sbu_priority_version"
    configuration_status:
      name: "inv configuration_status"
    scc_firmware_version:
      name: "inv scc_firmware_version"
    load_status:
      name: "inv load_status"
    battery_voltage_to_steady_while_charging:
      name: "inv battery_voltage_to_steady_while_charging"
    charging_status:
      name: "inv charging_status"
    scc_charging_status:
      name: "inv scc_charging_status"
    ac_charging_status:
      name: "inv ac_charging_status"
    charging_to_floating_mode:
      name: "inv charging_to_floating_mode"
    switch_on:
      name: "inv switch_on"
    dustproof_installed:
      name: "inv dustproof_installed"
    silence_buzzer_open_buzzer:
      name: "inv silence_buzzer_open_buzzer"
    overload_bypass_function:
      name: "inv overload_bypass_function"
    lcd_escape_to_default:
      name: "inv lcd_escape_to_default"
    overload_restart_function:
      name: "inv overload_restart_function"
    over_temperature_restart_function:
      name: "inv over_temperature_restart_function"
#    backlight_on:
#      name: "inv backlight_on"

sensor:
  - platform: pipsolar
    pipsolar_id: inverter0
    # QPIRI
#    grid_rating_voltage:
#      name: "inv grid_rating_voltage"
#    grid_rating_current:
#      name: "inv grid_rating_current"
#    ac_output_rating_voltage:
#      name: "inv ac_output_rating_voltage"
[... 201 lines cut ...]
        "10A": "MCHGC010"
        "20A": "MCHGC020"
        "30A": "MCHGC030"
        "40A": "MCHGC040"
        "50A": "MCHGC050"
        "60A": "MCHGC060"
        "70A": "MCHGC070"
        "80A": "MCHGC080"
        "90A": "MCHGC090"
        "100A": "MCHGC100"
        "110A": "MCHGC110"
        "120A": "MCHGC120"
      statusmap:
        "10": "10A"
        "20": "20A"
        "30": "30A"
        "40": "40A"
        "50": "50A"
        "60": "60A"
        "70": "70A"
        "80": "80A"
        "90": "90A"
        "100": "100A"
        "110": "110A"
        "120": "120A"`

Edmondo-diRonza, 2024-05-06:
> external_components:
>   - source: github://syssi/esphome-pipsolar@pip8048
>     refresh: 0s

Hi, sure! Sorry I'm really a newbie at these kinds of problems. I paste the compiler log, made with the home assistant esphome plugin.
[logs_esphome-web-57072c_run (1).txt](https://github.com/syssi/esphome-pipsolar/files/15219263/logs_esphome-web-57072c_run.1.txt)

I'm also posting the ESP32 test protocol log, in case it could be helpful. Thanks.
[logs_esphome-web-57072c_logs (3).txt](https://github.com/syssi/esphome-pipsolar/files/15219288/logs_esphome-web-57072c_logs.3.txt)