syssi/esphome-pipsolar #40: Connections for Voltronic MK IV

nuvolare12, 2022-11-24:
Hello Sissi, i wish to connect my Axpert MK IV with Home Assistant, i have a different connector, can you help me step by step to connect it? I attaced the my connection, i don't have the step down for the moment  thank you
[image]
[image]
[image]
[image]

nuvolare12, 2022-12-06:
I'm so getting timout 
[image]

nuvolare12, 2022-12-06:
[image]
And this are the connection

VirtualFab, 2022-12-08:
Ok, my fault, I didn't recognized in my logs that for example
[09:57:35][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[09:57:35][D][pipsolar:838]: Sending polling command : QPIGS with length 5
[09:57:35][D][uart_debug:158]: <<< "QPIGS\xB7\xA9\r"
I was receiving the same exact data I was sending, so it was clear Tx and Rx were somehow in a loop connection.
I simply reswapped Rx and Tx (that I swapped on the ESP32 side) and then swapped them on the MAX3232 pins on the RS-232 side-
This led to the correct connection:

> [11:11:39][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
> [11:11:39][D][pipsolar:838]: Sending polling command : QPIGS with length 5
> [11:11:40][D][uart_debug:158]: **<<< "(233.7 49.9 233.7 49.9 0771 0750 013 376 50.20 011 026 0027 01.9 334.0 00.00 00000 00010111 00 00 00663 010\xD3}\r**"
> [11:11:40][D][pipsolar:772]: checking crc on incoming message
> [11:11:40][D][pipsolar:775]: CRC OK

So now the whole thing is working, thank you!

nuvolare12, 2022-12-11:
Hello VirtualFab, i tried to invert RX with TX, and now it's working. For the Power supply i use the usb connector for the mobile phone, but i want to use a Step-down

syssi, 2023-04-14:
Summary: The Axpert MK IV / Voltronic MK IV is compatible with the `pipsolar` component.

closed 2023-04-14 by syssi