syssi/esphome-pipsolar #41: PIP 8048 wrong "current_max_ac_charging_current"

Gio-dot, 2022-12-10:
Hi, thank you for your work! Today i tested setting of ac charging current, but i noticed that inverter doesn't load the correct value. TO do so i created simply buttons to load 2A and 10A; in both cases a wrong value is loaded by the inverter.

[image]

[image]

[image]

Gio-dot, 2022-12-15:
Ok, solved with this mod.:
[image]

Gio-dot, 2022-12-24:
Hi, i need some help to solve completely "current_max_ac_charging_current" issue. As i said, original component command don't work in 8048 (this is because it has only 2 digit but 8048 uses 3 digit for ac charing current, due to the  max 120A). 
I partially solved with the mod in the previous post but unfortunately even if now the current set is working, often esp32 crash. I noticed that there is some issue in decoding MUCHGC command with my modification as you can see in attached screenshot. I'm not expert in c++ at all, so i can't solve this issue for myself. Can someone help to get MUCHGC command properly working? Thank you

[image]

altabrida, 2023-01-05:
Sorry for my English, I am a French Canadian.

CONF_CURRENT_MAX_CHARGING_CURRENT: ([10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120], "MCHGC0%03.0f"), --> ... "MCHGC%03.0f"

I had a similar problem and I solved it.
I removed the Zero in front of the sign of % because I noticed that it was normally served unit 0 in a parallel system.
But in certain version of the protocol it is not required.
[image]

But finally I had to put it back because my inverter model needs it. My problem came more early on the %02.0f -> %03.0f
So you help me a lot.

I ave 2 x MppSolar LV6548 for 120/240vac and i use the PIP 8048

altabrida, 2023-01-05:
sorry i a copy the worng line

CONF_CURRENT_MAX_AC_CHARGING_CURRENT: ([2, 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120], "MUCHGC**0**%03.0f"),
CONF_CURRENT_MAX_AC_CHARGING_CURRENT: ([2, 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120], "MUCHGC%03.0f"),

SeByDocKy, 2023-01-08:
> sorry i a copy the worng line
> 
> CONF_CURRENT_MAX_AC_CHARGING_CURRENT: ([2, 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120], "MUCHGC**0**%03.0f"), CONF_CURRENT_MAX_AC_CHARGING_CURRENT: ([2, 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120], "MUCHGC%03.0f"),

I updated the code with your modification but for my PIP8048 (120A), i still have bad values...

syssi, 2023-01-11:
Good job!

SeByDocKy, 2023-01-11:
> Good job!

I will introduce a new select with this output command since the interval [2, 10, 20, ...., 120] is a bit weird for a number...

SeByDocKy, 2023-01-11:
- Just added two select for max_ac_charging_current and max_charging_current for the 8048 branch
```
- platform: pipsolar
    pipsolar_id: pip8048
    current_max_ac_charging_current:
      id: ${name}_${inverter}_current_max_ac_charging_current_select 
      name: ${name}_${inverter}_current_max_ac_charging_current_select
      optionsmap:
        "2A": "MUCHGC0002"
        "10A": "MUCHGC0010"
        "20A": "MUCHGC0020"
        "30A": "MUCHGC0030"
        "40A": "MUCHGC0040"
        "50A": "MUCHGC0050"
        "60A": "MUCHGC0060"
        "70A": "MUCHGC0070"
        "80A": "MUCHGC0080"
        "90A": "MUCHGC0090"
        "100A": "MUCHGC0100"
        "110A": "MUCHGC0110"
        "120A": "MUCHGC0120"
      statusmap:
        "2": "2A"
        "10": "10A"
        "20": "20A"
        "30": "30A"       
        "40": "40A"
        "50": "50A"
        "60": "60A"
        "70": "70A"
        "80": "80A"
        "90": "90A"
        "100": "100A"
        "110": "110A"
        "120": "120A"

  - platform: pipsolar
    pipsolar_id: pip8048
    current_max_charging_current:
      id: ${name}_${inverter}_current_max_charging_current_select 
      name: ${name}_${inverter}_current_max_charging_current_select
      optionsmap:
        "10A": "MCHGC010"
        "20A": "MCHGC020"
        "30A": "MCHGC030"
        "40A": "MCHGC040"
        "50A": "MCHGC050"
        "60A": "MCHGC060"
        "70A": "MCHGC070"
        "80A": "MCHGC080"
        "90A": "MCHGC090"
        "100A": "MCHGC100"
        "110A": "MCHGC110"
        "120A": "MCHGC120"
      statusmap:
        "10": "10A"
        "20": "20A"
        "30": "30A"       
        "40": "40A"
        "50": "50A"
        "60": "60A"
        "70": "70A"
        "80": "80A"
        "90": "90A"
        "100": "100A"
        "110": "110A"
        "120": "120A"```

syssi, 2023-01-11:
@SeByDocKy Could you extend the two example configuration of the feature branch accordingly? Thanks in advance!

```
esp32-example.yaml
esp8266-example.yaml
```

SeByDocKy, 2023-01-11:
> @SeByDocKy Could you extend the two example configuration of the feature branch accordingly? Thanks in advance!
> 
> ```
> esp32-example.yaml
> esp8266-example.yaml
> ```

You mean you pushed also modifications for non 8048 branches ?

syssi, 2023-01-11:
No. :-) Your feature branch (`pip8084`) contains two example configurations:

https://github.com/syssi/esphome-pipsolar/blob/pip8048/esp32-example.yaml
https://github.com/syssi/esphome-pipsolar/blob/pip8048/esp8266-example.yaml

Could you extend these configurations to promote/document the new features here?

closed 2023-04-14 by syssi

WildeRNS, 2025-08-29:
> I also extend the buffer command to 12 chars to accept this new format

can you share, what exactly to change? because i have Daxtromn 6.2 and everything working except **current_max_charging_current command!** i see data from this command but can't change value - **everytime command not successful** from log. i try to change **CONF_CURRENT_MAX_CHARGING_CURRENT: ([ 10, 20, 30, 40, 50, 60, 70, 80], "MCHGC0%03.0f")**, to different values and no luck...