syssi/esphome-pipsolar #42: pip8048max CRC NOK

sergeyshadow, 2022-12-17:
I have https://www.mppsolar.com/v3/pip-max/
I'm trying to wire inverter-rs232tottl-esp32 and I'm getting the following messages in the log:

```
[23:52:20][D][uart_debug:158]: <<< "(230.0 47.8 230.0 50.0 47.8 11000 11000 48.0 46.0 44.0 56.8 55.7 2 010 150 0 0 2 9 01 0 0 54.0 0 1 480 0 000qO\r"
[23:52:20][D][pipsolar:839]: checking crc on incoming message
[23:52:20][D][pipsolar:849]: CRC NOK expected: 7F D4 but got: 0 0
[23:52:21][D][uart_debug:158]: >>> "QBATCD\xAA&\r"
[23:52:21][D][pipsolar:905]: Sending polling command : QBATCD with length 6
[23:52:21][D][uart_debug:158]: <<< "(NAKss\r"
[23:52:21][D][pipsolar:839]: checking crc on incoming message
[23:52:21][D][pipsolar:842]: CRC OK
[23:52:22][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[23:52:22][D][pipsolar:905]: Sending polling command : QPIGS with length 5
[23:52:22][D][uart_debug:158]: <<< "(201.6 50.0 201.6 50.0 03749 03724 034 435 55.60 001 100 0046 00.0 000.0 00.00 00000 00010101 00 00 00000 110<\xAD\r"
[23:52:27][D][pipsolar:822]: timeout command to poll: QPIGS
[23:52:27][D][uart_debug:158]: >>> "QPIGS2h-\r"
```
Is my inverter not supported or have I connected something wrong?

syssi, 2022-12-17:
Could you double check the setting `dummy_receiver` is turned off (`false`)?

```
uart:
  id: uart0
  baud_rate: 2400
  tx_pin: GPIO1
  rx_pin: GPIO3
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\r"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
```

sergeyshadow, 2022-12-17:
Yes, I checked it three times :)

```
    tx_pin: GPIO3
    rx_pin: GPIO1
    # most devices use 2400 as baud_rate
    baud_rate: 2400
    debug:
      direction: BOTH
      dummy_receiver: false
      after:
        delimiter: "\r"
      sequence:
        - lambda: UARTDebug::log_string(direction, bytes);
```

syssi, 2022-12-17:
Yes. It looks like the response payload of your device isn't supported yet. Could you provide a longer log capture? I'm a bit confused because of these responses:

```
(230.0 47.8 230.0 50.0 47.8 11000 11000 48.0 46.0 44.0 56.8 55.7 2 010 150 0 0 2 9 01 0 0 54.0 0 1 480 0 000qO\r
(201.6 50.0 201.6 50.0 03749 03724 034 435 55.60 001 100 0046 00.0 000.0 00.00 00000 00010101 00 00 00000 110<\xAD\r
```

sergeyshadow, 2022-12-17:
> 

```
[00:09:53][D][pipsolar:822]: timeout command to poll: QPIGS
[00:09:53][D][uart_debug:158]: >>> "QPIGS2h-\r"
[00:09:53][D][pipsolar:905]: Sending polling command : QPIGS2 with length 6
[00:09:53][D][uart_debug:158]: <<< "(00.0 000.0 00000 E\x93\r"
[00:09:53][D][pipsolar:839]: checking crc on incoming message
[00:09:53][D][pipsolar:842]: CRC OK
[00:09:53][D][pipsolar:503]: Decode QPIGS2
[00:09:53][V][sensor:076]: 'pipsolar pv2_input_current': Received new state 0.000000
[00:09:53][D][sensor:127]: 'pipsolar pv2_input_current': Sending state 0.00000 A with 1 decimals of accuracy
[00:09:53][V][json:031]: Attempting to allocate 512 bytes for JSON serialization
[00:09:53][V][json:051]: Size after shrink 88 bytes
[00:09:53][V][sensor:076]: 'pipsolar pv2_input_voltage': Received new state 0.000000
[00:09:53][D][sensor:127]: 'pipsolar pv2_input_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[00:09:53][V][json:031]: Attempting to allocate 512 bytes for JSON serialization
[00:09:53][V][json:051]: Size after shrink 88 bytes
[00:09:53][V][sensor:076]: 'pipsolar pv2_charging_power': Received new state 0.000000
[00:09:53][D][sensor:127]: 'pipsolar pv2_charging_power': Sending state 0.00000 W with 1 decimals of accuracy
[00:09:53][V][json:031]: Attempting to allocate 512 bytes for JSON serialization
[00:09:53][V][json:051]: Size after shrink 92 bytes
[00:09:53][V][component:200]: Component esphome.coroutine took a long time for an operation (0.06 s).
[00:09:53][V][component:201]: Components should block for at most 20-30ms.
[00:09:54][D][uart_debug:158]: >>> "QMODI\xC1\r"
[00:09:54][D][pipsolar:905]: Sending polling command : QMOD with length 4
[00:09:54][D][uart_debug:158]: <<< "(L\x06\a\r"
[00:09:54][D][pipsolar:839]: checking crc on incoming message
[00:09:54][D][pipsolar:842]: CRC OK
[00:09:54][D][pipsolar:514]: Decode QMOD
[00:09:54][V][text_sensor:016]: 'pipsolar device_mode': Received new state L
[00:09:54][D][text_sensor:067]: 'pipsolar device_mode': Sending state 'L'
[00:09:54][V][json:031]: Attempting to allocate 512 bytes for JSON serialization
[00:09:54][V][json:051]: Size after shrink 84 bytes
[00:09:55][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[00:09:55][D][pipsolar:905]: Sending polling command : QFLAG with length 5
[00:09:55][D][uart_debug:158]: <<< "(EbuvxzDajky\xC6\x00\r"
[00:09:55][D][pipsolar:839]: checking crc on incoming message
[00:09:55][D][pipsolar:842]: CRC OK
[00:09:55][D][pipsolar:522]: Decode QFLAG
[00:09:56][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[00:09:56][D][pipsolar:905]: Sending polling command : QPIRI with length 5
[00:09:57][D][uart_debug:158]: <<< "(230.0 47.8 230.0 50.0 47.8 11000 11000 48.0 46.0 44.0 56.8 55.7 2 010 150 0 0 2 9 01 0 0 54.0 0 1 480 0 000qO\r"
[00:09:57][D][pipsolar:839]: checking crc on incoming message
[00:09:57][D][pipsolar:849]: CRC NOK expected: BF AA but got: 0 0
[00:09:57][D][uart_debug:158]: >>> "QBATCD\xAA&\r"
[00:09:57][D][pipsolar:905]: Sending polling command : QBATCD with length 6
[00:09:57][D][uart_debug:158]: <<< "(NAKss\r"
[00:09:57][D][pipsolar:839]: checking crc on incoming message
[00:09:57][D][pipsolar:842]: CRC OK
[00:09:58][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[00:09:58][D][pipsolar:905]: Sending polling command : QPIGS with length 5
[00:09:59][D][uart_debug:158]: <<< "(201.4 49.9 201.4 49.9 03927 03919 035 436 55.60 001 100 0046 00.0 000.0 00.00 00000 00010101 00 00 00000 110nR\r"
```

syssi, 2022-12-17:
Just for illustration: The decoder expects a very specific pattern of numbers:

QPIRI: https://github.com/syssi/esphome-pipsolar/pull/17/files#diff-fa19b978d1ab115551a0f9422c5d2a76c1b48b901e4955aaad3da0ddecbc4e0cR430-R441
QPIGS: https://github.com/syssi/esphome-pipsolar/pull/17/files#diff-fa19b978d1ab115551a0f9422c5d2a76c1b48b901e4955aaad3da0ddecbc4e0cR449-R517

```
226.7 49.9 226.7 49.9 0498  0479  016 427 27.00 005 100 0035 01.9 255.1 00.00 00000 10010110 00 00 00510 110 # QPIGS response examples of the PIP 2424MSE1
232.6 50.0 229.9 49.9 0391  0312  007 402 54.40 042 072 0066 0042 284.6 00.00 00000 00010010 00 00 02901 010 # QPIGS response example of a PIP-5048Mg FW71.85
201.4 49.9 201.4 49.9 03927 03919 035 436 55.60 001 100 0046 00.0 000.0 00.00 00000 00010101 00 00 00000 110 # QPIGS response of your device
                                                             ^^^^ the parser fails here probably
```

Please switch to this branch and give it another try:

```
external_components:
  - source: github://syssi/esphome-pipsolar@2424mse1
    refresh: 0s
```

sergeyshadow, 2022-12-17:
> Please switch to this branch and give it another try:
> 
> ```
> external_components:
>   - source: github://syssi/esphome-pipsolar@2424mse1
>     refresh: 0s
> ```

Previously I used config for pip8048, what config can I use for this branch?

syssi, 2022-12-17:
There is a config at the branch with some less entities/features.

sergeyshadow, 2022-12-17:
> external_components:
>   - source: github://syssi/esphome-pipsolar@2424mse1
>     refresh: 0s

The result is the same (

```
[00:31:03][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[00:31:03][D][pipsolar:891]: Sending polling command : QPIRI with length 5
[00:31:03][D][uart_debug:158]: <<< "(230.0 47.8 230.0 50.0 47.8 11000 11000 48.0 46.0 44.0 56.8 55.7 2 010 150 0 0 2 9 01 0 0 54.0 0 1 480 0 000qO\r"
[00:31:03][D][pipsolar:825]: checking crc on incoming message
[00:31:03][D][pipsolar:835]: CRC NOK expected: 93 F but got: 0 0
[00:31:04][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[00:31:04][D][pipsolar:891]: Sending polling command : QPIGS with length 5
[00:31:04][D][uart_debug:158]: <<< "(204.3 49.9 204.3 49.9 03697 03668 033 435 55.60 001 100 0046 00.0 000.0 00.00 00000 00010101 00 00 00000 110_\xC2\r"
```

syssi, 2022-12-17:
Hmm.. I will have another look tomorrow.

syssi, 2022-12-18:
Could you ask your seller about some protocol details or software to control the inverter via rs232?

sergeyshadow, 2022-12-18:
> Could you ask your seller about some protocol details or software to control the inverter via rs232?

I asked, but it is unlikely that the seller will respond.
Perhaps this document contains the necessary information
https://forum.arduino.cc/uploads/short-url/9Fe8Tfap7KgZOveeQrtgJhSRmAR.pdf

Seb2311, 2023-01-03:
Hello,
i have the same trouble with Apert max 2 10 kw.
I read your documentation we can see the response of some command is not the same of others inverter.
but does as a impact on the CRC?

sergeyshadow, 2023-02-23:
Hello @syssi have any news for me? :)

sergeyshadow, 2023-02-24:
What am I doing wrong? :(

```
In file included from src/esphome/components/pipsolar/output/pipsolar_output.h:3,
                 from src/esphome/components/pipsolar/output/pipsolar_output.cpp:1:
src/esphome/components/pipsolar/output/../pipsolar.h:7:10: fatal error: esphome/components/pipsolar/select/pipsolar_select.h: No such file or directory
 #include "esphome/components/pipsolar/select/pipsolar_select.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/solar-inverter/.pioenvs/solar-inverter/src/esphome/components/pipsolar/output/pipsolar_output.cpp.o] Error 1
In file included from src/esphome/components/pipsolar/pipsolar.cpp:1:
src/esphome/components/pipsolar/pipsolar.h:7:10: fatal error: esphome/components/pipsolar/select/pipsolar_select.h: No such file or directory
 #include "esphome/components/pipsolar/select/pipsolar_select.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/solar-inverter/.pioenvs/solar-inverter/src/esphome/components/pipsolar/pipsolar.cpp.o] Error 1

syssi, 2023-02-24:
Please use the example YAML from the branch. You have to add one `select` entity at minimum to your YAML.

sergeyshadow, 2023-02-24:
> Yes! Please flash the `2424mse1` branch another time. The chance is high it's working out of the box now.

[image]

closed 2023-02-24 by syssi

sergeyshadow, 2023-02-25:
Thank you for your work!
May I ask a couple more questions?
Is it possible to get data on pv2 (Voltage, current, etc.?)
And is it possible to get the data of solar energy production for the day that is stored in the inverter?

syssi, 2023-02-26:
Feel free to ask more questions but could you create a new issue first? Thanks in advance!