syssi/esphome-pipsolar #43: ESP sending altered commands

PhilKloppers, 2022-12-23:
I'm having trouble connecting my Infini MkIV. It works when connected to the SolarPower app, and I can sniff the packets on the serial port on my PC (and can provide more info if needed).

However, when running the ESP with the protocol test config, the log shows different data than what is in the YAML file. For example:
YAML:
`      # PI18
      - uart.write: "^P005PIq\x8b\r"
      - delay: 2s

      # PI30/PI30MAX/PI30REVO
      - uart.write: "QPI\xbe\xac\r"
      - delay: 2s

      # MKS Test protocol
      - uart.write: "^P007PIRI\xee\x38\r"
      - delay: 2s

      # PI41
      - uart.write: "QDIq\x1b\r"
      - delay: 2s
      - uart.write: "QPIRI\xF8T\r"
      - delay: 2s
      - uart.write: "QPIGS\xB7\xA9\r"
      - delay: 2s
      - uart.write: "QMODI\xC1\r"
      - delay: 2s
      - uart.write: "QFLAG\x98t\r"`

shows as
`[10:25:56][D][uart_debug:158]: >>> "^P005PIq\xC2\x8B\r"
[10:25:58][D][uart_debug:158]: >>> "QPI\xC2\xBE\xC2\xAC\r"
[10:26:00][D][uart_debug:158]: >>> "^P007PIRI\xC3\xAE8\r"
[10:26:02][D][uart_debug:158]: >>> "QDIq\e\r"
[10:26:04][D][uart_debug:158]: >>> "QPIRI\xC3\xB8T\r"
[10:26:06][D][uart_debug:158]: >>> "QPIGS\xC2\xB7\xC2\xA9\r"
[10:26:08][D][uart_debug:158]: >>> "QMODI\xC3\x81\r"
[10:26:10][D][uart_debug:158]: >>> "QFLAG\xC2\x98t\r"
`

in the log. It seems the encoded characters are being added and/or changed somewhere, and I think this is preventing communication (perhaps because the CRC is now incorrect?)

Am I doing something wrong, or is there something I can change, or is this normal?

syssi, 2022-12-23:
If the CRC is wrong your inverter should respond with `<<< NAK`. I don't see any responses so I assume your inverter isn't connected properly to your ESP. Please double check the RX/TX lines between the ESP and the RS232 converter as well as the RX/TX lines between the RS232 converter and the inverter. How did you power the RS232 converter? In most cases these converter boards must be powered twice (low voltage level + high voltage level).

PhilKloppers, 2022-12-23:
I've tried alternately swapping RX/TX on both ends without success. The RS232 is a MAX232-based board. I'm providing power on the 3.3v side from the ESP, and it's getting 12v from the inverter. The datasheet says it doesn't actually need voltage on the RS232 side as it has a charge pump onboard to boost the level to 12v from the 3.3v supply, but it's connected anyway. The only response I've ever seen in the log was an echo of the sent data, and that only when I disconnected the ground line to the MAX232 board. I assumed that this means that the RX/TX on the ESP side are incorrect in that configuration. I will try to connect the RS232+ESP to a USB-Serial cable I have and try sending data to see if it gets through - at least I can then confirm orientation! Failing that I'll have to dig the oscilloscope out...

syssi, 2022-12-23:
Your first paragraphs sounds good. The log is a bit weird. What did you do exactly? In general the should be one response to one request. Could you capture some traffic between the SolarPower app and your inverter without noise of the third party (the ESPHome implementation)? This should help to identify the protocol of your inverter and to make sure the ESPHome component is compatible or not.

PhilKloppers, 2022-12-23:
Sure! (The ESP-Home log is just showing both sides trying to discover, so won't actually be a coherent exchange).

Here's some of the sniffed serial data between the app and the inverter:
```
+0.0000061	IRP_MJ_WRITE	DOWN	0x00000000	5e 50 30 30 36 56 46 57 f6 e6 0d 	^P006VFW...
+0.0040702	IRP_MJ_READ	UP	0x00000000	5e 44 30 32 30 30 35 34 30 32 2c 30 34 30 32 35 2c 30 30 30 …	^D02005402,04025,000…
+0.0000056	IRP_MJ_WRITE	DOWN	0x00000000	5e 50 30 30 37 50 49 52 49 ee 38 0d 	^P007PIRI.8.
+0.0039939	IRP_MJ_READ	UP	0x00000000	5e 44 30 38 39 32 34 30 30 2c 32 33 33 2c 32 34 30 30 2c 35 …	^D0892400,233,2400,5…
+0.0000093	IRP_MJ_WRITE	DOWN	0x00000000	5e 50 30 30 36 46 57 53 c5 43 0d 	^P006FWS.C.
+0.0041065	IRP_MJ_READ	UP	0x00000000	5e 44 30 33 39 30 30 2c 30 2c 30 2c 30 2c 30 2c 30 2c 30 2c …	^D03900,0,0,0,0,0,0,…
+0.0000063	IRP_MJ_WRITE	DOWN	0x00000000	5e 50 30 30 37 50 52 49 30 6d 9d 0d 	^P007PRI0m..
+0.0041183	IRP_MJ_READ	UP	0x00000000	5e 44 30 34 30 31 2c 31 34 2c 39 36 31 33 32 31 31 30 31 30 …	^D0401,14,9613211010…
+0.0000063	IRP_MJ_WRITE	DOWN	0x00000000	5e 50 30 30 37 50 47 53 30 29 b6 0d 	^P007PGS0)..
+0.0041749	IRP_MJ_READ	UP	0x00000000	5e 44 31 31 33 31 2c 35 2c 30 30 2c 32 33 35 37 2c 35 30 31 …	^D1131,5,00,2357,501…
+0.0000065	IRP_MJ_WRITE	DOWN	0x00000000	5e 50 30 30 37 50 52 49 30 6d 9d 0d 	^P007PRI0m..
+0.0040325	IRP_MJ_READ	UP	0x00000000	5e 44 30 34 30 31 2c 31 34 2c 39 36 31 33 32 31 31 30 31 30 …	^D0401,14,9613211010…
+0.0000092	IRP_MJ_WRITE	DOWN	0x00000000	5e 50 30 30 37 50 52 49 31 7d bc 0d 	^P007PRI1}..
+0.0040749	IRP_MJ_READ	UP	0x00000000	5e 44 30 34 30 31 2c 31 34 2c 39 36 31 33 32 31 31 30 31 30 …	^D0401,14,9613211010…
+0.0000061	IRP_MJ_WRITE	DOWN	0x00000000	5e 50 30 30 37 50 47 53 31 39 97 0d 	^P007PGS19..
+0.0040898	IRP_MJ_READ	UP	0x00000000	5e 44 31 31 33 31 2c 35 2c 30 30 2c 32 33 35 38 2c 35 30 31 …	^D1131,5,00,2358,501…
+0.0000074	IRP_MJ_WRITE	DOWN	0x00000000	5e 50 30 30 36 46 57 53 c5 43 0d 	^P006FWS.C.
+0.0040879	IRP_MJ_READ	UP	0x00000000	5e 44 30 33 39 30 30 2c 30 2c 30 2c 30 2c 30 2c 30 2c 30 2c …	^D03900,0,0,0,0,0,0,…
+0.0000057	IRP_MJ_WRITE	DOWN	0x00000000	5e 50 30 30 37 50 52 49 31 7d bc 0d 	^P007PRI1}..
+0.0040908	IRP_MJ_READ	UP	0x00000000	5e 44 30 34 30 31 2c 31 34 2c 39 36 31 33 32 31 31 30 31 30 …	^D0401,14,9613211010…
+0.0000059	IRP_MJ_WRITE	DOWN	0x00000000	5e 50 30 30 37 50 52 49 32 4d df 0d 	^P007PRI2M..
+0.0040996	IRP_MJ_READ	UP	0x00000000	5e 44 30 34 30 30 2c 30 30 2c 30 30 30 30 30 30 30 30 30 30 …	^D0400,00,0000000000…
+0.0000059	IRP_MJ_WRITE	DOWN	0x00000000	5e 50 30 30 37 50 52 49 33 5d fe 0d 	^P007PRI3]..
+0.0041905	IRP_MJ_READ	UP	0x00000000	5e 44 30 34 30 30 2c 30 30 2c 30 30 30 30 30 30 30 30 30 30 …	^D0400,00,0000000000…
+0.0000064	IRP_MJ_WRITE	DOWN	0x00000000	5e 50 30 30 37 50 52 49 34 2d 19 0d 	^P007PRI4-..
+0.0041518	IRP_MJ_READ	UP	0x00000000	5e 44 30 34 30 30 2c 30 30 2c 30 30 30 30 30 30 30 30 30 30 …	^D0400,00,0000000000…
+0.0000084	IRP_MJ_WRITE	DOWN	0x00000000	5e 50 30 30 37 50 52 49 35 3d 38 0d 	^P007PRI5=8.
+0.0041289	IRP_MJ_READ	UP	0x00000000	5e 44 30 34 30 30 2c 30 30 2c 30 30 30 30 30 30 30 30 30 30 …	^D0400,00,0000000000…
+0.0000066	IRP_MJ_WRITE	DOWN	0x00000000	5e 50 30 30 37 50 52 49 36 0e 5b 0d 	^P007PRI6.[.
+0.0041180	IRP_MJ_READ	UP	0x00000000	5e 44 30 34 30 30 2c 30 30 2c 30 30 30 30 30 30 30 30 30 30 …	^D0400,00,0000000000…
+0.0000064	IRP_MJ_WRITE	DOWN	0x00000000	5e 50 30 30 37 50 52 49 37 1d 7a 0d 	^P007PRI7.z.
+0.0040962	IRP_MJ_READ	UP	0x00000000	5e 44 30 34 30 30 2c 30 30 2c 30 30 30 30 30 30 30 30 30 30 …	^D0400,00,0000000000…
+0.0000066	IRP_MJ_WRITE	DOWN	0x00000000	5e 50 30 30 37 50 52 49 38 ec 95 0d 	^P007PRI8...
+0.0040332	IRP_MJ_READ	UP	0x00000000	5e 44 30 34 30 30 2c 30 30 2c 30 30 30 30 30 30 30 30 30 30 …	^D0400,00,0000000000…
```

syssi, 2022-12-23:
Unfortunately this protocol isn't supported by the `pipsolar` component.

PhilKloppers, 2022-12-23:
Ah! That would explain all the difficulties!😂 I did see the protocol listed in the documents in this issue https://github.com/syssi/esphome-pipsolar/issues/39 (specifically this one I think: https://github.com/ardupic/voltronic-inverter-communication-protocols/blob/main/Infini-Solar%20V%26VII%26VIII%26VIV%26%20V%20II%206K%20LV%20Split-Phase%26TWIN%20protocol%20202205....xlsx)

Are there any plans to include the protocol, and if not, could you suggest any package that could work? I also don't mind doing the legwork to get it included if that's feasible?

SeByDocKy, 2022-12-23:
As far I know for MPI models, there is no CRC  ....

syssi, 2023-04-14:
@PhilKloppers I've fixed and improved the protocol test YAML. Could you give it another try?

https://github.com/syssi/esphome-pipsolar/blob/main/tests/esp8266-test-protocols.yaml

closed 2023-04-14 by syssi