syssi/esphome-pipsolar #47: PIP 2424MSE1 - compilation error

Rebel3D-CZ, 2023-01-10:
Hi, I have been using working configuration for a long time. Now I needed to edit some detail in the entity description and the module cannot be compiled.

I have the current version of ESPHome 2022.12.3

I am using an external source:

external_components:
     - source: github://syssi/esphome-pipsolar@2424mse1
       refresh: 0 sec

If I don't use an external source, it compiles, but some entities don't work.

When compiling with an external source, the compilation ends with an error:

INFO Reading configuration /config/esphome/fve1-module.yaml...
INFO Updating https://github.com/syssi/esphome-pipsolar.git@2424mse1
INFO Detected timezone 'Europe/Prague'
INFO Generating C++ source...
INFO Compiling app...
Processing fve1-module (board: esp07s; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling /data/fve1-module/.pioenvs/fve1-module/src/esphome/components/pipsolar/pipsolar.cpp.o
Compiling /data/fve1-module/.pioenvs/fve1-module/src/esphome/components/pipsolar/switch/pipsolar_switch.cpp.o
In file included from src/esphome/components/pipsolar/pipsolar.cpp:1:
src/esphome/components/pipsolar/pipsolar.h:7:10: fatal error: esphome/components/pipsolar/select/pipsolar_select.h: No such file or directory
    7 | #include "esphome/components/pipsolar/select/pipsolar_select.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from src/esphome/components/pipsolar/switch/pipsolar_switch.h:3,
                 from src/esphome/components/pipsolar/switch/pipsolar_switch.cpp:1:
src/esphome/components/pipsolar/switch/../pipsolar.h:7:10: fatal error: esphome/components/pipsolar/select/pipsolar_select.h: No such file or directory
    7 | #include "esphome/components/pipsolar/select/pipsolar_select.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling /data/fve1-module/.pioenvs/fve1-module/src/esphome/components/time/real_time_clock.cpp.o
Compiling /data/fve1-module/.pioenvs/fve1-module/src/esphome/components/total_daily_energy/total_daily_energy.cpp.o
Compiling /data/fve1-module/.pioenvs/fve1-module/src/esphome/components/uart/uart.cpp.o
*** [/data/fve1-module/.pioenvs/fve1-module/src/esphome/components/pipsolar/pipsolar.cpp.o] Error 1
Compiling /data/fve1-module/.pioenvs/fve1-module/src/esphome/components/uart/uart_component.cpp.o
*** [/data/fve1-module/.pioenvs/fve1-module/src/esphome/components/pipsolar/switch/pipsolar_switch.cpp.o] Error 1
========================== [FAILED] Took 5.27 seconds ==========================

Evidently it's after some esphome or homeassistant upgrade because it was working before. I tried restoring to esphome 12.0.0 from the oldest backup, but it didn't help.

I also tried to copy the unavailable files to src, but when checking git and compilation, they are always deleted.

Can anyone help with this?
Thank you, Dan.

zbyndatvrd, 2023-01-10:
Same for me.

jowser7, 2023-01-11:
Same here. I've now updated esphome, esp-idf, platformio, arduino.  
In the pipsolar_output.h file the first include is written as #include "../pipsolar.h"   I'm wondering if this ../ works okay on windows compiles or needs more of the directory written into it like the other includes.
#include "esphome/components/pipsolar/select/pipsolar_select.h" in pipsolar_output.h

Then in debug output I see they use the src/ folders. Thats where I don't understand the compiling directories vs what was the base. But when I look in .esphome\build\pipsolar\src\esphome\components\pipsolar\select
this folder is empty, not having the same files as what I find in 
.esphome\external_components\6d791439\components\pipsolar\select  
[image]

syssi, 2023-01-11:
Adding one of the select entities to the YAML configuration should fix the build issue:

```
select:
  - platform: pipsolar
    pipsolar_id: inverter0
    output_source_priority:
      id: inverter0_output_source_priority_select
      name: inverter0_output_source_priority_select
      optionsmap:
        "Utility first": "POP00"
        "Solar only": "POP01"
        "Solar Battery Utility": "POP02"
      statusmap:
        "0": "Utility first"
        "1": "Solar only"
        "2": "Solar Battery Utility"
```

If it doesn't work out of the  box please try a clean build:

```
esphome clean config.yaml
esphome run config.yaml
```

Rebel3D-CZ, 2023-01-11:
Great, thanks for the help, it helped!
If there are no more questions by the end of the evening, I will close the issue.
Thank you very much.

closed 2023-01-11 by Rebel3D-CZ