syssi/esphome-pipsolar #48: Axpert VM IV 24v 3600w

Pokornypetr, 2023-01-16:
hello, the data from the converter is working, most of the values ​​look ok. Some problems are what I encountered with the values ​​I really need. I do not see the current power of the PV panel and the current voltage of the PV PANEL
    pv_input_voltage: has a value of 0v - 0,9v  at an almost real 180v
    pv_charging_power:   it is still at 0w

 pv_charging_power:
      name: "FV nabíjecí výkon - ${friendly_name}"
      id: pv_charging_power
pv_input_voltage:
      name: "FV napětí - ${friendly_name}"

syssi, 2023-01-17:
Could you enable the `debug` output of the `uart` and provide some logs including the raw traffic of your inverter (at a sunny day)?

Pokornypetr, 2023-01-17:
26.5 0 0\x127\r"
[10:48:15][D][pipsolar:772]: checking crc on incoming message
[10:48:15][D][pipsolar:775]: CRC OK
[10:48:15][D][pipsolar:429]: Decode QPIRI
[10:48:15][D][sensor:126]: 'Jmenovité napětí baterie - FVE Axpert': Sending state 24.00000 V with 1 decimals of accuracy
[10:48:24][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[10:48:24][D][pipsolar:836]: Sending polling command : QPIGS with length 5
[10:48:25][D][uart_debug:158]: <<< "(247.3 50.0 239.0 50.0 0931 0805 025 360 26.10 007 060 0017 04.6 179.2 00.00 00001 00010110 00 00 00831 011\xC4h\r"
[10:48:25][D][pipsolar:772]: checking crc on incoming message
[10:48:25][D][pipsolar:775]: CRC OK
[10:48:25][D][pipsolar:448]: Decode QPIGS
[10:48:25][D][sensor:126]: 'Činný výkon výstupu - FVE Axpert': Sending state 805.00000 W with 1 decimals of accuracy
[10:48:25][D][sensor:126]: 'Zatížení výstupu - FVE Axpert': Sending state 25.00000 % with 1 decimals of accuracy
[10:48:25][D][sensor:126]: 'Napětí baterie - FVE Axpert': Sending state 26.10000 V with 1 decimals of accuracy
[10:48:25][D][sensor:126]: 'Nabíjecí proud baterie - FVE Axpert': Sending state 7.00000 A with 1 decimals of accuracy
[10:48:25][D][sensor:126]: 'Kapacita baterie - FVE Axpert': Sending state 60.00000 % with 1 decimals of accuracy
[10:48:25][D][sensor:126]: 'FV vstupní proud do baterie - FVE Axpert': Sending state 4.00000 A with 1 decimals of accuracy
[10:48:25][D][sensor:126]: 'FV napětí - FVE Axpert': Sending state 0.60000 V with 1 decimals of accuracy
[10:48:25][D][sensor:126]: 'Vybíjecí proud baterie - FVE Axpert': Sending state 0.00000 A with 1 decimals of accuracy
[10:48:25][D][sensor:126]: 'FV nabíjecí výkon - FVE Axpert': Sending state 0.00000 W with 1 decimals of accuracy
[10:48:34][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[10:48:35][D][pipsolar:836]: Sending polling command : QPIRI with length 5
[10:48:35][D][uart_debug:158]: <<< "(240.0 15.0 240.0 50.0 15.0 3600 3600 24.0 24.0 23.5 29.2 29.0 2 010 100 0 2 3 1 01 0 0 26.5 0 0\x127\r"
[10:48:35][D][pipsolar:772]: checking crc on incoming message
[10:48:35][D][pipsolar:775]: CRC OK
[10:48:35][D][pipsolar:429]: Decode QPIRI
[10:48:35][D][sensor:126]: 'Jmenovité napětí baterie - FVE Axpert': Sending state 24.00000 V with 1 decimals of accuracy

syssi, 2023-01-17:
```
QPIGS: (247.3 50.0 239.0 50.0 0931 0805 025 360 26.10 007 060 0017 04.6 179.2 00.00 00001 00010110 00 00 00831 011
QPIRI: (240.0 15.0 240.0 50.0 15.0 3600 3600 24.0 24.0 23.5 29.2 29.0 2 010 100 0 2 3 1 01 0 0 26.5 0 0

Pokornypetr, 2023-01-17:
[image]
as I said, most of the data are ok, only pv charging power and pv voltage do not match or are without value.

Pokornypetr, 2023-01-17:
logo and photo are taken at the same time

Pokornypetr, 2023-01-18:
> ```
> QPIGS: (247.3 50.0 239.0 50.0 0931 0805 025 360 26.10 007 060 0017 04.6 179.2 00.00 00001 00010110 00 00 00831 011
> QPIRI: (240.0 15.0 240.0 50.0 15.0 3600 3600 24.0 24.0 23.5 29.2 29.0 2 010 100 0 2 3 1 01 0 0 26.5 0 0
> ```

so can I ask for some concrete advice. I'm a big llama so I really need a deeper explanation

syssi, 2023-01-18:
Please update your `external_components` section to:

```
external_components:
  - source: github://syssi/esphome-pipsolar@2424mse1
    refresh: 0s
```

and flash your ESP again. The `2424mse1` feature branch should fit better to your inverter. I assume this will solve your issue.

Pokornypetr, 2023-01-22:
[image]
In external_component, everything was listed as you wrote in the example, so I only put reload-groups, but nothing changed, the problem is still the same

ityian, 2023-02-05:
Peter, did you resolve this discrepancy? I have the same inverter (AXPERT VM IV 48V) and the data you are talking about are displayed in the wrong format too.

Pokornypetr, 2023-02-18:
Hi, sorry I haven't been here for a while. Unfortunately, I haven't been able to solve it yet. Some items could be extracted from other values, I have the impression that the voltage on the PV panels comes from the value of the battery voltage from the scc, others, I don't know yet. I would like to finish it successfully, but I don't have time for it now and I don't know how to do it myself. and the support is silent here :(

syssi, 2023-04-14:
- pv_input_voltage: has a value of 0v - 0,9v at an almost real 180v
- pv_charging_power: it is still at 0w

Pokornypetr, 2023-04-15:
> @Pokornypetr Let's give it another try. Providing support is time consuming.
> 
> Could you tell me the names of the sensor which doesn't provide the correct measurements / values? Do you own the protocol description of your inverter or could you ask the seller to provide one?

Hello, thanks, pisolar was updated some time ago and since then everything shows as it should even on an expert. Many thanks for your work!

syssi, 2023-04-15:
Summary: Trying to parse an float as integer was probably the root cause or increasing the RX buffer did the trick.

closed 2023-04-15 by syssi