syssi/esphome-pipsolar #50: battery_discharge_current showing zero

reflexsa, 2023-01-29:
Hi all,

I have been using pipsolar with a ESP32-cam for a while and it has been running well. My inverter is an Axpert VMII 1.2kW.
When I initially setup pipsolar I noticed that one missing sensor is my battery_discharge_current.
This always shows 0. I noticed that in the last_qpigs message it reads correctly there.
At the moment I am extracting the value using a sensor in my configuration.yaml. See below code:

```
#extract battery discharge current from qpigs message
  - sensor:
      - name: "inverter0_battery_discharge_qpigs"
        unit_of_measurement: "A"
        availability: >
          {{states('sensor.inverter0_last_qpigs')[78:82] not in ['unavailable', 'unknown']}}
        state: >
          {{states('sensor.inverter0_last_qpigs')[78:82] | float(0)}}
```

Doing it this way causes a lot of unwanted data to be transferred and also stored in my history.
Is there a way to only send the battery discharge current section of the qpigs (basically use this code on the ESP) or to get the original "battery_discharge_current" sensor working?

Thank you for any help and advice.

reflexsa, 2023-02-02:
Managed to solve this using a lambda filter. Perhaps there is a way to use the other text sensor filters and more efficient ways but this seemed to work.
On the ESP:
```
text_sensor:
  - platform: pipsolar
    pipsolar_id: inverter0
    last_qpigs:
      id: inverter0_last_qpigs
      name: inverter0_battery_discharge_current_qpigs
      filters:
        - lambda: |-
            return x.substr(78,4);
```

In the configuration.yaml:
```
#convert battery discharge current from to float
  - sensor:
      - name: "inverter0_battery_discharge_qpigs"
        unit_of_measurement: "A"
        availability: >
          {{states('sensor.inverter0_battery_discharge_current_qpigs') not in ['unavailable', 'unknown']}}
        state: >
          {{states('sensor.inverter0_battery_discharge_current_qpigs') | float(0)}}
```

syssi, 2023-03-18:
Do you use the official esphome component or one of the feature branches?

syssi, 2023-03-18:
Could you provide a QPIGS response example / the value of the `last_qpigs` sensor?

reflexsa, 2023-03-20:
As far as I know the official version. From what I remember I just included the below in the .yaml in ESPHome:
```
pipsolar:
  - uart_id: uart_bus
    id: inverter0
```

qpigs response:
```
[text_sensor:067]: 'inverter0_battery_discharge_current_qpigs': Sending state '(235.7 50.0 235.7 50.0 0000 0000 000 431 13.80 000 100 0036 00.0 000.0 00.00 00000 00010101 00 00 00000 110
```

abdulmannan, 2023-04-06:
I am having similar issue. 'battery_discharge_current' is always 0 and it never gets updated.
```
inverter0_battery_discharge_current      0.0 A
25 minutes ago
```

here is the output of last_qpigs
```
(000.0 00.0 219.7 50.0 0109 0077 003 344 24.90 000 067 0039 00.0 000.0 00.00 00004 00010000 00 00 00000 010
```

this output clearly shows that '00004' is the battery discharge current.

I am new to ESPHome. can anyone help me resolving this issue.