syssi/esphome-pipsolar #55: does it work with inverter easun smg II?

amgstone, 2023-02-15:
does it work with inverter easun smg II?

SeByDocKy, 2023-02-15:
Seems to be modbus based (over RS232 and not RS485)... so the answer is no .the current PIPsolar integration 
But is not so complex for example from ESPhome/HA to retrieve informations from Modbus register

amgstone, 2023-02-15:
and how to build a connection scheme to HA in order to find out the registers?

syssi, 2023-02-15:
The native `api` component is your friend. All ESPHome entities will be exposed using this component: https://esphome.io/components/api.html

You have to use the `modbus_controller` component to retrieve a value per register and expose this value as sensor measurement: https://esphome.io/components/modbus_controller.html

syssi, 2023-02-15:
This is a good example for orientation / to adapt: https://esphome.io/cookbook/tracer-an.html

amgstone, 2023-02-15:
I saw an interesting article about my inverter, but I don't understand what they installed in esphome so that the HA could be configured with a modbus. https://powerforum.co.za/topic/11341-i-want-to-know-protocol-used-in-isolar-smg-ii-5kw/

syssi, 2023-02-16:
It looks like the user burleanu is using a Tasmota based bridge to forward the RAW traffic to a Home Assistant instance. The modbus-rtu integration is used here to read the registers and populate sensor entities. If you have the choice you should implement the logic at the ESPHome node to keep your HA setup as simple as possible.

amgstone, 2023-02-16:
but I need your help to configure esphome. There is no such information on the Internet. I don't understand how to make esphome take sensor settings from HA and ask the inverter via esphome. If possible, make an example of how to implement it.

syssi, 2023-02-16:
I will provide some guidance. Do you want to use an ESP32 or ESP8266 to talk to the inverter?

amgstone, 2023-02-16:
I have everything available, as it will be more convenient for you

amgstone, 2023-02-16:
didn't you forget me?)))

syssi, 2023-02-16:
No but my free time is limited to the morning and the late evening (CEST). :-P