syssi/esphome-pipsolar #56: EASun problems

v00g100skr, 2023-02-20:
hi
i have a problem with easun inverter

i use default config form example

here is the log

[13:04:28][D][pipsolar:838]: Sending polling command : QMOD with length 4
[13:04:28][D][uart_debug:158]: <<< "QMODI\xC1\r"
[13:04:28][D][pipsolar:772]: checking crc on incoming message
[13:04:28][D][pipsolar:775]: CRC OK
[13:04:28][D][pipsolar:471]: Decode QMOD
[13:04:28][D][text_sensor:067]: 'pipsolar device_mode': Sending state 'M'
[13:04:28][D][uart_debug:158]: <<< "\xC2\xD4"
[13:04:29][D][uart_debug:158]: <<< "\x02\xD0"
[13:04:29][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[13:04:29][D][pipsolar:838]: Sending polling command : QFLAG with length 5
[13:04:29][D][uart_debug:158]: <<< "QFLAG\x98t\r"
[13:04:29][D][pipsolar:772]: checking crc on incoming message
[13:04:29][D][pipsolar:775]: CRC OK
[13:04:29][D][pipsolar:479]: Decode QFLAG
[13:04:29][D][uart_debug:158]: <<< "\x90\x00"
[13:04:30][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[13:04:30][D][pipsolar:838]: Sending polling command : QPIRI with length 5
[13:04:30][D][uart_debug:158]: <<< "QPIRI\xF8T\r"
[13:04:30][D][pipsolar:772]: checking crc on incoming message
[13:04:30][D][pipsolar:775]: CRC OK
[13:04:30][D][pipsolar:429]: Decode QPIRI
[13:04:30][D][uart_debug:158]: <<< "\x82\xF7"
[13:04:31][D][uart_debug:158]: <<< "\x82\xE0"
[13:04:31][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[13:04:31][D][pipsolar:838]: Sending polling command : QPIGS with length 5
[13:04:31][D][uart_debug:158]: <<< "QPIGS\xB7\xA9\r"
[13:04:31][D][pipsolar:772]: checking crc on incoming message
[13:04:31][D][pipsolar:775]: CRC OK
[13:04:31][D][pipsolar:448]: Decode QPIGS
[13:04:31][D][sensor:127]: 'pipsolar grid_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[13:04:31][D][sensor:127]: 'pipsolar grid_frequency': Sending state 0.00000 Hz with 1 decimals of accuracy
[13:04:31][D][sensor:127]: 'pipsolar ac_output_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[13:04:31][D][sensor:127]: 'pipsolar ac_output_frequency': Sending state 0.00000 Hz with 1 decimals of accuracy
[13:04:31][D][sensor:127]: 'pipsolar ac_output_apparent_power': Sending state 0.00000 VA with 1 decimals of accuracy
[13:04:31][D][sensor:127]: 'pipsolar ac_output_active_power': Sending state 0.00000 W with 1 decimals of accuracy
[13:04:31][D][sensor:127]: 'pipsolar output_load_percent': Sending state 0.00000 % with 1 decimals of accuracy
[13:04:31][D][sensor:127]: 'pipsolar bus_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[13:04:31][D][sensor:127]: 'pipsolar battery_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[13:04:31][D][sensor:127]: 'pipsolar battery_charging_current': Sending state 0.00000 A with 1 decimals of accuracy
[13:04:31][D][sensor:127]: 'pipsolar battery_capacity_percent': Sending state 0.00000 % with 1 decimals of accuracy
[13:04:31][D][sensor:127]: 'pipsolar inverter_heat_sink_temperature': Sending state 0.00000 °C with 1 decimals of accuracy
[13:04:31][D][sensor:127]: 'pipsolar pv_input_current_for_battery': Sending state 0.00000 A with 1 decimals of accuracy
[13:04:31][D][sensor:127]: 'pipsolar pv_input_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[13:04:31][D][sensor:127]: 'pipsolar battery_voltage_scc': Sending state 0.00000 V with 1 decimals of accuracy
[13:04:31][D][sensor:127]: 'pipsolar battery_discharge_current': Sending state 0.00000 A with 1 decimals of accuracy
[13:04:31][D][sensor:127]: 'pipsolar battery_voltage_offset_for_fans_on': Sending state 0.00000 V with 1 decimals of accuracy
[13:04:31][D][sensor:127]: 'pipsolar pv_charging_power': Sending state 0.00000 W with 1 decimals of accuracy
[13:04:31][D][uart_debug:158]: <<< "\xC2\xD8"
[13:04:32][D][uart_debug:158]: <<< "\x80\x00"
[13:04:32][D][uart_debug:158]: >>> "QMODI\xC1\r"
[13:04:32][D][pipsolar:838]: Sending polling command : QMOD with length 4
[13:04:32][D][uart_debug:158]: <<< "QMODI\xC1\r"
[13:04:32][D][pipsolar:772]: checking crc on incoming message
[13:04:32][D][pipsolar:775]: CRC OK
[13:04:32][D][pipsolar:471]: Decode QMOD
[13:04:32][D][text_sensor:067]: 'pipsolar device_mode': Sending state 'M'
[13:04:32][D][uart_debug:158]: <<< "\x80$"
[13:04:33][D][uart_debug:158]: <<< "\x02\xF0"
[13:04:33][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[13:04:33][D][pipsolar:838]: Sending polling command : QFLAG with length 5
[13:04:33][D][uart_debug:158]: <<< "QFLAG\x98t\r"
[13:04:33][D][pipsolar:772]: checking crc on incoming message
[13:04:33][D][pipsolar:775]: CRC OK
[13:04:33][D][pipsolar:479]: Decode QFLAG
[13:04:33][D][uart_debug:158]: <<< "B\xE8"
[13:04:34][D][uart_debug:158]: <<< "\x80\x84"
[13:04:34][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[13:04:34][D][pipsolar:838]: Sending polling command : QPIRI with length 5
[13:04:34][D][uart_debug:158]: <<< "QPIRI\xF8T\r"
[13:04:34][D][pipsolar:772]: checking crc on incoming message
[13:04:34][D][pipsolar:775]: CRC OK
[13:04:34][D][pipsolar:429]: Decode QPIRI
[13:04:34][D][uart_debug:158]: <<< "\x80\x80"
[13:04:35][D][uart_debug:158]: <<< "\x02\xE0"
[13:04:35][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[13:04:35][D][pipsolar:838]: Sending polling command : QPIGS with length 5
[13:04:35][D][uart_debug:158]: <<< "QPIGS\xB7\xA9\r"
[13:04:35][D][pipsolar:772]: checking crc on incoming message
[13:04:35][D][pipsolar:775]: CRC OK
[13:04:35][D][pipsolar:448]: Decode QPIGS
[13:04:35][D][sensor:127]: 'pipsolar grid_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[13:04:35][D][sensor:127]: 'pipsolar grid_frequency': Sending state 0.00000 Hz with 1 decimals of accuracy
[13:04:35][D][sensor:127]: 'pipsolar ac_output_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[13:04:35][D][sensor:127]: 'pipsolar ac_output_frequency': Sending state 0.00000 Hz with 1 decimals of accuracy
[13:04:35][D][sensor:127]: 'pipsolar ac_output_apparent_power': Sending state 0.00000 VA with 1 decimals of accuracy
[13:04:35][D][sensor:127]: 'pipsolar ac_output_active_power': Sending state 0.00000 W with 1 decimals of accuracy
[13:04:35][D][sensor:127]: 'pipsolar output_load_percent': Sending state 0.00000 % with 1 decimals of accuracy
[13:04:35][D][sensor:127]: 'pipsolar bus_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[13:04:35][D][sensor:127]: 'pipsolar battery_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[13:04:35][D][sensor:127]: 'pipsolar battery_charging_current': Sending state 0.00000 A with 1 decimals of accuracy
[13:04:35][D][sensor:127]: 'pipsolar battery_capacity_percent': Sending state 0.00000 % with 1 decimals of accuracy
[13:04:35][D][sensor:127]: 'pipsolar inverter_heat_sink_temperature': Sending state 0.00000 °C with 1 decimals of accuracy
[13:04:35][D][sensor:127]: 'pipsolar pv_input_current_for_battery': Sending state 0.00000 A with 1 decimals of accuracy
[13:04:35][D][sensor:127]: 'pipsolar pv_input_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[13:04:35][D][sensor:127]: 'pipsolar battery_voltage_scc': Sending state 0.00000 V with 1 decimals of accuracy
[13:04:35][D][sensor:127]: 'pipsolar battery_discharge_current': Sending state 0.00000 A with 1 decimals of accuracy
[13:04:35][D][sensor:127]: 'pipsolar battery_voltage_offset_for_fans_on': Sending state 0.00000 V with 1 decimals of accuracy
[13:04:35][D][sensor:127]: 'pipsolar pv_charging_power': Sending state 0.00000 W with 1 decimals of accuracy

syssi, 2023-02-20:
You are receiving a local echo of the request. This isn't a real response of your inverter. It looks like your wiring isn't correct.

v00g100skr, 2023-02-20:
thanks. i will try to find a mistake

syssi, 2023-02-20:
May be this schematics will help you to get a better idea how the data cable should look like: https://github.com/syssi/esphome-smg-ii#schematics

v00g100skr, 2023-02-20:
i use only rx-tx pins to connect because my rs232 is powered from inverter
is this right?

syssi, 2023-02-20:
Do you have a RS232-to-TTL converter module between the inverter and your ESP? You have to convert the logic level from -12/+12V to 0/3.3V.

syssi, 2023-02-20:
Could you provide a photo of your wiring?

v00g100skr, 2023-02-20:
i have a wifi dongle with cable to rs232 port in invertor
inside this dongle i found  "3.3-rx-tx-gnd" pins 

https://photos.app.goo.gl/XTy8T4vCptWV686V7

syssi, 2023-02-20:
I assume the pin header at the PCB is an UARR-TTL and not RS232. You should connect the inverter to the D-Sub 9pol connector and make sure the pinout matches.

v00g100skr, 2023-02-20:
i tried this connection

https://photos.app.goo.gl/txGkw8yFhJcKhHF86

but maybe its a wrong way because its not rs232

i will buy max3232 and do manual wiring through D-Sub 9pol (i have internal pinout in inverter manual)

Foroxon, 2023-02-20:
@syssi MAX3222 is also compatible or do we need only MAX3232-based converters?

syssi, 2023-02-20:
Is there a difference between the MAX3222 and MAX3232? Please make sure the RS232 converter can be driven using 3.3V. There are converter chips around which require 5V and 3.3V is out of spec.

syssi, 2023-03-18:
Any news here? I would like to close this issue.

closed 2023-04-14 by syssi

v00g100skr, 2023-06-06:
we find another way to connect - modbus through  usb port and manual mapping all data from this file
https://github.com/syssi/esphome-smg-ii/blob/main/esp32-example.yaml