syssi/esphome-pipsolar #57: Few qestions about PIP8048 Max

sergeyshadow, 2023-02-26:
Hello.
I use 2424mse1 branch.
Is it possible to get data on pv2 (Voltage, current, etc.)?
And is it possible to get the data of solar energy production for the day that is stored in the inverter?

syssi, 2023-02-26:
> Is it possible to get data on pv2 (Voltage, current, etc.)?

A second PV string is supported by the `pip8048` branch. I've pushed the support of long QPIGS responses here too now. Could you try this branch again? 

```
external_components:
  - source: github://syssi/esphome-pipsolar@pip8048
    refresh: 0s
```

Some entities/sensors are named differently here. This example configuration YAML should help:

https://github.com/syssi/esphome-pipsolar/blob/pip8048/esp32-example.yaml
https://github.com/syssi/esphome-pipsolar/blob/pip8048/esp8266-example.yaml

syssi, 2023-02-26:
> And is it possible to get the data of solar energy production for the day that is stored in the inverter?

The retrieval of this property isn't supported yet. I'm aware of voices that claim the values are not very accurate.

sergeyshadow, 2023-02-26:
Seems to work :)
But again there are a few questions:
1. My inverter has a maximum charge current of 150A and a minimum of 10A. can i just change the yaml?
2. I can not understand what the "charging discharging control" parameter is responsible for. I searched for information in the document from the comment, but without success.

syssi, 2023-02-26:
1. Yes. The default options are just defaults. You can override these defaults at the YAML. Just search the name of the select entity in the issues of this project. I assume you will find some more hints/examples.
2. I can just provide the issued command so you can try to look it up at the protocol description. Does this help?

sergeyshadow, 2023-02-27:
> 1. Yes. The default options are just defaults. You can override these defaults at the YAML. Just search the name of the select entity in the issues of this project. I assume you will find some more [...]

1. it's work, thanks
2. Yes :)

syssi, 2023-02-27:
```
  - platform: pipsolar
    pipsolar_id: inverter0
    charging_discharging_control:
      # See MAX_Communication_Protocol_20200526.pdf page 18
      name: "${name} charging discharging control"
      optionsmap:
        "111": "PBATCD111"
        "011": "PBATCD011"
        "101": "PBATCD101"
        "110": "PBATCD110"
        "010": "PBATCD010"
        "100": "PBATCD100"
        "001": "PBATCD001"
        "000": "PBATCD000"
      statusmap:
        "111": "111"
        "011": "011"
        "101": "101"
        "110": "110"
        "010": "010"
        "100": "100"
        "001": "001"
        "000": "000"
```

Please try to lookup the `PBATCD` command. See MAX_Communication_Protocol_20200526.pdf page 18.

closed 2023-02-27 by syssi