syssi/esphome-pipsolar #59: Easun smh 3k

Warloo93, 2023-03-09:
Hello. 
I am trying to configure my inverter to HA without luck. 
I think hardware wiring is good, here is picture : 
[image]
One from the inverter : 
[image]

And here is the debug from watchpower and log from esphome in zip. [Zip](https://drive.google.com/file/d/17_66NXF-hrRubD033psy2vlOuQ0S4Gh9/view?usp=drivesdk) 

I tried to add diferent external components but no one worked. 

Here is my current config : 

  esphome:
  name: solar
  platform: ESP8266
  board: d1_mini

# Enable logging
logger:
  level: VERY_VERBOSE
  logs:
    uart: VERY_VERBOSE
    uart.arduino_esp8266: VERY_VERBOSE
    
# Enable Home Assistant API
api:

ota:
  

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Solar Fallback Hotspot"

captive_portal:

uart:
  - id: uart_bus
    tx_pin: TX
    rx_pin: RX
    # most devices use 2400 as baud_rate
    baud_rate: 2400

pipsolar:
  - uart_id: uart_bus
    id: inverter0

sensor:
  - platform: pipsolar
    pipsolar_id: inverter0

    grid_rating_current:
      id: inverter0_grid_rating_current
      name: inverter0_grid_rating_current
    ac_output_active_power:
      id: inverter0_ac_output_active_power
      name: inverter0_ac_output_active_power
    ac_output_apparent_power:
      id: inverter0_ac_output_apparent_power
      name: inverter0_ac_output_apparent_power
    ac_output_voltage:
      id: inverter0_ac_output_voltage
      name: inverter0_ac_output_voltage
    battery_capacity_percent:
      id: inverter0_battery_capacity_percent
      name: inverter0_battery_capacity_percent
    battery_charging_current:
      id: inverter0_battery_charging_current
      name: inverter0_battery_charging_current
    battery_discharge_current:
      id: inverter0_battery_discharge_current
      name: inverter0_battery_discharge_current
    battery_voltage:
      id: inverter0_battery_voltage
      name: inverter0_battery_voltage
    grid_voltage:
      id: inverter0_grid_voltage
      name: inverter0_grid_voltage
    inverter_heat_sink_temperature:
      id: inverter0_inverter_heat_sink_temperature
      name: inverter0_inverter_heat_sink_temperature
    output_load_percent:
      id: inverter0_output_load_percent
      name: inverter0_output_load_percent
    pv_charging_power:
      id: inverter0_pv_charging_power
      name: inverter0_pv_charging_power
    pv_input_voltage:
      id: inverter0_pv_input_voltage
      name: inverter0_pv_input_voltage
    

    
text_sensor:
  - platform: pipsolar
    pipsolar_id: inverter0
    device_mode:
      id: inverter0_device_mode
      name: inverter0_device_mode
    last_qpigs:
      id: inverter0_last_qpigs
      name: inverter0_last_qpigs
    last_qpiri:
      id: inverter0_last_qpiri
      name: inverter0_last_qpiri
      
switch:
  - platform: pipsolar
    pipsolar_id: inverter0
    output_source_priority_utility:
      name: inverter0_output_source_priority_utility
    output_source_priority_solar:
      name: inverter0_output_source_priority_solar
    output_source_priority_battery:
      name: inverter0_output_source_priority_battery
    input_voltage_range:
      name: inverter0_input_voltage_range
    pv_power_balance:
      name: inverter0_pv_power_balance

Thank you for the support.

Warloo93, 2023-03-09:
And i cannot find where need to write this code : 

# Install esphome
pip3 install esphome

# Clone this external component
git clone https://github.com/syssi/esphome-pipsolar.git
cd esphome-pipsolar

# Create a secret.yaml containing some setup specific secrets
cat > secrets.yaml <<EOF
mqtt_host: MY_MQTT_HOST
mqtt_username: MY_MQTT_USERNAME
mqtt_password: MY_MQTT_PASSWORD

wifi_ssid: MY_WIFI_SSID
wifi_password: MY_WIFI_PASSWORD
EOF

# Validate the configuration, create a binary, upload it, and start logs
# If you use a esp8266 run the esp8266-examle.yaml
esphome run esp32-example.yaml

i tried into esp config, HA terminal but nothing works

syssi, 2023-03-18:
If you use RX/TX (GPIO1/GPIO3) instead of GPIO4/5 you have to set the `baud_rate` of the `logger` to `0`:

```
logger:
  baud_rate: 0
  level: VERY_VERBOSE
  logs:
    uart: VERY_VERBOSE
    uart.arduino_esp8266: VERY_VERBOSE
```

The logger is attached to GPIO1/GPIO3 per default.

syssi, 2023-03-18:
Please flash this YAML as first step:

https://raw.githubusercontent.com/syssi/esphome-pipsolar/main/tests/esp8266-test-protocols.yaml

Please attach the inverter/rs232 module to GPIO4/GPIO5 and provide the log output. It should show some outgoing and incoming traffic. So we get an idea which protocol version your inverter supports.

Warloo93, 2023-03-18:
Thank you for response.
My bad, i connected the rs232 revesed, and ttl side too. 
Now is working, only problem i have with temperature meter, its showing 400-700 °C.

syssi, 2023-03-19:
Do you know the command (QPIGS, QPIRI?) which returns the temperature? Could you provide an example response containing the temperature?

Warloo93, 2023-03-19:
Do you mean this log ? 

`[12:44:14][D][pipsolar:836]: Sending polling command : QPIGS with length 5
[12:44:14][D][pipsolar:772]: checking crc on incoming message
[12:44:14][D][pipsolar:775]: CRC OK
[12:44:14][D][pipsolar:448]: Decode QPIGS
[12:44:14][D][sensor:126]: 'easun grid_voltage': Sending state 241.00000 V with 1 decimals of accuracy
[12:44:14][D][sensor:126]: 'easun grid_frequency': Sending state 50.00000 Hz with 1 decimals of accuracy
[12:44:14][D][sensor:126]: 'easun ac_output_voltage': Sending state 228.00000 V with 1 decimals of accuracy
[12:44:14][D][sensor:126]: 'easun ac_output_frequency': Sending state 50.10000 Hz with 1 decimals of accuracy
[12:44:14][D][sensor:126]: 'easun ac_output_apparent_power': Sending state 0.00000 VA with 1 decimals of accuracy
[12:44:14][D][sensor:126]: 'easun ac_output_active_power': Sending state 0.00000 W with 1 decimals of accuracy
[12:44:14][D][sensor:126]: 'easun output_load_percent': Sending state 0.00000 % with 1 decimals of accuracy
[12:44:14][D][sensor:126]: 'easun bus_voltage': Sending state 419.00000 V with 1 decimals of accuracy
[12:44:14][D][sensor:126]: 'easun battery_voltage': Sending state 28.26000 V with 1 decimals of accuracy
[12:44:14][D][sensor:126]: 'easun battery_charging_current': Sending state 12.00000 A with 1 decimals of accuracy
[12:44:14][D][sensor:126]: 'easun battery_capacity_percent': Sending state 100.00000 % with 1 decimals of accuracy
[12:44:14][D][sensor:126]: 'easun inverter_heat_sink_temperature': Sending state 696.00000 °C with 1 decimals of accuracy
[12:44:14][D][sensor:126]: 'easun pv_input_current_for_battery': Sending state 13.00000 A with 1 decimals of accuracy
[12:44:14][D][sensor:126]: 'easun pv_input_voltage': Sending state 73.10000 V with 1 decimals of accuracy
[12:44:14][D][sensor:126]: 'easun battery_voltage_scc': Sending state 28.11000 V with 1 decimals of accuracy
[12:44:14][D][sensor:126]: 'easun battery_discharge_current': Sending state 0.00000 A with 1 decimals of accuracy
[12:44:14][D][sensor:126]: 'easun battery_voltage_offset_for_fans_on': Sending state 0.00000 V with 1 decimals of accuracy
[12:44:14][D][sensor:126]: 'easun pv_charging_power': Sending state 389.00000 W with 1 decimals of accuracy`

syssi, 2023-03-19:
Not exactly. I'm looking for the value of the `text_sensor.last_qpigs` sensor.

Warloo93, 2023-03-19:
I dont see .last_qpigs in my log 
[14:14:09][C][wifi:504]: WiFi:
[14:14:09][C][wifi:362]:   Local MAC: A0:20:A6:02:56:4C
[14:14:09][C][wifi:363]:   SSID: [redacted]
[14:14:09][C][wifi:364]:   IP Address: 192.168.1.15
[14:14:09][C][wifi:365]:   BSSID: [redacted]
[14:14:09][C][wifi:367]:   Hostname: 'easun'
[14:14:09][C][wifi:369]:   Signal strength: -90 dB ▂▄▆█
[14:14:09][C][wifi:373]:   Channel: 13
[14:14:09][C][wifi:374]:   Subnet: 255.255.255.0
[14:14:09][C][wifi:375]:   Gateway: 192.168.1.1
[14:14:09][C][wifi:376]:   DNS1: 0.0.0.0
[14:14:09][C][wifi:377]:   DNS2: 0.0.0.0
[14:14:09][C][logger:293]: Logger:
[14:14:09][C][logger:294]:   Level: DEBUG
[14:14:09][C][logger:295]:   Log Baud Rate: 0
[14:14:09][C][logger:296]:   Hardware UART: UART0
[14:14:09][C][uart.arduino_esp8266:102]: UART Bus:
[14:14:09][C][uart.arduino_esp8266:103]:   TX Pin: GPIO3
[14:14:09][C][uart.arduino_esp8266:104]:   RX Pin: GPIO1
[14:14:09][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[14:14:09][C][uart.arduino_esp8266:108]:   Baud Rate: 2400 baud
[14:14:09][C][uart.arduino_esp8266:109]:   Data Bits: 8
[14:14:09][C][uart.arduino_esp8266:110]:   Parity: NONE
[14:14:09][C][uart.arduino_esp8266:111]:   Stop bits: 1
[14:14:09][C][uart.arduino_esp8266:115]:   Using software serial
[14:14:09][C][pipsolar:860]: Pipsolar:
[14:14:09][C][pipsolar:861]: used commands:
[14:14:09][C][pipsolar:864]: QPIGS
[14:14:09][C][pipsolar:864]: QMOD
[14:14:09][C][pipsolar:864]: QFLAG
[14:14:09][C][pipsolar:864]: QPIRI
[14:14:09][C][pipsolar.switch:076]: Pipsolar Switch 'easun output_source_priority_utility'
[14:14:09][C][pipsolar.switch:078]:   Icon: 'mdi:power'
[14:14:09][C][pipsolar.switch:098]:   Restore Mode: restore defaults to OFF
[14:14:09][C][pipsolar.switch:076]: Pipsolar Switch 'easun output_source_priority_solar'
[14:14:09][C][pipsolar.switch:078]:   Icon: 'mdi:power'
[14:14:09][C][pipsolar.switch:098]:   Restore Mode: restore defaults to OFF
[14:14:09][C][pipsolar.switch:076]: Pipsolar Switch 'easun output_source_priority_battery'
[14:14:09][C][pipsolar.switch:078]:   Icon: 'mdi:power'
[14:14:09][C][pipsolar.switch:098]:   Restore Mode: restore defaults to OFF
[14:14:09][C][pipsolar.switch:076]: Pipsolar Switch 'easun input_voltage_range'
[14:14:09][C][pipsolar.switch:078]:   Icon: 'mdi:power'
[14:14:09][C][pipsolar.switch:098]:   Restore Mode: restore defaults to OFF
[14:14:09][C][pipsolar.switch:076]: Pipsolar Switch 'easun pv_ok_condition_for_parallel'
[14:14:09][C][pipsolar.switch:078]:   Icon: 'mdi:power'
[14:14:09][C][pipsolar.switch:098]:   Restore Mode: restore defaults to OFF
[14:14:09][C][pipsolar.switch:076]: Pipsolar Switch 'easun pv_power_balance'
[14:14:09][C][pipsolar.switch:078]:   Icon: 'mdi:power'
[14:14:09][C][pipsolar.switch:098]:   Restore Mode: restore defaults to OFF
[14:14:10][C][mdns:108]: mDNS:
[14:14:10][C][mdns:109]:   Hostname: easun
[14:14:10][C][ota:093]: Over-The-Air Updates:
[14:14:10][C][ota:094]:   Address: 192.168.1.15:8266
[14:14:10][C][api:138]: API Server:
[14:14:10][C][api:139]:   Address: 192.168.1.15:6053
[14:14:10][C][api:143]:   Using noise encryption: NO
[14:14:10][D][pipsolar:836]: Sending polling command : QMOD with length 4
[14:14:10][D][pipsolar:772]: checking crc on incoming message
[14:14:10][D][pipsolar:775]: CRC OK
[14:14:10][D][pipsolar:471]: Decode QMOD
[14:14:10][D][text_sensor:067]: 'easun device_mode': Sending state 'B'
[14:14:10][D][pipsolar:836]: Sending polling command : QFLAG with length 5
[14:14:10][D][pipsolar:772]: checking crc on incoming message
[14:14:10][D][pipsolar:775]: CRC OK
[14:14:10][D][pipsolar:479]: Decode QFLAG
[14:14:11][D][pipsolar:836]: Sending polling command : QPIRI with length 5
[14:14:12][D][pipsolar:772]: checking crc on incoming message
[14:14:12][D][pipsolar:775]: CRC OK
[14:14:12][D][pipsolar:429]: Decode QPIRI
[14:14:12][D][pipsolar:836]: Sending polling command : QPIGS with length 5
[14:14:13][D][pipsolar:772]: checking crc on incoming message
[14:14:13][D][pipsolar:775]: CRC OK
[14:14:13][D][pipsolar:448]: Decode QPIGS

syssi, 2023-03-19:
Did you remove this sensor from your YAML?

```
text_sensor:
  - platform: pipsolar
    pipsolar_id: inverter0
    last_qpigs:
      id: inverter0_last_qpigs
      name: inverter0_last_qpigs
```

Warloo93, 2023-03-19:
I have comma before this section.
now is here : 
[14:25:25][D][text_sensor:067]: 'easun last_qpigs': Sending state '(236.0 49.9 228.0 50.0 0091 0091 003 418 28.23 003 100 0680 0008 073.4 28.05 00000 10010110 00 04 00251 100'

Warloo93, 2023-03-21:
Do you need log or anything?

syssi, 2023-03-21:
No. Could you ask your seller about protocol details of your device? It assume the value/position reported as temperature isn't a temperature at your case.

Warloo93, 2023-03-21:
I wrote for the seller. I will notice you, if i get something. 
Thank you for support.

Warloo93, 2023-03-22:
Hi
I got the pdf from seller.

The problem:
 
temperature 
T is an integer ranging from 0 to 9. The units 
is ℃（NTC A/D value for Axpert 1~3K）

[Limk](https://wetransfer.com/downloads/dfdfc4990844e9e6cff312dd3a728ae920230322023252/8fa1c9?spm=a1z65.imconversation.0.0&statusBarHeight=113&_dognoseId=WVk2dlJhRXNrMW9EQUZiUGoraEJ4WXYBhweH1GthbGlleHByZQAAAz%2F1&_currency=EUR&_lang=en_SK&fromApp=true)

syssi, 2023-03-22:
I must admit I don't understand the protocol description here:

[image]

In other words are value between `0000` and `9999` is expected here but a resolution is missing here. Do you have the chance to find out the proper meaning of

```
[12:44:14][D][sensor:126]: 'easun inverter_heat_sink_temperature': Sending state 696.00000 °C with 1 decimals of accuracy
```

If we multiply this value by `0.1` the sensor would return `69.9 °C`. Does this match with the reality?

Warloo93, 2023-03-22:
I think is not that simple. [https://sparks.gogo.co.nz/ntc_thermistor.html](link) 

No, multiplyng not match the reality. I have this inverter in 0-15°C.

syssi, 2023-03-22:
This is an example how to transform a ADC voltage into a resistance into a temperature / sensor value:

https://github.com/syssi/esphome-xiaomi-philips-light/blob/main/xiaomi_light.yaml#L41-L69

I cannot help here. The math behind it is black magic for me. ;-)

avion23, 2023-03-22:
The voltage divider configuration greatly influences the behavior of the
NTC thermistor in the circuit.

The NTC can be connected:

- Above the fixed resistor (toward the input voltage)
- Or below the fixed resistor (toward ground)

It is most likely that the NTC is connected to ground in the circuit, but I
simply don't know.

Furthermore, the value of the fixed resistor shifts the usable range of the
temperature measurements. It is important to consider the type of NTC being
used, which in this case, is assumed to be a 100k thermistor.

This is what chatgpt proposes, I have no idea if it works:
```
#include <math.h>

#define REF_TEMP 25.0f
#define REF_RESISTANCE 100000.0f
#define B_CONSTANT 3950.0f
#define ADC_MAX_VALUE 1024
#define FIXED_RESISTOR 100000.0f

float convert_raw_adc_to_celsius(int adc_value) {
  float resistance_ntc = FIXED_RESISTOR * ((float)ADC_MAX_VALUE /
(float)adc_value - 1);
  float temperature_kelvin = B_CONSTANT / log(resistance_ntc /
REF_RESISTANCE) + 1 / (REF_TEMP + 273.15f);
  return temperature_kelvin - 273.15f;
}
```

closed 2023-04-14 by syssi