syssi/esphome-pipsolar #63: MPP Solar MPPT Charge Controller

adrusa, 2023-03-26:
I tried to use this for my MPP Solar Charge Controller PCM60X since I noticed it uses similar code to retrieve data, but I get some weird data. Is anyone using this for a charge controller?

adrusa, 2023-03-28:
> Could you provide an example response of your device? You could try to request it using this example:
> 
> ```
> substitutions:
>   name: pipsolar-protocol-test
>   tx_pin: GPIO4
>   rx_pin: GPIO5
> 
> esphome:
>   name: ${name}
>   platform: ESP8266
>   board: d1_mini
> 
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
> 
> api:
> ota:
> logger:
> 
> uart:
>   id: uart0
>   baud_rate: 2400
>   tx_pin: ${tx_pin}
>   rx_pin: ${rx_pin}
>   debug:
>     direction: BOTH
>     dummy_receiver: true
>     after:
>       delimiter: "\r"
>     sequence:
>       - lambda: UARTDebug::log_string(direction, bytes);
> 
> interval:
>   - interval: 30s
>     then:
>       - uart.write: "QPIGS\xB7\xA9\r"
> ```

`INFO Reading configuration /config/esphome/fcctest.yaml...
INFO Starting log output from pipsolar-protocol-test.local using esphome API
INFO Successfully connected to pipsolar-protocol-test.local
[15:03:37][I][app:102]: ESPHome version 2023.3.2 compiled on Mar 28 2023, 14:54:00
[15:03:37][C][wifi:504]: WiFi:
[15:03:37][C][wifi:362]:   Local MAC: 
[15:03:37][C][wifi:363]:   SSID: 
[15:03:37][C][wifi:364]:   IP Address: 
[15:03:37][C][wifi:365]:   BSSID: 
[15:03:37][C][wifi:367]:   Hostname: 'pipsolar-protocol-test'
[15:03:37][C][wifi:369]:   Signal strength: -34 dB ▂▄▆█
[15:03:37][C][wifi:373]:   Channel: 11
[15:03:37][C][wifi:374]:   Subnet: 255.255.255.0
[15:03:37][C][wifi:375]:   Gateway: 
[15:03:37][C][wifi:376]:   DNS1: 
[15:03:37][C][wifi:377]:   DNS2: 0.0.0.0
[15:03:37][C][logger:293]: Logger:
[15:03:37][C][logger:294]:   Level: DEBUG
[15:03:37][C][logger:295]:   Log Baud Rate: 115200
[15:03:37][C][logger:296]:   Hardware UART: UART0
[15:03:37][C][uart.arduino_esp8266:102]: UART Bus:
[15:03:37][C][uart.arduino_esp8266:103]:   TX Pin: GPIO4
[15:03:37][C][uart.arduino_esp8266:104]:   RX Pin: GPIO5
[15:03:37][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[15:03:37][C][uart.arduino_esp8266:108]:   Baud Rate: 2400 baud
[15:03:37][C][uart.arduino_esp8266:109]:   Data Bits: 8
[15:03:37][C][uart.arduino_esp8266:110]:   Parity: NONE
[15:03:37][C][uart.arduino_esp8266:111]:   Stop bits: 1
[15:03:37][C][uart.arduino_esp8266:115]:   Using software serial
[15:03:37][C][web_server:151]: Web Server:
[15:03:37][C][web_server:152]:   Address: pipsolar-protocol-test.local:80
[15:03:37][C][mdns:108]: mDNS:
[15:03:37][C][mdns:109]:   Hostname: pipsolar-protocol-test
[15:03:37][C][ota:093]: Over-The-Air Updates:
[15:03:37][C][ota:094]:   Address: pipsolar-protocol-test.local:8266
[15:03:37][C][api:138]: API Server:
[15:03:37][C][api:139]:   Address: pipsolar-protocol-test.local:6053
[15:03:37][C][api:143]:   Using noise encryption: NO
[15:03:38][D][uart_debug:158]: >>> "QPIGS\xC2\xB7\xC2\xA9\r"
[15:03:38][D][uart_debug:158]: <<< "(NAKss\r"
[15:04:08][D][uart_debug:158]: >>> "QPIGS\xC2\xB7\xC2\xA9\r"
[15:04:08][D][uart_debug:158]: <<< "(NAKss\r"`

Thanks for your quick and kind response. This is the log.

syssi, 2023-03-28:
Do you have a windows application of the manufacturer to capture some working commands?

adrusa, 2023-03-28:
> Do you have a windows application of the manufacturer to capture some working commands?

Yes, it is MPPTracker. http://www.mppsolar.com/manual/PCM-60X/MPPTracker%20(monitoring%20software)/installMPPTracker_Windows(v1.02sp3).rar

syssi, 2023-03-28:
Please try

```
      - uart.write: "QPI\xBE\xAC\r"
```

This is the periodic MPPTracker request.

adrusa, 2023-03-28:
> - uart.write: "QPI\xBE\xAC\r"

`[21:29:20][D][uart_debug:158]: >>> "QPI\xC2\xBE\xC2\xAC\r"
[21:29:20][D][uart_debug:158]: <<< "(NAKss\r"
[21:29:50][D][uart_debug:158]: >>> "QPI\xC2\xBE\xC2\xAC\r"
[21:29:50][D][uart_debug:158]: <<< "(NAKss\r"
[21:30:20][D][uart_debug:158]: >>> "QPI\xC2\xBE\xC2\xAC\r"
[21:30:20][D][uart_debug:158]: <<< "(NAKss\r"`

Basically same reply.

syssi, 2023-03-29:
Next try:

```
      - uart.write: [0x51, 0x50, 0x49, 0xbe, 0xac, 0x0d]
```

adrusa, 2023-03-29:
> - uart.write: [0x51, 0x50, 0x49, 0xbe, 0xac, 0x0d]

`[09:39:28][D][uart_debug:158]: >>> "QPI\xBE\xAC\r"
[09:39:28][D][uart_debug:158]: <<< "(PI34\xDA\x8F\r"
[09:39:58][D][uart_debug:158]: >>> "QPI\xBE\xAC\r"
[09:39:58][D][uart_debug:158]: <<< "(PI34\xDA\x8F\r"`

This is the new log.

syssi, 2023-03-29:
This is a valid response. Nice!

adrusa, 2023-03-29:
> This is a valid response. Nice!

Thanks, so how do I proceed?

syssi, 2023-04-07:
Could you try to use `mpp-solar` to talk to your inverter? Just to verify we are on the right track?

adrusa, 2023-04-09:
> Could you try to use `mpp-solar` to talk to your inverter? Just to verify we are on the right track?

How do I do that?

syssi, 2023-04-14:
I've improved the protocol test. Could you flash the updated YAML and provide another log? Thanks in advance!

https://github.com/syssi/esphome-pipsolar/blob/main/tests/esp8266-test-protocols.yaml

adrusa, 2023-04-14:
> I've improved the protocol test. Could you flash the updated YAML and provide another log? Thanks in advance!
> 
> https://github.com/syssi/esphome-pipsolar/blob/main/tests/esp8266-test-protocols.yaml

This is the new log. Thanks.
`[18:08:09][D][uart_debug:158]: >>> "^P005GSX\x14\r"
[18:08:09][D][uart_debug:158]: <<< "(NAKss\r"
[18:08:10][I][main:407]: Testing PI30/PI30MAX/PI30REVO/PI41 commands...
[18:08:10][I][main:410]: This is the set of commands supported by the pipsolar component!
[18:08:10][D][uart_debug:158]: >>> "QPI\xBE\xAC\r"
[18:08:10][D][uart_debug:158]: <<< "(PI34\xDA\x8F\r"
[18:08:11][D][uart_debug:158]: >>> "^P006MOD\xDD\xBE\r"
[18:08:11][D][uart_debug:158]: <<< "(NAKss\r"
[18:08:12][D][uart_debug:158]: >>> "QDIq\e\r"
[18:08:12][D][uart_debug:158]: <<< "(00 2000 02 14.40 13.60 00 +00.0 0000\xFDq\r"
[18:08:13][I][main:524]: Testing unsupported PI17 commands...
[18:08:13][D][uart_debug:158]: >>> "^P003PI\r"
[18:08:13][D][uart_debug:158]: <<< "(NAKss\r"
[18:08:14][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[18:08:14][D][uart_debug:158]: <<< "(NAKss\r"
[18:08:15][D][uart_debug:158]: >>> "^P004MOD\r"
[18:08:15][D][uart_debug:158]: <<< "(NAKss\r"
[18:08:16][D][uart_debug:158]: >>> "QMN\xBBd\r"
[18:08:16][D][uart_debug:158]: <<< "(NAKss\r"
[18:08:17][D][uart_debug:158]: >>> "^P005FLAG\r"
[18:08:17][D][uart_debug:158]: <<< "(NAKss\r"
[18:08:18][D][uart_debug:158]: >>> "QMODI\xC1\r"
[18:08:18][D][uart_debug:158]: <<< "(NAKss\r"
[18:08:19][I][main:548]: Testing unsupported PI16 commands...
[18:08:19][D][uart_debug:158]: >>> "QPI\r"
[18:08:19][D][uart_debug:158]: <<< "(NAKss\r"
[18:08:20][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[18:08:20][D][uart_debug:158]: <<< "(095.0 54.00 03.98 03.98 00.00 0214 +047 42.81 -001 0000 01000000\xF1@\r"
[18:08:21][D][uart_debug:158]: >>> "QMOD\r"
[18:08:21][D][uart_debug:158]: <<< "(NAKss\r"
[18:08:22][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[18:08:22][D][uart_debug:158]: <<< "(3000 48 20.0 13.85 13.85 02 01 +00.0 00 04 4 11.50 0\xED\xAE\r"
[18:08:23][D][uart_debug:158]: >>> "QPIGS\r"
[18:08:23][D][uart_debug:158]: <<< "(NAKss\r"
[18:08:24][D][uart_debug:158]: >>> "QPIWS\xB4\xDA\r"
[18:08:24][D][uart_debug:158]: <<< "(000000000000000000000000000000s*\r"
[18:08:25][D][uart_debug:158]: >>> "QPIRI\r"
[18:08:25][D][uart_debug:158]: <<< "(NAKss\r"
[18:08:26][D][uart_debug:158]: >>> "QT\'\xFF\r"
[18:08:26][D][uart_debug:158]: <<< "(NAKss\r"
[18:08:27][D][uart_debug:158]: >>> "QMOD\r"`

syssi, 2023-04-14:
@adrusa I've pushed a small change: https://github.com/syssi/esphome-pipsolar/commit/58cf88f1c5746250b3ad873d46785468481cd681

Could you update your protocol test yaml, compile, flash and provide another log again? Thanks! :-)

adrusa, 2023-04-14:
> @adrusa I've pushed a small change: [58cf88f](https://github.com/syssi/esphome-pipsolar/commit/58cf88f1c5746250b3ad873d46785468481cd681)
> 
> Could you update your protocol test yaml, compile, flash and provide another log again? Thanks! :-)

`[20:09:51][D][uart_debug:158]: <<< "\x00"
[20:10:22][I][main:413]: Testing PI30/PI30MAX/PI30REVO/PI41 commands...
[20:10:22][I][main:416]: This is the set of commands supported by the pipsolar component!
[20:10:22][D][uart_debug:158]: >>> "QPI\xBE\xAC\r"
[20:10:22][D][uart_debug:158]: <<< "(PI34\xDA\x8F\r"
[20:10:23][D][uart_debug:158]: >>> "QDIq\e\r"
[20:10:23][D][uart_debug:158]: <<< "(00 2000 02 14.40 13.60 00 +00.0 0000\xFDq\r"
[20:10:24][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[20:10:24][D][uart_debug:158]: <<< "(NAKss\r"
[20:10:25][D][uart_debug:158]: >>> "QMN\xBBd\r"
[20:10:25][D][uart_debug:158]: <<< "(NAKss\r"
[20:10:26][D][uart_debug:158]: >>> "QMODI\xC1\r"
[20:10:26][D][uart_debug:158]: <<< "(NAKss\r"
[20:10:27][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[20:10:28][D][uart_debug:158]: <<< "(002.6 53.06 00.00 00.00 00.00 0000 +038 00.06 +070 0000 10000000\xE4\xDB\r"
[20:10:28][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[20:10:29][D][uart_debug:158]: <<< "(3000 48 20.0 13.85 13.85 02 01 +00.0 00 04 4 11.50 0\xED\xAE\r"
[20:10:29][D][uart_debug:158]: >>> "QPIWS\xB4\xDA\r"
[20:10:30][D][uart_debug:158]: <<< "(000000000000000000010000000000\xABc\r"
[20:10:30][D][uart_debug:158]: >>> "QT\'\xFF\r"
[20:10:30][D][uart_debug:158]: <<< "(NAKss\r"
[20:10:31][I][main:482]: Testing PI41 split phase / multiple strings commands...
[20:10:31][D][uart_debug:158]: >>> "QPIGS2h-\r"
[20:10:32][D][uart_debug:158]: <<< "(NAKss\r"
[20:10:32][D][uart_debug:158]: >>> "QP2GS0\x14\x05\r"
[20:10:33][D][uart_debug:158]: <<< "(NAKss\r"
[20:10:34][D][uart_debug:158]: >>> "QP2GS1\x04$\r"
[20:10:34][D][uart_debug:158]: <<< "(NAKss\r"
[20:10:35][I][main:506]: Testing unsupported PI18 commands...
[20:10:35][D][uart_debug:158]: >>> "^P005PIq\x8B\r"
[20:10:35][D][uart_debug:158]: <<< "(NAKss\r"
[20:10:36][D][uart_debug:158]: >>> "^P005GSX\x14\r"
[20:10:36][D][uart_debug:158]: <<< "(NAKss\r"`

syssi, 2023-04-14:
Much better! I will need some more days to do some research.

adrusa, 2023-04-14:
> Much better! I will need some more days to do some research.

Great. Thank you for your great work. I wish I got the skill to help😣

adrusa, 2023-12-27:
> Much better! I will need some more days to do some research.

I'm still expecting your update on this. I understand that you are quiet busy, but kindly find time to help. Thanks.

syssi, 2023-12-29:
I've checked the responses of your inverter carefully and they aren't compatible / similar to other devices supporting the QPIGS/QPIRI commands. The response description above seems to be correct but isn't comparable to the response of supported inverters.

adrusa, 2023-12-29:
> I've checked the responses of your inverter carefully and they aren't compatible / similar to other devices supporting the QPIGS/QPIRI commands. The response description above seems to be correct but isn't comparable to the response of supported inverters.

Thank you very much for your work. I worked around it by using template to extract what I need from the last_qpigs.

It has worked well for several months. I guess I will continue to use it like that.

Thank you again

closed 2023-12-30 by adrusa