syssi/esphome-pipsolar #65: EASUN ISOLAR SML III 5.5kW - Problem

ZVlog, 2023-03-29:
I tested all possible configurations including tests. I get a NOK response every time. Connections also checked several times. Does anyone have this model and can provide yaml configuration?

syssi, 2023-04-05:
Please double check your wiring. The incoming bytes (garbage: `\xFF\xFF\xFF\xFF\xFF\xFF\xFF`) looks like bad grounding (`GND`).

syssi, 2023-04-07:
Is the inverter compatible with the SmartESS wifi dongle?

BeckoPopov, 2023-04-08:
If you use esp32:
tx_pin: GPIO16
rx_pin: GPIO17

or ESP8266:
tx_pin: GPIO1
rx_pin: GPIO3

`# Enable logging
logger:
  baud_rate: 0`
I saw the same on pins GPIO4 and GPIO5

syssi, 2023-04-14:
@ZVlog Any updates? I would like to close the issue. Did you see @BeckoPopov recommendation?

Please try to attach the inverter to GPIO1/GPIO3 and disable the `logger` on GPIO1/GPIO3 by `baud_rate: 0`.

mpodda2566, 2023-08-22:
Hi, I,ve the same inverter and the same problem, I checked wires but it is impossible to connect it to hone assistant by esp8266 and converter rs232 ttl. I asked easun about rj45 connector with Rs232/485 pinout but nothing answer me. Can anyone help me?

kreamuhely, 2024-03-08:
The search SOLAR2MQTT_D1_MINI_0.3.3.bin on the github. But problem, because in the MQTT message the "space" character, and Mosquitto mqtt broker does't . :(

SirPushek, 2025-11-17:
I have a very similar problem with this inverter and an ESP8266 using pipsolar. After a long struggle I managed to get the D1 Mini to communicate with the inverter on GPIO4 and GPIO5 (and a NodeMCU v3 on GPIO13 and GPIO15), but neither of them wants to talk to the inverter over GPIO1 and GPIO3 ☹.
I tried setting the logger to baud_rate: 0 and hardware_uart: UART1, but it doesn’t change anything. If I use GPIO1 and GPIO3 I only get “sending polling command Q….” and “Timeout command to poll”.

So if you want to run pipsolar with an EASUN ISOLAR SML III inverter, do it using an ESP8266 D1 Mini on GPIO4 and GPIO5 + an RS232-TTL converter + a 12V→5V/3.3V step-down converter. You can take 12V from pin 4 of the RJ45. Tested, works.
My problem is that I really need to get this communication running on GPIO1 and GPIO3. Why? In the meantime I found out that you can reflash the original datalogger supplied with the inverter [Wi-Fi Plug Pro-05], which has an ESP07S on board. The reflashing process itself went smoothly, but again I got stuck with no communication through GPIO1 and GPIO3, which are hardwired in the datalogger and cannot be changed…
Does anyone know why the ESP8266 refuses to communicate with this inverter on GPIO1 and GPIO3?