syssi/esphome-pipsolar #66: pipsolar >keine Daten im HA

KHackmann, 2023-03-29:
Seit Tagen suche ich den Fehler meines RS232 auf TTL Konverters im Zusammenhang mit einem ESP32. 

Hardware steht aber es kommen keine Daten vom Inverter an:

Wo sollte ich den Fehler suchen:  

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.178.94 using esphome API
INFO Successfully connected to 192.168.178.94
[23:03:51][I][app:102]: ESPHome version 2023.3.2 compiled on Mar 29 2023, 23:02:06
[23:03:51][C][wifi:504]: WiFi:
[23:03:51][C][wifi:362]:   Local MAC: 94:3C:C6:37:BB:7C
[23:03:51][C][wifi:363]:   SSID: [redacted]
[23:03:51][C][wifi:364]:   IP Address: 192.168.178.94
[23:03:51][C][wifi:366]:   BSSID: [redacted]
[23:03:51][C][wifi:367]:   Hostname: 'pipsolar'
[23:03:51][C][wifi:369]:   Signal strength: -50 dB ▂▄▆█
[23:03:51][C][wifi:373]:   Channel: 5
[23:03:51][C][wifi:374]:   Subnet: 255.255.255.0
[23:03:51][C][wifi:375]:   Gateway: 192.168.178.1
[23:03:51][C][wifi:376]:   DNS1: 0.0.0.0
[23:03:51][C][wifi:377]:   DNS2: 0.0.0.0
[23:03:51][C][logger:293]: Logger:
[23:03:51][C][logger:294]:   Level: DEBUG
[23:03:51][C][logger:295]:   Log Baud Rate: 115200
[23:03:51][C][logger:296]:   Hardware UART: UART0
[23:03:52][C][uart.arduino_esp32:108]: UART Bus 1:
[23:03:52][C][uart.arduino_esp32:109]:   TX Pin: GPIO18
[23:03:52][C][uart.arduino_esp32:110]:   RX Pin: GPIO19
[23:03:52][C][uart.arduino_esp32:112]:   RX Buffer Size: 256
[23:03:52][C][uart.arduino_esp32:114]:   Baud Rate: 2400 baud
[23:03:52][C][uart.arduino_esp32:115]:   Data Bits: 8
[23:03:52][C][uart.arduino_esp32:116]:   Parity: NONE
[23:03:52][C][uart.arduino_esp32:117]:   Stop bits: 1
[23:03:52][C][pipsolar:860]: Pipsolar:
[23:03:52][C][pipsolar:861]: used commands:
[23:03:52][C][pipsolar:864]: QPIRI
[23:03:52][C][pipsolar:864]: QPIGS
[23:03:52][C][pipsolar:864]: QMOD
[23:03:52][C][captive_portal:088]: Captive Portal:
[23:03:52][C][mdns:108]: mDNS:
[23:03:52][C][mdns:109]:   Hostname: pipsolar
[23:03:52][C][ota:093]: Over-The-Air Updates:
[23:03:52][C][ota:094]:   Address: 192.168.178.94:3232
[23:03:52][C][ota:097]:   Using Password.
[23:03:52][C][api:138]: API Server:
[23:03:52][C][api:139]:   Address: 192.168.178.94:6053
[23:03:52][C][api:141]:   Using noise encryption: YES
[23:03:53][D][api:102]: Accepted 192.168.178.79
[23:03:53][D][api.connection:920]: Home Assistant 2023.3.6 (192.168.178.79): Connected successfully
[23:03:56][D][pipsolar:755]: timeout command to poll: QPIGS
[23:03:56][D][pipsolar:838]: Sending polling command : QMOD with length 4
[23:03:56][D][uart_debug:114]: >>> 51:4D:4F:44:49:C1:0D
[23:04:01][D][pipsolar:755]: timeout command to poll: QMOD
[23:04:01][D][pipsolar:838]: Sending polling command : QPIRI with length 5
[23:04:01][D][uart_debug:114]: >>> 51:50:49:52:49:F8:54:0D
[23:04:06][D][pipsolar:755]: timeout command to poll: QPIRI
[23:04:06][D][pipsolar:838]: Sending polling command : QPIGS with length 5
[23:04:06][D][uart_debug:114]: >>> 51:50:49:47:53:B7:A9:0D
[23:04:11][D][pipsolar:755]: timeout command to poll: QPIGS
[23:04:11][D][pipsolar:838]: Sending polling command : QMOD with length 4
[23:04:11][D][uart_debug:114]: >>> 51:4D:4F:44:49:C1:0D
[23:04:16][D][pipsolar:755]: timeout command to poll: QMOD
[23:04:16][D][pipsolar:838]: Sending polling command : QPIRI with length 5
[23:04:16][D][uart_debug:114]: >>> 51:50:49:52:49:F8:54:0D
[23:04:21][D][pipsolar:755]: timeout command to poll: QPIRI
[23:04:21][D][pipsolar:838]: Sending polling command : QPIGS with length 5
[23:04:21][D][uart_debug:114]: >>> 51:50:49:47:53:B7:A9:0D

syssi, 2023-03-30:
Please provide a photo of your wiring!

KHackmann, 2023-03-30:
VGVon meinem Smartphone gesendet

KHackmann, 2023-03-30:
> <image0.jpeg><image1.jpeg><image2.png>VGVon meinem Smartphone gesendet

VGVon meinem Smartphone gesendet

KHackmann, 2023-03-30:
[image]

KHackmann, 2023-03-30:
[image]

KHackmann, 2023-03-30:
Inverter 
Braun - 8 GND
Grün - 2 Rx
Weis/grün - 1. Tx 

[image]

KHackmann, 2023-03-30:
Thank you so much for Support 

Danke Danke

KHackmann, 2023-03-30:
Idee
Must I Turn Arround the RJ45 ? 
Is GRN ather side?

syssi, 2023-03-30:
Ah, die Belegung deines RJ45-Steckers ist nach 586A und ganz außen ist Grün und Grün-Weiss? Dann sollte es in der Tat passen. Kannst du mir die Beschriftung deines ESP-Boards erklaeren? Oben im YAML sieht man, dass der UART auf `GPIO18` und `GPIO19` konfiguriert ist. Stecken "Rot" und "Orange" wirklich an diesen GPIOs? Braun würde ich auf 3.3V stecken anstatt 5V, sofern es sich beim RS232-Wandler-Chip um einen MAX3232 handelt.

KHackmann, 2023-03-30:
Danke für Feedback, ich habe das RJ45 Kabel zum D-Sub Stecker gerade noch mal kontrolliert. Das ist 100% richtig. 
Jetzt verfolge ich deine anderen Ideen 
Auf dem Board steht ESP32 -WROOM32

Rot  = TXD am ESP32 
orange = RXD am ESP32

ok braun kann ich auf 3,3 V stecken aber ich dachte der MAX3232 benötigt 5V

KHackmann, 2023-03-30:
oder erkennst du hier im .yamel einen Fehler 

sphome:
  name: pipsolar
  friendly_name: Pipsolar

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "W4NxSxaHXc1Sspqcb2QHxhfDO7+jRpma81w+mVxlm6Y="

ota:
#  password: "762e76beee236bc6bcbaada7a490f557"
  password: "7c79824ec1913a6b8e33b5d181716b1b"

wifi:
  ssid: "FritzBoxWLAN7490"
  password: "xxxxxxxxxxxxxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Pipsolar Fallback Hotspot"
    password: "kLs17vNLMARN"

# feste IP
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.178.94
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.178.1
        # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

captive_portal:

uart:
    tx_pin: GPIO18
    rx_pin: GPIO19
    baud_rate: 2400
    debug:
      direction: BOTH

pipsolar:
    id: inverter0
  
sensor:
  platform: pipsolar
  pipsolar_id: inverter0
  grid_rating_voltage:
    id: inverter0_grid_rating_voltage
    name: inverter0_grid_rating_voltage
  grid_rating_current:
    id: inverter0_grid_rating_current
    name: inverter0_grid_rating_current
  ac_output_apparent_power:
    id: inverter0_ac_output_apparent_power
    name: inverter0_ac_output_apparent_power
  ac_output_active_power:
    id: inverter0_ac_output_active_power
    name: inverter0_ac_output_active_power
  battery_bulk_voltage:
    id: inverter0_battery_bulk_voltage
    name: inverter0_battery_bulk_voltage

binary_sensor:
  - platform: pipsolar
    pipsolar_id: inverter0
    configuration_status:
      id: inverter0_configurations_status
      name: inverter0_configurations_status
    scc_charging_status:
      id: inverter0_scc_charging_status
      name: inverter0_scc_charging_status

text_sensor:
  - platform: pipsolar
    pipsolar_id: inverter0
    device_mode:
      id: inverter0_device_mode
      name: inverter0_device_mode
    last_qpigs:
      id: inverter0_last_qpigs
      name: inverter0_last_qpigs
    last_qpiri:
      id: inverter0_last_qpiri
      name: inverter0_last_qpiri

output:
  - platform: pipsolar
    pipsolar_id: inverter0
    battery_recharge_voltage:
      id: inverter0_battery_recharge_voltage_out

KHackmann, 2023-03-30:
Es ist verrückt aber ich weiß nicht wo ich noch suchen soll?  rot und orange haben ich noch mal umgesteckt. Auch das brachte keinen Erfolg. 
Die Entität - Pipsolar - ist auch nicht mehr verfügbar. 

irgendwie ist hier der Wurm drin.

Hast du noch einen Tipp zur Fehlersuche? 

Danke

syssi, 2023-03-31:
In der YAML ist konfiguriert, dass die Komponente an GPIO18/19 sprechen und hoeren soll. Angesteckt ist der RS232-Wandler aktuell an GPIO1/3. Vergleiche: https://www.mischianti.org/wp-content/uploads/2021/07/ESP32-DEV-KIT-DevKitC-v4-pinout-mischianti.jpg

Kannst du RX/TX des Wandlers umstecken? Es ist wichtig, dass du nicht die YAML aenderst und GPIO1/3 nutzt. Hier laeuft naemlich schon der Logger und schreibt Logmeldungen, so dass man sie ueber den USB-to-TTL-Wandler abrufen/empfangen kann.

KHackmann, 2023-03-31:
leider noch kein Erfolg. 

Immer noch diese LOGS:  das Umklemmen hat nix genützt. 

[15:15:15][D][uart_debug:114]: >>> 51:50:49:52:49:F8:54:0D
[15:15:20][D][pipsolar:755]: timeout command to poll: QPIRI
[15:15:20][D][pipsolar:838]: Sending polling command : QPIGS with length 5
[15:15:20][D][uart_debug:114]: >>> 51:50:49:47:53:B7:A9:0D
[15:15:25][D][pipsolar:755]: timeout command to poll: QPIGS
[15:15:25][D][pipsolar:838]: Sending polling command : QMOD with length 4
[15:15:25][D][uart_debug:114]: >>> 51:4D:4F:44:49:C1:0D
[15:15:30][D][pipsolar:755]: timeout command to poll: QMOD
[15:15:30][D][pipsolar:838]: Sending polling command : QPIRI with length 5
[15:15:30][D][uart_debug:114]: >>> 51:50:49:52:49:F8:54:0D
[15:15:35][D][pipsolar:755]: timeout command to poll: QPIRI
[15:15:35][D][pipsolar:838]: Sending polling command : QPIGS with length 5
[15:15:35][D][uart_debug:114]: >>> 51:50:49:47:53:B7:A9:0D
[15:15:40][D][pipsolar:755]: timeout command to poll: QPIGS
[15:15:40][D][pipsolar:838]: Sending polling command : QMOD with length 4
[15:15:40][D][uart_debug:114]: >>> 51:4D:4F:44:49:C1:0D
[15:15:45][D][pipsolar:755]: timeout command to poll: QMOD
[15:15:45][D][pipsolar:838]: Sending polling command : QPIRI with length 5
[15:15:45][D][uart_debug:114]: >>> 51:50:49:52:49:F8:54:0D

KHackmann, 2023-03-31:
[image]

KHackmann, 2023-03-31:
[image]

KHackmann, 2023-04-01:
Bilder anbei 

Fällt euch noch was ein wo ich den Fehler finden sollte? 

Danke für Feedback.

KHackmann, 2023-04-03:
Fehler gefunden in der Verkabelung 

Danke nochmal

syssi, 2023-04-03:
Wo lag der Fehler letztendlich?

KHackmann, 2023-04-03:
RJ 45 / D-SUB  hatte keine richtige Verbindung zum Inverter. 
Neu gelötet !

closed 2023-04-03 by syssi