syssi/esphome-pipsolar #67: EASUN SMP5KW  pipsplar

visionix24, 2023-03-30:
Hello,

Inverter is a EASUN smp5K 48Volt

 This is my Protocol-test.yaml

 substitutions:
  name: pipsolar-protocol-test
  tx_pin: GPIO16
  rx_pin: GPIO17

esphome:
  name: ${name}
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.0.222
    gateway: 192.168.0.100
    subnet: 255.255.255.0
api:
ota:
   password: "fcdc829ac919fca24e13186523b1a4e5"

logger:

uart:
  id: uart0
  baud_rate: 2400
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH
    dummy_receiver: true
    after:
      delimiter: "\r"
    sequence:
      - lambda: UARTDebug::log_string(d"QPIGS\xB7\xA9\r"irection, bytes);

interval:
  - interval: 30s
    then:

      # PI41
      - uart.write: "QDIq\x1b\r"
      - delay: 2s
      - uart.write: "QPIRI\xF8\r"
      - delay: 2s
      - uart.write: "QPIGS\xB7\r" # = meine Einstellung. orginale Einstellung ist "QPIGS\xB7\xA9\r"
      - delay: 2s
      - uart.write: "QMODI\xC1\r"
      - delay: 2s
      - uart.write: "QFLAG\x98t\r"

 This is the log from Protocol-test.yaml

 command QDiq QPIRI and QPIGS is ok

[14:22:44][D][uart_debug:158]: >>> "QDIq\e\r"
[14:22:44][D][uart_debug:158]: <<< "(230.0 50.0 0030 42.0 54.0 56.4 46.0 60 1 0 2 0 0 0 0 0 1 1 0 0 1 0 54.0 0 1 000p\xC4\r"
[14:22:46][D][uart_debug:158]: >>> "QPIRI\xC3\xB8\r"
[14:22:46][D][uart_debug:158]: <<< "(230.0 22.6 230.0 50.0 22.6 5200 5200 48.0 45.0 46.0 55.0 54.0 2 02 070 1 2 3 9 00 0 0 54.0 0 1 000 \x93\r"
[14:22:48][D][uart_debug:158]: >>> "QPIGS\xC2\xB7\r"
[14:22:48][D][uart_debug:158]: <<< "(231.2 50.0 230.3 49.9 0299 0270 005 374 50.60 005 044 0033 03.4 182.9 00.00 00000 00010010 00 00 00640 010 0 01 0000\xF4/\r"
[14:22:50][D][uart_debug:158]: >>> "QMODI\xC3\x81\r"
[14:22:50][D][uart_debug:158]: <<< "(NAKss\r"
[14:22:52][D][uart_debug:158]: >>> "QFLAG\xC2\x98t\r"
[... 234 lines cut ...]
[14:29:54][D][pipsolar:838]: Sending polling command : QMOD with length 4
[14:29:54][D][pipsolar:772]: checking crc on incoming message
[14:29:54][D][pipsolar:775]: CRC OK
[14:29:54][D][pipsolar:471]: Decode QMOD
[14:29:54][D][text_sensor:067]: 'pipsolar device_mode': Sending state 'B'
[14:29:55][D][pipsolar:838]: Sending polling command : QFLAG with length 5
[14:29:55][D][pipsolar:772]: checking crc on incoming message
[14:29:55][D][pipsolar:775]: CRC OK
[14:29:55][D][pipsolar:479]: Decode QFLAG
[14:29:56][D][pipsolar:838]: Sending polling command : QPIRI with length 5
[14:29:56][D][pipsolar:772]: checking crc on incoming message
[14:29:56][D][pipsolar:775]: CRC OK
[14:29:56][D][pipsolar:429]: Decode QPIRI
[14:29:57][D][sensor:127]: 'pipsolar grid_rating_voltage': Sending state 230.00000 V with 1 decimals of accuracy
[14:29:57][D][sensor:127]: 'pipsolar grid_rating_current': Sending state 22.60000 A with 1 decimals of accuracy
[14:29:57][D][sensor:127]: 'pipsolar ac_output_rating_voltage': Sending state 230.00000 V with 1 decimals of accuracy
[14:29:57][D][sensor:127]: 'pipsolar ac_output_rating_frequency': Sending state 50.00000 Hz with 1 decimals of accuracy
[14:29:57][D][sensor:127]: 'pipsolar ac_output_rating_current': Sending state 22.60000 A with 1 decimals of accuracy
[14:29:57][D][sensor:127]: 'pipsolar ac_output_rating_apparent_power': Sending state 5200.00000 VA with 1 decimals of accuracy
[14:29:57][D][sensor:127]: 'pipsolar ac_output_rating_active_power': Sending state 5200.00000 W with 1 decimals of accuracy
[14:29:57][D][sensor:127]: 'pipsolar battery_rating_voltage': Sending state 48.00000 V with 1 decimals of accuracy
[14:29:58][D][sensor:127]: 'pipsolar output_source_priority': Sending state 2.00000  with 1 decimals of accuracy
[14:29:58][D][sensor:127]: 'pipsolar charger_source_priority': Sending state 3.00000  with 1 decimals of accuracy
[14:29:58][D][sensor:127]: 'pipsolar topology': Sending state 0.00000  with 1 decimals of accuracy
[14:29:58][D][sensor:127]: 'pipsolar output_mode': Sending state 0.00000  with 1 decimals of accuracy

visionix24, 2023-03-30:
Hallo, jetzt kommt diese FehlermeldungINFO 

Reading configuration esp32-example.yaml...
INFO Updating https://github.com/syssi/esphome-pipsolar.git@pip8048
Failed config

sensor.pipsolar: [source esp32-example.yaml:58]
  platform: pipsolar
  pipsolar_id: inverter0
  grid_rating_voltage: 
    name: pipsolar grid_rating_voltage
  grid_rating_current: 
    name: pipsolar grid_rating_current
  ac_output_rating_voltage: 
    name: pipsolar ac_output_rating_voltage
  ac_output_rating_frequency: 
    name: pipsolar ac_output_rating_frequency
  ac_output_rating_current: 
    name: pipsolar ac_output_rating_current
  ac_output_rating_apparent_power: 
    name: pipsolar ac_output_rating_apparent_power
  ac_output_rating_active_power: 
    name: pipsolar ac_output_rating_active_power
  battery_rating_voltage: 
    name: pipsolar battery_rating_voltage
  output_source_priority: 
    name: pipsolar output_source_priority
  charger_source_priority: 
    name: pipsolar charger_source_priority
  topology: 
    name: pipsolar topology
  output_mode: 
    name: pipsolar output_mode
  grid_voltage: 
    name: pipsolar grid_voltage
  grid_frequency: 
    name: pipsolar grid_frequency
  ac_output_voltage: 
    name: pipsolar ac_output_voltage
  ac_output_frequency: 
    name: pipsolar ac_output_frequency
  ac_output_apparent_power: 
    name: pipsolar ac_output_apparent_power
  ac_output_active_power: 
    name: pipsolar ac_output_active_power
  output_load_percent: 
    name: pipsolar output_load_percent
  bus_voltage: 
    name: pipsolar bus_voltage
  battery_voltage: 
    name: pipsolar battery_voltage
  battery_charging_current: 
    name: pipsolar battery_charging_current
  battery_capacity_percent: 
    name: pipsolar battery_capacity_percent
  inverter_heat_sink_temperature: 
    name: pipsolar inverter_heat_sink_temperature
  
  [pv_input_current] is an invalid option for [sensor.pipsolar]. Did you mean [pv2_input_current], [pv1_input_current], [grid_rating_current]?
  pv_input_current: 
    name: pipsolar pv_input_current_for_battery
  
  [pv_input_voltage] is an invalid option for [sensor.pipsolar]. Did you mean [pv2_input_voltage], [pv1_input_voltage], [input_voltage_range]?
  pv_input_voltage: 
    name: pipsolar pv_input_voltage
  battery_voltage_scc: 
    name: pipsolar battery_voltage_scc
  battery_discharge_current: 
    name: pipsolar battery_discharge_current
  battery_voltage_offset_for_fans_on: 
    name: pipsolar battery_voltage_offset_for_fans_on
  
  [pv_charging_power] is an invalid option for [sensor.pipsolar]. Did you mean [pv2_charging_power], [pv1_charging_power]?
  pv_charging_power: 
    name: pipsolar pv_charging_power

syssi, 2023-03-30:
Please use this YAML as starting point: https://github.com/syssi/esphome-pipsolar/blob/pip8048/esp32-example.yaml

Some sensor names has changed here.

visionix24, 2023-03-30:
Hallo, Danke für deine schnelle Hilfe. Es funktioniert. Super Projekt .

closed 2023-03-30 by visionix24