syssi/esphome-pipsolar #70: Solar Inverter communication with max 232 not work properly for all inverter

MajdAli1a, 2023-04-16:
Hi, I am working on a small project depending on Arduino, max 232, and solar inverter. 
I do the wiring and upload the code, the problem is I have two brands of solar inverters, one of them a commercial blue sun inverter, 
the second one is the Falcon inverter. 
So, I could send a message and get a response from the Falcon inverter, but the blue sun inverter did not respond to all messages.
I check the Voltage of TX, and RX of the Rs232 cable of the blue sun inverter and it seems correct as -13V for TX and  0V for RX same as the Falcon inverter. 
I went to my friend and got the same problem with his solar inverter, is the problem with the wiring, or does each solar inverter brand has a unique commutation protocol? 
[SolarInverterCode.zip](https://github.com/syssi/esphome-pipsolar/files/11241491/SolarInverterCode.zip)

syssi, 2023-04-16:
Why do you expect both inverters are using the same protocol?

MajdAli1a, 2023-04-16:
Hello @syssi, I think that because I did a lot of research on the internet and all results indicate that it uses the rs232 protocol. 
like this file:
[RS232 communication Protocol for InfiniSolar 3K.pdf](https://github.com/syssi/esphome-pipsolar/files/11243077/RS232.communication.Protocol.for.InfiniSolar.3K.pdf)

syssi, 2023-04-16:
Please try to use the PC application Watchpower on each inverter. If Watchpower is working you should be able to replicate the communication. Furthermore Watchpower has a debug log so you can have a look at the traffic here.

MajdAli1a, 2023-04-16:
Thank you my friend for replying.
I tried Watch Power when I plug the Falcon inverter with a USB cable, it appears in the Watch Power app but when I plug second inverter it does not appear on my PC.

syssi, 2023-04-16:
Sounds like the second inverter isn't supported by Watchpower because it uses another protocol.

MajdAli1a, 2023-04-16:
Today, I went to another friend who works in an electronics store, he give me permission to try on different brands of inverters, 
The inverter that works default protocol is SAKO, FALCON, and PowerLand, RENOVO, but the inverters that not works are MUST, BlueSun.

MajdAli1a, 2023-04-16:
is there any possible way to know what is the protocol that the inverter works on ?? 
I have the catalog and all the paper, but there is nothing about communication only the use of rs232.

MajdAli1a, 2023-04-16:
is there any PC application for Solar inverter that use different protocol?

syssi, 2023-04-16:
> Is there any difference if I use esp2866 instead of Arduino?

Some Arduinos are using a logic level of 5V. The ESP has a logic level of 3.3V. UART-TTL communication should be pretty similar.

syssi, 2023-04-16:
> is there any PC application for Solar inverter that use different protocol?

I cannot help here.

MajdAli1a, 2023-04-16:
thank you, my friend. 
I will update the state of the issue if I get any solution.