syssi/esphome-pipsolar #77: pipsolar:755]: timeout command to poll: QPIRI

Desef, 2023-06-08:
I'm trying to connect my Mecer (Axpert)  3kw inverter to HA using ESPHome pipsolar via RS232, but the communication just times out. I'm pretty sure that the RS233 to TTL converter is wired correctly, but I just can't get this to work. Note that ESPHome is on the latest version.

The attached log shows the result of the configuration that tests the protocol, but I am a bit lost at this point. Some assistance to get onto the right track would be appreciated.

[logs_mecer-inverter_logs.txt](https://github.com/syssi/esphome-pipsolar/files/11692246/logs_mecer-inverter_logs.txt)

Desef, 2023-06-09:
Not to worry - the dumb MAX3232 board is not through-plated. There are solder tabs on both sides of the board, but only 1 side is actually connected. (Maybe is is the dumb nut behind the soldering iron!!!)

closed 2023-06-09 by Desef

syssi, 2023-06-09:
Good job and have fun! :-)