syssi/esphome-pipsolar #78: Add PI17 to support MPI 12KW WP and MPI 15KW WP

JakobTewes, 2023-07-08:
Heyhey @syssi,

I´ve got a MPI 15KW WP and currently use the python module https://github.com/jblance/mpp-solar/wiki.
As I got bunch of ESPHome stuff running and only want to ingest stuff into Home Assistant, using your pipsolar integration would be a big leap for me forward.

MPI 12KW WP and MPI 15KW WP use protocol "PI17" and I currently extract info from the inverters via

´´´
# mpp-solar -p /dev/hidraw0 -n 'BPVInverter2' -T 'BPVInverter2' -P PI17 -o screen -c MOD#BATS#ET#GS#PS
´´´
The commands implemented for PI17
´´´
# mpp-solar -P PI17 -c
MOD       Device working mode inquiry -- queries the device working mode
BATS      Query battery setting -- queries battery setting
ET        Query total generated energy -- queries total generated energy from the Inverter
GS        Query general status -- queries general status
PS        Device Power Status -- queries power status
´´´
Do you see an option, to implement the protocol and commands with pipsolar?

Thanks and regards

Jakob

WagnerSystems, 2024-03-24:
Hello,
PI17 would also support my MPI 10k so +1 for support of PI17

Volker

JakobTewes, 2025-02-12:
Things unwinded and the protocol is nowadays [defined as "pi17m058"](https://github.com/jblance/mpp-solar/issues/485) as it has slight differences to the original PI17

JakobTewes, 2025-02-12:
But still interested obviously ;-)

JakobTewes, 2025-03-02:
Got the [esp8266-test-protocols.yaml](https://github.com/syssi/esphome-pipsolar/blob/main/tests/esp8266-test-protocols.yaml) fired up. Getting this output:
`INFO ESPHome 2025.2.1`
`[12:14:12][I][main:603]: This is the set of commands supported by the pipsolar component!`
`[12:14:12][D][uart_debug:158]: >>> "QPI\xBE\xAC\r"`
`[12:14:12][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:13][D][uart_debug:158]: >>> "QDIq\e\r"`
`[12:14:13][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:14][D][uart_debug:158]: >>> "QFLAG\x98t\r"`
`[12:14:14][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:15][D][uart_debug:158]: >>> "QMN\xBBd\r"`
`[12:14:16][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:16][D][uart_debug:158]: >>> "QMODI\xC1\r"`
`[12:14:17][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:17][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"`
`[12:14:18][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:18][D][uart_debug:158]: >>> "QPIRI\xF8T\r"`
`[12:14:19][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:19][D][uart_debug:158]: >>> "QPIWS\xB4\xDA\r"`
`[12:14:20][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:20][D][uart_debug:158]: >>> "QT\'\xFF\r"`
`[12:14:21][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:21][D][uart_debug:158]: >>> "QPGS0?\xDA\r"`
`[12:14:22][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:23][D][uart_debug:158]: >>> "QPGS1/\xFB\r"`
`[12:14:23][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:24][D][uart_debug:158]: >>> "QPGS2\x1F\x98\r"`
`[12:14:24][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:25][I][main:690]: Testing PI41 split phase / multiple strings commands...`
`[12:14:25][D][uart_debug:158]: >>> "QPIGS2h-\r"`
`[12:14:25][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:26][D][uart_debug:158]: >>> "QP2GS0\x14\x05\r"`
`[12:14:26][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:27][D][uart_debug:158]: >>> "QP2GS1\x04$\r"`
`[12:14:27][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:28][I][main:714]: Testing unsupported PI18 commands...`
`[12:14:28][D][uart_debug:158]: >>> "^P005PIq\x8B\r"`
`[12:14:28][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:29][D][uart_debug:158]: >>> "^P005GSX\x14\r"`
`[12:14:29][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:30][D][uart_debug:158]: >>> "^P006MOD\xDD\xBE\r"`
`[12:14:30][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:31][I][main:738]: Testing unsupported PI17 commands...`
`[12:14:31][D][uart_debug:158]: >>> "^P003PI\r"`
`[12:14:31][D][uart_debug:158]: <<< "^D00517\xCA\xEC\r"`
`[12:14:32][D][uart_debug:158]: >>> "^P004MOD\r"`
`[12:14:32][D][uart_debug:158]: <<< "^D00505\xD9\x9F\r"`
`[12:14:33][D][uart_debug:158]: >>> "^P005FLAG\r"`
`[12:14:33][D][uart_debug:158]: <<< "^D0441,1,1,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0\xE6\x17\r"`
`[12:14:34][I][main:762]: Testing unsupported PI16 commands...`
`[12:14:34][D][uart_debug:158]: >>> "QPI\r"`
`[12:14:34][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:35][D][uart_debug:158]: >>> "QMOD\r"`
`[12:14:35][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:36][D][uart_debug:158]: >>> "QPIGS\r"`
`[12:14:36][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:37][D][uart_debug:158]: >>> "QPIRI\r"`
`[12:14:37][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:38][D][uart_debug:158]: >>> "QMOD\r"`
`[12:14:38][I][main:796]: Testing unsupported Qx commands...`
`[12:14:38][D][uart_debug:158]: >>> "Q1\r"`
`[12:14:38][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:39][D][uart_debug:158]: >>> "Q2\r"`
`[12:14:39][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:40][D][uart_debug:158]: >>> "Q4\r"`
`[12:14:40][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:41][D][uart_debug:158]: >>> "Q6\r"`
`[12:14:41][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:42][I][main:827]: Done. Repeating...`

rene-dev, 2025-03-13:
Im currently working on pip17 support, for my mpi10k
https://github.com/esphome/esphome/compare/dev...rene-dev:esphome:pip17

JakobTewes, 2025-03-13:
@rene-dev, sounds great! Are you planning to merge this back? What would you say, how close are you to a mvp?

JakobTewes, 2025-03-16:
Heyhey @rene-dev,

can you give me a hand on how to test your dev version please?

Thanks and regards

Jakob

rene-dev, 2025-03-16:
before merging it back it needs to be cleaned up.
unfortunateley supporting both protocols is difficult due to the way the component is written,
so I will probably make a new one.
It is working now, inclunding energy meter info for feedin control, and bms support.
this is my config, comment in lines 8-10 to use my branch.
https://github.com/rene-dev/solar/blob/master/mpi2.yaml

JakobTewes, 2025-03-17:
great - got it working:

[image]

Best thing: Its close to realtime ;-).

Currently I´m (maybe same as you, if I got the repo right) the mpp-solar package running. 
Until now it can do more - some of the features tend to me critical stuff for me. Maybe you can add some of them? ;-)

These Commands I use:

`/mpp-solar -p /dev/hidrawINV1 -n 'BPVInverter1' -T 'BPVInverter1' -P pi17m058 -o hass_mqtt --mqttbroker 127.0.0.1 --mqttuser bpvinverter --mqttpass 1234 -c MOD#HECS#BATS#GS#PS#WS`

-> Providing:

```
Command: MOD - Device working mode inquiry
--------------------------------------------------------------------------------
Parameter    Value          Unit
working_mode Hybrid mode (Line mode, Grid mode)
--------------------------------------------------------------------------------

Command: HECS - Query energy control status
--------------------------------------------------------------------------------
Parameter                                              Value          Unit
solar_energy_distribution_priority                     Load-Battery-Grid
solar_charge_battery                                   enabled
ac_charge_battery                                      disabled
feed_power_to_utility                                  disabled
battery_discharge_to_loads_when_solar_input_normal     enabled
battery_discharge_to_loads_when_solar_input_loss       enabled
battery_discharge_to_feed_grid_when_solar_input_normal disabled
battery_discharge_to_feed_grid_when_solar_input_loss   disabled
--------------------------------------------------------------------------------

Command: BATS - Query battery setting
--------------------------------------------------------------------------------
Parameter                                                                           Value          Unit
battery_maximum_charge_current                                                      100.0          A
battery_constant_charge_voltage                                                     56.5           V
battery_floating_charge_voltage                                                     54.6           V
battery_stop_charger_current_level_in_floating_charging                             0.0            A
keep_charged_time_of_battery_catch_stopped_charging_current_level                   60             Minutes
battery_voltage_of_recover_to_charge_when_battery_stop_charger_in_floating_charging 53.0           V
battery_under_voltage                                                               47.5           V
battery_under_voltage_release                                                       52.6           V
battery_weak_voltage_in_hybrid_mode                                                 47.5           V
battery_weak_voltage_release_in_hybrid_mode                                         52.6           V
battery_type                                                                        Ordinary
ac_charger_keep_battery_voltage_function_enable/diable                              Disabled
ac_charger_keep_battery_voltage                                                     48.0           V
battery_temperature_sensor_compensation                                             0.0            mV
max._ac_charging_current                                                            1.0            A
battery_discharge_max_current_in_hybrid_mode                                        100            A
battery_under_soc                                                                   10             %
battery_under_back_soc                                                              20             %
battery_weak_soc_in_hybrid_mode                                                     20             %
battery_weak_back_soc_in_hybrid_mode                                                80             %
unknown                                                                             0
--------------------------------------------------------------------------------

Command: GS - Query general status
--------------------------------------------------------------------------------
Parameter                    Value          Unit
solar_input_voltage_1        0.0            V
solar_input_voltage_2        0.0            V
solar_input_current_1        0.0            A
solar_input_current_2        0.0            A
battery_voltage              52.7           V
battery_capacity             87             %
[... 56 lines cut ...]
battery_disconnected                    disabled
battery_over_voltage                    disabled
battery_low_in_hybrid_mode              disabled
grid_voltage_high_loss                  disabled
grid_voltage_low_loss                   disabled
grid_frequency_high_loss                disabled
grid_frequency_low_loss                 disabled
ac_input_long-time_average_voltage_over disabled
ac_input_voltage_loss                   disabled
ac_input_frequency_loss                 disabled
ac_input_island                         disabled
ac_input_phase_dislocation              disabled
over_temperature                        disabled
over_load                               disabled
emergency_power_off_active              disabled
ac_input_wave_loss                      disabled
unknown_value_in_response_22            1
--------------------------------------------------------------------------------
```

`mpp-solar -p /dev/hidrawINV1 -n 'BPVInverter1' -T 'BPVInverter1' -P pi17m058 -o hass_mqtt --mqttbroker 192.168.177.32 --mqttuser bpvinverter --mqttpass j9PejFVhzVXAckEVvKiKmZiB -c ET#ED`date +"%Y%m%d"`#EM`date +'%Y%m' -d 'last month'``

-> Providing daily, monthly, total generation

What do ya think? ;-)

JakobTewes, 2025-03-17:
Also are you aware of the possibility to power the ESP via RJ45? Benefits:
- Device is off, if inverter goes off (me doing this)
- Obviously - no external power source ;-)
- Got problems with USB after some inverter powercycles. Resulting in loss of hidraw device, that can only be cured via powercycling the raspi, i currently use for mpp-solar (disconnecting the 5V power pin on the USB plug did btw. not solve this)

JakobTewes, 2025-03-17:
> Yes, its planned to add them. Im getting the hourly, daily and monthly stats from the energy dashboard. Currently im working on a bug caused by not handling the ack properly.

Really cool! Thanks so much in advance ;-)

JakobTewes, 2025-03-17:
> before merging it back it needs to be cleaned up. unfortunateley supporting both protocols is difficult due to the way the component is written, so I will probably make a new one. It is working now, [...]

Maybe you can give me a hand regarding some elements in your config:

- What do you do with CAN? Communicating to the BMS most likely? Is it an original pylontech or are you just using the protocol?
- The obis_power_total is pushed out to control feedin? Did you spare the ModBus card and SDM630 with this?

rene-dev, 2025-03-17:
Yes, can is connected to pylontech battery. obis_power_total is feedin control, value is pushed from my energy meter.
My esp is powered my the inverter, it's on a custom pcb in the intelligent slot :)

MarkusBec, 2025-03-17:
I have developed the hardware for rene and me

It supports:
BMS modbus 
Meter Modbus
BMS CAN
100Mbit ethernet 

[image]

JakobTewes, 2025-03-17:
> I have developed the hardware for rene and me
> 
> It supports: BMS modbus Meter Modbus BMS CAN 100Mbit ethernet
> 
> [image]

Kudos...this is amazing!

BMS means the Pylontech right?
The BMS/Meter RJ45 is used as a Splitter? CAN + ModBus? 
One ModBus for BMS and Meter? 
Why CAN (not ModBus)?

Personally I got a way more lame scenario ;-)
- 2 x MPI 15K (currently only one used)
- ModBus Card connected to SDM630
- 2 x JKBMS with 16x LiFePo4 each
- EASun Powerwall 

Your stuff sounds great when i thinking of multiple downsides of my setup:
- Inverter and Batts not talking to each other
- EASun Powerwall mainly to satisfy German regulators (local electricity provider)
- Feed-in regulation super slow due to laggy ModBus implementation

JakobTewes, 2025-03-17:
> Kudos...this is amazing!
>
> BMS means the Pylontech right? The BMS/Meter RJ45 is used as a Splitter? CAN + ModBus? One ModBus for BMS and Meter? Why CAN (not ModBus)?
>
> Personally I got a way more lame [...]

ah....now I realized...the right board is the splitter for ModBus and CAN right?

MarkusBec, 2025-03-17:
for the pylontec can is better but there are batterys with only modbus.
there are 2 independent modbus and the left board is just a splitter.

My setup are 2x EAsun 10kw with pylontec battery and a PAC4200 powermeter

JakobTewes, 2025-03-17:
> for the pylontec can is better but there are batterys with only modbus. there are 2 independent modbus and the left board is just a splitter.
> 
> My setup are 2x EAsun 10kw with pylontec battery and a PAC4200 powermeter

JKMBS and EASun Batt (they are great for the buck btw) can do both...downside...I´m not educated enough to sniff their protocol and reverse engineer ;-)

And wtf...PAC4200....this thing is a spaceship ;-)

sarzyk, 2025-03-18:
Hi. Maybe someone can help me to connect V-TAC battery (the same like Westwood or Narada probably) via rs485. If I connect battery we with solar assistant via converter it works well.
I can't get communication battery with inverter.
I have Anern 10,2kW inverter.
If someone can help me I will be very grateful.
Thanks

JakobTewes, 2025-03-18:
Heyhey @sarzyk,

not sure, if this thread matches for your setup ;-).
Is the Anern inverter another MPP/EASun clone?

Did you successfully try the mpp-solar python library? Is the protocol pi17?

If your inverter speaks pi17 protocol, can you elaborate on „solar assistant via converter“? Is „converter“ some sort of a software translation? Maybe you just describe, on what you did with solar assistant in hard- and software ;-)

JakobTewes, 2025-03-20:
@MarkusBec or @rene-dev, do you have ideas (ideally config) on how to interact (get voltages, charging % etc) from Pylontech inverters via RS485? The Pylontech examples, I´ve seen so far all go with the RS232 ports, but unfortunately the EASun Batt has a config only RS232 ;-(

WagnerSystems, 2025-04-10:
@MarkusBec would you provide some information on the PinOut from the Inverter to the extension Card. Which Circuit Lines is which Signal?

[image]

MarkusBec, 2025-04-10:
[image]

[image]

hope this helps

JakobTewes, 2025-04-20:
Heyhey @rene-dev,

seen, that you´ve updated your mpi1.yaml and mpi2.yaml to no longer contain your custom pipsolar component.
Is this due to depreciation of your custom pipsolar component approach? 

Happy Easter, thanks and regards

Jakob

rene-dev, 2025-04-20:
Its still using my custom component, I have not made an effort to merge it upstream.