syssi/esphome-pipsolar #83: output_source_priority_battery switch not staying turned on

streetwiz, 2023-08-30:
Forgive me if i am not reporting this correctly.. this is the first time i have reported an issue.

Since upgrading to esphome 2023.8.2 the 3 switches for output source priority are not functioning as they were before
solar works as expected
utility worked as expected
Battery turns back off again

I have not changed the esphome code from before the upgrade 

substitutions:
  name: solar-sensor
switch:
  - platform: pipsolar
    pipsolar_id: inverter
    output_source_priority_battery:
      name: "${name} output_source_priority_battery"
    output_source_priority_utility:
      name: "${name} output_source_priority_utility"
    output_source_priority_solar:
      name: "${name} output_source_priority_solar"

the outputs i'm getting when i try and toggle the switch is as below

[12:57:59][D][switch:012]: 'solar-sensor output_source_priority_battery' Turning ON.
[12:57:59][D][pipsolar:857]: got command: POP02
[12:57:59][D][pipsolar:849]: Command queued successfully: POP02 with length 5 at position 1
[12:57:59][D][pipsolar:043]: response length for command OK
[12:57:59][D][pipsolar:773]: checking crc on incoming message
[12:57:59][D][pipsolar:776]: CRC OK
[12:57:59][D][pipsolar:049]: command not successful
[12:57:59][D][pipsolar:808]: Sending command from queue: POP02 with length 5

I welcome any input so that I can return my setup back to it's intended operation

Thanks

SeByDocKy, 2023-08-30:
If I remember well some inverter need POPxx and some POPxxx ....

streetwiz, 2023-08-31:
Hi, I'm not sure I understand what it is you are referring to..

Also, the hardware and uncompiled esphome code hasn't changed.. just upgraded to the new version of esphome. 
Plus, why would it only affect 1 of the 3 switches.. that's the bit that doesn't make any sense.. if it had broken all 3, I would accept it

Very Odd..

streetwiz, 2023-09-18:
Just to update and close this off (i hate loose ends.) . since the upgrade to 2023.8.3 the functionality has returned.. Thanks

closed 2023-09-18 by streetwiz