syssi/esphome-pipsolar #85: charger_source_priority switch

raf9998, 2023-08-31:
Are there any plans to add support for changing charger_source_priority value as is the case with output_source_priority_solar?

magnetus26, 2023-08-31:
I will love to see charger_source_priority dropdown as well
thanks
M.

ans-gw, 2023-10-03:
I ll love too

magnetus26, 2023-10-03:
syssi if you, you could tell me wich files are involve, i could try to butch the files using the inverter output priority select as a template and try with my inverter.
thanks.
M.

WallK, 2023-12-02:
But it already has is
Here's a piece of my config
[image]

hosflow, 2024-03-03:
> But it already has is Here's a piece of my config [...]

Im using this and esphome return error, any help?
```

select:
  - platform: pipsolar
    pipsolar_id: ${id}
    charger_source_priority:
      id: ${id}_charger_source_priority_select
      name: $(name) Charger Source Priority Select
      optionsmap:
        "Solar first": "PCP01"
        "Solar and utility": "PCP02"
        "Solar charging only": "PCP03"
      statusmap:
        "1": "Solar first"
        "2": "Solar and utility"
        "3": "Solar charging only"
```

ERROR
 **Platform not found: 'select.pipsolar'.**

WallK, 2024-03-03:
@hosflow I think you don't have this installed
It should be a part of ESPHome by default for a while, maybe you have a very old version?
Do sensors work, for example?

hosflow, 2024-03-04:
> @hosflow I think you don't have this installed It should be a part of ESPHome by default for a while, maybe you have a very old version? Do sensors work, for example?

Everything works from pipsolar, it only stops working if you add this configuration, I don't know if I have to install another one, I'm new to esphome and what I used was a docker container that I uploaded less than 10 days ago, version 2024.2.1, I even tried run a clean build files, but it gives the error: Platform not found: 'select.pipsolar'. But if you remove that part everything works.

substitutions:
  id: inverter0

select:
  - platform: pipsolar
    pipsolar_id: ${id}
    charger_source_priority:
      id: ${id}_charger_source_priority_select
      name: $(name) Charger Source Priority Select
      optionsmap:
        "Solar first": "PCP01"
        "Solar and utility": "PCP02"
        "Solar charging only": "PCP03"
      statusmap:
        "1": "Solar first"
        "2": "Solar and utility"
        "3": "Solar charging only"

WallK, 2024-03-05:
Do you use this repo?
You'll need
```external_components:
  - source: github://syssi/esphome-pipsolar```
in your config
Or specify a branch with "@" after

hosflow, 2024-03-05:
> Do you use this repo? You'll need
> 
> ```
> external_components:
>   - source: github://syssi/esphome-pipsolar
> ```
> 
> in your config Or specify a branch with "@" after

```
    Could not find components folder for source. Please check the source contains a 'components' or 'esphome/components' folder.
    source: github://syssi/esphome-pipsolar@main
    refresh: 0s
```

I tried with @main and about 5 different branches and it didn't work, it gives the error above or others with components not found. I would like to take advantage of this and thank you for taking the time to try to help me. Thank you very much

syssi, 2024-03-05:
@hosflow Could you provide your complete YAML? :-)

hosflow, 2024-03-05:
> @hosflow Could you provide your complete YAML? :-)

Of course, follow below

```
substitutions:
  name: pipsolar
  tx_pin: GPIO1
  rx_pin: GPIO3
  id: inverter0

esphome:
  name: ${name}
  
esp8266:
  board: esp01_1m

external_components:
  - source: github://syssi/esphome-pipsolar@main
    refresh: 0s

logger:
  baud_rate: 0
  level: INFO

api:
  encryption:
    key: "PInurfsFYyPLd3S4Ek2/vKdMvfjeWr/4coB2h5GD7Dc="
ota:
  password: "8f2c8da7ac6d03005c844371073dd7dc"
 
wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
  - ssid: LIVE TIM_5201_2G
    password: dazegT53
  power_save_mode: none
  manual_ip:
    static_ip: 192.168.177.121
    gateway: 192.168.177.1
    subnet: 255.255.255.0

  ap:
    ssid: "${name} Hotspot"
    password: "U1SFCinhg9jy"

# Warning:
#
# If you configure a lot of the possible sensors etc. it could be that you run
# out of memory (on esp8266). If you configure nearly all sensors etc. you run
# in a stack-size issue. In this case you have to increase stack size!
#
#  https://github.com/esphome/issues/issues/855

uart:
  id: uart_0
  baud_rate: 2400
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\r"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

pipsolar:
  uart_id: uart_0
[... 176 lines cut ...]
      name: $(name) Charger Source Priority Select
      optionsmap:
        "Solar first": "PCP01"
        "Solar and utility": "PCP02"
        "Solar charging only": "PCP03"
      statusmap:
        "1": "Solar first"
        "2": "Solar and utility"
        "3": "Solar charging only"
# select:
#   - platform: pipsolar
#     pipsolar_id: ${id}
#     output_source_priority:
#       id: ${id}_output_source_priority_select
#       name: ${name} Output Source Priority Select
#       optionsmap:
#         "Utility first": "POP00"
#         "Solar only": "POP01"
#         "Solar Battery Utility": "POP02"
#       statusmap:
#         "0": "Utility first"
#         "1": "Solar only"
#         "2": "Solar Battery Utility"

```

syssi, 2024-03-06:
I've changed the branch to `pip8084` and made some minor changes because of the updated code base. Please give it a try:

```
substitutions:
  name: pipsolar
  tx_pin: GPIO1
  rx_pin: GPIO3
  id: inverter0

esphome:
  name: ${name}
  
esp8266:
  board: esp01_1m

external_components:
  - source: github://syssi/esphome-pipsolar@pip8048
    refresh: 0s

logger:
  baud_rate: 0
  level: INFO

api:
  encryption:
    key: "PInurfsFYyPLd3S4Ek2/vKdMvfjeWr/4coB2h5GD7Dc="
ota:
  password: "8f2c8da7ac6d03005c844371073dd7dc"
 
wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
  - ssid: LIVE TIM_5201_2G
    password: dazegT53
  power_save_mode: none
  manual_ip:
    static_ip: 192.168.177.121
    gateway: 192.168.177.1
    subnet: 255.255.255.0

  ap:
    ssid: "${name} Hotspot"
    password: "U1SFCinhg9jy"

# Warning:
#
# If you configure a lot of the possible sensors etc. it could be that you run
# out of memory (on esp8266). If you configure nearly all sensors etc. you run
# in a stack-size issue. In this case you have to increase stack size!
#
#  https://github.com/esphome/issues/issues/855

uart:
  id: uart_0
  baud_rate: 2400
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\r"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

pipsolar:
  uart_id: uart_0
  id: inverter0

[... 173 lines cut ...]
      id: ${id}_charger_source_priority_select
      name: $(name) Charger Source Priority Select
      optionsmap:
        "Solar first": "PCP01"
        "Solar and utility": "PCP02"
        "Solar charging only": "PCP03"
      statusmap:
        "1": "Solar first"
        "2": "Solar and utility"
        "3": "Solar charging only"
# select:
#   - platform: pipsolar
#     pipsolar_id: ${id}
#     output_source_priority:
#       id: ${id}_output_source_priority_select
#       name: ${name} Output Source Priority Select
#       optionsmap:
#         "Utility first": "POP00"
#         "Solar only": "POP01"
#         "Solar Battery Utility": "POP02"
#       statusmap:
#         "0": "Utility first"
#         "1": "Solar only"
#         "2": "Solar Battery Utility"
```

hosflow, 2024-03-06:
> I've changed the branch to `pip8084` and made some minor changes because of the updated code base. Please give it a try:
>
> ```
> substitutions:
>   name: pipsolar
>   tx_pin: GPIO1
>   rx_pin: GPIO3
>   id: [...]

Many thanks, all working now, more one question if i want do set current_max_ac_charging_current i can create similar bellow? Or all options is always mapped to 1,2,3... list?

```
select:
  - platform: pipsolar
    pipsolar_id: ${id}
    current_max_ac_charging_current:
      id: ${id}_current_max_ac_charging_current
      name: $(name) Charging AC max current
      optionsmap:
        "10A": "10"
        "20A": "20"
        "30A": "30"
      statusmap:
        "10": "10A"
        "20": "20A"
        "30": "20A"
```

syssi, 2024-03-07:
@hosflow Could you create a new issue? Your new question is off-topic.

unknown1818, 2024-03-23:
Im trying use:

`external_components:
  - source: github://syssi/esphome-pipsolar@pip8048
    refresh: 0s`

But at compiling i got error:

`INFO ESPHome 2024.3.0
INFO Reading configuration /config/test2.yaml...
INFO Updating https://github.com/syssi/esphome-pipsolar.git@pip8048
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Europe/Warsaw'
INFO Generating C++ source...
INFO Compiling app...
Processing test (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/test/src/esphome/components/pipsolar/pipsolar.cpp.o
Compiling .pioenvs/test/src/esphome/components/pipsolar/text_sensor/pipsolar_textsensor.cpp.o
Compiling .pioenvs/test/src/esphome/components/select/select_call.cpp.o
Compiling .pioenvs/test/src/esphome/components/sensor/automation.cpp.o
In file included from src/esphome/components/pipsolar/pipsolar.cpp:1:
src/esphome/components/pipsolar/pipsolar.h:7:10: fatal error: esphome/components/pipsolar/select/pipsolar_select.h: No such file or directory
 #include "esphome/components/pipsolar/select/pipsolar_select.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/test/src/esphome/components/pipsolar/pipsolar.cpp.o] Error 1
In file included from src/esphome/components/pipsolar/text_sensor/pipsolar_textsensor.h:3,
                 from src/esphome/components/pipsolar/text_sensor/pipsolar_textsensor.cpp:1:
src/esphome/components/pipsolar/text_sensor/../pipsolar.h:7:10: fatal error: esphome/components/pipsolar/select/pipsolar_select.h: No such file or directory
 #include "esphome/components/pipsolar/select/pipsolar_select.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/test/src/esphome/components/pipsolar/text_sensor/pipsolar_textsensor.cpp.o] Error 1
========================== [FAILED] Took 3.51 seconds ==========================`

....

ouinouin, 2024-03-27:
just a bit of help, to have the syntax validated on the esphome dashboard, you shall tell which external component you retrieve on github apparently otherwise the editor will throw a lot of errors that are not real.
@syssi  : i noticed that only the 8048 branch has a folder components/pipsolar. is the 8048   intended to be the future main branch ? 

external_components:
  - source: github://syssi/esphome-pipsolar@pip8048
    components : pipsolar

thanks for this useful component.

sabinescu2006, 2024-09-29:
Hello!  I can't activate external components for change charge source priority
I have :
external_components:
 - source: 
    type: git
    url: https://github.com/syssi/esphome-pipsolar#pip8048-git

select:
  - platform: pipsolar
    pipsolar_id: ${id}
    charger_source_priority:
      id: ${id}_charger_source_priority_select
      name: $(name) Charger Source Priority Select
      optionsmap:
        "Solar first": "PCP01"
        "Solar and utility": "PCP02"
        "Solar charging only": "PCP03"
      statusmap:
        "1": "Solar first"
        "2": "Solar and utility"
        "3": "Solar charging only"

and error : 
INFO ESPHome 2024.9.1
INFO Reading configuration /config/esphome/inverter.yaml...
WARNING Found '${id}' (see select->0->pipsolar_id) which looks like a substitution, but 'id' was not declared
WARNING Found '${id}_charger_source_priority_select' (see select->0->charger_source_priority->id) which looks like a substitution, but 'id' was not declared
WARNING Found '${id}' (see select->0->pipsolar_id) which looks like a substitution, but 'id' was not declared
WARNING Found '${id}_charger_source_priority_select' (see select->0->charger_source_priority->id) which looks like a substitution, but 'id' was not declared
INFO Cloning https://github.com/syssi/esphome-pipsolar#pip8048-git@None
Failed config

external_components: [source /config/esphome/inverter.yaml:25]
  - 
    https://github.com/syssi/esphome-pipsolar#pip8048-git/info/refs not valid: is this a git repository?.
    source: 
      type: git
      url: https://github.com/syssi/esphome-pipsolar#pip8048-git

Thank You for help!

syssi, 2024-09-29:
The `external_components` section doesn't look good. Please use:

```
external_components:
  - source: github://syssi/esphome-pipsolar@pip8048
    refresh: 0s
```

sabinescu2006, 2024-09-29:
Thank you for your answer!
external_components:
 - source: github://syssi/esphome-pipsolar@pip8048
   refresh: 0s
Now I have many errors !: 
INFO ESPHome 2024.9.1
INFO Reading configuration /config/esphome/inverter.yaml...
WARNING Found '${id}' (see select->0->pipsolar_id) which looks like a substitution, but 'id' was not declared
WARNING Found '${id}_charger_source_priority_select' (see select->0->charger_source_priority->id) which looks like a substitution, but 'id' was not declared
WARNING Found '${id}' (see select->0->pipsolar_id) which looks like a substitution, but 'id' was not declared
WARNING Found '${id}_charger_source_priority_select' (see select->0->charger_source_priority->id) which looks like a substitution, but 'id' was not declared
INFO Updating https://github.com/syssi/esphome-pipsolar.git@pip8048
Failed config

sensor.pipsolar: [source /config/esphome/inverter.yaml:74]
  platform: pipsolar
  pipsolar_id: inverter0
  grid_rating_voltage: 
    name: grid_rating_voltage
  grid_rating_current: 
    name: grid_rating_current
  ac_output_rating_voltage: 
    name: ac_output_rating_voltage
  ac_output_rating_frequency: 
    name: ac_output_rating_frequency
  ac_output_rating_current: 
    name: ac_output_rating_current
  ac_output_rating_apparent_power: 
    name: pipsolar ac_output_rating_apparent_power
  ac_output_rating_active_power: 
    name: pipsolar ac_output_rating_active_power
  battery_rating_voltage: 
    name: pipsolar battery_rating_voltage
  battery_recharge_voltage: 
    name: pipsolar battery_recharge_voltage
  battery_under_voltage: 
    name: pipsolar battery_under_voltage
  battery_bulk_voltage: 
    name: pipsolar battery_bulk_voltage
  battery_float_voltage: 
    name: pipsolar battery_float_voltage
  battery_type: 
    name: pipsolar battery_type
  current_max_ac_charging_current: 
    name: pipsolar current_max_ac_charging_current
  current_max_charging_current: 
    name: pipsolar current_max_charging_current
  input_voltage_range: 
    name: pipsolar input_voltage_range
  output_source_priority: 
    name: pipsolar output_source_priority
  charger_source_priority: 
    name: charger_source_priority
  parallel_max_num: 
    name: pipsolar parallel_max_num
  machine_type: 
    name: pipsolar machine_type
  topology: 
    name: pipsolar topology
  output_mode: 
    name: output_mode
  battery_redischarge_voltage: 
    name: pipsolar battery_redischarge_voltage
  pv_ok_condition_for_parallel: 
    name: pipsolar pv_ok_condition_for_parallel
  pv_power_balance: 
    name: pipsolar pv_power_balance
  grid_voltage: 
    name: pipsolar grid_voltage
  grid_frequency: 
    name: pipsolar grid_frequency
  ac_output_voltage: 
    name: pipsolar ac_output_voltage
  ac_output_frequency: 
    name: pipsolar ac_output_frequency
  ac_output_apparent_power: 
    name: pipsolar ac_output_apparent_power
  ac_output_active_power: 
    name: pipsolar ac_output_active_power
  output_load_percent: 
    name: pipsolar output_load_percent
  bus_voltage: 
    name: pipsolar bus_voltage
  battery_voltage: 
    name: pipsolar battery_voltage
  battery_charging_current: 
    name: pipsolar battery_charging_current
  battery_capacity_percent: 
    name: pipsolar battery_capacity_percent
  inverter_heat_sink_temperature: 
    name: pipsolar inverter_heat_sink_temperature
  
  [pv_input_current_for_battery] is an invalid option for [sensor.pipsolar]. Did you mean [pv2_input_current], [pv1_input_current]?
  pv_input_current_for_battery: 
    name: pipsolar pv_input_current_for_battery
  
  [pv_input_voltage] is an invalid option for [sensor.pipsolar]. Did you mean [pv2_input_voltage], [pv1_input_voltage], [input_voltage_range]?
  pv_input_voltage: 
    name: pipsolar pv_input_voltage
  battery_voltage_scc: 
    name: pipsolar battery_voltage_scc
  battery_discharge_current: 
    name: pipsolar battery_discharge_current
  battery_voltage_offset_for_fans_on: 
    name: pipsolar battery_voltage_offset_for_fans_on
  
  [pv_charging_power] is an invalid option for [sensor.pipsolar]. Did you mean [pv2_charging_power], [pv1_charging_power]?
  pv_charging_power: 
    name: pipsolar pv_charging_power

syssi, 2024-09-29:
Just changing the source of the external component isn't enough because a lot of sensor names has changed etc. Do you use a ESP32 or ESP8266? You should pick one of the YAML examples of the feature branch to quick start.

sabinescu2006, 2024-09-29:
Used ESP32!
This is my code : 
    # Warning:
#
# If you configure a lot of the possible sensors etc. it could be that you run
# out of memory (on esp8266). If you configure nearly all sensors etc. you run
# in a stack-size issue. In this case you have to increase stack size!
#
#  https://github.com/esphome/issues/issues/855

substitutions:
  name: pipsolar
  tx_pin: GPIO17
  rx_pin: GPIO16
  

esphome:
  name: must-inverter
  friendly_name: Must Inverter

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  baud_rate: 0
  level: DEBUG
  
external_components:
  - source: github://syssi/esphome-pipsolar@pip8048
    refresh: 1d

# Enable Home Assistant API
api:
  encryption:
    key: "BaJg7la45ZZNbPNXdKp7hBPxuK4BhAbR/L//CWuyTHQ="

ota:
  - platform: esphome
    password: "04799bd8631c23c78debb4abf46fc834"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Must-Inverter Fallback Hotspot"
    password: "IEct2iEKypgY"

uart:
  - id: uart_0
    baud_rate: 2400
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
      direction: BOTH
      dummy_receiver: false
      after:
        delimiter: "\r"
      sequence:
        - lambda: UARTDebug::log_string(direction, bytes);

pipsolar:
  uart_id: uart_0
  id: inverter0

sensor:
[... 181 lines cut ...]
    charger_source_priority:
      id: ${id}_charger_source_priority_select
      name: $(name) Charger Source Priority Select
      optionsmap:
        "Solar first": "PCP01"
        "Solar and utility": "PCP02"
        "Solar charging only": "PCP03"
      statusmap:
        "1": "Solar first"
        "2": "Solar and utility"
        "3": "Solar charging only"
# select:
#   - platform: pipsolar
#     pipsolar_id: ${id}
#     output_source_priority:
#       id: ${id}_output_source_priority_select
#       name: ${name} Output Source Priority Select
#       optionsmap:
#         "Utility first": "POP00"
#         "Solar only": "POP01"
#         "Solar Battery Utility": "POP02"
#       statusmap:
#         "0": "Utility first"
#         "1": "Solar only"
#         "2": "Solar Battery Utility"

syssi, 2024-09-29:
This is the examle configuration you should start from: https://github.com/syssi/esphome-pipsolar/blob/pip8048/esp32-example.yaml

sabinescu2006, 2024-09-29:
Thank You ! It works with this example! Thank You!

Alarico77, 2024-09-30:
Hi, I entered the yaml code for the esp8266-example.yaml, but it gives me this error, as if it can't find the 8048...

INFO ESPHome 2024.9.1
INFO Reading configuration /config/esphome/voltronic-axpert.yaml...
INFO Updating https://github.com/syssi/esphome-pipsolar.git@pip8084
Failed config

external_components: [source /config/esphome/voltronic-axpert.yaml:23]
  - 
    couldn't find remote ref pip8084.
    source: github://syssi/esphome-pipsolar@pip8084
    refresh: 0s

gptech2444, 2024-10-04:
Hi, I'm also trying to get my Sumry 10.2kw inverter to change from "SUB" to "SBU" mode but I also get the "Could not find components folder for source. Please check the source contains a 'components' or 'esphome/components' folder."
I'm using the ESP32 example yaml with Select section added.
I've tried @main and @pip8048 but neither work. Any ideas?
Thanks.

syssi, 2024-10-04:
@gptech2444 Please create a new issue and provide your complete YAML configuration.

gptech2444, 2024-10-04:
> @gptech2444 Please create a new issue and provide your complete YAML configuration.

I was able to get it to work. I didn't realise I needed to use MQTT and then when I did there was some formatting issues in my secrets.yaml
Thanks @syssi this is such a useful resource.
 
Here's my YAML in case it helps anyoe else.

# Warning:
#
# If you configure a lot of the possible sensors etc. it could be that you run
# out of memory (on esp8266). If you configure nearly all sensors etc. you run
# in a stack-size issue. In this case you have to increase stack size!
#
#  https://github.com/esphome/issues/issues/855

substitutions:
  name: sumry10kw
  tx_pin: GPIO17
  rx_pin: GPIO16
  id: inverter0

esphome:
  name: ${name}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-pipsolar@pip8084"
    version: 1.0.0

esp32:
  board: upesy_wroom
  framework:
    type: esp-idf

external_components:
  - source: github://syssi/esphome-pipsolar@pip8048
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome

logger:
#  baud_rate: 0

mqtt:
  broker: !secret mqtt_broker
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client

uart:
  - id: uart_0
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    baud_rate: 2400

pipsolar:
  uart_id: uart_0
  id: inverter0

sensor:
  - platform: pipsolar
    pipsolar_id: inverter0
    # QPIRI
#    grid_rating_voltage:
#      name: "${name} grid_rating_voltage"
#    grid_rating_current:
[... 265 lines cut ...]
#        "10A": "MCHGC010"
#        "20A": "MCHGC020"
#        "30A": "MCHGC030"
#        "40A": "MCHGC040"
#        "50A": "MCHGC050"
#        "60A": "MCHGC060"
#        "70A": "MCHGC070"
#        "80A": "MCHGC080"
#        "90A": "MCHGC090"
#        "100A": "MCHGC100"
#        "110A": "MCHGC110"
#        "120A": "MCHGC120"
#      statusmap:
#        "10": "10A"
#        "20": "20A"
#        "30": "30A"
#        "40": "40A"
#        "50": "50A"
#        "60": "60A"
#        "70": "70A"
#        "80": "80A"
#        "90": "90A"
#        "100": "100A"
#        "110": "110A"
#        "120": "120A"