syssi/esphome-pipsolar #87: QPIGS CRC NOK    on  AXPERT VM IV Twin 4KVA

ZdzislawC, 2023-09-09:
Hello, I can not read QPIGS data from my Voltronic Axpert VM IV Twin 4KVA Inverter. Other parameters are read correctly.
Here is example of logs:

[16:35:26][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[16:35:26][D][pipsolar:837]: Sending polling command : QPIGS with length 5
[16:35:27][D][uart_debug:158]: <<< "(230.0 49.9 230.0 49.9 0805 0751 020 448 29.00 000 100 0037 02.3 271.1 00.00 00007 00010000 00 00 00647 011 0 00 0000\x00\xE7\r"
[16:35:27][D][pipsolar:773]: checking crc on incoming message
[16:35:27][D][pipsolar:782]: CRC NOK expected: DA 8A but got: 0 E7
[16:35:27][VV][scheduler:225]: Running interval 'update' with interval=1000 last_execution=100553597 (now=100554597)

There was no problem with QPIGS on my old 'burned' EAsun inverter :-(

I tried to add lines:

external_components:
  - source: github://syssi/esphome-pipsolar@2424mse1
    refresh: 0s

but compiler errors occur.  Please help.

raivonigul, 2023-09-30:
Hello. I have same issue with my invertes. After investigation I find the issue. The inverter protocol differs from expected. My Inverter protocol is descriebed in PI30_HS_MS_MSX_RS232_Protocol_20140822_after_current_upgrade.pdf (last reference in readme.md)
QPIRI
Inverter output:
(230.0 23.9 230.0 50.0 23.9 5500 5500 48.0 49.0 48.0 57.9 57.9 2 02 060 1 2 3 1 01 0 0 00.0 0 1\x9EQ\r"
Expected:
(BBB.B CC.C DDD.D EE.E FF.F HHHH IIII JJ.J KK.K JJ.J KK.K LL.L O PP Q0
O P Q R SS T U VV.V W X<CRC><cr>
current upgraded
(BBB.B CC.C DDD.D EE.E FF.F HHHH IIII JJ.J KK.K JJ.J KK.K LL.L O PP QQ0
O P Q R SS T U VV.V W X<CRC><cr>
Same for QPIGS
Is there possible to configure version of used communication protocol?

raivonigul, 2023-10-12:
I do not know why, but it start working for me, I reinstalled ESP32 device and it starts work.
[17:04:00][D][pipsolar:838]: Sending polling command : QPIGS with length 5
[17:04:00][D][pipsolar:772]: checking crc on incoming message
[17:04:00][D][pipsolar:775]: CRC OK
[17:04:00][D][pipsolar:448]: Decode QPIGS
[17:04:00][D][sensor:094]: 'pipsolar grid_voltage': Sending state 240.89999 V with 1 decimals of accuracy
[17:04:00][D][sensor:094]: 'pipsolar grid_frequency': Sending state 50.00000 Hz with 1 decimals of accuracy
[17:04:00][D][sensor:094]: 'pipsolar ac_output_voltage': Sending state 230.10001 V with 1 decimals of accuracy
[17:04:00][D][sensor:094]: 'pipsolar ac_output_frequency': Sending state 50.00000 Hz with 1 decimals of accuracy
[17:04:00][D][sensor:094]: 'pipsolar ac_output_apparent_power': Sending state 322.00000 VA with 1 decimals of accuracy
[17:04:00][D][sensor:094]: 'pipsolar ac_output_active_power': Sending state 265.00000 W with 1 decimals of accuracy
[17:04:00][D][sensor:094]: 'pipsolar output_load_percent': Sending state 5.00000 % with 1 decimals of accuracy
[17:04:00][D][sensor:094]: 'pipsolar bus_voltage': Sending state 394.00000 V with 1 decimals of accuracy
[17:04:00][D][sensor:094]: 'pipsolar battery_voltage': Sending state 53.00000 V with 1 decimals of accuracy
[17:04:00][D][sensor:094]: 'pipsolar battery_charging_current': Sending state 0.00000 A with 1 decimals of accuracy
[17:04:00][D][sensor:094]: 'pipsolar battery_capacity_percent': Sending state 48.00000 % with 1 decimals of accuracy
[17:04:00][D][sensor:094]: 'pipsolar inverter_heat_sink_temperature': Sending state 56.00000 °C with 1 decimals of accuracy
[17:04:00][D][sensor:094]: 'pipsolar pv_input_current_for_battery': Sending state 0.00000 A with 1 decimals of accuracy
[17:04:00][D][sensor:094]: 'pipsolar pv_input_voltage': Sending state 120.10000 V with 1 decimals of accuracy
[17:04:00][D][sensor:094]: 'pipsolar battery_voltage_scc': Sending state 0.00000 V with 1 decimals of accuracy
[17:04:00][D][sensor:094]: 'pipsolar battery_discharge_current': Sending state 7.00000 A with 1 decimals of accuracy
[17:04:00][D][sensor:094]: 'pipsolar battery_voltage_offset_for_fans_on': Sending state 0.00000 V with 1 decimals of accuracy
[17:04:00][D][sensor:094]: 'pipsolar pv_charging_power': Sending state 0.00000 W with 1 decimals of accuracy

ZdzislawC, 2023-10-25:
Hi raivinigul, thanks for your information, it was an inspiration for me. I have changed the code to pip8048  and now everything is OK.

closed 2023-10-25 by ZdzislawC