syssi/esphome-pipsolar #88: Mpi10k mppsolar how to do

Raydox187, 2023-09-10:
Hey i am Raymond i am absolutely new in homeassistant (os) and esphome ....and i Test and search since 14 days to Run MPI 10 k inverter from mppsolar ....and nothing works .....i testen Ned Kelly Bit Not works ( because after long Test IT IS Not foor OS ) than i Test pipsolar for ESP No working for MPI 10 k ) than Test smg-ii from Sissy and i think i am zu stupid to Install or to silly for wiring ) now i See for mpi10k is  pi 18 and modbus ..... I think i start 😭..... WHO can helps me please .... This evening i will Test syssi , Wish me luck that this works 😭

Help 😁😗😘

unknown1818, 2024-03-23:
Run debug config and post here log from uart

JakobTewes, 2025-03-02:
Thanks @unknown1818, took me quite some time to get back to this. Hope, you´re still willing to help ;-).
Got the [esp8266-test-protocols.yaml](https://github.com/syssi/esphome-pipsolar/blob/main/tests/esp8266-test-protocols.yaml) fired up and this back:

`INFO ESPHome 2025.2.1`
`[12:14:12][I][main:603]: This is the set of commands supported by the pipsolar component!`
`[12:14:12][D][uart_debug:158]: >>> "QPI\xBE\xAC\r"`
`[12:14:12][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:13][D][uart_debug:158]: >>> "QDIq\e\r"`
`[12:14:13][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:14][D][uart_debug:158]: >>> "QFLAG\x98t\r"`
`[12:14:14][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:15][D][uart_debug:158]: >>> "QMN\xBBd\r"`
`[12:14:16][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:16][D][uart_debug:158]: >>> "QMODI\xC1\r"`
`[12:14:17][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:17][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"`
`[12:14:18][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:18][D][uart_debug:158]: >>> "QPIRI\xF8T\r"`
`[12:14:19][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:19][D][uart_debug:158]: >>> "QPIWS\xB4\xDA\r"`
`[12:14:20][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:20][D][uart_debug:158]: >>> "QT\'\xFF\r"`
`[12:14:21][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:21][D][uart_debug:158]: >>> "QPGS0?\xDA\r"`
`[12:14:22][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:23][D][uart_debug:158]: >>> "QPGS1/\xFB\r"`
`[12:14:23][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:24][D][uart_debug:158]: >>> "QPGS2\x1F\x98\r"`
`[12:14:24][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:25][I][main:690]: Testing PI41 split phase / multiple strings commands...`
`[12:14:25][D][uart_debug:158]: >>> "QPIGS2h-\r"`
`[12:14:25][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:26][D][uart_debug:158]: >>> "QP2GS0\x14\x05\r"`
`[12:14:26][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:27][D][uart_debug:158]: >>> "QP2GS1\x04$\r"`
`[12:14:27][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:28][I][main:714]: Testing unsupported PI18 commands...`
`[12:14:28][D][uart_debug:158]: >>> "^P005PIq\x8B\r"`
`[12:14:28][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:29][D][uart_debug:158]: >>> "^P005GSX\x14\r"`
`[12:14:29][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:30][D][uart_debug:158]: >>> "^P006MOD\xDD\xBE\r"`
`[12:14:30][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:31][I][main:738]: Testing unsupported PI17 commands...`
`[12:14:31][D][uart_debug:158]: >>> "^P003PI\r"`
`[12:14:31][D][uart_debug:158]: <<< "^D00517\xCA\xEC\r"`
`[12:14:32][D][uart_debug:158]: >>> "^P004MOD\r"`
`[12:14:32][D][uart_debug:158]: <<< "^D00505\xD9\x9F\r"`
`[12:14:33][D][uart_debug:158]: >>> "^P005FLAG\r"`
`[12:14:33][D][uart_debug:158]: <<< "^D0441,1,1,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0\xE6\x17\r"`
`[12:14:34][I][main:762]: Testing unsupported PI16 commands...`
`[12:14:34][D][uart_debug:158]: >>> "QPI\r"`
`[12:14:34][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:35][D][uart_debug:158]: >>> "QMOD\r"`
`[12:14:35][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:36][D][uart_debug:158]: >>> "QPIGS\r"`
`[12:14:36][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:37][D][uart_debug:158]: >>> "QPIRI\r"`
`[12:14:37][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:38][D][uart_debug:158]: >>> "QMOD\r"`
`[12:14:38][I][main:796]: Testing unsupported Qx commands...`
`[12:14:38][D][uart_debug:158]: >>> "Q1\r"`
`[12:14:38][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:39][D][uart_debug:158]: >>> "Q2\r"`
`[12:14:39][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:40][D][uart_debug:158]: >>> "Q4\r"`
`[12:14:40][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:41][D][uart_debug:158]: >>> "Q6\r"`
`[12:14:41][D][uart_debug:158]: <<< "^0\e\xE3\r"`
`[12:14:42][I][main:827]: Done. Repeating...`

Is this what we need?

syssi, 2025-07-13:
See https://github.com/syssi/esphome-pipsolar/issues/78

closed 2025-07-13 by syssi