syssi/esphome-pipsolar #92: esp01 supported?

shamirahim111, 2023-11-09:
Hi.

can we use an ESP01 because its a bit cheaper?

shamirahim111, 2023-11-09:
https://robu.in/product/esp-01-esp8266-serial-wifi-wireless-transceiver-module/

like this?

syssi, 2023-11-09:
It's cheaper but you have to care about a stable power supply and it's more complicated to flash the ESP + apply a proper boot configuration. The ESP is compatible in general.

shamirahim111, 2023-11-09:
@syssi okay cool. i will try to purchase both and see if https://robu.in/product/d1-mini-v2-nodemcu-4m-bytes-lua-wifi-internet-of-things-development-board-based-esp8266/ works any better. it also has in built power so yeah.

one question i have.  
this is the configuration that seems to be working for me. 
https://solaranzeige.de/phpBB3/download/Phocos%20Any-Grid.json

[image]

you think this would work?

syssi, 2023-11-09:
I don't know. I cannot find any details about the used protocol in the `Phocos Any-Grid.json`. Do I miss something?

shamirahim111, 2023-11-18:
@syssi  hey. what about config from this repo? https://github.com/wolffshots/phocus

shamirahim111, 2023-11-18:
@syssi i have bought one esp8266 nodemcu v3, will test it out

shamirahim111, 2023-11-18:
@syssi what about this specific line 
https://github.com/DeBaschdi/solar_config/blob/f329f2a034c52264c6883e0be857b70dde9807a5/www/html/multi_usb_init.php#L345

shamirahim111, 2023-11-24:
thanks.  i think i am able to connect via esp8266. only thing missing is to figure out protocol. thanks

closed 2023-11-24 by shamirahim111