syssi/esphome-pipsolar #95: rs-232 to rsp8266

shamirahim111, 2023-11-18:
hi.

i have an inverter that has rs-232 port. i have bought one rs-232 to TTL serial converter https://robu.in/product/rs232-to-ttl-serial-interface-module/

this has just 4 pinouts,

VCC, gnd, RX, Tx

in your photo, D1 mini pins are G, D1, D2 and SV3? so these pins aren't all visible in https://robu.in/product/nodemcu-esp8266-v3-lua-ch340-wifi-dev-board/

so which should be connected where?

any help would be appreciated

wojtbar, 2023-11-18:
RS-233 to ttl pin vcc to  3.3 esp8266 ,
RS-232 to ttl pin gnd to G esp8266
Rs232 to ttl pin txd to  D2 esp8266
Rs232 to ttl pin rxd to  D1 of ESP8266

wojtbar, 2023-11-18:
https://github.com/syssi/esphome-pipsolar/tree/main/images

syssi, 2023-11-18:
> in your photo, D1 mini pins are G, D1, D2 and SV3? so these pins aren't all visible in https://robu.in/product/nodemcu-esp8266-v3-lua-ch340-wifi-dev-board/

Are you sure? This image of your board shows `GND`, `3.3V`, `D1 = GPIO5`, `D2 = GPIO4` :

https://robu.in/wp-content/uploads/2018/03/62160-03.jpg

closed 2023-11-24 by shamirahim111