syssi/esphome-pipsolar #99: Add Voltronic MK KING PIP30 support

saviothecnic, 2023-12-12:
You can add sensors that are only present on the MK inverter that speaks with the PIP30 protocol such as the Voltronic MAX
But it has sensors that the max doesn't
has the same as that of Operational Logic
If you see in this manual page 4 option Z and on page 30

https://github.com/jblance/mpp-solar/blob/master/docs/protocols/PI30_PIP-GK_MK-Protocol.pdf

You can also add parallel support so you can add and see sensors
of inverters connected in parallel

`select:
  - platform: pipsolar
    pipsolar_id: inverter0
    output_source_priority:
      id: inverter0_operation_logic_select
      name: inverter0_operation_logic_select
      optionsmap:
        "Auto": "POPLG00"
        "On Line": "POPLG01"
        "ECO": "POPLG02"
      statusmap:
        "0": "Auto"
        "1": "On Line"
        "2": "ECO"`

saviothecnic, 2023-12-15:
Here is the file where I added the sensors which obviously are now not seen by the source and are commented with # at the end of the code

[Voltronic-Athom-RS232.yaml.txt](https://github.com/syssi/esphome-pipsolar/files/13684462/Voltronic-Athom-RS232.yaml.txt)

unknown1818, 2024-03-23:
Hi,

I use default config i got PI30 and on QPIGS protocol i get timeout but inverter send string with correct data other protocols like QMOD QPIRI working

baturinivan, 2024-03-30:
> Hi,
> 
> I use default config i got PI30 and on QPIGS protocol i get timeout but inverter send string with correct data other protocols like QMOD QPIRI working

Found similar issue with other inverter. QPIGS response was longer than expected. I have found the solution in other issues there. This is what was working for me: https://github.com/syssi/esphome-pipsolar/issues/113#issuecomment-2028003885