syssi/esphome-pipsolar #100: inverter suddenly stopped working

test2a, 2023-12-18:
inverter was working till 3pm during the day and suddenly am getting unknown response

```
INFO ESPHome 2023.11.6
INFO Reading configuration /config/esphome/esphome-web-5f9e4d.yaml...
INFO Updating https://github.com/syssi/esphome-pipsolar.git@pip8048
INFO Starting log output from esphome-web-5f9e4d.local using esphome API
INFO Successfully connected to esphome-web-5f9e4d in 0.203s
INFO Successful handshake with esphome-web-5f9e4d in 2.422s
[22:26:05][I][app:102]: ESPHome version 2023.11.4 compiled on Nov 28 2023, 23:52:53
[22:26:05][C][wifi:559]: WiFi:
[22:26:05][C][wifi:391]:   Local MAC: 48:3F:DA:5F:9E:4D
[22:26:05][C][wifi:396]:   SSID: 'wifi'[redacted]
[22:26:05][C][wifi:397]:   IP Address: 192.168.29.26
[22:26:05][C][wifi:398]:   BSSID: 58:95:D8:24:49:81[redacted]
[22:26:05][C][wifi:400]:   Hostname: 'esphome-web-5f9e4d'
[22:26:05][C][wifi:402]:   Signal strength: -82 dB ▂▄▆█
[22:26:05][C][wifi:406]:   Channel: 6
[22:26:05][C][wifi:407]:   Subnet: 255.255.255.0
[22:26:05][C][wifi:408]:   Gateway: 192.168.29.1
[22:26:05][C][wifi:409]:   DNS1: 192.168.29.85
[22:26:05][C][wifi:410]:   DNS2: 192.168.29.85
[22:26:05][D][pipsolar:843]: timeout command to poll: QPIRI
[22:26:05][C][logger:416]: Logger:
[22:26:05][C][logger:417]:   Level: DEBUG
[22:26:05][C][logger:418]:   Log Baud Rate: 115200
[22:26:05][C][logger:420]:   Hardware UART: UART0
[22:26:05][D][pipsolar:924]: Sending polling command : QBATCD with length 6
[22:26:05][C][uart.arduino_esp8266:102]: UART Bus:
[22:26:05][C][uart.arduino_esp8266:103]:   TX Pin: GPIO4
[22:26:05][C][uart.arduino_esp8266:104]:   RX Pin: GPIO5
[22:26:05][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[22:26:05][C][uart.arduino_esp8266:108]:   Baud Rate: 2400 baud
[22:26:05][C][uart.arduino_esp8266:109]:   Data Bits: 8
[22:26:05][C][uart.arduino_esp8266:110]:   Parity: NONE
[22:26:05][C][uart.arduino_esp8266:111]:   Stop bits: 1
[22:26:05][C][uart.arduino_esp8266:115]:   Using software serial
[22:26:05][C][pipsolar:948]: Pipsolar:
[22:26:05][C][pipsolar:949]: used commands:
[22:26:05][C][pipsolar:952]: QPIGS
[22:26:05][C][pipsolar:952]: QPIGS2
[22:26:05][C][pipsolar:952]: QMOD
[22:26:05][C][pipsolar:952]: QFLAG
[22:26:05][C][pipsolar:952]: QPIRI
[22:26:05][C][pipsolar:952]: QBATCD
[22:26:05][C][pipsolar.switch:068]: Pipsolar Switch '${name} output_source_priority_utility'
[22:26:05][C][pipsolar.switch:070]:   Icon: 'mdi:power'
[22:26:05][C][pipsolar.switch:090]:   Restore Mode: always OFF
[22:26:05][C][pipsolar.switch:068]: Pipsolar Switch '${name} output_source_priority_solar'
[22:26:05][C][pipsolar.switch:070]:   Icon: 'mdi:power'
[22:26:05][C][pipsolar.switch:090]:   Restore Mode: always OFF
[22:26:05][C][pipsolar.switch:068]: Pipsolar Switch '${name} output_source_priority_battery'
[22:26:05][C][pipsolar.switch:070]:   Icon: 'mdi:power'
[22:26:05][C][pipsolar.switch:090]:   Restore Mode: always OFF
[22:26:05][C][pipsolar.switch:068]: Pipsolar Switch '${name} input_voltage_range'
[22:26:05][C][pipsolar.switch:070]:   Icon: 'mdi:power'
[22:26:05][C][pipsolar.switch:090]:   Restore Mode: always OFF
[22:26:05][C][pipsolar.switch:068]: Pipsolar Switch '${name} pv_ok_condition_for_parallel'
[22:26:05][C][pipsolar.switch:070]:   Icon: 'mdi:power'
[22:26:05][C][pipsolar.switch:090]:   Restore Mode: always OFF
[22:26:05][C][pipsolar.switch:068]: Pipsolar Switch '${name} pv_power_balance'
[22:26:05][C][pipsolar.switch:070]:   Icon: 'mdi:power'
[22:26:05][C][pipsolar.switch:090]:   Restore Mode: always OFF
[22:26:05][C][pipsolar.select:009]: pipsolar.selectPipsolar Controller Select 'inverter0_output_source_priority_select'
[22:26:06][C][pipsolar.select:009]: pipsolar.selectPipsolar Controller Select '${name} charging discharging control'
[22:26:06][C][pipsolar.select:009]: pipsolar.selectPipsolar Controller Select '${name} charger_source_priority_select'
[22:26:06][C][pipsolar.select:009]: pipsolar.selectPipsolar Controller Select '${name} current_max_ac_charging_current_select'
[22:26:06][C][pipsolar.select:009]: pipsolar.selectPipsolar Controller Select '${name} current_max_charging_current_select'
[22:26:06][C][captive_portal:088]: Captive Portal:
[22:26:06][C][mdns:115]: mDNS:
[... 186 lines cut ...]
[22:33:38][D][pipsolar:924]: Sending polling command : QBATCD with length 6
[22:33:42][D][pipsolar:843]: timeout command to poll: QBATCD
[22:33:42][D][pipsolar:924]: Sending polling command : QPIGS with length 5
[22:33:47][D][pipsolar:843]: timeout command to poll: QPIGS
[22:33:47][D][pipsolar:924]: Sending polling command : QPIGS2 with length 6
[22:33:52][D][pipsolar:843]: timeout command to poll: QPIGS2
[22:33:52][D][pipsolar:924]: Sending polling command : QMOD with length 4
[22:33:57][D][pipsolar:843]: timeout command to poll: QMOD
[22:33:57][D][pipsolar:924]: Sending polling command : QFLAG with length 5
[22:34:02][D][pipsolar:843]: timeout command to poll: QFLAG
[22:34:02][D][pipsolar:924]: Sending polling command : QPIRI with length 5
[22:34:07][D][pipsolar:843]: timeout command to poll: QPIRI
[22:34:07][D][pipsolar:924]: Sending polling command : QBATCD with length 6
[22:34:12][D][pipsolar:843]: timeout command to poll: QBATCD
[22:34:12][D][pipsolar:924]: Sending polling command : QPIGS with length 5
[22:34:17][D][pipsolar:843]: timeout command to poll: QPIGS
[22:34:17][D][pipsolar:924]: Sending polling command : QPIGS2 with length 6
[22:34:22][D][pipsolar:843]: timeout command to poll: QPIGS2
[22:34:22][D][pipsolar:924]: Sending polling command : QMOD with length 4
[22:34:27][D][pipsolar:843]: timeout command to poll: QMOD
[22:34:27][D][pipsolar:924]: Sending polling command : QFLAG with length 5
[22:34:32][D][pipsolar:843]: timeout command to poll: QFLAG
[22:34:32][D][pipsolar:924]: Sending polling command : QPIRI with length 5

```

syssi, 2023-12-18:
This looks like an issue of the inverter and is unrelated to this component.

1. Please restart the inverter. Does it work again?
2. Double check your data cable
3. Replace the RS232 converter module
4. Replace the RS232 linecard of your inverter

test2a, 2023-12-26:
Thanks. Found the issue. Bad wiring. Closing the issue

closed 2023-12-26 by test2a