syssi/esphome-pipsolar #111: Question regarding ttl RS485 adapter

brycelarge, 2024-03-13:
Hi there,

I have a RCT-AXPERT VM2 Premium 1200W 12V I’m inverter that I’m trying to setup with my ESP32 dev board.

I have one of these https://www.communica.co.za/products/hkd-ttl-to-rs-485-module-max485 ttl rs485 adapters and I’m just wondering what the wiring would be to this https://www.communica.co.za/products/hkd-esp-32-wifi-b-t-dev-board ESP32 board

I have tried but with no success. I also removed the R7 resistor from the ttl rs485 to see if that would work but no success.

any advice would be appreciated.

Thanks

brycelarge, 2024-03-13:
I also tried 

https://files.waveshare.com/upload/9/95/RS485-Board-UserManual.pdf

brycelarge, 2024-03-13:
Seems I answered my own question. I need a rs232 not rs485

syssi, 2024-03-13:
That's the point! ;-)

closed 2024-03-13 by syssi