syssi/esphome-pipsolar #113: Qoltec 6KW

unknown1818, 2024-03-23:
Hi,

Im trying to get data from rs232 my debug:

`[08:13:30][I][main:468]: Testing PI30/PI30MAX/PI30REVO/PI41 commands...
[08:13:30][I][main:471]: This is the set of commands supported by the pipsolar component!
[08:13:30][D][uart_debug:158]: >>> "QPI\xBE\xAC\r"
[08:13:31][D][uart_debug:158]: <<< "(PI30\x9A\v\r"
[08:13:31][D][uart_debug:158]: >>> "QDIq\e\r"
[08:13:32][D][uart_debug:158]: <<< "(230.0 50.0 0030 42.0 54.0 56.4 46.0 60 0 0 2 0 0 0 0 0 1 1 1 0 1 0 54.0 0 1\xFD\xD8\r"
[08:13:32][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[08:13:33][D][uart_debug:158]: <<< "(EakxyzDbdjlnuv|\x0E\r"
[08:13:34][D][uart_debug:158]: >>> "QMN\xBBd\r"
[08:13:34][D][uart_debug:158]: <<< "(VMIII-60003\xFB\r"
[08:13:35][D][uart_debug:158]: >>> "QMODI\xC1\r"
[08:13:35][D][uart_debug:158]: <<< "(L\x06\a\r"
[08:13:36][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[08:13:36][D][uart_debug:158]: <<< "(235.4 50.0 235.4 50.0 0000 0000 000 384 03.20 000 000 0018 00.0 109.0 00.00 00000 00010000 00 00 00000 011 0 00 0000-f\r"
[08:13:37][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[08:13:37][D][uart_debug:158]: <<< "(230.0 26.0 230.0 50.0 26.0 6000 6000 48.0 46.0 43.0 56.4 54.0 2 030 060 0 2 3 1 01 0 0 54.0 0 1\a\xCB\r"
[08:13:38][D][uart_debug:158]: >>> "QPIWS\xB4\xDA\r"
[08:13:38][D][uart_debug:158]: <<< "(000000000000000001000010000000000000\x99\xF0\r"
[08:13:39][D][uart_debug:158]: >>> "QT\'\xFF\r"
[08:13:39][D][uart_debug:158]: <<< "(20231206230229\xBFN\r"
[08:13:40][D][uart_debug:158]: >>> "QPGS0?\xDA\r"
[08:13:40][D][uart_debug:158]: <<< "(NAKss\r"
[08:13:41][D][uart_debug:158]: >>> "QPGS1/\xFB\r"
[08:13:41][D][uart_debug:158]: <<< "(NAKss\r"
[08:13:42][D][uart_debug:158]: >>> "QPGS2\x1F\x98\r"
[08:13:42][D][uart_debug:158]: <<< "(NAKss\r"`

But script can only get: 
get inverter_grid_rating_voltage: 230V
inverter_grid_rating_current: 26A
inverter_battery_type: 2.0

Cant decode other from:
235.4 50.0 235.4 50.0 0000 0000 000 384 03.20 000 000 0018 00.0 109.0 00.00 00000

I think there is all info

baturinivan, 2024-03-30:
Just testing Sunlife VM III Twin, 24V, 4.2kW (looks exactly as Axpert VM III TWIN 4KVA/6KVA)
Seems to be the same QPIGS problem (timeout) while other commands works fine. QPIGS response has more values than expected in the code. 

Also format seems to be similar to described in MAX.Communication.Protocol20210217.pdf found elsewhere in issues.

There is the testing log:
[I][main:451]: Testing PI30/PI30MAX/PI30REVO/PI41 commands...
[I][main:454]: This is the set of commands supported by the pipsolar component!
[D][uart_debug:158]: >>> "QPI\xBE\xAC\r"
[D][uart_debug:158]: <<< "(PI30\x9A\v\r"
[D][uart_debug:158]: >>> "QDIq\e\r"
[D][uart_debug:158]: <<< "(230.0 50.0 0030 21.0 27.0 28.2 23.0 60 0 0 2 0 0 0 0 0 1 1 1 0 1 0 27.0 0 1)F\r"
[D][uart_debug:158]: >>> "QFLAG\x98t\r"
[D][uart_debug:158]: <<< "(EauvxyzDbdjk\xBF!\r"
[D][uart_debug:158]: >>> "QMN\xBBd\r"
[D][uart_debug:158]: <<< "(VMIII-4000\xDE\x93\r"
[D][uart_debug:158]: >>> "QMODI\xC1\r"
[D][uart_debug:158]: <<< "(B\xE7\xC9\r"
[D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[D][uart_debug:158]: <<< "(000.0 00.0 229.9 49.9 0276 0265 006 397 27.70 075 100 0039 08.3 250.7 00.00 00000 00010110 00 00 02093 111 0 00 0000p\xB1\r"
[D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[D][uart_debug:158]: <<< "(230.0 17.3 230.0 50.0 17.3 4000 4000 24.0 23.0 21.0 28.6 28.2 2 080 120 0 0 2 1 01 0 0 27.0 0 1m\xB0\r"
[D][uart_debug:158]: >>> "QPIWS\xB4\xDA\r"
[D][uart_debug:158]: <<< "(000001000000000000000000000000000000\xB2\xA7\r"
[D][uart_debug:158]: >>> "QT\'\xFF\r"
[D][uart_debug:158]: <<< "(20240330151815]\v\r"
[D][uart_debug:158]: >>> "QPGS0?\xDA\r"
[D][uart_debug:158]: <<< "(NAKss\r"
[D][uart_debug:158]: >>> "QPGS1/\xFB\r"
[D][uart_debug:158]: <<< "(NAKss\r"
[D][uart_debug:158]: >>> "QPGS2\x1F\x98\r"
[D][uart_debug:158]: <<< "(NAKss\r"

baturinivan, 2024-03-30:
> Seems to be the same QPIGS problem (timeout) while other commands works fine. QPIGS response has more values than expected in the code.

Finally, after a day of troubleshooting I have found a solution.

Took this example: https://github.com/syssi/esphome-pipsolar/blob/pip8048/esp32-example.yaml
I have changed esphome section and added esp32 with esp-idf framework, version recommended (latest didn't work for me). 
`esphome:
  name: ${name}

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: recommended`

syssi, 2024-03-30:
Good job! `recommended` is the default. So you could drop this line if you like.