syssi/esphome-pipsolar #114: Adding external componen not found

unknown1818, 2024-03-23:
Im trying to add axternal component to my conf in esphome

external_components:
  - source: github://syssi/esphome-pipsolar@pip8048

but get error when compiling

In file included from src/esphome/components/pipsolar/pipsolar.cpp:1:
src/esphome/components/pipsolar/pipsolar.h:7:10: fatal error: esphome/components/pipsolar/select/pipsolar_select.h: No such file or directory
 #include "esphome/components/pipsolar/select/pipsolar_select.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/test/src/esphome/components/pipsolar/pipsolar.cpp.o] Error 1

syssi, 2024-03-23:
A clean build is required. If you use the ESPHome dashboard there should be a `clean` button somewhere. If you use the CLI:

```
esphome clean yourconfig.yaml
esphome run yourconfig.yaml
```

unknown1818, 2024-03-23:
I got it working couple seconds ago with this default yaml

https://github.com/syssi/esphome-pipsolar/blob/pip8048/esp32-example.yaml

But:

`pipsolar pv1_input_current	0.0 A	
pipsolar pv1_input_voltage	112.2 V`

Why input current showing 0.0 A im currently using it at 150watts

unknown1818, 2024-03-23:
`pipsolar ac_charging_status
wyłączono
pipsolar ac_output_active_power
0,0 W
pipsolar ac_output_apparent_power
0,0 VA
pipsolar ac_output_frequency
49,9 Hz
pipsolar ac_output_voltage
234,9 V
pipsolar add_sbu_priority_version
wyłączono
pipsolar battery_capacity_percent
0,0%
pipsolar battery_charging_current
0,0 A
pipsolar battery_discharge_current
0,0 A
pipsolar battery_voltage
3,3 V
pipsolar battery_voltage_offset_for_fans_on
0,0 V
pipsolar battery_voltage_scc
0,0 V
pipsolar battery_voltage_to_steady_while_charging
wyłączono
pipsolar bus_voltage
395,0 V
pipsolar charging_status
wyłączono
pipsolar charging_to_floating_mode
wyłączono
pipsolar configuration_status
wyłączono
pipsolar device_mode
L
pipsolar grid_frequency
49,9 Hz
pipsolar grid_voltage
234,9 V
pipsolar inverter_heat_sink_temperature
18,0 °C
pipsolar lcd_escape_to_default
włączono
pipsolar load_status
włączono
pipsolar output_load_percent
0,0%
pipsolar over_temperature_restart_function
wyłączono
pipsolar overload_bypass_function
wyłączono
pipsolar overload_restart_function
wyłączono
pipsolar pv1_charging_power
1,0 W
pipsolar pv1_input_current
0,0 A
pipsolar pv1_input_voltage
117,0 V
pipsolar pv2_charging_power
nieznany
pipsolar pv2_input_current
nieznany
pipsolar pv2_input_voltage
nieznany
pipsolar scc_charging_status
wyłączono
pipsolar silence_buzzer_open_buzzer
włączono
pipsolar switch_on
włączono`

unknown1818, 2024-03-23:
pipsolar ac_charging_status
wyłączono
pipsolar ac_output_active_power
0,0 W
pipsolar ac_output_apparent_power
0,0 VA
pipsolar ac_output_frequency
49,9 Hz
pipsolar ac_output_voltage
234,9 V
pipsolar add_sbu_priority_version
wyłączono
pipsolar battery_capacity_percent
0,0%
pipsolar battery_charging_current
0,0 A
pipsolar battery_discharge_current
0,0 A
pipsolar battery_voltage
3,3 V
pipsolar battery_voltage_offset_for_fans_on
0,0 V
pipsolar battery_voltage_scc
0,0 V
pipsolar battery_voltage_to_steady_while_charging
wyłączono
pipsolar bus_voltage
395,0 V
pipsolar charging_status
wyłączono
pipsolar charging_to_floating_mode
wyłączono
pipsolar configuration_status
wyłączono
pipsolar device_mode
L
pipsolar grid_frequency
49,9 Hz
pipsolar grid_voltage
234,9 V
pipsolar inverter_heat_sink_temperature
18,0 °C
pipsolar lcd_escape_to_default
włączono
pipsolar load_status
włączono
pipsolar output_load_percent
0,0%
pipsolar over_temperature_restart_function
wyłączono
pipsolar overload_bypass_function
wyłączono
pipsolar overload_restart_function
wyłączono
pipsolar pv1_charging_power
1,0 W
pipsolar pv1_input_current
0,0 A
pipsolar pv1_input_voltage
117,0 V
pipsolar pv2_charging_power
nieznany
pipsolar pv2_input_current
nieznany
pipsolar pv2_input_voltage
nieznany
pipsolar scc_charging_status
wyłączono
pipsolar silence_buzzer_open_buzzer
włączono
pipsolar switch_on
włączono

unknown1818, 2024-03-23:
Its a hybrid inventer working without battery, do i need battery to show wattage from pv ?

closed 2024-08-12 by syssi