syssi/esphome-pipsolar #115: MPI 10k integration

WagnerSystems, 2024-03-23:
Hi, 
I would like to integrate my MPI 10k to hass with esphome. 
I ran the testing script but this seems to gives me answers at the PI17 section.

`[17:10:29][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:10:45][I][main:464]: Testing PI30/PI30MAX/PI30REVO/PI41 commands...
[17:10:45][I][main:467]: This is the set of commands supported by the pipsolar component!
[17:10:45][D][uart_debug:158]: >>> "QPI\xBE\xAC\r"
[17:10:45][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:10:46][D][uart_debug:158]: >>> "QDIq\e\r"
[17:10:47][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[17:10:47][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:10:48][D][uart_debug:158]: >>> "QMN\xBBd\r"
[17:10:48][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:10:49][D][uart_debug:158]: >>> "QMODI\xC1\r"
[17:10:49][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:10:50][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[17:10:50][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:10:51][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[17:10:51][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:10:52][D][uart_debug:158]: >>> "QPIWS\xB4\xDA\r"
[17:10:52][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:10:53][D][uart_debug:158]: >>> "QT\'\xFF\r"
[17:10:53][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:10:54][D][uart_debug:158]: >>> "QPGS0?\xDA\r"
[17:10:54][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:10:55][D][uart_debug:158]: >>> "QPGS1/\xFB\r"
[17:10:55][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:10:56][D][uart_debug:158]: >>> "QPGS2\x1F\x98\r"
[17:10:56][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:10:57][I][main:554]: Testing PI41 split phase / multiple strings commands...
[17:10:57][D][uart_debug:158]: >>> "QPIGS2h-\r"
[17:10:57][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:10:58][D][uart_debug:158]: >>> "QP2GS0\x14\x05\r"
[17:10:58][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:10:59][D][uart_debug:158]: >>> "QP2GS1\x04$\r"
[17:10:59][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:11:00][I][main:578]: Testing unsupported PI18 commands...
[17:11:00][D][uart_debug:158]: >>> "^P005PIq\x8B\r"
[17:11:01][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:11:01][D][uart_debug:158]: >>> "^P005GSX\x14\r"
[17:11:02][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:11:02][D][uart_debug:158]: >>> "^P006MOD\xDD\xBE\r"
[17:11:03][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:11:03][I][main:602]: Testing unsupported PI17 commands...
[17:11:03][D][uart_debug:158]: >>> "^P003PI\r"
[17:11:04][D][uart_debug:158]: <<< "^D00517\xCA\xEC\r"
[17:11:04][D][uart_debug:158]: >>> "^P004MOD\r"
[17:11:05][D][uart_debug:158]: <<< "^D00505\xD9\x9F\r"
[17:11:05][D][uart_debug:158]: >>> "^P005FLAG\r"
[17:11:06][D][uart_debug:158]: <<< "^D0301,0,0,0,0,0,0,0,1,0,0,0,0,0\\\x8D\r"
[17:11:06][I][main:626]: Testing unsupported PI16 commands...
[17:11:06][D][uart_debug:158]: >>> "QPI\r"
[17:11:07][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:11:08][D][uart_debug:158]: >>> "QMOD\r"
[17:11:08][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:11:09][D][uart_debug:158]: >>> "QPIGS\r"
[17:11:09][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:11:10][D][uart_debug:158]: >>> "QPIRI\r"
[17:11:10][D][uart_debug:158]: <<< "^0\e\xE3\r"
[17:11:11][D][uart_debug:158]: >>> "QMOD\r"
[17:11:11][I][main:660]: Done. Repeating...
[17:11:11][D][uart_debug:158]: <<< "^0\e\xE3\r"`

Can somebody point me to the wright direction on how I can communicate with my MPI?

JakobTewes, 2025-02-12:
Hi @WagnerSystems,

did you get it to work? Got a MPI 15 KW and also want to switch to ESPHome ;-).
Also can you provide a config that selects PI17 or even better pi17m058 as protocol?

Reference: https://github.com/jblance/mpp-solar/issues/485 

Thanks and regards

Jakob

WagnerSystems, 2025-02-13:
Hi Jakob,
I am still working on creating a proper module.
For now I use some custom scripts within ESPHome, which gives me only the daily production to integrate it in the energy dash.

Volker

JakobTewes, 2025-03-02:
Just did my own experiment with [esp8266-test-protocols.yaml ](https://github.com/syssi/esphome-pipsolar/blob/main/tests/esp8266-test-protocols.yaml).

This readings flow back:

`INFO ESPHome 2025.2.1
[12:14:12][I][main:603]: This is the set of commands supported by the pipsolar component!
[12:14:12][D][uart_debug:158]: >>> "QPI\xBE\xAC\r"
[12:14:12][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:13][D][uart_debug:158]: >>> "QDIq\e\r"
[12:14:13][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:14][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[12:14:14][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:15][D][uart_debug:158]: >>> "QMN\xBBd\r"
[12:14:16][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:16][D][uart_debug:158]: >>> "QMODI\xC1\r"
[12:14:17][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:17][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[12:14:18][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:18][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[12:14:19][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:19][D][uart_debug:158]: >>> "QPIWS\xB4\xDA\r"
[12:14:20][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:20][D][uart_debug:158]: >>> "QT\'\xFF\r"
[12:14:21][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:21][D][uart_debug:158]: >>> "QPGS0?\xDA\r"
[12:14:22][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:23][D][uart_debug:158]: >>> "QPGS1/\xFB\r"
[12:14:23][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:24][D][uart_debug:158]: >>> "QPGS2\x1F\x98\r"
[12:14:24][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:25][I][main:690]: Testing PI41 split phase / multiple strings commands...
[12:14:25][D][uart_debug:158]: >>> "QPIGS2h-\r"
[12:14:25][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:26][D][uart_debug:158]: >>> "QP2GS0\x14\x05\r"
[12:14:26][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:27][D][uart_debug:158]: >>> "QP2GS1\x04$\r"
[12:14:27][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:28][I][main:714]: Testing unsupported PI18 commands...
[12:14:28][D][uart_debug:158]: >>> "^P005PIq\x8B\r"
[12:14:28][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:29][D][uart_debug:158]: >>> "^P005GSX\x14\r"
[12:14:29][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:30][D][uart_debug:158]: >>> "^P006MOD\xDD\xBE\r"
[12:14:30][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:31][I][main:738]: Testing unsupported PI17 commands...
[12:14:31][D][uart_debug:158]: >>> "^P003PI\r"
[12:14:31][D][uart_debug:158]: <<< "^D00517\xCA\xEC\r"
[12:14:32][D][uart_debug:158]: >>> "^P004MOD\r"
[12:14:32][D][uart_debug:158]: <<< "^D00505\xD9\x9F\r"
[12:14:33][D][uart_debug:158]: >>> "^P005FLAG\r"
[12:14:33][D][uart_debug:158]: <<< "^D0441,1,1,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0\xE6\x17\r"
[12:14:34][I][main:762]: Testing unsupported PI16 commands...
[12:14:34][D][uart_debug:158]: >>> "QPI\r"
[12:14:34][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:35][D][uart_debug:158]: >>> "QMOD\r"
[12:14:35][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:36][D][uart_debug:158]: >>> "QPIGS\r"
[12:14:36][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:37][D][uart_debug:158]: >>> "QPIRI\r"
[12:14:37][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:38][D][uart_debug:158]: >>> "QMOD\r"
[12:14:38][I][main:796]: Testing unsupported Qx commands...
[12:14:38][D][uart_debug:158]: >>> "Q1\r"
[12:14:38][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:39][D][uart_debug:158]: >>> "Q2\r"
[12:14:39][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:40][D][uart_debug:158]: >>> "Q4\r"
[12:14:40][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:41][D][uart_debug:158]: >>> "Q6\r"
[12:14:41][D][uart_debug:158]: <<< "^0\e\xE3\r"
[12:14:42][I][main:827]: Done. Repeating...
`

Steef-the-Beef, 2025-04-09:
Hi, 
I also need the Protocol 17 vor my MPI 15k.
did you get it to work?

Can I help implenting the protocol?

Best regards
Stefano

JakobTewes, 2025-04-09:
> Hi [@Steef-the-Beef](https://github.com/Steef-the-Beef), maybe you can give this a try. It worked with my MPI 10k.
> 
> [https://github.com/rene-dev/solar/blob/master/mpi2.yaml](url)
> 
> Volker

This also works for me. It is fast and works (great). Until now the feature set is relatively limited (only few sensors and it is very much tailored to the creatores requirements ([MarkusBec](https://github.com/MarkusBec) and [rene-dev](https://github.com/rene-dev). Good news - they are planning to [add more to it ](https://github.com/syssi/esphome-pipsolar/issues/78#issuecomment-2731103199).

For "production" I still use a raspi with [mpp-solar](https://github.com/jblance/mpp-solar)

Read Configuration Parameters
`/usr/bin/python3 /usr/local/bin/mpp-solar -p /dev/hidrawINV1 -n 'Inverter1' -T 'Inverter1' -P pi17m058 -o hass_mqtt --mqttbroker 192.168.177.32 --mqttuser inverter --mqttpass pass -c MOD#HECS#BATS#GS#PS#WS`

Read Energy Generation
`/usr/bin/python3 /usr/local/bin/mpp-solar -p /dev/hidrawINV1 -n 'Inverter1' -T 'Inverter1' -P pi17m058 -o hass_mqtt --mqttbroker 192.168.177.32 --mqttuser inverter --mqttpass pass -c ET#ED`date +"%Y%m%d"`#EM`date +'%Y%m' -d 'last month'`

Sync Time
`/usr/bin/python3 /usr/local/bin/mpp-solar -p /dev/hidrawINV1 -n 'Inverter1' -T 'Inverter1' -P pi17m058 -c DAT`date '+%g%m%d%H%M%S'``

Configure to "Charge Battery"
`/usr/bin/python3 /usr/local/bin/mpp-solar -p /dev/hidrawINV1 -n 'Inverter1' -T 'Inverter1' -P pi17m058 -c SEP01#EDB0`

Configure to "Prefer Load"
`/usr/bin/python3 /usr/local/bin/mpp-solar -p /dev/hidrawINV1 -n 'Inverter1' -T 'Inverter1' -P pi17m058 -c SEP00#EDB0#EDC0#EDE1

Steef-the-Beef, 2025-04-09:
Hi Volker & Jakob,
did you flash it on a ESP32 or ESP8266?

JakobTewes, 2025-04-09:
Currently preparing a self-sufficient setup with:

- Power via RJ45 and [stepdown](https://de.aliexpress.com/item/1005007092498838.html?gps-id=pcStoreLeaderboard&scm=1007.22922.271278.0&scm_id=1007.22922.271278.0&scm-url=1007.22922.271278.0&pvid=c6e738b1-3b91-47df-8cc4-52aa7bde33e1&_t=gps-id%3ApcStoreLeaderboard%2Cscm-url%3A1007.22922.271278.0%2Cpvid%3Ac6e738b1-3b91-47df-8cc4-52aa7bde33e1%2Ctpp_buckets%3A668%232846%238115%232000&pdp_ext_f=%7B%22order%22%3A%22995%22%2C%22eval%22%3A%221%22%2C%22sceneId%22%3A%2212922%22%7D&pdp_npi=4%40dis%21EUR%211.09%211.09%21%21%218.65%218.65%21%40211b628117441920865576919eb729%2112000039378058524%21rec%21DE%21890928512%21X&spm=a2g0o.store_pc_home.smartLeaderboard_2008173214601.1005007092498838&gatewayAdapt=glo2deu) 
- [RS232 to TTL converter ](https://de.aliexpress.com/item/32946986183.html?spm=a2g0o.order_list.order_list_main.116.6f375c5fsi8Sxh&gatewayAdapt=glo2deu)
- [ESP32 S2 Mini ](https://de.aliexpress.com/item/1005006985001838.html?spm=a2g0o.order_list.order_list_main.12.6f375c5fsi8Sxh&gatewayAdapt=glo2deu)

Steef-the-Beef, 2025-04-09:
Hi, 

did you test the mpi2.yaml already with a esp32?

JakobTewes, 2025-04-09:
Yep, had it running...

Steef-the-Beef, 2025-04-10:
Hi, thank you for the fast response :-)
can you share the yaml file with me?
Did you flash it with esphome comand?

WagnerSystems, 2025-04-11:
Hi @Steef-the-Beef ,
i flashed it directly from ESPHome. Even ota.
Here is the relevant part of my yaml-File.

` uart:
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 2400
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\r"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

substitutions:
  name: mpi1

external_components:
  - source: github://rene-dev/esphome@pip17
    components: [pipsolar]

pipsolar:
  - id: inverter0

sensor:
  - platform: pipsolar
    pipsolar_id: inverter0
    inverter_heat_sink_temperature:
      name: ${name} inverter_heat_sink_temperature
    component_max_temp:
      name: ${name} component_max_temp
    energy:
      name: ${name} energy
    pv_voltage1:
      name: ${name} pv_voltage1
    pv_voltage2:
      name: ${name} pv_voltage2
    pv_current1:
      name: ${name} pv_current1
    pv_current2:
      name: ${name} pv_current2
    grid_frequency:
      name: ${name} grid_frequency
    pv_power1:
      name: ${name} pv_power1
    pv_power2:
      name: ${name} pv_power2
    battery_power:
      name: ${name} battery_power
    ac_input_total_active_power:
      name: ${name} ac_input_total_active_power
    eminfo:
     id: eminfo
     internal: true`

Steef-the-Beef, 2025-04-23:
Hi,

I have tried to correct the indentations but I am not able to do that. I am noob with VSCode. :-D
Can you help me please?

```
esphome:
  name: mpi5
  friendly_name: mpi5

#debug:
#  direction: BOTH
#  dummy_receiver: false
#  after:
#    delimiter: "\r"
#    sequence:
#     - lambda: UARTDebug::log_string(direction, bytes);

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

wifi:
  ssid: XXX
  password: XXX

mqtt:
 broker: "X.X.X.X"
 username: ""
 password: ""
 id: mqtt_client

external_components:
 - source: github://rene-dev/esphome@pip17
   components: [pipsolar]

uart:
 tx_pin: GPIO17
 rx_pin: GPIO16
 baud_rate: 2400

#pipsolar:
#  - id: inverter0

sensor:
  - platform: pipsolar
    pipsolar_id: inverter0
    inverter_heat_sink_temperature:
      name: ${name} inverter_heat_sink_temperature
    component_max_temp:
      name: ${name} component_max_temp
    energy:
      name: ${name} energy
    pv_voltage1:
      name: ${name} pv_voltage1
    pv_voltage2:
      name: ${name} pv_voltage2
    pv_current1:
      name: ${name} pv_current1
    pv_current2:
      name: ${name} pv_current2
    grid_frequency:
      name: ${name} grid_frequency
    pv_power1:
      name: ${name} pv_power1
    pv_power2:
      name: ${name} pv_power2
    battery_power:
      name: ${name} battery_power
    ac_input_total_active_power:
      name: ${name} ac_input_total_active_power
    eminfo:
     id: eminfo
     internal: true

```