syssi/esphome-pipsolar #119: Effecta AX-P1 whit USB2.0 typ B COM port

tiimsvk, 2024-03-30:
Hi, I would like some advice. I have a solar inverter: Effecta AX-P1
It has a USB2.0 type B data port that can be connected directly to a USB PC and I can monitor the data using the WachPower program.

How would I connect it via ESP32 and esphome?

Well thank you.

tiimsvk, 2024-04-01:
[image]
This is comunication board.

is it possible to use the white connector for communication via uart?

syssi, 2024-04-01:
Could you provide some more photos (top and bottom side of the PCB). I'm looking for some labels and connections between the different components.

tiimsvk, 2024-04-02:
[image]
[image]
[image]

i have already started the pin mapping. When I have it, at least from the white connector, I will send it

tiimsvk, 2024-04-02:
[image]

so far on paper.  mapped to optocouplers

syssi, 2024-04-02:
Do you have access to a logic analyzer?

tiimsvk, 2024-04-02:
What do you mean?  that I can get inside the inverter?  or analyze the digital output because I have a handheld oscilloscope.

tiimsvk, 2024-04-04:
Just for fun, I tested solar-assistant and it works without problems.

tiimsvk, 2024-04-06:
today there was a failure of the connection between the USB and the solar assistant. So it made me look for another solution again.
I am looking for a similar board for communication with my converter, see:
https://powerforum.co.za/topic/2359-axpert-5kva-pc-board-burned/

And I also found a similar scheme:
https://forums.aeva.asn.au/viewtopic.php?p=66395#p66395

I tried one more thing, namely communication on Pine1 TX and pine 2 RX. If USB is connected, both of them show communication using the oscillator. As soon as I disconnect both lines are silent. But I would expect the Sender pin to still be up and running.

It looks like some kind of communication between them must take place at the beginning for the converter to start sending something. Does it work the same for other converters with RS232 communication port?

Would it be enough to connect the isolation of the RX TX line (pin 1 and 2) to the esphome through the opoto? Or is it more complicated than that?

syssi, 2024-04-06:
> Would it be enough to connect the isolation of the RX TX line (pin 1 and 2) to the esphome through the opoto? Or is it more complicated than that?

No. That's all! You could enable the debug mode of the `uart` component to dump all traffic and start with a single pin (RX).

tiimsvk, 2024-05-08:
Hello. I'm back to testing again today.
So far I have been working via solar-assitant and USB port, but the reliability of the connection fluctuates and I can't imagine adding more peripherals to the RPI zero2. That's why it made me try esphome again.

Here is the result:
I physically connected 4 wires to the output connector of the converter.
From the mountain:
Pin1 RX           -> AI   -> AO -> TX
Pin2 TX           -> BO  -> BI  -> RX
Pin3 VCC 5V   -> V2  -> V1 -> 3V3
Pin4 GND       -> G2  -> G1 -> GND
Pin5 NC
Pin6 NC

Subsequently, I tried different types of TTL=RS232
In the end I stuck with this one, which is also an optical isolator and works:
https://vi.aliexpress.com/item/1005005643287661.html

But I have a problem. Output values go to uart debug:
```
[06:55:42][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[06:55:42][D][pipsolar:926]: Sending polling command : QPIRI with length 5
[06:55:43][D][uart_debug:158]: <<< "(230.0 13.0 230.0 50.0 13.0 3000 3000 24.0 24.0 21.7 29.0 27.6 2 20 030 0 2 1 1 01 0 0 28.0 0 1\x13\x05\r"
[06:55:43][D][pipsolar:860]: checking crc on incoming message
[06:55:43][D][pipsolar:870]: CRC NOK expected: 6F A6 but got: 13 5
[06:55:43][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[06:55:43][D][pipsolar:926]: Sending polling command : QPIGS with length 5
[06:55:44][D][uart_debug:158]: <<< "(246.2 49.9 230.0 49.9 0322 0224 010 368 26.02 002 100 0050 0002 084.5 25.96 00005 00110110 00 00 00073 010\xB6\x05\r"
[06:55:44][D][pipsolar:860]: checking crc on incoming message
[06:55:44][D][pipsolar:870]: CRC NOK expected: 99 DB but got: B6 5
[06:55:44][D][uart_debug:158]: >>> "QPIGS2h-\r"
[06:55:44][D][pipsolar:926]: Sending polling command : QPIGS2 with length 6
[06:55:44][D][uart_debug:158]: <<< "(NAKss\r"
[06:55:49][D][pipsolar:843]: timeout command to poll: QPIGS2
[06:55:49][D][uart_debug:158]: >>> "QMODI\xC1\r"
[06:55:49][D][pipsolar:926]: Sending polling command : QMOD with length 4
[06:55:49][D][uart_debug:158]: <<< "(B\xE7\xC9\r"
[06:55:54][D][pipsolar:843]: timeout command to poll: QMOD
[06:55:54][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[06:55:54][D][pipsolar:926]: Sending polling command : QFLAG with length 5
[06:55:55][D][uart_debug:158]: <<< "(EbkyDajuvxz\xAB\x80\r"
[06:55:59][D][pipsolar:843]: timeout command to poll: QFLAG
[06:55:59][D][uart_debug:158]: >>> "QPIWS\xB4\xDA\r"
[06:55:59][D][pipsolar:926]: Sending polling command : QPIWS with length 5
[06:56:00][D][uart_debug:158]: <<< "(00000000000000000100000000000000E\x18\r"
[06:56:04][D][pipsolar:843]: timeout command to poll: QPIWS
[06:56:04][D][uart_debug:158]: >>> "QT\'\xFF\r"
[06:56:04][D][pipsolar:926]: Sending polling command : QT with length 2
[06:56:05][D][uart_debug:158]: <<< "(NAKss\r"
[06:56:09][D][pipsolar:843]: timeout command to poll: QT
[06:56:09][D][uart_debug:158]: >>> "QMN\xBBd\r"
[06:56:09][D][pipsolar:926]: Sending polling command : QMN with length 3
[06:56:10][D][uart_debug:158]: <<< "(VM-3000b\xCB\r"
[06:56:14][D][pipsolar:843]: timeout command to poll: QMN
```
But they are not sent to esphome and HA, can you advise?

Here is the yaml.
esphome:
  name: energy-solar
  friendly_name: energy-solar

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: recommended

external_components:
  - source: github://syssi/esphome-pipsolar@pip8048
    refresh: 0s
[... 136 lines cut ...]
      name: inverter0_pv_ok_condition_for_parallel
    pv_power_balance:
      name: inverter0_pv_power_balance

# Example configuration entry
output:
  - platform: pipsolar
    pipsolar_id: inverter0
    battery_recharge_voltage:
      id: inverter0_battery_recharge_voltage_out

select:
  - platform: pipsolar
    pipsolar_id: inverter0
    output_source_priority:
      id: inverter0_output_source_priority_select
      name: inverter0_output_source_priority_select
      optionsmap:
        "Utility first": "POP00"
        "Solar only": "POP01"
        "Solar Battery Utility": "POP02"
      statusmap:
        "0": "Utility first"
        "1": "Solar only"
        "2": "Solar Battery Utility"

tiimsvk, 2024-05-09:
Haha half a day behind me and this nonsense.
just set the uart to
`dummy_receiver: false`
And the CRC check is ok.
Tahnks for topic: https://github.com/syssi/esphome-pipsolar/issues/42#issuecomment-1356332733